morrownr/USB-WiFi #514: Alfa AWUS036AXM as 6GHz AP?

snaens, 2024-09-26:
Hey all,
I'd like to create an AP on 6GHz using a compact USB or PCIe module with external antennas.
(It will be used for a robot in areas where 2.4 and 5GHz are saturated)

I read the **Alfa AWUS036AXM** *(MT7921AUN)* and **EDUP EP-AX1672** *(MT7921AU)* are good candidates, 
but I can't find out if they can be used as an AP on 6GHz.
*Can someone clarify this? (send the output of `iw list`)*
I'm curious, because the the following 2 devices use the same chipset and neither work:

The **Alfa AWUS036AXML** *(MT7921AUN)* apparently doesn't like 6GHz AP mode (NO-IR).
[https://github.com/FinchSec/wifi-adapter-testing/Kali2023.1/iw-phy-info.txt#L511-L570](https://github.com/FinchSec/wifi-adapter-testing/blob/5733bcb99546dbdd15db94728bb5636729317703/AWUS036AXML/Kali2023.1/iw-phy-info.txt#L511-L570)

And in my tests a **RZ608** *(MT7921K)* PCIe module didn't want to do 6GHz at all
\- `iw list` shows 'disabled' for all, no matter what country i set it to
\- note that I got it second-hand, no idea if it is some OEM stuff

<details>

<summary>iw list output</summary>

```
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		* managed
		* AP
		* AP/VLAN
		* monitor
		* P2P-client
		* P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
[... 495 lines cut ...]
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ ACK_SIGNAL_SUPPORT ]: ack signal level support
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

</details>

---
Through your `iw list` database I (finally) found a module that [multiple](https://github.com/morrownr/USB-WiFi/blob/37d4ce952877d5acf0a6af8be5d1afe8ce30e4e0/home/iw_list/Qualcomm_QCNFA765_m.2.txt) [sources](https://forum.openwrt.org/t/how-can-i-run-nfa765-a-6ghz-ap/166193/4) confirm should be able to do 6GHz AP:  
 the **QCNFA765**
I may go buy one to find out.

morrownr, 2024-09-26:
Hi @snaens 

> I'd like to create an AP on 6GHz using a compact USB or PCIe module with external antennas.

Me too. We have seen many reports from users of adapters with mt7921au chips that have successfully created a 6 GHz AP. They were all in the EU with country codes like DE and FR. This tends to indicate the issue is regulatory and not with hardware of driver. Tell us what country you are in.

FYI: Recently there was one person that reported a 6 GHZ WiFi 7 AP with a mt7925 chip in the US. The report is here somewhere and I used some info from his report but I have been busy on other things.

The example hostapd.conf that I have up for WiFi 6 is in really good shape. What you might want to do is use WiFi 6 with the least congested channel you can find on either 2.4 or 5 GHz until such time as this is sorted out. Something to keep in mind given that you have a robotics project going is that the range drops off greatly with 6 GHz so it may not be the holy grail.

snaens, 2024-09-27:
> My opinion is 6ghz is not ready for prime time on any platform. Too many countries haven't approved it yet.

This is for a competition/research robot in an academic context,
I mean someone has to try this stuff out, no? ;)

Ofc 6GHz needs work, but I don't mind bandwidth/hi-speed performance,
we compress the heck out of our images anyway.
Stability could be a snag though.

> Some people got the "disabled" channels to work. I don't know how, trial and error with country codes I think.

I tried that, but I feel like my specific module may be locked-out.
The *RZ608* is often used by OEMs and as I said, I have no idea where mine came from.

---

> Me too. We have seen many reports from users of adapters with mt7921au chips that have successfully created a 6 GHz AP. They were all in the EU with country codes like DE and FR. This tends to indicate the issue is regulatory and not with hardware of driver. Tell us what country you are in.

**Conveniently I am in the country with the code DE ;)**

I did however only run short-length tests.
(tried many settings and kept restarting the module **(OS: openWRT))**
**Does the mt7921 scan surrounding networks to check what county it is in?**

> FYI: Recently there was one person that reported a 6 GHZ WiFi 7 AP with a mt7925 chip in the US. The report is here somewhere and I used some info from his report but I have been busy on other things.

Yea I think I read that at some point, and promptly was confused when my modules (and those of some others on the net) straight up disabled or didn't support 6GHz.
It really doesn't help that the only difference between modules with & without 6GHz support is a singular 'K' at the end 🙄
(MT7921K vs MT7921)

> The example hostapd.conf that I have up for WiFi 6 is in really good shape.

I tried using it on **OpenWRT on a Pi4** with my weird *RZ608*, but it always fails to initialize (illegal channel combination) and then hangs in AP-NoIR mode.
This is on channel 37 (worked for someone else here), with everything set correctly.
`iw list` shows the channels as disabled, so I'm not surprised it's broken.

> What you might want to do is use WiFi 6 with the least congested channel you can find on either 2.4 or 5 GHz until such time as this is sorted out.

So far we have been sneakily running on the top-end of the 5GHz spectrum, as many devices can't go that high and there are 1~2 free channels most of the time.
As a compromise we are running on a single channel, so very low bandwidth.

This is usually fine, except we have 2 robots, and there is only really bandwidth for one.
So they both work fine alone, but start lagging when they are both simultaneously in operation :/

We have no chance contesting for 'regular' channels, as we are in a sort of "antweight" challenge, and share the room/hall with univerity robots 10x our size, which all have crazy overpowered APs
(sit behind one and you have no more signal kinda stuff)

> Something to keep in mind given that you have a robotics project going is that the range drops off greatly with 6 GHz so it may not be the holy grail.

It is in a competition scenario.
Think of it like a trade fair/convention where everyone has a high-power AP, but you can't have one.
Point-blank range (<10m), insane congestion, people running ancient hardware on 2.4GHz.
All things WiFi6 and 6GHz can improve/avoid.

If you actually wanted to use the robots long-range outdoors, you'd probably be using different bands and radios.
Our specific competition only permits 802.11x <all the wifis> anyway.

---

I'm going to try my RZ608 on more standard hardware on debian testing/trixie and see what happens.
If it doesn't work I'll try my luck with the *QCNFA765* and *AWUS036AXM*, I have to purchase then first though.

morrownr, 2024-09-29:
> Conveniently I am in the country with the code DE ;)

Cool. You should be up and running according to what I have read from others in DE.

>  it always fails to initialize (illegal channel combination) and then shuts down.

You do have WPA3 selected, right? It is required for band 4 (6 Ghz). Mixed won't work and WPA2 won't work.

> I'll try my luck with the QCNFA765 and AWUS036AXM...

My favorite USB WiFi adapter in the WiFi 6, mt7921au, class is the Edup adapter as shown in the mt7921au section of the Plug and Play List. Darn good little adapter that has reasonable range, a decent price and seems to be widely available these days... make sure you get the right one. The model number is in the Plug and Play List.

snaens, 2024-09-30:
> > it always fails to initialize (illegal channel combination) and then shuts down.
> 
> You do have WPA3 selected, right? It is required for band 4 (6 Ghz). Mixed won't work and WPA2 won't work.

Yep, I saw your comment in the conf :+1: 
It worked perfectly fine with 5GHz and WiFi 5 mode.

The `x_oper_centr_freq_seg0_idx` confused me a bit though - why can't the driver calculate this from the selected channel and channel width?
I also couldn't get 40 MHz ch_width to work - it only did 20 and 80 MHz, maybe my module is at fault?

bjlockie wasn't kidding when they said WiFi 6 mode isn't ready:
Setting the channel and ch_width in WiFi 6 doesn't work,
but those settings are somewhat retained when switching from WiFi 5 to WiFi 6 mode mode.
(I was trying to get it to run at the aforementioned 'high' channels)

FYI I switched modes by setting `ieee80211xx=0` to disable either one.

> My favorite USB WiFi adapter in the WiFi 6, mt7921au, class is the Edup adapter as shown in the mt7921au section of the Plug and Play List. Darn good little adapter that has reasonable range, a [...]

I went looking for that one, the _EDUP EP-AX1672_ at first, but you can't get it in Europe apparently.
Alfa modules on the other hand are easily found.

So I either get the EDUP for ~45$ from the USA or the _AWUS036AXM_ locally for ~30$. (both after tax and shipping) 
Keep in mind the EDUP comes from the USA and therefore may be programmed for the wrong country.
(in this case perhaps beneficial, as DE limits tx power on 6GHz to 13db, and US doesn't)

---

Anyhow, @morrownr, since you seem to have an _EDUP EP-AX1672_ from the US, can you please run `iw list` and share what it says for 6GHz?
(if you are feeling generous also try it after setting the region to `iw reg set DE`)
It'd save me the whole hassle 😁

morrownr, 2024-10-05:
> since you seem to have an EDUP EP-AX1672 from the US, can you please run iw list and share what it says for 6GHz?

```
		Frequencies:
			* 5955 MHz [1] (12.0 dBm) (no IR)
			* 5975 MHz [5] (12.0 dBm) (no IR)
			* 5995 MHz [9] (12.0 dBm) (no IR)
			* 6015 MHz [13] (12.0 dBm) (no IR)
			* 6035 MHz [17] (12.0 dBm) (no IR)
			* 6055 MHz [21] (12.0 dBm) (no IR)
			* 6075 MHz [25] (12.0 dBm) (no IR)
			* 6095 MHz [29] (12.0 dBm) (no IR)
			* 6115 MHz [33] (12.0 dBm) (no IR)
			* 6135 MHz [37] (12.0 dBm) (no IR)
			* 6155 MHz [41] (12.0 dBm) (no IR)
			* 6175 MHz [45] (12.0 dBm) (no IR)
			* 6195 MHz [49] (12.0 dBm) (no IR)
			* 6215 MHz [53] (12.0 dBm) (no IR)
			* 6235 MHz [57] (12.0 dBm) (no IR)
			* 6255 MHz [61] (12.0 dBm) (no IR)
			* 6275 MHz [65] (12.0 dBm) (no IR)
			* 6295 MHz [69] (12.0 dBm) (no IR)
			* 6315 MHz [73] (12.0 dBm) (no IR)
			* 6335 MHz [77] (12.0 dBm) (no IR)
			* 6355 MHz [81] (12.0 dBm) (no IR)
			* 6375 MHz [85] (12.0 dBm) (no IR)
			* 6395 MHz [89] (12.0 dBm) (no IR)
			* 6415 MHz [93] (12.0 dBm) (no IR)
			* 6435 MHz [97] (12.0 dBm) (no IR)
			* 6455 MHz [101] (12.0 dBm) (no IR)
			* 6475 MHz [105] (12.0 dBm) (no IR)
			* 6495 MHz [109] (12.0 dBm) (no IR)
			* 6515 MHz [113] (12.0 dBm) (no IR)
			* 6535 MHz [117] (12.0 dBm) (no IR)
			* 6555 MHz [121] (12.0 dBm) (no IR)
			* 6575 MHz [125] (12.0 dBm) (no IR)
			* 6595 MHz [129] (12.0 dBm) (no IR)
			* 6615 MHz [133] (12.0 dBm) (no IR)
			* 6635 MHz [137] (12.0 dBm) (no IR)
			* 6655 MHz [141] (12.0 dBm) (no IR)
			* 6675 MHz [145] (12.0 dBm) (no IR)
			* 6695 MHz [149] (12.0 dBm) (no IR)
			* 6715 MHz [153] (12.0 dBm) (no IR)
			* 6735 MHz [157] (12.0 dBm) (no IR)
			* 6755 MHz [161] (12.0 dBm) (no IR)
			* 6775 MHz [165] (12.0 dBm) (no IR)
			* 6795 MHz [169] (12.0 dBm) (no IR)
			* 6815 MHz [173] (12.0 dBm) (no IR)
			* 6835 MHz [177] (12.0 dBm) (no IR)
			* 6855 MHz [181] (12.0 dBm) (no IR)
			* 6875 MHz [185] (12.0 dBm) (no IR)
			* 6895 MHz [189] (12.0 dBm) (no IR)
			* 6915 MHz [193] (12.0 dBm) (no IR)
			* 6935 MHz [197] (12.0 dBm) (no IR)
			* 6955 MHz [201] (12.0 dBm) (no IR)
			* 6975 MHz [205] (12.0 dBm) (no IR)
			* 6995 MHz [209] (12.0 dBm) (no IR)
			* 7015 MHz [213] (12.0 dBm) (no IR)
			* 7035 MHz [217] (12.0 dBm) (no IR)
			* 7055 MHz [221] (12.0 dBm) (no IR)
			* 7075 MHz [225] (12.0 dBm) (no IR)
			* 7095 MHz [229] (12.0 dBm) (no IR)
			* 7115 MHz [233] (12.0 dBm) (no IR)

```

 $ uname -r
6.6.47+rpt-rpi-v8

$ iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

-----

> (if you are feeling generous also try it after setting the region to iw reg set DE)

$ sudo iw reg set DE

$ iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

```
		Frequencies:
			* 5955 MHz [1] (23.0 dBm)
			* 5975 MHz [5] (23.0 dBm)
			* 5995 MHz [9] (23.0 dBm)
			* 6015 MHz [13] (23.0 dBm)
			* 6035 MHz [17] (23.0 dBm)
			* 6055 MHz [21] (23.0 dBm)
			* 6075 MHz [25] (23.0 dBm)
			* 6095 MHz [29] (23.0 dBm)
			* 6115 MHz [33] (23.0 dBm)
			* 6135 MHz [37] (23.0 dBm)
			* 6155 MHz [41] (23.0 dBm)
			* 6175 MHz [45] (23.0 dBm)
			* 6195 MHz [49] (23.0 dBm)
			* 6215 MHz [53] (23.0 dBm)
			* 6235 MHz [57] (23.0 dBm)
			* 6255 MHz [61] (23.0 dBm)
			* 6275 MHz [65] (23.0 dBm)
			* 6295 MHz [69] (23.0 dBm)
			* 6315 MHz [73] (23.0 dBm)
			* 6335 MHz [77] (23.0 dBm)
			* 6355 MHz [81] (23.0 dBm)
			* 6375 MHz [85] (23.0 dBm)
			* 6395 MHz [89] (23.0 dBm)
			* 6415 MHz [93] (23.0 dBm)
			* 6435 MHz [97] (disabled)
			* 6455 MHz [101] (disabled)
			* 6475 MHz [105] (disabled)
			* 6495 MHz [109] (disabled)
			* 6515 MHz [113] (disabled)
			* 6535 MHz [117] (disabled)
			* 6555 MHz [121] (disabled)
			* 6575 MHz [125] (disabled)
			* 6595 MHz [129] (disabled)
			* 6615 MHz [133] (disabled)
			* 6635 MHz [137] (disabled)
			* 6655 MHz [141] (disabled)
			* 6675 MHz [145] (disabled)
			* 6695 MHz [149] (disabled)
			* 6715 MHz [153] (disabled)
			* 6735 MHz [157] (disabled)
			* 6755 MHz [161] (disabled)
			* 6775 MHz [165] (disabled)
			* 6795 MHz [169] (disabled)
			* 6815 MHz [173] (disabled)
			* 6835 MHz [177] (disabled)
			* 6855 MHz [181] (disabled)
			* 6875 MHz [185] (disabled)
			* 6895 MHz [189] (disabled)
			* 6915 MHz [193] (disabled)
			* 6935 MHz [197] (disabled)
			* 6955 MHz [201] (disabled)
			* 6975 MHz [205] (disabled)
			* 6995 MHz [209] (disabled)
			* 7015 MHz [213] (disabled)
			* 7035 MHz [217] (disabled)
			* 7055 MHz [221] (disabled)
			* 7075 MHz [225] (disabled)
			* 7095 MHz [229] (disabled)
			* 7115 MHz [233] (disabled)

```

It looks like you should be good to go on the lower channels.

snaens, 2024-10-06:
Alright, fantastic!
My laptop's *AX210* also only goes up to channel 93,
so that seems to be the limit here.

Order has gone out, shipping takes a week.
I hope I'll have good news by then :crossed_fingers:

morrownr, 2024-10-07:
@snaens 

I am slowly doing some deep research into using band 4 (6 GHz) for AP mode on Linux. I am finding some good info so check back in with me when ready.

morrownr, 2024-11-10:
Hi @Andy-45 

>  I wonder how EDUP disabled the BT !

This is a good question. I have one of the Edup EP=AX1672 adapters and it does not suffer from the BT issue. I took the Alfa AXML out of the The Plug and Play List until such time as the problem is resolved. I thought the list was clean of adapters with the BT problem. Are you saying that the AXM has the BT problem?

Andy-45, 2024-11-10:
yes, the AWUS036AXM has the BT problem, look at my original post -> https://github.com/morrownr/USB-WiFi/issues/495 
there is a workaround, deleting the BT_RAM_CODE firmware file, which works on my RasPi 5 (kernel 6.6.51) - but it did not work on Ubuntu 24.04.1 - so you may really remove the adapter from your list.

morrownr, 2024-11-10:
> so you may really remove the adapter from your list.

Going to do it right now. That makes 3 adapters that have shown the problem and all were early adapters. The other was the Comfast 953. This problem did not show until late last year and it is a BT stack issue as best I can make of it. Hopefully it can be fixed.

snaens, 2024-11-11:
Hi, sorry for disappearing like that.

I purchased both an *Alfa AWUS036AXM* and a *EDUP EP-AX1672*.
Like `Andy-45` said, they share identical boards.

**right: Alfa AWUS036AXM, left: EDUP EP-AX1672**
[image]
[image]

Note that the Alfa chips are sanded or something, and almost impossible to read.

Unfortunately my Alfa came DOA (bought on amazon '2nd hand', had been returned before).
It didn't even register as a USB device.

---

I was able to create an AP on 6GHz, scans on other 6GHz-capable devices also picked it up. (Debian testing/trixie with hostapd and [WiFi7 config](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi7.conf))
But connecting didn't work. (From a framework laptop with Intel AX210 on Fedora 40) (note the AX210 is WiFi 6e).
I would have tried it with the two Mediatek adapters, had the Alfa been ok.

I see `morrownr` created [a config for WiFi 6e specifically](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6e.conf), i will try it out now.

snaens, 2024-11-11:
Well, I'm still trying to get an AP functioning :P

The 6e config seems to work.
But my PC still fails to connect.

NetworkManager on my PC (mac addr censored):
```
device (wlp166s0): Activation: starting connection 'myPI-WiFi6e' (dd22dfb7-4590-4018-993b-6857ed4ff5b4)
device (wlp166s0): supplicant interface state: completed -> disconnected
device (p2p-dev-wlp166s0): supplicant management interface state: completed -> disconnected
device (wlp166s0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
manager: NetworkManager state is now CONNECTING
device (wlp166s0): set-hw-addr: set-cloned MAC address to xx:xx:xx:xx:xx:xx (stable-ssid)
device (wlp166s0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
device (wlp166s0): Activation: (wifi) access point 'myPI-WiFi6e' has security, but secrets are required.
device (wlp166s0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
device (wlp166s0): supplicant interface state: disconnected -> interface_disabled
device (p2p-dev-wlp166s0): supplicant management interface state: disconnected -> interface_disabled
device (wlp166s0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
device (wlp166s0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
device (wlp166s0): Activation: (wifi) connection 'myPI-WiFi6e' has security, and secrets exist.  No new secrets needed.
Config: added 'ssid' value 'myPI-WiFi6e'
Config: added 'scan_ssid' value '1'
Config: added 'bgscan' value 'simple:30:-70:86400'
Config: added 'key_mgmt' value 'SAE FT-SAE'
Config: added 'psk' value '<hidden>'
Config: added 'ieee80211w' value '2'
device (wlp166s0): supplicant interface state: interface_disabled -> disconnected
device (p2p-dev-wlp166s0): supplicant management interface state: interface_disabled -> disconnected
device (wlp166s0): supplicant interface state: disconnected -> scanning
device (p2p-dev-wlp166s0): supplicant management interface state: disconnected -> scanning
device (wlp166s0): Activation: (wifi) association took too long, failing activation
device (wlp166s0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
manager: NetworkManager state is now CONNECTED_LOCAL
device (wlp166s0): set-hw-addr: set MAC address to xx:xx:xx:xx:xx:xx (scanning)
device (wlp166s0): Activation: failed for connection 'myPI-WiFi6e'
device (wlp166s0): supplicant interface state: scanning -> disconnected
device (p2p-dev-wlp166s0): supplicant management interface state: scanning -> disconnected
device (wlp166s0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
```

Cli: `Error: Connection activation failed: The Wi-Fi network could not be found`

---

<details><summary>My hostapd config (on raspi)</summary>
<p>

```shell
# /etc/hostapd/hostapd-WiFi6e.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2024-10-09
# Purpose: hostapd.conf example for WiFi 6e configuration for USB WiFi adapters using the mt7921au chip.
# Status: alpha1, please test and report.

# Requirements for WiFi-6e (ieee80211ax) on 6GHz:
# - WPA3 for authentication
# - SAE key management required for WPA3
# - Management Frame Protection required for WPA3
# - Radar awareness required for some RegDomains on 5GHz and 6GHz
# - WME/WMM required

# Information:
# - tested on RasPi4B with 64 bit RasPiOS
# - see [Main Menu](https://github.com/morrownr/USB-WiFi) for...
#   -- make sure mt7921 firmware files are installed/updated
#   [How to install firmware for Mediatek-based USB WiFi adapters](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md)
#   -- you need hostap 2.10 or later for WiFi 6 support: $ hostapd -v
#   [Upgrade hostapd to support WiFi 6](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md)
# - improvements are welcome
[... 239 lines cut ...]
# current wireless-regdb packages delivered with VyOS. As of now, 5GHz channels would
# not allow access point operation (iw list showing no-ir on all channels) and 6GHz
# channels would be completely disabled. Also, switching country codes (iw reg set DE)
# would not have any effect. For now, I have this workaround:
# Download and install wireless-regdb from the Ubuntu Packages page, and
# (not needed anymore with newest 1.5-rolling) manually configure your regulatory
# domain as option for the cfg80211 module by editing
# /etc/modprobe.d/cfg80211.conf and adding this line: options cfg80211 ieee80211_regdom=DE

# End of hostapd-WiFi6.conf
```

</p>
</details>

I note that selecting a frequency was very odd, especially for 80MHz.
The default channel of 55 (centr_freq 61) results in `IEEE 802.11 Hardware does not support configured mode (2) (hw_mode in hostapd.conf) Could not select hw_mode and channel. (-2)`
But channel 53 and centr_freq 47 works at 80MHz (I have nothing that can scan and verify the bandwidth, though).
Channel 53 with centr_freq 55 @ 40MHz bandwith also works fine (used in config) .

---

I wonder if the edup being up can convince my AX210 to do the same (it says NO-IR on all channels)
Then I could try doing this in the opposite direction.
Or I could just get another WiFi adaper...

morrownr, 2024-11-11:
> The 6e config seems to work.

I am working through a usb wifi adapter as AP using 6 GHz right now. 6 GHz settings for hostapd are hard to figure out but I am on the way. I can't provide more details right now.

> I wonder if the edup being up can convince my AX210 to do the same (it says NO-IR on all channels)

Note likely.

For 40 GHz, try channels 65 and 67.

There are likely some he specific settings that need to be address on the mt7921au chip for hostapd.conf. I'll try to sort that out.

Andy-45, 2024-11-11:
not sure if network-manager supports Wifi 6e. 
I followed the example from https://dev.to/wataniguchi
and was able to connect via wpa_supplicant

and for the AX210 look here: https://medium.com/@barry_wu.me/how-to-connect-wi-fi-6e-ap-over-the-6-ghz-band-with-ubuntu-21-04-2a0b4f071ac2

snaens, 2024-11-12:
I tried using wpa_supplicant instead of NetworkManager
and it yields similar results;

[`ap_scan` docs](https://gist.github.com/penguinpowernz/ce4ed0e64ce0fa99a5e335c1a4c954b3#file-wpa_supplicant-conf-L88-L115)
If I set `ap_scan` to `1` it simply can't find the AP in it's scans.
With `ap_scan` set to `0` or `2` it doesn't scan, but triggering a scan (with wavemon) makes the ap show up in wpa_supplicant's output. However since this scan was triggered externally it refuses to use it.
When set to `0` it additionally complains about "Could not read SSID from driver"

Triggering a scan using `iw` won't detect the AP.

> [!NOTE]
> - This is with an AX210 now. EDUP is the AP.
> - AX210 as AP did not work at all, just like morrownr predicted.
> - The rest of my config is identical to [the Article Andy-45 recommended](https://medium.com/@barry_wu.me/how-to-connect-wi-fi-6e-ap-over-the-6-ghz-band-with-ubuntu-21-04-2a0b4f071ac2). 

This all seems a bit like two driver interfaces/APIs being used simultaneously, and only one suppoting 6GHz, no?

---

The EDUP AP seems to be working alright, It's been active and connectable all day.

This would all be so much easier if I just had another mt7921au device...

snaens, 2024-11-12:
>  not sure if network-manager supports Wifi 6e.

It sure shows up in my WiFi menu (on KDE)
But yes, the tools (nmtui and KDE network settings) both don't list 6GHz frequencies.

snaens, 2024-11-12:
Yesterday I ordered a QCNFA765 card for 13€
I've seen it can do AP mode, but heard the driver sucks.

But anything with a mt7925 costs like ~30€, and I just want something else to test this AP on.

I also gave Windows 11 a shot just now (VM with PCIe passthrough), but that didn't show anything on 6GHz, even in scans.

morrownr, 2024-11-12:
> Yesterday I ordered a QCNFA765 card for 13€

Someone else was asking me if I knew anything about this chip recently and I could not offer anything. I watch linux-wireless and OpenWRT. Qualcomm and Mediatek dominate the OpenWRT supported wifi routers. I support 2 routers with Qualcomm SOCs and one with a Mediatek SOC. Both companies do a good job but in my limited sample size, Mediatek drivers are better.

I'd like to hear about what you find out.

snaens, 2024-11-30:
Still no luck, the new card won't connect either.
I'll try it as an AP next.

morrownr, 2024-12-01:
Won't connect to what? An AP/wifi router?

snaens, 2024-12-01:
Yup, the pi + edup AP.
Same issue as with the ax210: it can't find the SSID for some reason.

Honestly I think this is an issue with the NetworkManager backend;
it interfaces with the driver in a way that doesn't support 6GHz.
(i.e. 6GHz APs don't show up in scans)

How did you connect with wpa_supplicant?
Would you mind sharing your config?

Andy-45, 2024-12-04:
@snaens 
try harder :-) 

My setup:
2 Raspberry,  1 Edup-AX1672 and 1 Awus036AXM.
the Edup is AP "foo-ax" , the Awus works as client

this time I used nmcli to connect:

[image]

[image]

client wpa_supplicant-wlan1.conf:

> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=DE
pmf=2
sae_pwe=1
network={
    scan_ssid=1
    ssid="foo-ax"
    key_mgmt=SAE
    proto=RSN
    ieee80211w=2
    sae_password="foobar1234"
    pairwise=CCMP
    group=CCMP
}

AP hostapd.conf  (not optimized for bandwith, just for working demo):

> ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
country_code=DE
interface=wlan1
bridge=br0
driver=nl80211
ssid=foo-ax
sae_password=foobar1234
hw_mode=a
channel=57
ieee80211ax=1
wmm_enabled=1
op_class=131
ieee80211w=2
auth_algs=3
wpa=2
wpa_key_mgmt=SAE
rsn_pairwise=CCMP
rsn_preauth=1
sae_require_mfp=1
sae_pwe=1
wpa_group_rekey=1800
group_cipher=CCMP
wpa_psk_radius=0