morrownr/USB-WiFi #516: Built on your foundation

briantbutton, 2024-10-08:
A year ago, I popped up with [questions](https://github.com/morrownr/USB-WiFi/issues/296) about drivers, adapters and such.&nbsp;   Anyway, we built a product and I thought I would mention it here since it uses one of your drivers.&nbsp;  It is based on the Raspberry Pi CM 4, uses that Nano A baseboard and exposes two radios.

The whole thing weighs a pound and a half and will run 12 hours plus or minus.&nbsp; ***(first image)***&nbsp;
I discovered _(here, I think)_ that the Pi internal radio won't support many users and, after some study, adopted the Alfa AWUS036ACH as the workhorse radio.&nbsp;  It has performed fine.&nbsp; 

We don't want all the consumer packaging and accessories and so were able to get Alfa to just ship us the board.&nbsp; With some additional wiring and **hilarious** molded plastics, we adhere the second radio to the Pi subsystem with suitable mounting spots.&nbsp;&nbsp; ***(second image)***&nbsp; The height of the combined unit is a bit more than a cell phone, so we are able to get batteries with the same silhouette to fit right behind it.

[image]
[image]

morrownr, 2024-10-10:
Brian,

That is cool. FYI: The in-kernel, RTW88 rtw8812au driver is in review right now and should go into 6.13 or not long after that.

briantbutton, 2025-02-15:
Thank you for that info.  We meticulously create a software image that has exactly the goods we seek and then flash them to units in one motion.  So there is no technical advantage to a driver being in-kernel.  At least I cannot think of one.

I heard a rumor that the Raspberry Pi 5 has a stronger WiFi chipset than the 4.  The Pi CM4 could only support a handful of simultaneous users.  Does anyone here know anything about the Pi5 integral WiFi?

It is not a pivotal bit of knowledge but I am curious.  Thought to ask here.

morrownr, 2025-02-16:
Hi Brian,

> So there is no technical advantage to a driver being in-kernel. At least I cannot think of one.

Let me help with that:

- If all else was equal, then there would be no technical advantage to a driver being in-kernel in your situation. However all else is far from equal. 

In-kernel drivers are regularly scanned by computer programs looking for coding errors. This is an automated process. This process regularly finds issues and reports them allowing for developers to work the issues in a proactive manner. This does not happen for out-of-kernel driver.

In-kernel drivers are regularly compiled by computer programs that look for compilation errors. This is an automated process. This process regularly finds compilation errors and reports them allowing for developers to work the issues in a proactive manner.

morrownr, 2025-02-16:
@briantbutton 

I accidentally sent the above message early. I had several more issues to bring up but I think you can see where I was going with this.

The in-kernel drivers are maintained much better and we have access to the devs to get help or report issues. When is the last time I was able to report an issue with a Realtek out-of-kernel driver directly to Realtek? Never. 

The in-kernel drivers go through a quality control process on their way into the kernel. There is a person for each brand that handles this. It is common for many code changes to happen during this process.

The in-kernel drivers are subjected to a lot of testing whereas it is obvious Realtek's USB team does not have an effective testing program. Realtek's out-of-kernel drivers can take years to get into good enough shape to be considered reliable. This is just now happening with some of their WiFi 6 drivers. I'm not saying the Realtek WiFi 6 USB drivers are good but they are finally getting to the point that they are somewhat usable.

I remember many times I have been asked to help companies with a project. They make decisions without proper testing and without anyone with the proper knowledge involved and then they run into problems. One of the bigger reasons for problems like this is that Realtek's out-of-kernel drivers do not always function like would be expected in their drivers follow mac80211 guidelines. Have you tried to put 2 adapters with the same chipset in the same computer using a Realtek out-of-kernel driver? It won't work. That comes up often. The in-kernel drivers do not have this problem. I could go on but you get the idea. You have shown that you are more careful than most so you have run into fewer problems.

Cheers

briantbutton, 2025-02-17:
Thank you.  Consider my understanding upgraded.  Higher scrutiny.  More verification.  So, then, that's good news.

We have used the existing rtl8812au driver for some time with no problems but our feature usage is narrow and we are only running one flavor of hardware right now.