morrownr/USB-WiFi #517: [Help]: MT7927 - I see it's pending in the kernel

trip1, 2024-10-11:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

NA

### lsusb

NA

### rfkill

NA

### dkms

```text
NA
```

### iw

```text
NA
```

### What happened?

So I bought a TP Link Archer wifi card to go super fast. The reviews mentioned Linux support and I assumed it'd be fine, it was not.

So how can I help? I see on the readme it's pending in the the kernel but curious if there's more to do or where I should be looking.

I completely understand kernel ( and driver ) code is super complex, but I just wanna see it honestly and understand the process.

Thanks for any info, and for this repo! Super helpful.

morrownr, 2024-10-11:
Hi @trip1 

> So I bought a TP Link Archer wifi card to go super fast.

Is that a PCIe or m.2 card? Are you saying that TP-Link has made a card with a Mediatek mt7927 chip?

> So how can I help?

There is not much us mere mortals can do until Mediatek adds the driver to the kernel. Mediatek has a Linux programming staff that handles development and we can't help until the driver is in the kernel. I wish I had a timeline but I don't.

I have a mt7925 based M.2 card and it does go fast. I don't have a WiFi 7/ 6Ghz router yet but it handles 1 Gbps and more easily with my WiFi 6 dual band router using 160 MHz channel width.

Kiwimannen, 2024-10-12:
Its an M.2 card - the TP-Link card is called: TP-Link Archer TBE550E- BE9300 WIFI 7 / Bluetooth 5.4 PCIe Adapter

Here is a review where you can see the mt7927 M.2 card:

https://einfoldtech.com/2024/02/tp-link-archer-tbe550e-be9300-wifi-7-bluetooth-5-4-pcie-adapter/

It also comes bundled with some motherboards, but as far as I know its never been sold only as a M.2 card.

And there is only Windows drivers available.

trip1, 2024-10-14:
Ah ok thanks for the info. I should've double checked the drivers, not trust the reviews.

Guess I'll just watch for the driver to drop.

morrownr, 2024-10-14:
@Kiwimannen 

Your card is a good looking product. If the M.2 card was based on the mt7925 chip, you would be in business right now. The mt7927 chip is simply not supported yet. This slow support for the mt7927 seems to be an anomaly for Mediatek as they have been doing a good job of getting their drivers in the kernel with enough time before product is available that products are supported on launch... kinda like Intel and AMD. Hopefully the driver comes soon.

This is not an isolated issue. I have heard from a couple of Linux users who bought Realtek rtl8912au (WiFi 7) based adapters because Linux support was advertised by the seller. There is no Linux support for the rtl8912au unless it is really well hidden.

I have a M.2 card based on the mt7925 and works well in managed mode but I see no reason that it would not work well in AP or monitor modes since previous Mediatek drivers do so. As far as I can tell, the only difference between the mt7925 and mt7927 is the mt7925 can do 160 MHz channel width and the mt7927 can do 320 MHz channel width. My little M.2 card connecting to my WiFi 6 router using 160 MHz channel width can easily exceed 1 Gbps.

castillofrancodamian, 2024-10-14:
> Es una tarjeta M.2: la tarjeta TP-Link se llama: TP-Link Archer TBE550E- BE9300 WIFI 7 / Bluetooth 5.4 PCIe Adapter
> 
> A continuación os dejamos una review donde podéis ver la tarjeta M.2 mt7927:
> 
> https://einfoldtech.com/2024/02/tp-link-archer-tbe550e-be9300-wifi-7-bluetooth-5-4-pcie-adapter/
> 
> También viene incluido con algunas placas base, pero hasta donde yo sé, nunca se vendió solo como una tarjeta M.2.
> 
> Y solo hay controladores de Windows disponibles.

The person who made that post put MediaTek as "Cons". He doesn't know what he's talking about 😂😂😂

IanWorrall, 2024-10-15:
> Its an M.2 card - the TP-Link card is called: TP-Link Archer TBE550E- BE9300 WIFI 7 / Bluetooth 5.4 PCIe Adapter
>
> Here is a review where you can see the mt7927 M.2 card: [...]

One of those motherboards is my Gigabyte X670E Aorus Pro X.
I'd swap it out for something that works, but it's buried under the VRM heatsink.

Fear113, 2024-11-19:
How are things going with drivers? I also have an MT7927 and I'm also sitting without wi-fi and bluetooth... :(

morrownr, 2024-11-20:
I'm watching. I'll post a new issue when the driver goes in.

I started a new list last week. It is for Bluetooth adapters. I only have one adapter listed so far but it does not cost much and I think it pretty good:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Bluetooth_Adapters_for_Linux.md

jdecoste, 2025-01-16:
Commenting here to get notified when this gets updated. 

Built a PC with an asus proart x870e MB, and it ships with MT7927. The board has been on sale for awhile now, so disappointing its taking this long to get the linux driver for it.

boburxon, 2025-01-19:
Commenting here to get notified when this gets updated too.

Built a PC based on X670E AORUS PRO X motherboard, which also has a MT7927 WiFi chip.

pacak, 2025-01-19:
> Commenting here to get notified when this gets updated.

> Commenting here to get notified when this gets updated too.

There's a big button in the right column that says "Subscribe", this way you can get your notifications without notifying everybody else.

morrownr, 2025-01-19:
Here is an update on what I think that I know:

In preparation for the start of the kernel 6.14 cycle, many patches for Mediatek based cards and adapters have been flowing in. Unfortunately nothing for the mt7927 chip. The mt7925 chip seems to be maturing nicely. The driver has been in the kernel since 6.7 and the bluetooth support has been in the kernel since 6.12. Not sure what happened there.

The mt7925 can be found in M.2 and PCIe formats. I have a M.2 card. I only use it in managed mode currently. It seems to be super dependable and fast.

If you are interested in mt7927 support, it might speed things along if you mention Linux support to the place where you buy your product.

Cheers

M3t4Data, 2025-01-20:
Hi everyone,

Could this be good news for MT7927 support ? I came across this patch that seems to address an interoperability issue with the MT7927 chipset: https://patchwork.kernel.org/project/linux-wireless/patch/20250120114551.1542812-1-pmartin-gomez@freebox.fr/

morrownr, 2025-01-20:
Hi @M3t4Data 

I saw that but I'm not sure if it is a hint. The person submitting the patch is not a known Mediatek employee or contractor but how would he know what he says without having a copy of the 7927 driver. Maybe he is a tester?

kudos, 2025-01-20:
> Hi @M3t4Data 
> 
> I saw that but I'm not sure if it is a hint. The person submitting the patch is not a known Mediatek employee or contractor but how would he know what he says without having a copy of the 7927 driver. Maybe he is a tester?

Reverse engineering (even partial) is very common for hardware support in Linux.

morrownr, 2025-01-21:
Hi @kudos 

> Reverse engineering (even partial) is very common for hardware support in Linux.

Generally you see that in cases where the device to be reversed is relatively simple and when the maker of the product does not support Linux. In my opinion, neither apply to this situation. WiFi 7 is a lot of things but simple is not one of them and Mediatek's Linux support has basically been right up there with Intel and AMD over the last few years. The support for the 7925 chip was in the kernel 6 months before we saw product. Mediatek has to do Linux right. If you look at their offerings, there are several that are basically Linux only. Mediatek is a big player in the smartphone market. They need good Linux drivers and support for the SoCs to include wireless. It would be a waste of time to make Windows drivers. Same thing with their hardware for WiFi routers and the like. They are also a big supplier to TV and other appliance makers. Many markets where Linux dominates and Windows has basically zero presence.

I can't explain the delay with the mt7927 driver but I see a lot of evidence that the Mediatek devs are going as fast as they can. A lot of new products have been released over the last year so it may simply be a matter of "we'll get to it when we can?" You can't just hire good Linux wireless devs off of the street. We'll see.

Bouska, 2025-01-21:
> Hi everyone,
> 
> Could this be good news for MT7927 support ? I came across this patch that seems to address an interoperability issue with the MT7927 chipset: https://patchwork.kernel.org/project/linux-wireless/patch/20250120114551.1542812-1-pmartin-gomez@freebox.fr/

Hello,

I'm the author of that patch. Unfortunately, this patch is not related to the support of the MT7927 chipset in the kernel. Its purpose is for a AP based on Linux to support correctly the Association Request that a Windows STA using a MT7927 chipset sends.

IanWorrall, 2025-01-21:
The networking pull request for kernel 6.14 has appeared.
MT7927 owners can continue crying. No mention of it whatsoever.

https://lore.kernel.org/lkml/20250121125748.37808-1-pabeni@redhat.com/

markmann447, 2025-01-28:
> The networking pull request for kernel 6.14 has appeared. MT7927 owners can continue crying. No mention of it whatsoever.
> 
> https://lore.kernel.org/lkml/20250121125748.37808-1-pabeni@redhat.com/

I hope we get it by summer at this point.

morrownr, 2025-01-28:
@markmann447 

The amount of wireless driver improvements and specifically the improvements and additions for USB WiFi adapters has been large over the last year. With the recent addition of the new drivers for rtl8812au and rtl8821/11au chips and the soon to go in driver for rtl8814au, that will pretty much mean full support for Realtek WiFi 5 class adapters. Mediatek's drivers continue to improve as well and Mediatek has the lead on Realtek in that Mediatek has good driver for their WiFi 6 chips and for the mt7925 WiFi 7 chip. 

What we want and expect to see sometime this year:

- driver for the mt7927 chip

- USB WiFi adapters based on the mt7925 chip (M2 and PCIe cards are already available).

I think is too much to expect Realtek WiFi 6 and 7 USB support in the kernel this year. Realtek is making big improvements with their support for their chips for cards but not so much with USB. That is unfortunate.

mah3uz, 2025-03-31:
Anyone have any update on this? eagerly wating for some update.

TZECHIN6, 2025-04-10:
seem its still not supporting it at all...

morrownr, 2025-04-10:
The mt7927 is not supported yet. I saw a new driver for the mt7990 flow in last week but still nothing on the mt7927. There is one heck of a lot of work by the Mediatek devs flowing in which may give us a hint as to the delay. They simply may have more on their plates than they can eat right now.

The best I can do right now, if you have a system with a mt7927 chip, is suggest that you get a USB WiFi adapter to help out for now. Once the mt7927 driver is available, you can use the adapter for other things. They can be quite handy:

https://github.com/morrownr/USB-WiFi

That is the Main Menu of this site. Menu items 1 and 2 have a lot of good information. The Plug and Play List gives information about known  good USB WiFi adapters.

bkovacev, 2025-04-11:
Can we contribute and do the actual development in any way?

TZECHIN6, 2025-04-11:
As I have a extra router that able to act as a repeater, I have used LAN to connect the PC to that repeater wifi and solve the connection issue without pulling a very long LAN cable.

red-smeg, 2025-05-01:
Can mediatek make the mt7927 device information open source and we can write one ourselves !!

morrownr, 2025-05-01:
Intellectual property theft is a real problem in this industry. None of the wifi hardware makers reveal all of the details of their hardware. As much help as Mediatek is to the Linux community, I do not see an all included release happening.

I wish I knew when driver will be released but I did even know about the mt7920 chip or its driver until last month when the driver went into the kernel.

If you are in a hurry, the M.2 cards with the mt7925 chips are readily available at low prices. You could switch out the cards until the mt7927 driver is available.

secure-byte, 2025-05-07:
I fell for marketing tricks with buying Archer TBE550E/BE9300 card just to learn that there are no drivers for it (yet). I also bought a very expensive TP Link wifi 7 router that now does the same job as my old wifi 6e router.

Anyway, for all you out there thinking about going wifi 7 on linux - don't. Not yet. Wait for about a year, year and a half to have decent support for decent hardware.

sa-mendez, 2025-05-23:
Add me to the list of disgruntled early adopters.  Bought a new ASUS motherboard with Wifi 7, and I couldnt even install my distro until I broke out the usb Wifi (yes, its very useful to have one on hand as others have said). 

I sent an email to MediaTek asking about an ETA if they have one.  A shot in the dark I know.

TZECHIN6, 2025-05-23:
@sa-mendez 

Have you tried the latest 6.14 kernel? Which distribution you are planning to install? Seem Ubuntu 24 with this kernel update could use the MT7925 chip driver. I haven't tested it out as some project turning required Ubuntu 22.

sa-mendez, 2025-05-23:
@TZECHIN6 

I already managed to install it using the usb wifi adapter.  I am currently on 6.14.6-arch1-1 (which is the latest as of today).  I was not aware I could use the  MT7925.  

If that driver is in the kernel, the motherboard mt7927 is not picking it up as compatible, though lspci definitely lists it on the  PCI bus:

_09:00.0 Network controller: MEDIATEK Corp. Device 7927_

morrownr, 2025-05-23:
> though lspci definitely lists it on the PCI bus: 09:00.0 Network controller: MEDIATEK Corp. Device 7927

The output of `lspci` and `lsusb` comes from their own db and this has nothing to do with a driver.

> I was not aware I could use the MT7925.

Me either. When last I checked in the source, no device IDs for he 7927 are in the 7925 driver. Somebody have a hack?

jbbandos, 2025-05-23:
6.14.7-1-liquorix-amd64 mt7925e module supports only the MT7925 chip. I have added a pcie adapter with it to my Asus MB (Proart x870) and have both chips now, only the MT7925 is working.

420coupe, 2025-06-01:
Just another request for  drivers for 7927.

morristai, 2025-06-02:
Just bought a ROG Strix X870-I Gaming WiFi, found there's no Wifi/Bluetooth driver on Linux. Requesting drivers for MT7927.

420coupe, 2025-06-02:
> The mt7927 is not supported yet. I saw a new driver for the mt7990 flow in last week but still nothing on the mt7927. There is one heck of a lot of work by the Mediatek devs flowing in which may [...]

Do you know if it's just as simple as adding in the device id to the 7925 drivers. Do they not use the same m.2 protocol for both cards, so should it not be the same? 

How would I go about reverse engineering the windows drivers over to Linux

samTrent, 2025-06-06:
Also voicing request for Linux drivers.

bjlockie, 2025-06-07:
What is the differences between the mt7925 and mt7927?

Is it worth upgrading from a mt7921k to a mt7925 when the router is wifi6?

morrownr, 2025-06-09:
@bjlockie 

> What is the differences between the mt7925 and mt7927?

The only difference that I can see is that the mt7925 supports channel width 160 MHz. The mt7927 supports channels width up to 320 MHz.

> Is it worth upgrading from a mt7921k to a mt7925 when the router is wifi6?

I have a M.2 mt7925 card in a Mini PC. The answer to you question depends on what your wifi 6 router can do. Can it do 160 MHz channel width? If so, you could see significant increases in speed depending on what you are doing. Using a mt7921au based adapter from my Mini PC, I see speeds up to around 650 Mbps. With the mt7925 card, I see speeds of up to 1.2 Gbps.

CaptainGlac1er, 2025-06-09:
The funny thing is one time since I was dual booted and windows bsod I went into Linux and the 7927 actually worked and was able to connect to my headphones but after I restarted it didn't work again. (but works in windows)

CaptainGlac1er, 2025-06-13:
@morrownr it is built into the ASUS X870E-E board, I did already pick up that other one already in the meantime lol

dubhater, 2025-06-21:
@bjlockie The chip will probably reject that firmware, but the firmware extracted from the Windows MT7927 driver could work.

sudomemorizor, 2025-06-22:
> The best way to speed up support is to contact 3 companies and make a request. Be specific which chip needs Linux WiFi driver support (mt7927).
> 
> https://www.mediatek.com/ https://www.amd.com/en.html https://www.asus.com/
> 
> I think AMD uses the following part number: AMD RZ738 . So it might be good to use both mt7927 and rz738 in correspondence.
> 
> Leave messages with all 3 often.

Should you contact TP-Link as well? My device is a TP-Link PCIE adapter.

sudomemorizor, 2025-06-22:
> I wonder if renaming the 7925 firmware will work .:-)

No, it was stated earlier in the thread that it doesn't work, unfortunately.

lukasz1992, 2025-06-28:
Maybe someone can start working with this commit: https://github.com/greearb/linux-ct-6.10/commit/1fe0c0394c18fb01d1070d5c092c22627d56bf50
firmwares need to be extracted from Windows drivers.

JoseG42, 2025-06-28:
I was confidently trying to install Arch after testing it in virtual machines. I never expected that my network card would stop me at step two

morrownr, 2025-06-28:
Hi @JoseG42 

I think I am going to make example letters that can be sent to AMD and Mediatek. In my opinion, the best solution, until the mt7927 driver is available, is to get a USB WiFi adapter. USB adapters provide flexibility and are very handy to have in a drawer or toolkit as a backup. If you follow the guidelines that I use for making The Plug and Play List, you should be able to get a dependable USB adapter that fits your budget:

https://github.com/morrownr/USB-WiFi

dubhater, 2025-06-29:
@lukasz1992 Sounds like you could make it work, if you have the skills to ``~review~`` the Windows driver. But I guess you don't have the card?

red-smeg, 2025-07-07:
any update on this ?

morrownr, 2025-07-09:
Nothing yet.

Michaelhess17, 2025-07-12:
commenting so I get notified when something happens. I just built my dream PC and went with an Asus motherboard because they've always had good linux support. I have a PCIe Wifi 6/Bluetooth 5.2 card for now, but it's still annoying!

kudos, 2025-07-12:
There's a subscribe function on every issue, please use that. Also please stop posting +1 type messages, some of use used the subscribe function for meaningful updates.

anthony-barricelli, 2025-08-27:
adding this here for visibility

https://github.com/ehausig/mt7927

gamelaster, 2025-08-27:
@anthony-barricelli that's (looks almost) full AI generated thing. I am really not sure how much we can believe it at all.

I also checked what @dubhater sent, there wasn't much references in the code than that one about MT7927. Either is really that very close to 7925, or it's not complete.

The best bet is to reverse engineer Windows driver and compare it with 7925 Windows or Linux driver.

My honest opinion is, that if it would be that similar to 7925 (which is already nicely supported) why wouldn't Mediatek done it already? Like yeah, whatever can happen in such big silicon company, but still... I think it is more different, but only way to figure out is to do the hard work of reverse engineering of the drivers.

lirtle, 2025-09-14:
I got tired of waiting for a driver and the USB adapters I tried didn't have good enough range, so I replaced the MT7927 with an AX210 in my X870E ProArt board. It was a trivial swap in case anyone else wants to go that route while we wait for support. 

I originally tried with the MT7925, but its Linux drivers don't feel mature yet and I had a lot of issues when dual booting. If you can live without WiFi 7, I highly recommend swapping for an AX210 until support for the MT7927 improves.

gamelaster, 2025-09-15:
@lirtle thanks! And eugh... I hoped I will not need to take out motherboard out of the case... This will be "fun". 😂

sengokudaikon, 2025-09-15:
> 
> Instead of swapping Wi-Fi cards, I went a different route. I picked up a Wi-Fi 7 router, set it up in client mode, and now not only my main PC but also a NAS and another computer all run through it. No adapters, no waiting — just letting the router do all the heavy lifting.

That, unfortunately, only works until you need Bluetooth to work on your Linux machine. Which is a lot of times for me, so I had to buy a BT dongle. 

It's baffling that we are even having this issue in 2025 😭

noobmaster777, 2025-09-28:
Is this the moment we've all been waiting for?

morrownr, 2025-09-28:
@noobmaster777 

Can you be specific?

VIPQualityPost, 2025-11-03:
That guy came in here a while ago all excited and dumped it in here, it's mostly chatgpt code.

[44 more replies]