morrownr/USB-WiFi #520: [Help]: suggested Wifi6 5Ghz hostapd.conf results in "80/80+80 MHz: no second channel offset" error

DurvalMenezes, 2024-10-13:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux REDACTED 6.10.6+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.10.6-1~bpo12+1 (2024-08-26) x86_64 GNU/Linux

### lsusb

Bus 001 Device 077: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

8: phy6: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
zfs/2.2.6, 6.10.6+bpo-amd64, x86_64: installed
```

### iw

```text
phy#6
        Interface wlan1
                ifindex 12
                wdev 0x600000001
                addr e8:4e:06:ac:f4:1f
                type managed
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       115     0       0       0       0       20416           121
```

### What happened?

I just bought this adapter: https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L, as recommended here: https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

I'm trying to use it with hostapd using this configuration file:  https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf (only lines changed: `ssid=`, `wpa_passphrase=` and `interface=`. 

When I run hostapd with this configuration file, I get a `80/80+80 MHz: no second channel offset` error:

```
# /usr/sbin/hostapd -B /etc/hostapd/hostapd_liliput_5ghz.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
80/80+80 MHz: no second channel offset
Could not set channel for kernel driver
Interface initialization failed
```

Further details: 
* Distro: Devuan Daedalus (Debian Bookworm without systemd)
* Machine: Geekom A7 configured with Ryzen 9 7940HS and 32GB RAM
* The adapter is connected via a USB 3.0 powered hub

Thanks in advance for any and all help and tips.

morrownr, 2024-10-13:
Hi @DurvalMenezes 

> Distro: Devuan Daedalus (Debian Bookworm without systemd)

My ability to help is limited because of your use of Devuan Daedalus. It would be easier for me to help if you used Debian 12 Bookworm or any of the many distros directly based on it that retain systemd. That they have pulled out systemd makes me wonder about what else they have messed with.

The  errors you are seeing are not something I have had to deal with so I'm not sure where to start. I have kept a guide to building an ap up on this site for several years and it has evolved over time as I test and improve it as I can. The guide currently makes heavy use of systemd and has proven to be a really solid guide. The guide is on the Main Menu just above the link to the WiFi 6 hostapd.conf example you are using. While the guide uses RaspBerry Pi OS, it works well with Debian 12 as well. The guide currently is tilted toward WiFi 4 and WiFi 5 but it is simple to substitute the WiFi 6 example.

If you do try Debian 12, it defaults to kernel 6.1 but it has additional more modern kernels to choose from. You can start synaptic and search for linux-image. I recommend kernel 6.6. You might also want to follow the Firmware guide on the Main Menu so as to make sure the firmware for your adapter is up to date. Once you are set up, you can follow the AP guide and if you have any questions, come back here and ask.

I have the same Edup adapter that you have and it is currently pulling duty as an AP following my AP guide and the WiFi 6 guide you are trying so I am well equipped to help if you need it.

Regards

DurvalMenezes, 2024-10-14:
Hi @morrownr, thanks for your prompt response and sorry for the delay in getting back to you. 

> My ability to help is limited because of your use of Devuan Daedalus. It would be easier for me to help if you used Debian 12 Bookworm or any of the many distros directly based on it that retain systemd

This is not a problem, it's not much trouble for me to try and reproduce the issue under Debian.

That said, I do not understand why or how systemd could be in play here: you see, the problem comes up with me executing `hostapd` directly, so no init system is involved (be it systemd, sysvinit, openrc or whatever).

> I have kept a guide to building an ap up on this site for several years and it has evolved over time as I test and improve it as I can. The guide currently makes heavy use of systemd and has proven [...]

You mean this one, right?  https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

The reason I have not followed it is because my AP does not operate in "bridge" mode, but rather in "router" mode and with some complexities involved (ipfilter rules, virtual machines running and being routed/filtered, etc). 

I will try and follow this guide of yours and try to reproduce the issue, as many of the steps involved apply only to a Raspberry Pi setup ("Reduce overall power consumption", etc) I will skip these steps. 

I will get back to you with the results. Thanks again.

morrownr, 2024-10-14:
@DurvalMenezes 

>  I do not understand why or how systemd could be in play here...

It may not be in play but it is easier for me to help you if I can play along and test things here. I have various test systems but all are based on distros that use systemd.

> You mean this one, right? 

Yes

> The reason I have not followed it is because my AP does not operate in "bridge" mode, but rather in "router" mode...

I understand. My guide is not for "router" mode but what I do is offer my opinion as to what might be the best solution for "router" mode in the document. I recommend using OpenWRT if you need "router" mode.

OpenWRT has the driver for your adapter. In fact, some of the Mediatek kernel devs hang out at OpenWRT so anything Mediatek is supported better than anything else. The soon to be released OpenWRT One router is based on the Mediatek mt7981 SOC.

> I will try and follow this guide of yours and try to reproduce the issue...

If you are looking for "router" mode, my guide will not get you where you want to do as even the networking is handled by systemd. Think about using OpenWRT. I'll be glad to assist if you want as I have time.

DurvalMenezes, 2024-10-15:
Again, thanks for your kind and prompt response, @morrownr! Much appreciated!!

> Think about using OpenWRT.

Thank you for your suggestion, but unfortunately I can't use OpenWRT because being an AP is only one of this machine's functions -- it must also run software that does not run in OpenWRT. Debian is mandatory for my use case. 

> It may not be in play but it is easier for me to help you if I can play along and test things here. I have various test systems but all are based on distros that use systemd.

Thanks for the detailed explanation, and I have good news: I brought up a machine using plain Debian 12 Bookworm and your aforementioned guide -- and `hostapd` works perfectly with the `/etc/hostapd/hostapd-WiFi5.conf` you indicate in the guide. But when I use it with your `/etc/hostapd/hostapd-WiFi6.conf` I pointed to in my OP, `hostapd` fails to start with *exactly* the same error message I was getting under Devuan, this time as recorded in ~/hostapd.log: 

```
80/80+80 MHz: no second channel offset
Could not set channel for kernel driver
Interface initialization failed
```

The only things I changed on your `/etc/hostapd/hostapd-WiFi6.conf` were `interface=` (to reflect the correct interface name in my system, ie `wlxXXXXXXXXXXXX` as attributed to it by default by Debian based on its MAC address) and commenting-out the `ctrl_interface=` and `ctrl_interface_group=` lines (otherwise `hostapd` fails with a `invalid line` message mentioning the  `ctrl_interface_group=` line). 

Attached please find my  `/etc/hostapd/hostapd-WiFi6.conf` (renamed with a TXT extension so Github won't reject it as an 'unsupported file type') and my `~/hostapd.log` files; the latter comprises the successful invocation with your 
`/etc/hostapd/hostapd-WiFi5.conf` including successful connection of an WiFi client to that AP, followed by the unsuccessful invocations with your `/etc/hostapd/hostapd-WiFi6.conf`, first with the wrong `interface=`, then with the error caused  by `ctrl_interface=`, and finally with the same error I was getting with Devuan, depicted above. 

Please let me know if you need anything else, and thanks again for all your help and kindness.

[hostapd-WiFi6.conf.txt](https://github.com/user-attachments/files/17373848/hostapd-WiFi6.conf.txt)
[hostapd.log](https://github.com/user-attachments/files/17373847/hostapd.log)

morrownr, 2024-10-15:
Hmmm...

Post the contents of the following file:

$ sudo nano /etc/NetworkManager/NetworkManager.conf

Post the result of the following command:

$ iw dev

DurvalMenezes, 2024-10-17:
Sorry for the delay, something happened and I didn't get a notification for your comment above, just saw it now when I came back here to check.

Here's the info you asked for: 

* /etc/NetworkManager/NetworkManager.conf: no such file here (this machine doesn't even have X installed, much less a Desktop Environment and therefore no NetworkManager, as it's so far headless; if you need to simulate this condition in your test environment, it would be just like "Option 2: // Uninstall NetworkManager" as per your guide; 

* `iw dev`: 
```
phy#0
        Interface wlxXXXXXXXXXXXX
                ifindex 3
                wdev 0x1
                addr XX:XX:XX:XX:XX:XX
                type managed
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       81      0       0       0       0       18456           90
```

Please let me know if you need anything else, I will be camping on this page so as to be able to respond faster.

morrownr, 2024-10-17:
This is a puzzle. What does the following command give you?

$ hostapd -v

DurvalMenezes, 2024-10-17:
> $ hostapd -v
```
hostapd v2.10
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2022, Jouni Malinen <j@w1.fi> and contributors
```

And here's the exact installed package: 
```
# dpkg -l hostapd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version           Architecture Description
+++-==============-=================-============-=============================================================
ii  hostapd        2:2.10-12+deb12u2 amd64        access point and authentication server for Wi-Fi and Ethernet
```

Anything else I can provide?

morrownr, 2024-10-17:
> hostapd v2.10

That is the same version I am running. My AP setup:

RasPi4b
RasPiOS 64 bit (based on Debian 12) (it does use Network Manager so I tell NM to ignore the wireless interface)
Kernel 6.6
Using the hostapd.conf and AP guide that I mentioned.

I am not seeing this issue and, infact, I cannot say that I have ever seen that specific set of issues. The only obvious difference in our setups is that you are running a `routed` setup and I am running a `bridged` setup but if that had something to do with it, I would expect to see different issues. This message is going through a connection to my AP and it is. It is on channel 36, 80 MHz channel width and is using WiFi 6.

If I come up with something, I'll let you know but I am stumped for now.

DurvalMenezes, 2024-10-17:
Could you please post the exact hostapd package you have running with that adapter and WiFi6? I'm pretty convinced it has something to do with it.

morrownr, 2024-10-18:
@DurvalMenezes 

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version           Architecture Description
+++-==============-=================-============-=============================================================
ii  hostapd        2:2.10-12+deb12u2 arm64        access point and authentication server for Wi-Fi and Ethernet

```

It appears that they are the same but...

I agree that hostapd could be a problem because almost all of the capabilities can be compiled in or not... so we could compile the repo and make sure we have the right settings but at this point, it is not obvious that this could be the problem.

DurvalMenezes, 2024-10-19:
> It appears that they are the same but...

Not quite: mine is amd64, yours arm64 -- of course since I'm trying this on a PC and you are running on a Raspberry Pi.

Do you have a x86_64 machine you can try this on? Then we could discard this as a possible source of the problem.

DurvalMenezes, 2024-10-19:
> we could compile the repo and make sure we have the right settings but at this point, it is not obvious that this could be the problem.

I just ran that down: downloaded the source .tar.gz for the latest (2.11) version from https://w1.fi/hostapd/, compiled it with all options turned on, then renamed & replaced /usr/sbin/hostapd with the new binary:

```
# /usr/sbin/hostapd -v
hostapd v2.11
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2024, Jouni Malinen <j@w1.fi> and contributors
```

Running it produces *exactly* the same error: 
```
80/80+80 MHz: no second channel offset
Could not set channel for kernel driver
Interface initialization failed
```

Attached please find the complete hostapd.log file for this run.
[hostapd.log](https://github.com/user-attachments/files/17445652/hostapd.log)

Anything else you can suggest of think we should try? 

TIA!

buckazoidbob, 2024-10-22:
I can confirm that I am too running into this issue. The device driver both on a standard Debian 12 install and manually downloaded firmware. 

My iw list output shows this however:
**Supported Channel Width: neither 160 nor 80+80**

@morrownr can you provide the iw list for your device and see if it says the same? I think if device driver doesn't play nice with 80+80 this isn't going to work in WiFi 6 mode, maybe there is a difference between Raspian and stock Debian though?

Tested both the stock 6.1 and 6.10 backported kernels.

morrownr, 2024-10-22:
Hi @buckazoidbob 

> I think if device driver doesn't play nice with 80+80...

Um... if you are talking about a mt7921au chip based adapter then it is not going to play nice with 80+80 because it can only do 80. Period. That is the design limitation of that chip. A new WiFi 7 chip, the mt7925, is capable of 160 MHz channel width but no adapters are on the market yet. Then there is the mt7927 chips and it can do 320 MHz channel width and there are no adapters on the market with it either.

Programmerino, 2024-10-22:
On my machine, this was caused by not having `[HT40+]` in my `HT_CAPAB`. Maybe that will help?

pavelhrubec, 2024-10-22:
I don't understand all the context, but setting ht_capab helped me solve the same problem. Thank you everyone!

morrownr, 2024-10-22:
> On my machine, this was caused by not having [HT40+] in my ht_capab. Maybe that will help?

I was doing a little clean up on my example hostapd.conf for WiFi 6 this morning. I had some notes that needed to be taken care of. It has been a beta or rc for nearly a year and I think it is good enough to be considered a release. Would appreciate everyone looking it over or testing it. Your feedback would be appreciated:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

DurvalMenezes, 2024-10-22:
@buckazoidbob:   
 > **Supported Channel Width: neither 160 nor 80+80**

Just for the record, not my case here: 
```
# iw list
   (...)
            Band 2:
   (...)
                   VHT Capabilities (0x339071f6):
   (...)
                          Supported Channel Width: 160 MHz
```

@morrownr: 
> Um... if you are talking about a mt7921au chip based adapter then it is not going to play nice with 80+80 because it can only do 80. Period. That is the design limitation of that chip.

Mine is the EDUP EP-AX1672, so mt7921 'plain' (no letters after the 4 digits), so it should support 160Mhz, as show by the `iw list` output above, correct?

DurvalMenezes, 2024-10-22:
@Programmerino:
> On my machine, this was caused by not having `[HT40+]` in my `ht_capab`. Maybe that will help?

@morrownr:
> I was doing a little clean up on my example hostapd.conf for WiFi 6 this morning. I had some notes that needed to be taken care of. It has been a beta or rc for nearly a year and I think it is good [...]

It works!!!!

```
# /usr/sbin/hostapd /etc/hostapd/hostapd-WiFi6.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan1: interface state COUNTRY_UPDATE->HT_SCAN
wlan1: interface state HT_SCAN->ENABLED
wlan1: AP-ENABLED 
wlan1: STA XX:XX:XX:XX:XX:XX IEEE 802.11: authenticated
wlan1: STA XX:XX:XX:XX:XX:XX IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED XX:XX:XX:XX:XX:XX
wlan1: STA XX:XX:XX:XX:XX:XX RADIUS: starting accounting session 7920531F2DE3CF2F
wlan1: STA XX:XX:XX:XX:XX:XX WPA: pairwise key handshake completed (RSN)
wlan1: EAPOL-4WAY-HS-COMPLETED XX:XX:XX:XX:XX:XX
wlan1: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
wlan1: STA XX:XX:XX:XX:XX:XX IEEE 802.11: authenticated
wlan1: STA XX:XX:XX:XX:XX:XX IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED XX:XX:XX:XX:XX:XX
wlan1: STA XX:XX:XX:XX:XX:XX RADIUS: starting accounting session 09004FFDAE261DE9
wlan1: STA XX:XX:XX:XX:XX:XX WPA: pairwise key handshake completed (RSN)
wlan1: EAPOL-4WAY-HS-COMPLETED XX:XX:XX:XX:XX:XX
wlan1: AP-STA-DISCONNECTED XX:XX:XX:XX:XX:XX
```

What did you do? Just add  `[HT40+]` to `ht_capab` as pointed out by @Programmerino (and confirmed by @buckazoidbob), or something else/more? 

TIA!

morrownr, 2024-10-22:
@buckazoidbob 

Here is what I get with my Edup 1672 when I run `iw list`. The distro is RasPiOS 64 bit with kernel 6.6. The computer is a RasPi4B. 

```
Band 2:
...
VHT Capabilities (0x339071b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency

```

> ...as show by the iw list output above, correct?

That is interesting. Coding errors are made at times. I really can't accound for what you posted. Here is what I think I know:

Chip

mt7921 - WiFi 6, dual band, max 80 MHz channel width
mt7921k - WiFi 6, tri band, max 80 MHz channel width
mt7921au -  WiFi 6, tri band, max 80 MHz channel width
mt7922 -  WiFi 6, tri band, max 160 MHz channel width in client mode, 80 otherwise
mt7925 - Wifi 7, tri band,max 160 MHz channel width
mt7927 - Wifi 7, tri band,max 320 MHz channel width

The mt7921au, mt7925 and mt7927 are available for USB though we have not seen any wifi 7 adapters with the mt7925 yet. The driver for the mt7927 is not in the kernel yet.

I have a mt7925 based M.2. It works well. The in-kernel driver works well. Wish we had some usb adapters available.

DurvalMenezes, 2024-10-23:
Thanks for the additional data, @morrownr -- much appreciated!

In fact, things are getting really complicated in wifi-land. And the cryptic error message from `hostapd` certainly didn't help.

Right now I'm testing for performance, and things are looking good! Connection speed from my Android smartphone is reported as 1200Mbps both Tx and Rx, and initial tests indicate very good performance.

Will make a final post with my results.

Thanks again! 👍👍👍

buckazoidbob, 2024-10-23:
> I was doing a little clean up on my example hostapd.conf for WiFi 6 this morning. I had some notes that needed to be taken care of. It has been a beta or rc for nearly a year and I think it is good [...]

Just tested it out on mine and it is working! Neat!

Not to derail this as I'm a total n00b when it comes to fiddling with hostapd, what kind of throughput is everyone getting with the mt7921au? I'm getting like...180 Mbps down and 150 up with this config. Seems like others maybe getting more, obviously lot of factors can play here (I have an existing AP running in the 36 and 149 range).

morrownr, 2024-10-23:
> what kind of throughput is everyone getting with the mt7921au?

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 52126 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  66.9 MBytes   561 Mbits/sec    0    850 KBytes       
[  5]   1.00-2.00   sec  68.8 MBytes   577 Mbits/sec    0    963 KBytes       
[  5]   2.00-3.00   sec  67.5 MBytes   566 Mbits/sec    0    963 KBytes       
[  5]   3.00-4.00   sec  70.0 MBytes   587 Mbits/sec    0    963 KBytes       
[  5]   4.00-5.00   sec  70.0 MBytes   587 Mbits/sec    0    963 KBytes       
[  5]   5.00-6.00   sec  70.0 MBytes   587 Mbits/sec    0    963 KBytes       
[  5]   6.00-7.00   sec  70.0 MBytes   587 Mbits/sec    0    963 KBytes       
[  5]   7.00-8.00   sec  68.8 MBytes   577 Mbits/sec    0    963 KBytes       
[  5]   8.00-9.00   sec  68.8 MBytes   577 Mbits/sec    0    963 KBytes       
[  5]   9.00-10.00  sec  71.2 MBytes   598 Mbits/sec    0    963 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   692 MBytes   580 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   689 MBytes   578 Mbits/sec                  receiver

```

So, to answer the question, about 580 Mbps. It could go as high as around 650 but the signal is going through 2 walls and is is not on a clean channel. This little AP is a RasPi4B using RasPiOS 64 bit. It is the Edup 1672 adapter.

pavelhrubec, 2024-10-23:
This FENVI mt7921au with WIFI 6 hostapd configuration offers a speed of only 270 Mbps. Does anyone have a similar experience?
[image]

DurvalMenezes, 2024-10-23:
> This FENVI mt7921au with WIFI 6 hostapd configuration offers a speed of only 270 Mbps. Does anyone have a similar experience?

For other adapters, please open a separate issue. We are al dealing with the EDUP 1672 here.

DurvalMenezes, 2024-10-23:
> Right now I'm testing for performance, and things are looking good! Connection speed from my Android smartphone is reported as 1200Mbps both Tx and Rx, and initial tests indicate very good performance.
> 
> Will make a final post with my results.

Running this on my Android phone, its wifi adapter is built-in the Qualcomm Snapdragon 778G (SM7325) SoC, under Termux, with iperf installed from the Termux repo, the iperf server `XXX.XXX.XXX.XXX` is the Linux machine with the EDAP 1732 adapter installed : 
```
~ $ iperf3 -v
iperf 3.15 (cJSON 1.7.15)Linux localhost 5.4.242-XXXXXXXX #1 SMP PREEMPT Tue May 21 03:18:42 UTC 2024 aarch64  Optional features available: CPU affinity setting, IPv6 flow label, TCP congestion algorithm setting, sendfile / zerocopy, socket pacing, authentication, bind to device, support IPv4 don't fragment

~$ iperf3 -c XXX.XXX.XXX.XXX
(...)
[ ID] Interval           Transfer     Bitrate         Retr                                      
[  5]   0.00-10.00  sec   255 MBytes   214 Mbits/sec    0             sender                    
[  5]   0.00-10.03  sec   255 MBytes   213 Mbits/sec                  receiver
```

Quite disappointing :-1: specially when compared to the exact same client and the exact same server, but connected through a Starlink Gen2 router which is then connected to the same server via 1Gbps wired Ethernet: 

```
~ $ iperf3 -c XXX.XXX.XXX.XXX
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   631 MBytes   529 Mbits/sec    0             sender                    
[  5]   0.00-10.00  sec   631 MBytes   529 Mbits/sec                  receiver
```

More than twice the speed! And to make things worse: 

* The Android "Network details" screen reports the Starlink router as being only Wi-Fi 5, with maximum connection speed 866Mbps, and when connected to the EDUP, it reports Wi-Fi 6 and 1200Mbps connection speed as I mentioned before; 
* the EDUP adapter is configured to channel 149, which a wifi sniffer shows is not being used by anything else, while the Starlink router is using channel 36 which is shared with the neighbors' AP (I'm in a rural setting so only one neighbor).

So the EDUP adapter shows less than half (40%) of the Starlink router performance, despite having everything in its favor :-( :-(

@morrownr, anything I could/should try to get better performance?

DurvalMenezes, 2024-10-23:
> The first thing I would check is whether the Edup adapter is running in USB2 or USB3 mode:
> 
> $ lsusb -t
> 
> I know you are going to think "it should be" but those are famous last words. Check it.

You are absolutely correct: knowing is always better than supposing. And, at the present case, you are even more right:  

```
$ lsusb -t
    (...)
                |__ Port 3: Dev 14, If 0, Class=Vendor Specific Class, Driver=mt7921u, 480M
```

WTaF?! I checked further and it seems to be a cabling/port issue, moved it to another port and voilá:

```
$ lsusb -t
    (...)
            |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=mt7921u, 5000M
```

So now here's what I get, same conditions as above (except the placement/position of the EDUP adapter, which I tried and compensated for by rotating its antennas to be in about the same orientation): 

```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   601 MBytes   504 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   601 MBytes   504 Mbits/sec                  receiver
```

So, not that far from the numbers you are seeing, but still missing 580/504-1= 15% of your EDUP performance. 

Any more hints/tips to achieve your numbers?  

Thank you very much for the assist!

morrownr, 2024-10-23:
@DurvalMenezes 

> Any more hints/tips to achieve your numbers?

I am currently working on a couple of drivers and in about a week, I will need to do AP mode testing. When I do this, I do a full clean reinstallation along with blowing the dust out and checking connections. This gives me a clean platform to test my AP guide and example hostapd.conf files. That is also a good time to tweak performance. Give me a shout here in about a week and we can work on this.

In the meantime, use your AP and see if any problems come up. One day does not make a stable setup. But 500 Mbps is pretty darn good.

Question: Do you have any other modern adapters? ... WiFi 6 or WiFi 6e?

buckazoidbob, 2024-10-24:
> So, to answer the question, about 580 Mbps. It could go as high as around 650 but the signal is going through 2 walls and is is not on a clean channel. This little AP is a RasPi4B using RasPiOS 64 bit. It is the Edup 1672 adapter.

Interesting. After doing some testing I realized something pretty dumb, I was testing this upstairs. After a moment I realized even my "real" 5Ghz AP was only pulling 230-270 Mbps up there. Downstairs, where the AP is, a device with a newer wifi card gives me an a pretty darn good 450-500 Mbps, much like @morrownr.

DurvalMenezes, 2024-10-24:
> Give me a shout here in about a week and we can work on this.

Thanks, will do! Next wednesday I will create a new issue titled "mt7921au performance tuning", mentioning this comment, and give you a shout there. 

> In the meantime, use your AP and see if any problems come up. One day does not make a stable setup. But 500 Mbps is pretty darn good.

Will do!  I've been using this EDUP adapter on Wifi6 intensively and with all my devices,  and so far no issues at all, it doesn't even warm up. Excellent purchase, thanks to your great write-up and recommendation. :+1:

> Question: Do you have any other modern adapters? ... WiFi 6 or WiFi 6e?

I do have a few: 
* Azurewave AW-XB591NF M.2 2230 wireless card with a MediaTek 7922 chip. It supports WiFi 6E and Bluetooth 5.3; came installed on my Geekom A7, not being used with Wifi at the moment (due to unrelated matters related to its bluetooth support which I'm trying to debug).
* Fenvi FU-AX1800 WiFi6-only USB, also as per your recommendation. I use it with WiFi4 (for telemetry purposes, accepting connections from Tasmota devices I have in my home) and as a cheaper backup for the EDUP in case it ever fails. 
* Intel AX210 NGW Bluetooth 5.3 M.2 2230 wireless card; not installed yet, will go either into my Geekom A7 (replacing its Azurewave/Mediatek above in case I can't get it to work as I want) or into my ZBook 17 laptop to upgrade its old Intel 8265 NGW. 

If I and/or my adapters can be of any help with your testing, please let me know -- I'm a little short on available time, but if I can it will be my honor to assist.

DurvalMenezes, 2024-10-24:
> Downstairs, where the AP is, a device with a newer wifi card gives me an a pretty darn good 450-500 Mbps, much like @morrownr.

With less than 1m of distance I'm getting ~500Mbps which is about 15% less than the 580Mbps @morrownr manages with his setup. I'm not complaining of course, but I do look forward to further optimizations. 

Keep an eye here for the issue I'm going to open next week (and point to here) regarding that, if you want those extra 15%.

arrio464, 2025-02-27:
Although I don't know what happened, adding [HT40+][HT40-] to ht_capab fix the problem.
I am using `Realtek Semiconductor Corp. RTL8812AU-VS 802.11a/b/g/n/ac 2T2R DB WLAN Adapter` with rtw88 driver.
here is my hostapd config:
```
beacon_int=100
ssid=ssid
interface=x0wlan0
driver=nl80211
channel=149
vht_oper_centr_freq_seg0_idx=155
ctrl_interface=/dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.yrW99O/hostapd_ctrl
ctrl_interface_group=0
ignore_broadcast_ssid=0
ap_isolate=0
country_code=CN
ieee80211d=1
hw_mode=a
ieee80211n=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]
vht_oper_chwidth=1
wmm_enabled=1
wpa=2
wpa_passphrase=password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
```

closed 2025-11-02 by morrownr