morrownr/USB-WiFi #521: Quick question: Are the firmware updates permanent / flashed to the device?

rorar, 2024-10-13:
Hi,
are the firmware updates https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.mdfor MT7922 devices flashed to the device or e.g. are they loaded to the kernel?
I'm pretty new to this and like to understand how this works. :)
Thanks!

morrownr, 2024-10-14:
Hi @rorar 

@ieffuie4 is correct but your question can have different meanings so please allow me to provide a longer story:

Modules, commonly called drivers, generally consist of 2 or more files. All are part of the "driver" because if one or more driver files is missing, it is like no driver is installed... the hardware simply does not function. To separate out the files, one or more files will be called firmware. The firmware also goes by the name of binary blob(s). Firmware is not part of the kernel, it is part of the distro and is added and updated by the maintainers of your distro. The driver part of the driver, am I making sense?, is part of the kernel and is regularly updated by those kernel devs that maintain kernel drivers and by folks like you and I that know something about programming and figure out a fix for a bug. Everytime you see a new kernel flow into your system as an update, you are getting a new driver, however, you are not getting new firmware with that kernel upgrade.

Distro maintainers can flow a firmware upgrade into your system just like anything else that can be upgraded. Generally kernel upgrades, and the drivers they contain, are upgraded more often than firmware. Firmware that has stabilized may not need an update for years.

Firmware files are easier for users to upgrade, just follow the guide that I have on the Main Menu. The kernel part of the driver can be a challenge if you have a patch and only want to apply the patch. I guess I should include a guide for that but it would not be easy if you know know some things about programming.

During the boot process, all required firmware files and the "driver" for any hardware will loaded the firmware and "driver". If any of the firmware files or driver file is missing, the hardware won't work.

Why firmware, this is where the stuff that companies deem to be kept secret is placed. Intellectual property would be an example. It is common for community programmers to have to ask the company gatekeeper for help when working on a driver because we don't know what is not available to us and this stuff is complicated.

rorar, 2024-10-14:
That's some nice Linux magic sprinkles topping for flawless hardware/firmware patching! Thanks for the explanations!
I believe I did something (more or less) similiar for my unRAID system, adding a newer microcode update pre-boot inside syslinux config.

 Is there an actual way to flash the firmware bin files permanently to my wifi cards firmware rom? 
 Background: (On Windows)  If I end hibernation xor standby mode, the wifi card won't "wake up". Or is there a similiar way you've mentioned for Windows?

morrownr, 2024-10-14:
@rorar 

> That's some nice Linux magic sprinkles topping for flawless hardware/firmware patching! 

Thanks. I try to keep this site loaded with practical information that will benefit users.

> Is there an actual way to flash the firmware bin files permanently to my wifi cards firmware rom?

No. There ROM on our adapters is usually really small and when you are looking at firmware that exceed 1 mb in some cases, there is no room but that does not matter because ROM stands for Read Only Memory which means burn it once and that is it. No more writes. Things like BIOSs are flashable memory, ROM is not.

> Background: (On Windows) If I end hibernation xor standby mode, the wifi card won't "wake up".

Let me get this correct: On Windows 11 you are seeing a problem with mt7922 based M.2 or PCIe card not waking up from suspend properly? Is that right? If so, your best bet is to investigate the version of the Windows driver you are using and see if you can find an updated version of the driver so as to install it.

Keep in my that my technical knowledge of Windows is lacking as my history  with small computers includes CP/M, PC-DOS, DR-DOS, OS/2 and Linux. Note the lack of Windows but I think I am correct. You do not have to worry about firmware on Windows as the entire OS is binary blobs... nothing is available for the user community to work on.

Hope this helps.