morrownr/USB-WiFi #524: EDUP AX1672 can't connect to an ssid with channel 149 in 5 GHz

Shallisteria, 2024-10-14:
Hello guys, I'm a newbie from Indonesia in here so please go easy on me

I decided to buy EDUP AX1672 after reading @morrownr review and suddenly got this problem. It won't connect on both windows and linux using the default setting in the router, after i changed the region and set the channel to "Auto (DFS)" then the router set the channel to 36 at 80 MHz that the adapter can connect.

Maybe the router is the problem? I haven't got any chance to try with other router. The router in question is a second-hand cheap Totolink A3002R. but if that's the case my Intel AX210 doesn't have this problem and can connect before i changed the settings

morrownr, 2024-10-14:
Hi @Shallisteria 

With the computer that is running Linux and the Edup adapter, open a terminal (command prompt) and run the following:

$ iw reg get

Post the results.

What do you think are the allow 5 GHz channels in Indonesia?
What Linux distro and version are you using?

This likely not an issue with adapter but rather a regulatory issue.

Shallisteria, 2024-10-14:
Here's some addtional info:

$ hostnamectl   
 Static hostname: kali
       Icon name: computer-laptop
         Chassis: laptop 💻
      Machine ID: be363409e43b4f868cb670ee9e283e6b
         Boot ID: c4d28b4220fc4e839248b89a9c6b5995
Operating System: Kali GNU/Linux Rolling           
          Kernel: Linux 6.10.9-amd64
    Architecture: x86-64
 Hardware Vendor: ASUSTeK COMPUTER INC.
  Hardware Model: VivoBook_ASUSLaptop X412DA_A412DA
Firmware Version: X412DA.316
   Firmware Date: Fri 2021-10-08
    Firmware Age: 3y 6d
=======================================================================
$ iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country ID: DFS-UNSET
        (2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
        (2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
        (2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
        (5170 - 5190 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5190 - 5210 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5210 - 5230 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5230 - 5250 @ 80), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5250 - 5270 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5270 - 5290 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5290 - 5310 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5310 - 5330 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
        (5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ
        (5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ
        (5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ
        (5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ
=======================================================================
$ iwlist wlan0 freq
wlan0     25 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 149 : 5.745 GHz
          Channel 153 : 5.765 GHz
          Channel 157 : 5.785 GHz
          Channel 161 : 5.805 GHz
          Current Frequency=5.18 GHz (Channel 36)

This is my Intel AX210
=======================================================================
iwlist wlan1 freq     
wlan1     32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 14 : 2.484 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Current Frequency=5.18 GHz (Channel 36)

This is from the newly bought EDUP adapter
=======================================================================

Shallisteria, 2024-10-14:
Regarding regulations i can't find a data that is good enough to show it here but i found multiple people that says Indonesia regulation Indonesia permits up to VHT40 for 2.4 Ghz (1-13) and VHT80 for 5 Ghz (32-68 and 149-161). And I agree with this since every router or access point that I encounter when working, on college, or public facility uses channel 149-161.

morrownr, 2024-10-14:
@Shallisteria 

The Intel card is missing the DFS channels and the Mediatek adapter is missing the upper channels. This is telling us something but we need to figure out what that is. Does your router support DFS channels? If so, you could try DFS and see what happens.

Shallisteria, 2024-10-14:
Here's some more additional info:
[image]
That's the list when i set the region to "Other"

[image]
That's Indonesian Region

So I tried to set the channel to 128 and AX210 didn't find the SSID, the EDUP found the SSID but won't connect, and after that the router automatically changed the channel to 60(the settings didn't change tho, still set on 128 but broadcasting on channel 60)

May I see what your EDUP adapter iwlist freq looks like sir? If it's different i can try to argue with the seller to maybe replace my adapter

I often use the adapter normally not just for learning tho so it's a bummer if i can't connect no matter how good the adapter is

morrownr, 2024-10-15:
> May I see what your EDUP adapter iwlist freq looks like sir?

Sure. Here is `iw list` output with everything but the channels removed: (Country Code = US)

```
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)

		Frequencies:
			* 5180 MHz [36] (23.0 dBm)
			* 5200 MHz [40] (23.0 dBm)
			* 5220 MHz [44] (23.0 dBm)
			* 5240 MHz [48] (23.0 dBm)
			* 5260 MHz [52] (24.0 dBm) (radar detection)
			* 5280 MHz [56] (24.0 dBm) (radar detection)
			* 5300 MHz [60] (24.0 dBm) (radar detection)
			* 5320 MHz [64] (24.0 dBm) (radar detection)
			* 5500 MHz [100] (24.0 dBm) (radar detection)
			* 5520 MHz [104] (24.0 dBm) (radar detection)
			* 5540 MHz [108] (24.0 dBm) (radar detection)
			* 5560 MHz [112] (24.0 dBm) (radar detection)
			* 5580 MHz [116] (24.0 dBm) (radar detection)
			* 5600 MHz [120] (24.0 dBm) (radar detection)
			* 5620 MHz [124] (24.0 dBm) (radar detection)
			* 5640 MHz [128] (24.0 dBm) (radar detection)
			* 5660 MHz [132] (24.0 dBm) (radar detection)
			* 5680 MHz [136] (24.0 dBm) (radar detection)
			* 5700 MHz [140] (24.0 dBm) (radar detection)
			* 5720 MHz [144] (24.0 dBm) (radar detection)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (27.0 dBm) (no IR)
			* 5865 MHz [173] (27.0 dBm) (no IR)
			* 5885 MHz [177] (27.0 dBm) (no IR)

		Frequencies:
			* 5955 MHz [1] (12.0 dBm) (no IR)
			* 5975 MHz [5] (12.0 dBm) (no IR)
			* 5995 MHz [9] (12.0 dBm) (no IR)
			* 6015 MHz [13] (12.0 dBm) (no IR)
			* 6035 MHz [17] (12.0 dBm) (no IR)
			* 6055 MHz [21] (12.0 dBm) (no IR)
			* 6075 MHz [25] (12.0 dBm) (no IR)
			* 6095 MHz [29] (12.0 dBm) (no IR)
			* 6115 MHz [33] (12.0 dBm) (no IR)
			* 6135 MHz [37] (12.0 dBm) (no IR)
			* 6155 MHz [41] (12.0 dBm) (no IR)
			* 6175 MHz [45] (12.0 dBm) (no IR)
			* 6195 MHz [49] (12.0 dBm) (no IR)
			* 6215 MHz [53] (12.0 dBm) (no IR)
			* 6235 MHz [57] (12.0 dBm) (no IR)
			* 6255 MHz [61] (12.0 dBm) (no IR)
			* 6275 MHz [65] (12.0 dBm) (no IR)
			* 6295 MHz [69] (12.0 dBm) (no IR)
			* 6315 MHz [73] (12.0 dBm) (no IR)
			* 6335 MHz [77] (12.0 dBm) (no IR)
			* 6355 MHz [81] (12.0 dBm) (no IR)
			* 6375 MHz [85] (12.0 dBm) (no IR)
			* 6395 MHz [89] (12.0 dBm) (no IR)
			* 6415 MHz [93] (12.0 dBm) (no IR)
			* 6435 MHz [97] (12.0 dBm) (no IR)
			* 6455 MHz [101] (12.0 dBm) (no IR)
			* 6475 MHz [105] (12.0 dBm) (no IR)
			* 6495 MHz [109] (12.0 dBm) (no IR)
			* 6515 MHz [113] (12.0 dBm) (no IR)
			* 6535 MHz [117] (12.0 dBm) (no IR)
			* 6555 MHz [121] (12.0 dBm) (no IR)
			* 6575 MHz [125] (12.0 dBm) (no IR)
			* 6595 MHz [129] (12.0 dBm) (no IR)
			* 6615 MHz [133] (12.0 dBm) (no IR)
			* 6635 MHz [137] (12.0 dBm) (no IR)
			* 6655 MHz [141] (12.0 dBm) (no IR)
			* 6675 MHz [145] (12.0 dBm) (no IR)
			* 6695 MHz [149] (12.0 dBm) (no IR)
			* 6715 MHz [153] (12.0 dBm) (no IR)
			* 6735 MHz [157] (12.0 dBm) (no IR)
			* 6755 MHz [161] (12.0 dBm) (no IR)
			* 6775 MHz [165] (12.0 dBm) (no IR)
			* 6795 MHz [169] (12.0 dBm) (no IR)
			* 6815 MHz [173] (12.0 dBm) (no IR)
			* 6835 MHz [177] (12.0 dBm) (no IR)
			* 6855 MHz [181] (12.0 dBm) (no IR)
			* 6875 MHz [185] (12.0 dBm) (no IR)
			* 6895 MHz [189] (12.0 dBm) (no IR)
			* 6915 MHz [193] (12.0 dBm) (no IR)
			* 6935 MHz [197] (12.0 dBm) (no IR)
			* 6955 MHz [201] (12.0 dBm) (no IR)
			* 6975 MHz [205] (12.0 dBm) (no IR)
			* 6995 MHz [209] (12.0 dBm) (no IR)
			* 7015 MHz [213] (12.0 dBm) (no IR)
			* 7035 MHz [217] (12.0 dBm) (no IR)
			* 7055 MHz [221] (12.0 dBm) (no IR)
			* 7075 MHz [225] (12.0 dBm) (no IR)
			* 7095 MHz [229] (12.0 dBm) (no IR)
			* 7115 MHz [233] (12.0 dBm) (no IR)
```

> I often use the adapter normally not just for learning tho so it's a bummer if i can't connect no matter how good the adapter is

I do not think this issue is the adapters fault. My adapter is showing channel 149+. Hang in there and lets figure this out.

Tell me what Linux distro, and kernel, you are using and also what version of Windows.

Also, When the router changed the channel from 128 DFS to 60, that is normal behavior if you have something conflicting. Now try channel 100.

Shallisteria, 2024-10-15:
Cmiiw but isn't this output tells my Linus distro and version or am i wrong?

$ hostnamectl
Static hostname: kali
Icon name: computer-laptop
Chassis: laptop 💻
Machine ID: be363409e43b4f868cb670ee9e283e6b
Boot ID: c4d28b4220fc4e839248b89a9c6b5995
Operating System: Kali GNU/Linux Rolling
Kernel: Linux 6.10.9-amd64
Architecture: x86-64
Hardware Vendor: ASUSTeK COMPUTER INC.
Hardware Model: VivoBook_ASUSLaptop X412DA_A412DA
Firmware Version: X412DA.316
Firmware Date: Fri 2021-10-08
Firmware Age: 3y 6d

This is the version of windows that I'm using:

Edition	Windows 11 Pro
Version	23H2
Installed on	‎22/‎07/‎2023
OS build	22631.4112
Experience	Windows Feature Experience Pack 1000.22700.1034.0

additional info: This problem also happens to my little sister's Windows 10 Pro 22H2 Laptop

When I set the channel to 100 the same happens like before, except the channel doesn't automatically change to any other channels

morrownr, 2024-10-15:
> ... isn't this output tells my Linus distro and version or am i wrong?

You are correct but think about who you are working with. You are dealing with one problem and the problem is first hand. At any given time I may be dealing with 20 or more problems second hand so even though I try to take time to scan back up the threads to remind myself what the situation is, sometimes I simply miss the details. Thanks for posting again.

> Kernel: Linux 6.10.9-amd64

Kernel 6.10 has not been kind to wifi. There have been a lot of patches to the wifi stack going in.

Are you up for trying a different distro for testing? You don't even need to install the distro. Just boot it from a flash drive and let's see what we get. Download and burn the newly released Ubuntu 24.10 to a flash drive. It uses kernel 6.11. Boot the flash drive. Don't install, just use it. Check the results with your Edup adapter to see if we get different results.

> When I set the channel to 100 the same happens like before, except the channel doesn't automatically change to any other channels

The channel not changing indicates that is a clean DFS channel that you can use without getting bumped off. So you are saying the Edup adapter would not connect?

Shallisteria, 2024-10-15:
You sound a little upset with the first reply, I'm sorry if what i typed seems a litlle condescending but i'm just trying to make sure the information is correct, i'm just copy-pasting so i don't mind repeating the information if asked. Thanks for taking the time to helping

As for changing distro I can't do it this week since i ran out of allowance to buy flashdrive, will do after getting the money next week

And lastly channel 100 won't even show up in scan but it connected to channel 52(i took time to try with Auto again and the router set the channel to 52 and it connected, not because the channel bounced off from channel 100) I even made sure to reboot and wait a while for each change since this is a cheap router so it's slow to scan the area

I can only check the upper channel problem in Thursday or Friday in other router to make sure this whole problem is not because of my shitty router

morrownr, 2024-10-15:
> You sound a little upset with the first reply...

Oh, no. Aggravated with myself because the older I get, the harder it gets to keep up with things.

Try this:

Manually configure your regulatory domain as option for the cfg80211 module by editing
/etc/modprobe.d/cfg80211.conf and adding this line: options cfg80211 ieee80211_regdom=ID

$ sudo nano /etc/modprobe.d/cfg80211.conf

paste  `options cfg80211 ieee80211_regdom=ID` into the document and then save:

Ctrl-X, y, Enter
Reboot
Check what channels are shown with `iw list`

-----

> because of my shitty router...

Don't worry. We'll figure something out and learn something on the way. Knowing about wifi has its benefits. My family has enjoyed wifi that always works for many years.

gdluca, 2024-10-16:
Hi @Shallisteria ,

I am not a stable internet and can't do much reasearch but I have a few things that you might check:

- on Windows, what drivers did you use? If you used the official EDUP one, you can uninstall them and try updated drivers from Netgear → https://www.downloads.netgear.com/files/GDC/A8000/A8000_Win_V1.0.0.125.zip
 Note: EDUP has 1.0.0.68 version. Be aware that during the installation, it prompts you if you want disable other Wifi adapters if present. You can just say no(in case you click Yes, you can still reenable adapters from Device Manager).

- have you tried updating the router firmware? I am not sure if yours is a ISP branded one so it might not work.
  https://www.totolink.net/home/menu/detail/menu_listtpl/download/id/193/ids/36.html

Shallisteria, 2024-10-17:
> * have you tried updating the router firmware? I am not sure if yours is a ISP branded one so it might not work.
>   https://www.totolink.net/home/menu/detail/menu_listtpl/download/id/193/ids/36.html

My version is v2 tho
[image]

I'll try your suggested driver later today, thanks for the suggestion

Shallisteria, 2024-10-19:
Ok so Me and the seller decided to return the adapter since the window for returning is closing, I also tried the adapter on campus and public wifi and it won't connect since the channel for 5 GHz is in the upper range (149-173). It's a shame since i actually has many plans to do with the adapter since it has good range and comes with removeable antenna

morrownr, 2024-10-20:
@Shallisteria 

Well, we tried.

closed 2024-10-28 by morrownr