morrownr/USB-WiFi #526: Why MT7921 USB Not Supporting Mesh Point Mode? Hardware / Firmware or Driver Issue?

marsCatXdu, 2024-10-21:
I've searched a lot around, only got a conclusion ["MT7921 does not support mesh mode"](https://github.com/openwrt/mt76/issues/653#issuecomment-1166270724). But I wonder why, is it a problem that can't be solved in driver? 
If it can be implemented by extending the driver, I'm willing to learn and give it a try. I really need a WiFi6 Mesh...

Looking forward to a reply from experts, thank you so much.

morrownr, 2024-10-21:
Hi @marsCatXdu 

> But I wonder why, is it a problem that can't be solved in driver?

As far as I can tell, this is simply a case where the feature has not been added. The Mediatek team is receptive to patches. Do you subscribe to linux-wireless?

> I really need a WiFi6 Mesh...

I am not aware of any WiFi 6 adapters that have Mesh support. I think it is available if you drop down to WiFi 5.

marsCatXdu, 2024-10-22:
> Hi @marsCatXdu
>
>
> As far as I can tell, this is simply a case where the feature has not been added. The Mediatek team is receptive to patches. Do you subscribe to linux-wireless?
>
>
> I am not aware of [...]

Thank you for the information!

I'm new to kernel and driver, not familiar with how community runs and didn't subscribe it before. I will find the list try to reach them to ask. Thank you very much!

closed 2024-10-22 by marsCatXdu

morrownr, 2024-10-22:
If you need a link to learn about and subscribe to linux-wireless, I can help. Recommend you not just jump right it or you will likely be ignored. It is a busy place so learning first and talking later is a good idea. You can use the time read up on wireless development.

marsCatXdu, 2024-10-22:
> If you need a link to learn about and subscribe to linux-wireless, I can help. Recommend you not just jump right it or you will likely be ignored. It is a busy place so learning first and talking later is a good idea. You can use the time read up on wireless development.

Thanks, I've already subscribed. Thank you for your kind advice!

morrownr, 2024-10-22:
You are welcome. Reading for a while should give you are idea of how kernel development is done.

One of the first skills you need to learn is how to make and submit a patch. I am preparing some patches right now. I am in no hurry so if you want to learn by sending one in... we can do that. You will need to do some homework first so let me know when you are ready.

How are your C programming skills?

marsCatXdu, 2024-10-23:
Wow amazing, so kind of you, but will this directly taking advantage of your work? Maybe you can show me a simple start point to work on, maybe some parts of your work, so that I can do it myself and also help you a bit. 

I'm ready now, maybe we can directly communicate by emails?

I have no much experience on C programming, never touched driver. I'm good at C++ and capable for design and build a multi-thread program consists of about 10k lines of C++ code project myself, So I believe I can catch up with you!

morrownr, 2024-10-25:
Sorry for the delay. I have been very busy.

> Maybe you can show me a simple start point to work on, maybe some parts of your planning future work, so that I can do it myself and also help you a bit.

A simple start point would be good. Do you have any other usb wifi adapters? I am trying to think of ways for you to learn about git.

I'll get my email address to you soon.

If you are familiar with c++, that helps. C is not my favorite language but I keep a couple of reference books handy. Don't worry about that for now. You need to first learn git and then we can look at drivers.

marsCatXdu, 2024-10-26:
Thanks, I'm quite familiar with git, I also use it for work. I'm have necessary common programming skills, also tried build kmods, we can skip some steps.

I have ralink 3070, netgear a6210, tplink wn722n, rtl8821cu and EDUP EP-N8566.

marsCatXdu, 2024-10-26:
I'm just really interested in learning how to write drivers, getting deeper understanding in WLAN, and making some contributions to community. I will think some other ways to deal with my research about mesh so I'm not in a hurry. 

So we can do it in a relaxed way, it would be great if I can learn something while helping you and making some contributions.

marsCatXdu, 2024-10-28:
I have an idea about learning - how about I read and try to write documents for your existing WLAN drivers? I can learn WLAN drivers, I can make contributions, and you can help me on feedback. I will try to make things brief and simple to save your time.