morrownr/USB-WiFi #528: Possible New Adapter with mt7921au chip - Edup EP-AX1673

morrownr, 2024-10-25:
I do not have this adapter, at least not yet, so this information should be considered an educated guess until we have solid information to confirm the details.

The Edup EP-AX1673 came on the market recently. I have been watching it. It appears to be a single-state (no windows driver onboard) and single-function (no bluetooth). Those are both qualities that we want to help ensure solid stable operation. It also appears to use the mt7921au chip, WiFi 6, which is supported with a good in-kernel driver. Based on a couple of reviews, it likely uses the standard Mediatek VID/PID for this chip making it immediately plug and play on any distros using kernel 5.19 or later.

https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0DCBWDTG5

This is a low cost adapter. The Edup EP-AX1672 adapter is already listed in the Plug and Play list. Numerous Linux user have reported buying the EP-AX1672 and the reports generally indicate good results. It appears that the only major difference between these 2 adapters is that this new 1673 is smaller so if a smaller adapter better meets your needs, you might consider the new 1673. However, I will not recommend or place the 1673 in the Plug and Play List until we have confirmation that this new adapter is what we think it is.

-----

Changing subject: I need some help keeping an eye open for new usb adapters with the mt7925 chip. The mt7925 chip is a WiFi 7 with 160 MHz wide channel width. Cards with this chip are already on the market, I have a little M.2 card with this chip. I like it. The driver for this chip has been in the Linux kernel since v6.7. USB adapter with the mt7925 chip should be advertized as BE6500 as opposed to AX3000 or AXE3000 for the mt7921au chip. Please let us know if you come across an adapter using the mt7925 chip.

Thanks.

Kiwimannen, 2024-10-26:
Yes, its available on Edups official Aliexpress store and it says in the specifications that it is a MTK7921AU chip: 
https://www.aliexpress.com/item/1005007693983500.html

BTW, the lowest priced MTK7921AU must be this one:
https://www.aliexpress.com/item/1005006200962032.html
I can confirm that it actually is a MTK7921AU inside.

Have not seen any 7925 USB yet.

morrownr, 2024-10-26:
@Kiwimannen 

Thanks for the info. I added the AliExpress link in the original message.

The cheap adapter was once in the Plug and Play List but some reports about Comfast changing the chipset for a v2 of the adapter caused me to pull the adapter out of the Plug and Play List. Comfast is acting like TP-Link and that does not work well for Linux users.

Help me keep an eye out for mt7925 based adapters, They will come. USB always comes after the cards and the cards have been out for a while now.

kylefmohr, 2024-10-27:
Hey, I thought that model name looked familiar

[image]

I haven't had a chance to test it but I'll start working on that now. Anything in particular you'd like me to test or benchmark with it?

kylefmohr, 2024-10-27:
Hey, I thought that model name looked familiar

[image]

I haven't had a chance to test it but I'll start working on that now. Anything in particular you'd like me to test or benchmark with it? @morrownr

morrownr, 2024-10-27:
@kylefmohr 

Hey, thanks for the post. Just what we need.

The main thing I am looking for is does it meet the criteria to be included in the Plug and Play List. Here are some of the details that I look for:

- It must use a chip that is supported by a good in-kernel, standards compliant driver. If the chip is the mt7921au, then that is good. You can post the results of the following command to verify:

$ lsusb

- The adapter must be single-state. That is no windows driver in storage on the adapter as that can be problematic. You can test that by turning usb_modeswitch off, it comes up as a wifi adapter after usb_moderswitch is off, it is a single state adapter:

```
How to deactivate usb_modeswitch:

$ sudo nano /etc/usb_modeswitch.conf

Change

DisableSwitching=0

to

DisableSwitching=1

sudo reboot

```

- Provide a link to where you bought it and whether you had any problems.

- Use it for 2-3 days and describe the performance.

- Give us an opinion as the quality, size and appearance.

- Would you recommend it to others? Should it be in the Plug and Play List?

That is about it. Looking forward to your report.

@morrownr

kylefmohr, 2024-10-27:
> It must use a chip that is supported by a good in-kernel, standards compliant driver. If the chip is the mt7921au, then that is good.

<details><summary>Output of `hwinfo --network` confirms it's mt7921u</summary>
<p>

```
hwinfo --network
58: None 00.0: 1070a WLAN                                       
  [Created at net.126]
  Unique ID: AYEt.QXn1l67RSa1
  Parent ID: LybY.poFsugP8YwE
  SysFS ID: /class/net/wlan0
  SysFS Device Link: /devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-4/3-4.1/3-4.1:1.0
  Hardware Class: network interface
  Model: "WLAN network interface"
  Driver: "mt7921u"
  Driver Modules: "mt7921u"
  Device File: wlan0
  HW Address: 52:57:5e:<redacted>
  Permanent HW Address: e8:4e:06:<redacted>
  Link detected: no
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #52 (Unclassified device)

```
I redacted the second half the MAC addresses, not sure if that's necessary but thought it wouldn't hurt.
</p>
</details> 

> The adapter must be single-state. That is, there is no windows driver in storage on the adapter. That can be problematic. You can test that by turning usb_modeswitch off, it comes up as a wifi adapter after usb_moderswitch is off, it is a single state adapter:

Changed value in /etc/usb_modeswitch.conf to 1, rebooted, still just shows up as a WiFI adapter with no external storage device appearing.

> Provide a link to where you bought it and whether you had any problems.

[This Amazon listing](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0DCBWDTG5) (sidenote: I received this adapter for free in exchange for an honest review of the product. I was not otherwise compensated in any way)

I'll use the adapter in the next few days and add another comment once I get a better feel for its performance. @morrownr

morrownr, 2024-10-28:
@kylefmohr 

Thank you for confirming the chipset and that the adapter is single-state (no storage onboard). I think I am going to go ahead and start adding this to the Plug and Play List as reviews have been positive so users looking for a smaller adapter with have another options. Please take your time and post a review after you have used the adapter for a while.

nightshark, 2025-04-29:
@kylefmohr How is the adapter working for you?

nightshark, 2025-05-12:
I bought the EP-AX1673 for OpenWRT on a raspberry PI 4. It's been working great!

nightshark, 2025-05-12:
@morrownr I can confirm band 4 (6Ghz) does not really work at all. I did not realize it was so restricted in the US until I really read your repo! Thanks for the detailed documentation!

morrownr, 2025-05-14:
@nightshark 

> I can confirm band 4 (6Ghz) does not really work at all.

The good news is that WiFi 6 on band 2 (5 GHz) is really fast. We are waiting on mt7925 chip based adapters and they will have support for 160 GHz channel width. I actually have a little M.2 card in a Mini PC that is based on the mt7925 chip and it shows 1.2 Gbps throughput in managed mode with my wifi router on channel 100, WiFi 6 and 160 MHz channel width. Yes, it smokes 1 gb ethernet.