morrownr/USB-WiFi #533: [Help]:

KaliAssistant, 2024-11-07:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux KALIworkNB1 6.11.2-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.11.2-1kali1 (2024-10-15) x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0408:4033 Quanta Computer, Inc. ACER HD User Facing Bus 001 Device 003: ID 04ca:3802 Lite-On Technology Corp. Wireless_Device ^C (endless)

### rfkill

0: acer-wireless: Wireless LAN 	Soft blocked: no 	Hard blocked: no 1: acer-bluetooth: Bluetooth 	Soft blocked: no 	Hard blocked: no 2: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 3: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
8812au/5.6.4.2_35491.20191025, 6.10.11-amd64, x86_64: installed
8812au/5.6.4.2_35491.20191025, 6.11.2-amd64, x86_64: installed
acpi_ec/v1.0.4, 6.10.11-amd64, x86_64: installed
acpi_ec/v1.0.4, 6.11.2-amd64, x86_64: installed
acpi_ec/v1.0.4, 6.6.9-amd64, x86_64: installed
```

### iw

```text
phy#0
	Interface wlan0 <-----(not this, it is my pcie nic)
		ifindex 3
		wdev 0x1
		addr 9c:2f:9d:79:4c:ab
		type managed
		channel 161 (5805 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

I have a ALFA AWUS036AXML, but when i plug in to my laptop (kali linux 6.11.2) USB3.0, driver crashed:
```
[ 3289.963601] usb 1-5: new high-speed USB device number 4 using xhci_hcd
[ 3290.113355] usb 1-5: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 3290.113360] usb 1-5: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 3290.113361] usb 1-5: Product: Wireless_Device
[ 3290.113362] usb 1-5: Manufacturer: MediaTek Inc.
[ 3290.113363] usb 1-5: SerialNumber: 000000000
[ 3290.196559] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 3317.419927] Bluetooth: hci1: command 0xfd98 tx timeout
[ 3317.419939] Bluetooth: hci1: failed to reset (-19)
[ 3317.419987] Bluetooth: hci1: Failed to apply iso setting (-110)
[ 3317.419991] Bluetooth: hci1: Device setup in 26651826 usecs
[ 3317.419992] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 3319.435952] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[ 3319.436229] list_del corruption, ffff8f73d475a000->next is LIST_POISON1 (dead000000000100)
[ 3319.436242] ------------[ cut here ]------------
[ 3319.436243] kernel BUG at lib/list_debug.c:56!
[ 3319.436248] Oops: invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
[ 3319.436252] CPU: 8 UID: 0 PID: 685 Comm: kworker/8:2 Tainted: G           OE      6.11.2-amd64 #1  Kali 6.11.2-1kali1
[ 3319.436255] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[ 3319.436260] Hardware name: Acer Aspire A515-57/Ganymede_ADU, BIOS V1.16 08/16/2022
[ 3319.436262] Workqueue: usb_hub_wq hub_event [usbcore]
[ 3319.436294] RIP: 0010:__list_del_entry_valid_or_report.cold+0x5c/0x6f
[ 3319.436298] Code: e8 68 60 fe ff 0f 0b 48 89 fe 48 89 ca 48 c7 c7 40 34 54 8d e8 54 60 fe ff 0f 0b 48 89 fe 48 c7 c7 08 34 54 8d e8 43 60 fe ff <0f> 0b 48 89 fe 48 c7 c7 d8 33 54 8d e8 32 60 fe ff 0f 0b 4d 8b 4c
[... 58 lines cut ...]
[ 3319.436573]  ? __pfx_worker_thread+0x10/0x10
[ 3319.436575]  kthread+0xcf/0x100
[ 3319.436578]  ? __pfx_kthread+0x10/0x10
[ 3319.436580]  ret_from_fork+0x31/0x50
[ 3319.436582]  ? __pfx_kthread+0x10/0x10
[ 3319.436585]  ret_from_fork_asm+0x1a/0x30
[ 3319.436588]  </TASK>
[ 3319.436589] Modules linked in: mt7921u mt792x_usb mt76_usb xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo xt_addrtype nft_compat nf_tables libcrc32c br_netfilter bridge stp llc ppdev vmnet(OE) parport_pc parport vmw_vsock_vmci_transport vsock vmw_vmci vmmon(OE) ccm rfcomm cmac algif_hash algif_skcipher af_alg overlay qrtr bnep sunrpc binfmt_misc nls_ascii nls_cp437 vfat fat snd_ctl_led snd_soc_skl_hda_dsp snd_soc_hdac_hdmi snd_sof_probes snd_soc_intel_hda_dsp_common snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component snd_soc_dmic snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda_common snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_hdac_hda snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus intel_uncore_frequency intel_uncore_frequency_common
[ 3319.436633]  mt7921e x86_pkg_temp_thermal snd_soc_avs intel_powerclamp mt7921_common kvm_intel mt792x_lib mt76_connac_lib snd_soc_hda_codec snd_hda_ext_core snd_soc_core mt76 btusb kvm btrtl snd_compress btintel snd_pcm_dmaengine mac80211 btbcm snd_hda_intel btmtk snd_intel_dspcfg snd_intel_sdw_acpi videobuf2_vmalloc uvc bluetooth snd_hda_codec videobuf2_memops libarc4 videobuf2_v4l2 videodev mei_pxp mei_hdcp intel_rapl_msr snd_hda_core rapl snd_hwdep cfg80211 processor_thermal_device_pci intel_cstate videobuf2_common snd_pcm processor_thermal_device iTCO_wdt mc acer_wmi processor_thermal_wt_hint intel_pmc_bxt intel_uncore platform_profile wmi_bmof snd_timer processor_thermal_rfim iTCO_vendor_support ee1004 mei_me pcspkr processor_thermal_rapl watchdog snd rfkill intel_rapl_common ucsi_acpi soundcore mei typec_ucsi processor_thermal_wt_req processor_thermal_power_floor typec igen6_edac processor_thermal_mbox roles int3403_thermal intel_pmc_core int340x_thermal_zone intel_vsec pmt_telemetry int3400_thermal intel_hid
[ 3319.436672]  joydev ac pmt_class acpi_thermal_rel acpi_pad sparse_keymap acer_wireless serio_raw evdev vmwgfx drm_ttm_helper coretemp acpi_ec(OE) configfs efi_pstore nfnetlink efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic ahci libahci libata scsi_mod nvme hid_multitouch hid_generic nvme_core scsi_common i915 drm_buddy i2c_algo_bit drm_display_helper crct10dif_pclmul cec crc32_pclmul rc_core crc32c_intel ttm i2c_hid_acpi i2c_hid drm_kms_helper hid xhci_pci r8169 xhci_hcd realtek ghash_clmulni_intel mdio_devres sha512_ssse3 drm usbcore libphy thunderbolt sha256_ssse3 intel_lpss_pci i2c_i801 sha1_ssse3 i2c_smbus intel_lpss idma64 vmd usb_common fan video battery button wmi aesni_intel gf128mul crypto_simd cryptd [last unloaded: uvcvideo]
[ 3319.436723] ---[ end trace 0000000000000000 ]---
[ 3319.878109] RIP: 0010:__list_del_entry_valid_or_report.cold+0x5c/0x6f
[ 3319.878116] Code: e8 68 60 fe ff 0f 0b 48 89 fe 48 89 ca 48 c7 c7 40 34 54 8d e8 54 60 fe ff 0f 0b 48 89 fe 48 c7 c7 08 34 54 8d e8 43 60 fe ff <0f> 0b 48 89 fe 48 c7 c7 d8 33 54 8d e8 32 60 fe ff 0f 0b 4d 8b 4c
[ 3319.878118] RSP: 0000:ffffadcbc0a67778 EFLAGS: 00010246
[ 3319.878120] RAX: 000000000000004e RBX: ffff8f73d475a000 RCX: 0000000000000000
[ 3319.878121] RDX: 0000000000000000 RSI: ffff8f78d7621780 RDI: ffff8f78d7621780
[ 3319.878122] RBP: ffff8f73d475a4d0 R08: 0000000000000000 R09: ffffadcbc0a67608
[ 3319.878123] R10: ffffffff8dcb41e8 R11: 0000000000000003 R12: ffff8f73d475a000
[ 3319.878124] R13: ffffffffc15e5278 R14: ffffffffc15e5278 R15: ffff8f7350ff3c50
[ 3319.878125] FS:  0000000000000000(0000) GS:ffff8f78d7600000(0000) knlGS:0000000000000000
[ 3319.878127] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3319.878128] CR2: 00007f2d59e173e8 CR3: 000000010c44c000 CR4: 0000000000f50ef0
[ 3319.878129] PKRU: 55555554
[ 3319.878131] note: kworker/8:2[685] exited with preempt_count 1
```

morrownr, 2024-11-07:
Hi @KaliAssistant 

Several individuals with this adapter have experienced this. It appears that this is a problem caused by a change in bluetooth support which should have nothing to do with this adapter but it does for some reason. You can try deleting the bluetooth firmware:

$ sudo rm /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin

Let me know if this helps.

KaliAssistant, 2024-11-08:
@morrownr 
My laptop pcie wifi is mt7921"e" , wifi and bt(pcie)  is working good.
I tried:
`mv  rm /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin  rm /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin.bak`

morrownr, 2024-11-08:
I have almost exactly the same laptop that you have and it has an M.2 card with the mt7921 chip. I also have an AXML adapter so I will test here. I have seen a long thread where it appeared Mediatek devs were working on this and they were working on the bluetooth subsystem. I am not sure if the patch has gone in yet. 

I'm pretty sure or at least it seems that the AXML and one of the early Comfast adapters with the same chipset are the only adapters that show this problem. Yes, blacklisting bluetooth fixes the crash but kills all bluetooth which is not what we want so hopefully this is solved soon.

KaliAssistant, 2024-11-09:
ok, thanks

wellitsmez, 2024-11-15:
@KaliAssistant have you tried monitor mode at this kernel? is it stable or crashing just as mine? how many networks did you get when you use air monitor because for me i made stand alone installation not using vm here for this kernel and getting about 20 percentage of what i am getting with kernel 6.8 at vm am trying to figure out what makes the performance that bad in my end

KaliAssistant, 2024-11-15:
@wellitsmez i tried hcxdumptool,and its good for me, no chash ,no errors

luckylinux, 2024-11-24:
@morrownr: I am also getting crazy with this Issue :cry:.

I originally purchased the Fenvi Adapter which more-or-less works OK. But then, sometimes, out of the blue, the Rock 5B SBC where this is installed completely crashes / disappears from the Network. Sometimes it works for like 1 Month or so. Sometimes it crashes after 10 Minutes. Last Night I managed to get it back running (apparently it was "just" the Mediatek Driver, without a Kernel Panic/Hang) simply by doing `systemctl restart hostapd@wlan0`. But this Morning it stopped working again :disappointed:.

Now I was trying to get this fixed on a Raspberry Pi 2, since I don't want such critical piece of Network to go down because of a weird Driver / Firmware Issue.

But also on the Raspberry Pi 2 with Raspbian it seems all Hell is breaking loose.
The Raspberry Pi 2 reboots every 45 Seconds or so.

I tried removing the Bluetooth Firmware as you recommended:
```
sudo rm /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
```
I blacklisted (all ?) the Relevant Drivers:
Contents of `/etc/modprobe.d/blacklist-bluetooth.conf`:
```
# Blacklist the Bluetooth Part of mt7921u
blacklist btmtk
blacklist btusb
blacklist btrtl
```

I forced the Raspberry Pi to Kernel 6.12 (since it would NOT let me update to anything else basically and I thought that the Issue was linked to poor Kernel 6.6 Support):
```
sudo rpi-update next
```

I also tried to installing the `firmware-mediatek` from `bookworm-backports` Repository as the Version in Raspbian / Raspberry Pi OS is VERY Old.

Nothing seems to Help.

Is there some other troubleshooting Steps that I missed ?

morrownr, 2024-11-24:
@luckylinux 

> I am also getting crazy with this Issue

It happens.

> I originally purchased the Fenvi Adapter

Can you be specific about which Fenvi adapter you have?

> Now I was trying to get this fixed on a Raspberry Pi 2, since I don't want such critical piece of Network to go down because of a weird Driver / Firmware Issue.

Okay, let's concentrate on the RasPi2. We can come back to the Rock5B.

Can you provide a link to the hardware specifics of the RasPi2 that you have?

Can you also provide me with a link to the exact RasPiOS distro you are using?

> But also on the Raspberry Pi 2 with Raspbian it seems all Hell is breaking loose.
The Raspberry Pi 2 reboots every 45 Seconds or so.

I need to know what you are doing. Is this an AP mode setup? If so, what are the basics of your setup? hostapd? where is the internet connection coming from?

> I tried removing the Bluetooth Firmware as you recommended:

That solution is for a different problem that only happens on 3 adapters. I am not aware of it being an issue on the Fenvi adapters.

> I thought that the Issue was linked to poor Kernel 6.6 Support):

Kernel 6.6 is about as solid as it gets. I run an AP on my RasPi4B with RasPiOS 64 bit and it is really support solid.

> I also tried to installing the firmware-mediatek from bookworm-backports

I would not recommend that. You can install the newest firmware directly from the Linux Kernel Firmware repo:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

> Is there some other troubleshooting Steps that I missed ?

Carefully answer my questions so that I know the situation and maybe I can help.

luckylinux, 2024-11-25:
@morrownr 

> @luckylinux
> 
> > I am also getting crazy with this Issue
> 
> It happens.

Far too often :rofl: 

> > I originally purchased the Fenvi Adapter
> 
> Can you be specific about which Fenvi adapter you have?

Sure. Pretty sure it was the one you [reccomended](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md) in one of your Discussions/Pages/Issues a few Months ago:
https://www.aliexpress.com/item/1005005935638503.html

Note that the Driver is `mt7921u` and NOT `mt7921` as you stated though:

`dmesg` Output when plugging the Adapter in (trimmed):
```
[   47.278735] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[   47.410572] usb 1-1.4: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[   47.410616] usb 1-1.4: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[   47.410638] usb 1-1.4: Product: Wireless_Device
[   47.410656] usb 1-1.4: Manufacturer: MediaTek Inc.
[   47.410673] usb 1-1.4: SerialNumber: 000000000
[   48.028817] usb 1-1.4: reset high-speed USB device number 4 using dwc_otg
[   48.198421] usbcore: registered new interface driver mt7921u
[   48.206624] mt7921u 1-1.4:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a

[   48.479505] mt7921u 1-1.4:1.0: WM Firmware Version: ____010000, Build Time: 20240219111038
[   53.741673] cryptd: max_cpu_qlen set to 1000
```

Output of `lsusb -vvv` when plugging the Adapter in:
```

Bus 001 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0e8d MediaTek Inc.
  idProduct          0x7961 
  bcdDevice            1.00
  iManufacturer           2 MediaTek Inc.
  iProduct                3 Wireless_Device
  iSerial                 4 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0051
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5 Config_01
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
[... 537 lines cut ...]
Alright, I uninstalled `firmware-realtek` using APT, then I installed using your Method (that I Scripted):
```bash
#!/bin/bash

mkdir -p /lib/firmware/mediatek
cd /lib/firmware/mediatek
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin -O WIFI_MT7961_patch_mcu_1_2_hdr.bin
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/WIFI_RAM_CODE_MT7961_1.bin -O WIFI_RAM_CODE_MT7961_1.bin
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin -O BT_RAM_CODE_MT7961_1_2_hdr.bin
```

Currently I'm rebuilding my Initramfs with `update-initramfs -k all -u` then Reboot and try again.

I'm NOT very hopeful however based on previous Experiences ...

> > Is there some other troubleshooting Steps that I missed ?
> 
> Carefully answer my questions so that I know the situation and maybe I can help.

Thanks for your Help. Is there other Info you need ?

Currently running Kernel:
```
Linux raspberrypi2-02 6.6.51-v7+ #1802 SMP Tue Oct  8 17:23:58 BST 2024 armv7l GNU/Linux
```

morrownr, 2024-11-25:
@luckylinux 

> Pretty sure it was the one you recomended in one of your Discussions/Pages/Issues a few Months ago

Okay, you link shows the Fenvi FU-AX1800. The Plug and Play List also has a Fenvi FU-AX1801D. I am just trying to understand the situation. Your post should have been in a new Issue as this issue is about a different adapter.

> Note that the Driver is mt7921u and NOT mt7921 as you stated though...

I did a quick look around the site to see what I said but did not find where I said this. If you can point me to the problem, I can correct it.

> Not sure what you mean exactly. It was like 10 Years ago and the Seller removed the Article from their Website. The Email Invoice just says "Raspberry Pi 2 Model B - 1GB RAM".

I am trying to figure out the capabilities of your RasPi2B. It appears to me that this model is using a 32 bit cpu and is USB2 only. That means you have to use the 32 bit version of RasPiOS and it appears that you are. For me to test, I have a RasPi3B that is available, it is 64 bit but I could install the 32 bit version and I'll have to lock the use of the 64 bit kernel as RasPi has made it such that the 64 bit kernel will be used, even on 32 bit installations if the hardware is capable of 64 bit operation. 

It is curious why you did not have any success with a new 32 bit installation. I'll admit that I grow more skeptical of using 32 bit distros as time passes as 32 bit simply does not get the testing that it once did.

> So I basically grabbed an oldish (6 Months out of Date ?) SD Card with Rasberry Pi OS Installed on that specific Raspberry Pi 2, applied all updates, and here I am. Yes, it boots with latest Updates and it's Stable If the Fenvi Adapter is NOT plugged In.

Do you have any other USB WiFi adapters available for testing? If so, please let me know what you have so we can pick out one or two for you to test. 

> I tried disabling the hostapd Service now and see if it survives longer with the Fenvi Adapter plugged in. But nope, still crashes in about 45-60 Seconds or so after I plugged it in.

Good information. Have you tested the Fenvi adapter on a desktop or laptop with x86_64/amd64 cpu and a recent mainstream distro such as Ubuntu 24.04 or 24.10. I'd like to rule out a manufacturing problem. Even testing on a Windows system would help eliminate a manufacturing problem.

> Alright, I uninstalled firmware-realtek...

Realtek?

> Thanks for your Help. Is there other Info you need ?

I think the keys right now are you testing to eliminate a manufacturing problem and me testing a 32 bit RasPiOS installation. Let me know what you find out.

>  I must emphasize, at least on the Rock 5B, that the Fenvi USB Adapter (and possible the USB Hub that I'm using) seems to be EXTREMELY finicky: one light touch on the Rock 5B is enough to trigger a USB Disconnection (whereas my main worry would be the RJ45 Cable getting "ejected", since the retaining Tab on the Cable is broken). But nope, touch lightly the Rock 5B, and you can see USB Disconnection Messages in dmesg.

I think that you testing the adapter on a desktop/laptop system that is known to perform well, even if you need to do it on a Windows system is very important to this investigation. The issue is whether your adapter has a manufacturing problem.

> with the Firmware installed from Linux Kernel git Repository as per your Instruction, the dmesg Output changes to (indicating a new Firmware Version of 20241106):

Updating firmware is best performed with the method I outlined because the location is THE location where the new firmware is placed for distro maintainers to get the updates. Most maintainers of distros are not aggressive about getting updates of firmware.

luckylinux, 2024-11-25:
Damn I see that this morning I forgot to post the most important Comment (I was making an Edit of the Original Comment but never pressed Save) :disappointed:.

For some weird Reason `netconsole` refuses to do anything. Not sure why I cannot get remote Logging working over Netcat :disappointed:.

Anyways, I thought I'd check the Watchdog Configuration and sure enough it was enabled and `/etc/watchdog.conf` (copied from a Raspberry Pi 3 which does have WiFi !) it was set to Monitor Network Interface `wlan0`.

Which means, as soon as I plugged the Fenvi in the USB Port, after a few 15-seconds Attempts, the Watchdog would trigger a restart of the Raspberry Pi.

Whereas if `wlan0` didn't exist, then the Watchdog gets disabled automatically. Weird. I didn't see that coming :rofl:.

I now changed the Watchdog Monitor Interface to `eth0` instead and the Raspberry Pi survives "OK".

Still weird that, no Matter what I do, `netcnosole` + `netcat` refuse to work, so that I can get remote logging to work.

luckylinux, 2024-11-25:
> @luckylinux
>
>
> Okay, you link shows the Fenvi FU-AX1800. The Plug and Play List also has a Fenvi FU-AX1801D. I am just trying to understand the situation. Your post should have been in a new Issue as [...]

Sorry. Should I create one new Issue then ?

> > Note that the Driver is mt7921u and NOT mt7921 as you stated though...
> 
> I did a quick look around the site to see what I said but did not find where I said this. If you can point me to the problem, I can correct it.

Here I meant that it should be `mt7921u` instead of `mt7921`:
[image]

> I am trying to figure out the capabilities of your RasPi2B. It appears to me that this model is using a 32 bit cpu and is USB2 only. That means you have to use the 32 bit version of RasPiOS and it [...]

Yeah 32bit Raspberry Pi 2 is a bit of a Freak. But I cannot see myself just throwing it away like this :cry:. Still good for a very simple AP Setup. It **should** also work with Mainline Debian, but also there Things are moving towards 64-bit mainly/only.

> Do you have any other USB WiFi adapters available for testing? If so, please let me know what you have so we can pick out one or two for you to test.

Oh I had that VERY BAD nowadays supported RTL Adapter (8814au) but I don't think it's a good Idea to try that. You (rightly so) pointed me towards the Fenvi / Mediatek Chipset for better HW Support :smile:.

> Good information. Have you tested the Fenvi adapter on a desktop or laptop with x86_64/amd64 cpu and a recent mainstream distro such as Ubuntu 24.04 or 24.10. I'd like to rule out a manufacturing [...]

No, not yet at least.

> > Alright, I uninstalled firmware-realtek...
> 
> Realtek?

Sorry, `firmware-mediatek`. I also uninstalled other Firmware that were NOT needed though including NVIDIAv Graphics etc.

> > Thanks for your Help. Is there other Info you need ?
> 
> I think the keys right now are you testing to eliminate a manufacturing problem and me testing a 32 bit RasPiOS installation. Let me know what you find out.

It might have been the Watchdog all along on the Raspberry Pi 2.

That still does NOT however explain the finickiness on the Rock 5B though with very sensitive USB Disconnection.
 
> I think that you testing the adapter on a desktop/laptop system that is known to perform well, even if you need to do it on a Windows system is very important to this investigation. The issue is [...]

I have several Fenvi around (4x ? 5x ? not sure). So yeah at least 1 is "free" to play around with.

> Updating firmware is best performed with the method I outlined because the location is THE location where the new firmware is placed for distro maintainers to get the updates. Most maintainers of [...]

Debian even less so :rofl:.

morrownr, 2024-12-05:
@luckylinux 

> Should I create one new Issue then?

I think it would help.

> Here I meant that it should be mt7921u instead of mt7921:

I looked at the pic you posted and still do not understand. I did not mention the driver name at all. Tell me what you think should be changed.

> Yeah 32bit Raspberry Pi 2 is a bit of a Freak. But I cannot see myself just throwing it away...

I am not suggesting that you recycle it. I am trying to point out that as time passes, it may become more problematic getting some things to work on 32 bit. The mt7921au chip and it's driver came about long after we were solidly in a 64 bit world. Was the driver tested for 32 bit operation? Maybe.

steev, 2024-12-06:
Hey @morrownr, steev from the kali team here.  do you happen to know which unrelated bluetooth commit it is that breaks this?

morrownr, 2024-12-06:
Hi @steev 

> do you happen to know which unrelated bluetooth commit it is that breaks this?

I do not. I would dissected this problem but I have been very busy. I think it was about the time that kernel 6.7 was released that I got the first message about this problem.

I have an AXML and can easily duplicate the problem here. I also have another adapter with the same chip that does not show the problem. There are 3 known adapters that show this problem: Alfa's AXML and AXM as well as the Comfast 953. None of the other adapters with the same chip show the problem.

This is a problem that may be hard to solve without a dissection. I suspect that the bad adapters may have problematic eeproms in that they are showing bluetooth support when it is actually turned off. This leads to the BAD behavior in the bluetooth stack, which also kills wifi. This did not happen until around kernel 6.7 so something changed.

Keep me posted if you find someone to dissect this problem.

wellitsmez, 2024-12-06:
@steev hello steev the Bluetooth is not the only thing is broken here also the VIF mode under the monitor mode you can have a look at some work arounds in here https://github.com/morrownr/USB-WiFi/issues/539 and also some discussion about what is broken when it comes to kernels bigger than 6.1 in here https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-11407608

Warhammar, 2025-02-12:
I have tried so much last day.. dont know what do do more. 

[image]

Iwconfig does not show the device.. i have installed everything. Or what have i missed?

luckylinux, 2025-02-12:
@Warhammar 
>> Iwconfig does not show the device

More importantly why are you in a `rtw88` Folder (probably related to e.g. https://github.com/lwfinger/rtw88) when you have a Mediatek Device 🙄 ?

Otherwise ...

Isn't it `wlan0` ? Or do you have another Wireless Device installed ?

If Kernel Module has been installed and `insmod` / `modprobe`, the other Possibility is that you miss the `linux-firmware` Package required by the Device.

After installing firmware a `update-initramfs -k all -u` plus `reboot` and it should show up.

Try `dmesg` and see if there are some Messages about issues loading the Driver, missing Firmware, ...

Warhammar, 2025-02-12:
─$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.44-Re4son-v8l+ (root@kali-raspberry-pi) (gcc (Debian 11.2.0-19) 11.2.0, GNU ld (GNU Binutils for Debian) 2.38) #1 SMP PREEMPT Debian kali-pi (2022-07-03)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 400 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:27:A1:72 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=7ba34c71-02 rootfstype=ext4 fsck.repair=yes rootwait net.ifnames=0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3815980K/4050944K available (11712K kernel code, 1958K rwdata, 4112K rodata, 3776K init, 978K bss, 169428K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36806 entries in 144 pages
[    0.000000] ftrace: allocated 144 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000310] Console: colour dummy device 80x25
[    0.000827] printk: console [tty1] enabled
[    0.000896] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000936] pid_max: default: 32768 minimum: 301
[    0.001066] LSM: Security Framework initializing
[    0.001329] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001414] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002642] cgroup: Disabling memory control group subsystem
[    0.005329] rcu: Hierarchical SRCU implementation.
[    0.006413] EFI services will not be available.
[    0.006957] smp: Bringing up secondary CPUs ...
[    0.008070] Detected PIPT I-cache on CPU1
[    0.008147] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.009293] Detected PIPT I-cache on CPU2
[    0.009343] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.010455] Detected PIPT I-cache on CPU3
[    0.010503] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.010651] smp: Brought up 1 node, 4 CPUs
[... 351 lines cut ...]
[    7.255323] Bluetooth: HCI device and connection manager initialized
[    7.256777] Bluetooth: HCI socket layer initialized
[    7.258267] Bluetooth: L2CAP socket layer initialized
[    7.270515] Bluetooth: SCO socket layer initialized
[    7.499910] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.502070] bcmgenet fd580000.ethernet eth0: Link is Down
[    7.837521] usbcore: registered new interface driver btusb
[... cut ...]
[   13.875057] Bluetooth: HCI UART protocol Broadcom registered
[   28.302407] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   28.302437] Bluetooth: BNEP filters: protocol multicast
[   28.302454] Bluetooth: BNEP socket layer initialized
[   28.319334] NET: Registered PF_ALG protocol family
[   28.394545] Bluetooth: RFCOMM TTY layer initialized
[   28.394585] Bluetooth: RFCOMM socket layer initialized
[   28.394607] Bluetooth: RFCOMM ver 1.11
[   31.710367] cam1-reg: disabling
[   31.710417] cam-dummy-reg: disabling

luckylinux, 2025-02-12:
@Warhammar : EXTREMELY OLD Kernel at Linux 5.15.44. And also EXTREMELY OLD Raspberry Pi Firmware from 2020/2022.

Pretty sure the Mediatek Device won't work. But why again are you in a `rtw88` Folder trying to build a Realtek Driver for a Mediatek Device 🙄 ?

Warhammar, 2025-02-13:
I was in there to look if it has something to do with.. 
So what should i do? I can easily start from zero. Re-install the newest software and so. But i need guidance.. im still a noob..

morrownr, 2025-02-13:
Hi @Warhammar 

You may laugh at me but I thought it might be a good idea to review this issue... bad idea. The problem issue has changed multiple times. I then came back down here to work my way back to when the current activity started. The moral to the story is adding a new problem onto the end of an old issue makes it hard for us to help you.

> im still a noob...

That is not a problem. I remember when I started with Linux in 1994. Man, we did not even have installers back then. Today's Linux is like a dream come true.

> So what should i do?

@luckylinux is correct when he points out that your kernel is too old... Linux 5.15.44. The driver for your adapter initially went into kernel 5.18 so you are dealing with no driver as your problem.

FYI: I am working with another guy to backport the driver to kernels 5.15 and 5.10 but that is not ready.

What you need to do is `start from zero. ` Your Pi400 can support very modern distros. I'd recommend the latest RasPiOS 64 bit. Do you know how to download and burn a sd card? If not, I can walk you through it.

Warhammar, 2025-02-14:
I get to work and give you the infos in a while.. thanks.

Warhammar, 2025-02-14:
Heureka! I did install now the Raspi OS and it did work instandly.. withouth any other installation. Proof of concept with Wifite...  Previous i did install Kali ARM. and there was the problem.

morrownr, 2025-02-14:
That is good. If you have any problems, let me know.

For your information: You can install nearly all of the apps that are in Kali on the RasPiOS. Kali is a very unstable distro in my experience and then the ARM version dates back to the Bronze Age. I am not a fan of Kali. Give me Debian 12, Ubuntu, or RasPiOS any day.

steev, 2025-02-14:
> That is good. If you have any problems, let me know.
>
> For your information: You can install nearly all of the apps that are in Kali on the RasPiOS. Kali is a very unstable distro in my experience [...]

Gonna chime in because you're besmirching my good name ;)  I do not understand why you say it's very unstable.  I've been a dev on Kali for 11 years, and can count on 1 hand the amount of times we've had instability issues.  The ARM version does not at all date back to the Bronze Age, however for the RaspberryPi, yes, we use the 5.15 kernels because of the Nexmon firmware project.  Our testing on newer kernels was (surprisingly enough) that it was very ustable.  I'm working on changing that, but it's quite slow going when I'm the only one working on it.  I'm tracking that https://gitlab.com/kalilinux/build-scripts/kali-arm/-/issues/314  but do not have anything for anyone to test yet.  I appreciate the feedback though.

steev, 2025-02-14:
> Hi [@steev](https://github.com/steev)
> 
> > I do not understand why you say it's very unstable.
> 
> I said that because, in my experience, it is unstable. I used to try to keep an installation on a test system in order to better help users of USB WiFi. From about 4 years ago until about a year ago, it was just one damn thing after another. Update the system and it won't boot plus many other issues. Then about a year ago, I tried to do a clean install of a new release and it would not even install. That was the last straw. I am not a noob. I've been using Linux since 1994 and other distros were installing on that test system just fine.
> 

Not questioning your qualifications whatsoever :) It is unfortunate that that is your experience, it is not the common one.

> Then there is the kernel 5.15 on RasPis. I have lost count of how many people have stopped by here asking for help. In The Plug and Play List, I have made extensive documentation regarding what kernel is required for the mt7921au based adapters. I think most users that run into problems with the 5.15 kernel got their advice on what to buy somewhere else because I am trying to get the word to the users that stop by here. The bottom line is that Kali is causing me pain and I WILL continue to push users to other distros as long as that pain is there. Period.
> 
> If you cannot do better with the RasPi version than this then base Kali for RasPi on the RasPiOS.
> 
> Regards,
> 
> [@morrownr](https://github.com/morrownr)

As much as I would love to base on RaspiOS, that is simply not possible to do with the way the Kali repositories (based on Debian testing) work.  It would essentially be forking Kali just for the RaspberryPi devices, and I already work far more hours than I should.  That said, if a Kali user does show up, you are more than welcome to @ me to rope me in, or redirect them to our [Kali Forums](https://forums.kali.org) instead of dealing with it.

Regards,
- steev