morrownr/USB-WiFi #534: [New Adapter] tp-link Archer TXE50UH (uses the mt7921au chip, or so it would appear)

morrownr, 2024-11-08:
I normally pay no attention to tp-link usb wifi products for reasons outlined in menu item 1 on the main Menu but I do hear from people around the world where usb wifi adapter choices can be limited. I often hear them say that tp-link products are available so I am posting this message and will consider adding this product to The Plug and Play List IF I get good reports from users of this adapter and people express their option that it should be included in the List. Given that Realtek does not make a AXE3000 class chip, it would be hard for tp-link to change the chip and keep the product number and packaging so this might be a case where this is a safe pick if it is a good adapter. With that said:

The tp-link site to the TXE50UH is as follows:

https://www.tp-link.com/us/home-networking/usb-adapter/archer-txe50uh

It is a very recent product almost no reviews. Evidence points to it being a single-state (no windows driver onboard which is great). The evidence is that the ads I read said the package contains a cd. Why ship a cd if the driver is in the adapter? 

It comes with a nice looking extension cable/cradle which is very welcome in order to place the adapter at a location where it can get good reception. The chipset is the mt7921au so it should be plug and play. Something confusing on tp-link's support site for this product is that they list a v1 and a v1.6. tp-link is notorious for changing chipsets without changing product numbers which is a killer for us Linux users but in this case, the limited evidence I can find points to both versions using the same Mediatek chip. I have no idea what the real difference is in the 2 versions.

WARNING: If you decide you want to buy this adapter, I noticed that tp-link has numerous adapters with similar product names. I see the TX50UH name but that is NOT this product. You need to be very careful to get the TXE50UH

I'll wait for feedback before taking any action so please let me know what you think and if you have this adapter, please post the details and your opinion as to whether this is a good adapter that Linux users should consider.

@morrownr

morrownr, 2024-12-09:
G'day @dogkeeper886 

Thanks for the report. Not exactly plug and play but it is a start.

What vid/pid does the adapter have?

I can help get the vid/pid into the kernel with a bit more testing and confirmation from another user. I hate to be picky but mistakes are a pain to clean up.

Did you use the latest firmware files?

While 6.1 should be okay, I generally recommend 6.6 or later if that is doable.

Keep me posted. I want to know how this adapter is working out.

dogkeeper886, 2024-12-10:
Here is the lsusb info.
`
Bus 001 Device 003: ID 35bc:0107 MediaTek Inc. Wireless_Device
`
I think the kernel or linux distribution did a great job. We just missing a udev rule. I remember trying it on Fedora before, which already has the mt7921u module, but it still wasn't plug-and-play. At the time, I didn't realize where the problem was.

morrownr, 2024-12-10:
@dogkeeper886 

Okay, so the vid/pid is `ID 35bc:0107` . Do you mind running and posting the results of the following just to double check:

$ lsmod | grep mt79

> I think the kernel or linux distribution did a great job. We just missing a udev rule.

We have Mediatek to thank for doing their Linux driver the right way. In fact, Mediatek already has the mt7925u driver in the kernel. That is a WiFi 7 chip. I run a M.2 card with a mt7925 chip and it runs well. It uses the mt7925e driver.

For educational purposes:

It is best if the adapter makers use a chip maker standard vid/pid but some, like Netgear and TP-Link, use their own. Edup, Panda, Alfa and Fenvi generally use the standard vid/pid. You can see this in the Plug and Play List in the mt7921au section. Right now, the Netgear adapter is the only one where we had to add a udev rule and send a patch to linux-wireless.

Adding a udev rule is a temp fix. The real fix is to send a patch to linux-wireless to get the new vid-pid included in the mainline kernel. Would you like to assist me in doing that?

Back to business: Can I get you to do a test on the adapter to see if it is a single-state (no storage for Windows driver) adapter? If so, we can turn usb_modeswwitch off and if you reboot, your adapter should still come right up and be useable. Here is how to turn usb_modeswitch off:

How to deactivate usb_modeswitch:

sudo nano /etc/usb_modeswitch.conf

Change

DisableSwitching=0

to

DisableSwitching=1

sudo reboot

dogkeeper886, 2024-12-10:
Thanks, bro! I never had the chance to understand that before. I'll give the usb_modeswitch thing a try.

`HTML
<pre><font color="#C01C28"><b>mt79</b></font>21u                24576  0
<font color="#C01C28"><b>mt79</b></font>21_common          94208  1 <font color="#C01C28"><b>mt79</b></font>21u
mt76_connac_lib        81920  2 <font color="#C01C28"><b>mt79</b></font>21_common,<font color="#C01C28"><b>mt79</b></font>21u
mt76_usb               40960  1 <font color="#C01C28"><b>mt79</b></font>21u
mt76                  106496  4 mt76_usb,<font color="#C01C28"><b>mt79</b></font>21_common,mt76_connac_lib,<font color="#C01C28"><b>mt79</b></font>21u
mac80211             1261568  3 mt76,<font color="#C01C28"><b>mt79</b></font>21_common,mt76_connac_lib
cfg80211             1110016  5 mt76,mac80211,<font color="#C01C28"><b>mt79</b></font>21_common,mt76_connac_lib
</pre>

`

morrownr, 2024-12-10:
Thanks for confirming that the adapter is using the mt7921u driver. Let me know the results of the usb_modeswitch test.

>  After creating a udev rule, I ran into a hardware initialization error. Luckily, I came across a post suggesting that adjusting the USB version to 2.0 might help.

As you have time, give me more details about this issue What were you seeing when in USB3 mode?

Let me know when you are ready to step through the process of sending a patch to linux-wireless and I will start sending information so that you can walk through it with me. If we can finish within 2 weeks or so then the patch should go into kernel 6.14 just in time for Ubuntu 25.04 and if we say the right things, the patch will be backported to the 6.12, 6.6 and 6.1 LTS kernels. Cool huh? This will make your adapter truly plug and play going forward.

dogkeeper886, 2024-12-11:
I connected it to another Windows 10 PC, but it doesn't have the storage folder, similar to what happened with my Rocky Linux setup. Is this the information you need?

By the way. I verified the USB3 initialization problem on another desktop, and it didn't occur.

morrownr, 2024-12-11:
> I connected it to another Windows 10 PC, but it doesn't have the storage folder, similar to what happened with my Rocky Linux setup. Is this the information you need?

Yes, I am looking for confirmation that this adapter is single-state (no onboard Windows driver). It sounds like you are confirmation that the adapter is single-state.

I go through a list of bad things to filter out adapters that should not go into the Plug and play List. Another on the list is multi-function (wifi and bluetooth support). Can you confirm that no bluetooth support is turned on?

So far, this adapter looks like a good candidate to go into the Plug and Play List. Are you seeing anything, other than the vid/pid not being in the kernel yet, that would cause you to tell me not to put this adapter in the Plug and Play List? If not, let me know when you are ready to work with me on getting the vid/pid in the kernel. We can take it one step at a time.

Since you are compiling a mainline kernel for Rocky anyway, that would make it easy for you to test the patch that we come up with.

dogkeeper886, 2024-12-12:
My TP-Link Archer TXE50UH doesn't have Bluetooth. It would be nice if we can use it without all the extra effort.

morrownr, 2024-12-12:
>  It would be nice if we can use it without all the extra effort.

We can fix that. It would have been best if TP-Link had used the Mediatek standard vid/pid for the mt7921au chip. This would have meant the adapter would have worked immediately in modern Linux distro from the beginning. But they did not do this so we need to submit the vid/pid to have it included in the kernel. At that point, anyone else that buys the same adapter will have an adapter that works...plug and play as long as they are using the right kernels. We can have the patch backported all the way back to kernel 6.1.

If you want to go ahead and test my patch, I will tell you which file in the kernel source so that you can modify it and compile it.

morrownr, 2024-12-16:
@dogkeeper886 

>  It would be nice if we can use it without all the extra effort.

I'm willing to submit a patch so that the "extra effort" goes away. I need you to test the patch. Are you willing to do that?

dogkeeper886, 2024-12-16:
Sure, no problem. I can do it tomorrow.

morrownr, 2024-12-16:
For the test:

Edit the following file in the kernel source:

drivers/net/wireless/mediatek/mt76/mt7921.usb.c

Starting on or about line 15, note the section of code below:

```
static const struct usb_device_id mt7921u_device_table[] = {
	{ USB_DEVICE_AND_INTERFACE_INFO(0x0e8d, 0x7961, 0xff, 0xff, 0xff),
		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	/* Comfast CF-952AX */
	{ USB_DEVICE_AND_INTERFACE_INFO(0x3574, 0x6211, 0xff, 0xff, 0xff),
		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	/* Netgear, Inc. [A8000,AXE3000] */
	{ USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	/* TP-Link TXE50UH */
	{ USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0107, 0xff, 0xff, 0xff),
		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	{ },
};

```

Note the line `/* TP-Link TXE50UH */` , add it and the next 2 lines and save.

Compile and install the kernel.

Remove the udev rule that you had installed.

Boot and see if the adapter comes to life

Let me know how it goes.

Thanks.

dogkeeper886, 2024-12-17:
I found the relevant code at /usr/src/kernels/linux-6.1.120/drivers/net/wireless/mediatek/mt76/mt7921/usb.c. The device ID for the MT7921U is mt7921u_device_table. Although the format is slightly different, it's still functional.

```c
static const struct usb_device_id mt7921u_device_table[] = {
        { USB_DEVICE_AND_INTERFACE_INFO(0x0e8d, 0x7961, 0xff, 0xff, 0xff) },
        /* TP-Link TXE50UH */
        { USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0107, 0xff, 0xff, 0xff) },
        { },
};
```

I found the code at /usr/src/kernels/linux-6.12.4/drivers/net/wireless/mediatek/mt76/mt7921/usb.c. The code matches what you previously provided, and it's working as expected.
```c
static const struct usb_device_id mt7921u_device_table[] = {
        { USB_DEVICE_AND_INTERFACE_INFO(0x0e8d, 0x7961, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        /* Comfast CF-952AX */
        { USB_DEVICE_AND_INTERFACE_INFO(0x3574, 0x6211, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        /* Netgear, Inc. [A8000,AXE3000] */
        { USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        /* TP-Link TXE50UH */
        { USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0107, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        { },
};
```

morrownr, 2024-12-17:
@dogkeeper886 

Okay, so it tested good. I'll make a patch to send in. I'd like to add a Report-by line and a Tested by line and use your name and email address if that is okay.

morrownr, 2024-12-17:
@dogkeeper886 

FYI: Here is the patch minus your information in Report-by: and Tested-by:

Forward your info to me as you can so we can get this patch in mainline.

```
From 647960fed3629e0ae112a6c76c8ed2d5dcd3a727 Mon Sep 17 00:00:00 2001
From: Nick Morrow <usbwifi2024@gmail.com>
Date: Tue, 17 Dec 2024 06:40:50 -0600
Subject: [PATCH] wifi:mt76:mt7921au:Add new VID/PID 35bc/0107 for TP-Link TXE50UH
Add new VID/PID 35bc/0107 for TP-Link TXE50UH

Reported-by:
Tested-by:

Cc: stable@vger.kernel.org
---
 drivers/net/wireless/mediatek/mt76/mt7921/usb.c | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
index 8aa4f0203208..e3459295ad88 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
@@ -21,6 +21,9 @@ static const struct usb_device_id mt7921u_device_table[] = {
 	/* Netgear, Inc. [A8000,AXE3000] */
 	{ USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
 		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
+	/* TP-Link TXE50UH */
+	{ USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0107, 0xff, 0xff, 0xff),
+		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
 	{ },
 };
 
-- 
2.39.5

```

dogkeeper886, 2024-12-18:
Sure, go ahead.

morrownr, 2024-12-18:
I'd like to give you credit for reporting and testing. See the following section of the above message:

```
Reported-by:
Tested-by:

```

With your name and email address, you will get credit for reporting and testing this issue. Cool. If that is not what you want to do then I can leave it off and submit the message.

worklkt, 2024-12-19:
I read this  post and buy TXE500UH for my openwrt 23.05.5  device on raspiberry pi 4.(build in AC wireless range is so sort)

Easy way to use this device is download mt7921 and pain file put in right place,write device it to new_id file.
It is easy as morrownrs say.
Some information give you know.

Device use USB 3 port is fast than USB2 port (test by iferf3),USB2 never over 300Mbps USB3 can run 700Mbps maybe is hardward limit.
6Ghz not run,I change country code to US,it cat see 6G in luci control,but my 6E mobile device can't see SSID.

Very thanks both.

dogkeeper886, 2024-12-19:
> I'd like to give you credit for reporting and testing. See the following section of the above message:
> 
> ```
> Reported-by:
> Tested-by:
> Signed-off-by: Nick Morrow <usbwifi2024@gmail.com>
> ```
> 
> With your name and email address, you will get credit for reporting and testing this issue. Cool. If that is not what you want to do then I can leave it off and submit the message.

Reported-by: Nick Morrow <usbwifi2024@gmail.com>
Tested-by: Shang Chieh Tseng <shangchieh.tseng@tsengsy.com>

morrownr, 2024-12-19:
@dogkeeper886 

Message sent to linux-wireless. If things go as expected, the patch will initially go into kernel 6.14 and then will be also be applied to kernels 6.13,  6.12, 6.6 and 6.1. So, depending on the distro you use, within maybe 2 months. you may not need the udev rule anymore and your adapter will be plug and play for as long as you use it. Thanks for the report and the help.

One last thing for now: As you use the adapter, let me know if you think the TXE50UH should be included in The Plug and Play List.

BlanceIsBetter, 2024-12-20:
@morrownr

> I hope the adapter works well for you. Give us reports about it as you have time. We need to determine if this adapter should be in the Plug and Play List.

I have tested the Plug and Play as below describes. 

Power off the Pi and unplug usb dongle, then restart Pi. I plug in a new usb port that never used, It will generate new interface and default wifi, just need to setup it can run well.

Then I direct to pull out usb dongle to another used usb port,it can run well too.

> 
> The current regdb may not allow AP mode with the adapter in the US. You can run the following to see what it says about the 6 GHz channels:
> 
> # iw list
> If it says `No IR` then AP mode is not going to work for now. Given the serious reduction in txpwr for 6 GHz, it is of somewhat limited benefit.

As you say,  6 GHz list No IR....(Country code:US JP TW)
I think it's a pity, it would be better if it can operate at full frequency.

Thanks for teaching, I learned more.

====

To correct the previous test.
USB2 port can rut faster about 400Mbps, it need run at high frequency.

dogkeeper886, 2024-12-24:
> I read this post and buy TXE500UH(US) Ver.1 for my openwrt 23.05.5 device on raspiberry pi 4.(build in AC wireless range is so short.)
> 
> Easy way to use this device is download mt7921 and pain file put in right place,write device id to new_id file. It is easy as morrownrs say. Some information give you know.
> 
> TXE500UH use USB 3 port is fast than USB2 port (test by iferf3,android and ios device),USB2 never over 300Mbps USB3 can run 6-700Mbps maybe is hardward limit. Pi 4 tempture will effert wifi speed. 6Ghz not run,I change country code to US,it can see 6G in luci control,but my 6E mobile device can't see SSID.
> 
> Very thanks both.

Nice to hear that. Thank you.

morrownr, 2025-01-02:
The patch to add the ID for this TXE50UH adapter has been accepted as you can see a the below link:

https://patchwork.kernel.org/project/linux-wireless/patch/e797f105-9ca8-41e9-96de-7d25dec09943@gmail.com/

This means it should go into kernel 6.14 which should go into testing early next month. I also requested that the patch be applied to the existing supported kernels which as 6.13, 6.12, 6.6, 6.1. Can't go back further than that as the driver went into kernel 5.18.

So, this adapter should be plug and play soon.

I do need reports from everyone that has this adapter: The question that I needed answered is:

Should this adapter be in the Plug and Play List?

morrownr, 2025-01-08:
I have added this adapter to The Plug and Play List based on information here and in other locations. If any problems come up, please let me know so that I can reevaluate whether the adapter should be included in The Plug and Play List.

razvanphp, 2025-01-31:
Can we maybe backport the mt79 driver to kernel 5.15?

Nvidia Jetson is stuck with this kernel even in the latest version....

morrownr, 2025-01-31:
Hi @razvanphp 

> Can we maybe backport the mt79 driver to kernel 5.15?

Already done. The Mediatek devs that also work on OpenWRT did that a long time ago:

https://github.com/openwrt/mt76

razvanphp, 2025-01-31:
Thanks for the heads up!

Unfortunately, that one does not compile on my Jetson, are you sure it's supposed to work on this kernel?

```
$ make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/usr/src/linux-headers-5.15.136-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
  CC [M]  /home/jetson/mt76/mmio.o
In file included from /home/jetson/mt76/mmio.c:6:
/home/jetson/mt76/mt76.h:16:10: fatal error: linux/soc/mediatek/mtk_wed.h: No such file or directory
   16 | #include <linux/soc/mediatek/mtk_wed.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:295: /home/jetson/mt76/mmio.o] Error 1
make: *** [Makefile:1912: /home/jetson/mt76] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.136-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source'
```

then even more errors, not sure it can compile without `wed`. Also, can I skip `mt76` and only compile `mt79`?

morrownr, 2025-01-31:
> are you sure it's supposed to work on this kernel? 

The current OpenWRT stable, 23.05, uses kernel 5.15 and this driver is available and works. I'm currently trying to setup a mt76 repo myself. one that can be used for testing and backporting. I am making progress but am not there yet. I'd have to investigate what you ran into.

> Also, can I skip mt76 and only compile mt79?

I'd have to look to see what changes would be required.

razvanphp, 2025-01-31:
Thank you for looking into thos. I think I'm missing some patches that nvidia did not include?

For example this one: https://github.com/openwrt/openwrt/blob/4e1d1b7df0ce6fa96d7462dc883917682f428046/target/linux/generic/pending-5.10/701-02-net-ethernet-mtk_eth_soc-add-support-for-Wireless-Et.patch

```bash
$ ls -al /usr/src/linux-headers-5.15.136-tegra-ubuntu22.04_aarch64/3rdparty/canonical/linux-jammy/kernel-source/include/linux/soc/mediatek/
total 40
drwxr-xr-x  2 root root 4096 Jan 14 16:31 .
drwxr-xr-x 17 root root 4096 Jan 14 16:25 ..
-rw-r--r--  1 root root 6836 Jun 26  2024 infracfg.h
-rw-r--r--  1 root root 9420 Jun 26  2024 mtk-cmdq.h
-rw-r--r--  1 root root 1110 Jun 26  2024 mtk-mmsys.h
-rw-r--r--  1 root root  758 Jun 26  2024 mtk-mutex.h
-rw-r--r--  1 root root  653 Jun 26  2024 mtk_sip_svc.h
$
```

morrownr, 2025-01-31:
Here is my mt76 repo:

https://github.com/morrownr/mt76

I started it about 6 months ago but have not been able to work on it due to a lack of time. Today I updated the code to match wireless-next. We had changed the main Makefile and some others files to allow it to compile and install when we started but it appears my update has broken that... so, it appears the biggest thing that needs to be done is to backport it.  You are welcome to help.

martinellison, 2025-07-24:
I have bought a TXEU70UH (the model 50 seems not to be on sale in Sham Shui Po) and it is *not* working for me (on 6.15.7).

morrownr, 2025-07-24:
Hi @martinellison 

The TXE50UH is currently the only TP-Link USB WiFi adapter that is plug and play.

I think the TP-Link ARCHER TXE70UH AXE5400 Wi-Fi 6E High Gain Wireless USB Adapter uses a Realtek rtl8832cu chipset. Wor is underway on a driver to go into the kernel:

https://github.com/morrownr/rtw89

If you could give a good report about you experience in that repo in Issues in that repo, it would be greatly appreciated.

QinCai-rui, 2025-10-04:
Hey! just wondering if anybody has done iperf speedtests in AP mode. currently deciding between this and the [EDUP AX3000](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L). Thanks!

morrownr, 2025-10-04:
Hi @QinCai-rui 

> just wondering if anybody has done iperf speedtests in AP mode...

I have. I wish I had the time to post a document on the Main Menu on this topic but so far I have not had the time. I have a EP-AX1672 (like you provided the link for). I consider it to be an outstanding performer in AP mode. I would have to do some changing around here to test again but, if memory serves me correctly, I was seeing about 650 Mbps on  a consistent basis from the Edup adapter with band 2 (5 GHz), channel 100 (DFS-US) and 80 MHz channel width. Bands 1 and 2 work well in almost countries, however, band 4 (6 GHz) is hit and miss when it comes to AP mode. AP mode with 6 GHz using a USB WiFi adapters works well in EU countries whereas it does not work in the US and some other countries for a variety of reasons. This is not a problem with specific adapters but rather an issue with laws, regdb and other factors.

I have not tested the TXE50UH as I do not have one. It uses the same chipset as the EP-AX1672 but there can be differences depending on the AMP and antennas in use. We can probably get some idea of the differences by looking at the product reviews.

The product reviews for the EP-AX1672 are mostly positive (73% 5 star) overall but probably higher for Linux users as Linux has had 6 GHz support longer than Windows has. Here is a review from the link you posted:

```
Tested on both windows 11 and linux, the card itself preforms wonderfully. Works very well with tools like airodump-ng, aireplay-ng, and hcxtools on linux. Monitor, master, managed, and ad-hoc modes work out of the box with a modern linux kernel. However, the driver for windows 11 only supports managed mode. Additionally, there is no support for arm-based linux builds, such as the raspberry pi. Overall, excellently built, but the lack of support for the device holds it back.
```

My thoughts on this review:

> there is no support for arm-based linux builds, such as the raspberry pi.

This is not correct. The mt7921u driver and USB WiFi adapters that use this driver work very well with Raspberry Pi systems and the RasPiOS. I use a RasPi4B as an AP operational and test system to test things for users here. 

The comments `performs wonderfully` and `excellently built` are spot on in my experience. 

Regarding the TXE50UH, using the Amazon link that I have in The Plug and Play List, I see only one review by a Linux user:

> I didn’t see it advertised as having linux support, but it’s worked flawlessly for me!

Sounds like a happy camper. The TXE50UH will have fewer reviews by Linux users as it has only been on the market for a short time while the Edup adapter has been available for over 2 years. 

Hopefully this reply helps.

@morrownr

QinCai-rui, 2025-10-04:
Ahhh thank you so much @morrownr! I'm probably gonna go with the EDUP AX3000, since you (and others) have tested it on a Pi

morrownr, 2025-10-04:
A couple of follow on things:

- you have been writing `EDUP AX3000`. That does not point to the EP-AX1672 like in your link. Edup does make AX3000 class adapters but they generally contain Realtek chips. The correct class for the EP-AX1672 is AXE3000. It is better to use the model number, EP-AX1672, when talking about adapters.

- The review that said the EP-AX1672 supports ad-hoc mode is incorrect. ad-hoc has basically died.

The modern equivalent of an ad-hoc wireless network is Wi-Fi Direct (P2P). You can see it referred to as P2P and all Mediatek drivers support 3 different P2P modes. Modern Realtek drivers generally support 2 P2P modes.

Good luck and let us know how it goes.

ExplosivePresent, 2025-10-27:
I'm unsure if I'm facing a different problem than what I think due to my ignorance, but there might be an error in considering TXE50UH plug and play. Maybe its just naming convention problems.

According to the bottom of my adapter, which matches the image in your guide [here](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md), my adapter is a TP-Link TXE5400UH. I thought this is the same as a TXE50UH, but when I run 

$ lsusb 

I find: `Bus 002 Device 002: ID 35bc:0102 Realtek 802.11ax WLAN Adapter`
Which is quite different from 

> Here is the lsusb info. `Bus 001 Device 003: ID 35bc:0107 MediaTek Inc. Wireless_Device` 

I am quite new to Linux and this is just the adapter I had lying around. I have a freshly updated  install of Arch 6.17.5 through EndeavorOS. Is this a problem with me or with the adapter I have? I feel like I've tried every solution I could find so far, and it would make sense if none of it worked because I have been trying to use the MediaTek drivers on a RealTek chipset...