morrownr/USB-WiFi #535: [Help]: Alfa AWUS036AXML fake version at the market

wellitsmez, 2024-11-10:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

uname -a 

### lsusb

sud lssusb

### rfkill

sudo rfkill list

### dkms

```text
sudo dkms status
```

### iw

```text
sudo iw dev
```

### What happened?

would you please confirm with me the FCCID for this model i think i have fake one my FCCID is 2AB8788321can you please verify with me is it the same as the one you have or i should return it ?

mbaltrusitis, 2024-11-11:
i just received one from amazon today and have the same `fccid` you listed.

wellitsmez, 2024-11-14:
@mbaltrusitis yes i had a word with alfa guys and seems like its a legit version my bad

closed 2024-11-14 by wellitsmez