morrownr/USB-WiFi #536: [Help]: AWUS036AXML not showing up Raspberry Pi 3B+

neuterscooter, 2024-11-12:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux faradaypi 5.15.44-Re4son-v7+ #1 SMP Debian kali-pi (2022-07-03) armv7l GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter Bus 001 Device 011: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

1: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no 2: hci1: Bluetooth         Soft blocked: no         Hard blocked: no 9: hci0: Bluetooth         Soft blocked: no         Hard blocked: no

### dkms

```text
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8, 5.15.44-Re4son-v7+, armv7l: installed
realtek-rtl88xxau/5.6.4.2~git20240726.63cf0b4, 5.15.44-Re4son-v7+, armv7l: installed
rtl8821au/5.12.5.2, 5.15.44-Re4son-v7+, armv7l: installed
```

### iw

```text
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr 0e:bb:83:8c:ec:b1
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 22:54:8d:7d:b0:3b
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm
```

### What happened?

Trying to get a alfa awus036axml network adapter working on kali 32-bit for rpi 3B+. I have two other network adapters (AWUS036ACS and ACH), both had their drivers installed from the 8812au and 8821au repos and are able to undergo injection mode, but this adapter only lights up and appears in lsusb, but not in iwconfig. dmesg displays this when i plug it in:
[ 1899.301176] usb 1-1.5: USB disconnect, device number 10
[ 1900.764859] hwmon hwmon1: Voltage normalised
[ 1904.714874] usb 1-1.5: new high-speed USB device number 11 using dwc_otg
[ 1904.857089] usb 1-1.5: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 1904.857130] usb 1-1.5: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 1904.857156] usb 1-1.5: Product: Wireless_Device
[ 1904.857178] usb 1-1.5: Manufacturer: MediaTek Inc.
[ 1904.857199] usb 1-1.5: SerialNumber: 000000000

Not sure how to proceed with this issue, I want to use kali for the os, but if raspbian is the only solution then i will just have to do that

morrownr, 2024-11-12:
Hi @neuterscooter 

> Linux faradaypi 5.15.44-Re4son-v7+ #1 SMP Debian kali-pi (2022-07-03) armv7l GNU/Linux

There is the problem. The mt7921u driver first appeared in kernel 5.18. More information available in and below the mt7921au section in the following document:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

My recommendation is for you to burn a new sd card for testing with the current RasPiOS 64 bit. It uses kernel 6.6 and works well with the AXML or any other adapter with the mt7921au chip.

> I want to use kali for the os...

I cannot keep up with everything but I think a user recently mentioned a new version Kali for RasPi's that has a newer kernel. You would have to check. The 3+ is a 64 bit Pi so a newer 64 bit Kali would work if available.

FYI: The new rtl8812au and rtl8821/11au rtw88 drivers have now been approved to be merged into kernel 6.13 so there is something to look forward to.

morrownr, 2024-11-17:
Status?

neuterscooter, 2024-11-18:
Hi there, I got busy with some school projects. The driver works perfectly now with 64-bit raspiOS. The issue I'm having now is with the 8812au driver on rpi5, not going into monitor mode. Can i close this issue and make a new one?

morrownr, 2024-11-18:
Yes, please start new issue about the 8812au not going into monitor mode.

closed 2024-11-18 by morrownr