morrownr/USB-WiFi #538: Please help to choose a good card

SuperUser-exe, 2024-11-13:
I'm looking for a WiFi adapter compatible with Kali Linux for WiFi hacking purposes, specifically supporting WPA2 and WPA3 (personal and enterprise mixed modes). The adapter should also support monitor mode, mesh, IBSS, and packet injection.

I found a suitable card (Alfa AWUS036AXML) via the link below, but it appears to be delisted due to lack of support. Alfa's website also recommends this model.

Could you please advise whether I should purchase this model or consider an alternative?

https://alfa-network.eu/wi-fi/kali-linux-compatible/alfa-usb-adapter-awus036axml

Thank you."

morrownr, 2024-11-13:
Hi @SuperUser-exe 

> Could you please advise whether I should purchase this model or consider an alternative?

I would consider an alternative. The AXML was delisted from The Plug and Play List and the Recommended Adapters for Kali Linux list a few months ago due to an ongoing issue that actually seems to be a problem with the bluetooth stack but this issue has been outstanding for some time. We know work is ongoing to find a solution but until the solution is in place and tested, I cannot recommend the AXML for new purchase.

> supporting WPA2 and WPA3 (personal and enterprise mixed modes). The adapter should also support monitor mode, mesh, IBSS, and packet injection.

Based on this list of requirements, my suggestion is the Alfa ACHM:

https://github.com/morrownr/USB-WiFi/blob/main/home/iw_list/ALFA_AWUS036ACHM_mt7610u.md

An alternative that would meet your list of requirements would be the Alfa ACM:

https://github.com/morrownr/USB-WiFi/blob/main/home/iw_list/ALFA_AWUS036ACM.txt

The ACHM has an edge in that it has particularly LONG RANGE.

The location of the Recommended Adapters for Kali Linux list is:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

The full Plug and Play List is located at:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

The site Main Menu:

https://github.com/morrownr/USB-WiFi

If you have more questions, let us know.

SuperUser-exe, 2024-11-14:
Thanks for your reply and giving detailed information, 

So I am going to buy this model this is what you suggested properly working right, check the below link and please confirm this come with

closed 2024-11-14 by SuperUser-exe

morrownr, 2024-11-14:
This is another location. Known good seller and good price:

https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter

I'm not sure what your location is.

> ALFA AWUS036ACH Long-Range

That is a good adapter but I can't recommend it for Kali... at least not at this time. It uses a rtl8812au chip. There is a new in-kernel driver coming to kernel 6.13 and it is a good overall driver but I have not tested the driver for use with Kali yet.

SuperUser-exe, 2024-11-14:
Thanks much your reply 

and as per your test (https://github.com/morrownr/USB-WiFi/blob/main/home/Speed_Comparison_Test.md) i can see this model **(ALFA AWUS036ACH Long-Range)** is very old and low speed So as per Alfa recommendation also your recommendation (https://www.alfa.com.tw/collections/kali-linux-compatible) iam going to buy ALFA AWUS036ACM (https://www.alfa.com.tw/products/awus036acm_1)
so my final question is this model will support All the kali linux application right also the WPA,WPA2,WPA2-PSK,WPA3 Monitor, injection everything is working fine right , this is only get answer from your side because you are well experienced this that why