morrownr/USB-WiFi #539: [Help]: Wifi adapter not showing up ALFA AWUS036AXML

wellitsmez, 2024-11-14:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux Test 6.11.2-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.11.2-1kali1 (2024-10-15) x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 413c:2106  QuietKey Keyboard Bus 003 Device 003: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

output is blank

### dkms

```text
nvidia-current/535.183.06, 6.11.2-amd64, x86_64: installed
```

### iw

```text
also the output is blank
```

### What happened?

well i have installed kali linux with kernel 6.8 then i made full upgrade also i have added the new firmware for MediaTech 7921 as you have mentioned at https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md when i run lsusb the command doesn't complete and get stuck with the info i showed above i have no clue really what to do will be appreciated if there's some assist from you thank you also that's the result for iwconfig iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.
no sign of the wireless adapter

wellitsmez, 2024-11-15:
i got it working without removing the BT file however what i did to make it work at kernel 6.11 was the following
sudo echo "blacklist btmtk \n blacklist btusb" >> /etc/modprobe.d/blacklist.conf
everything went fine BUT the performance of monitor mode is really bad 
quick compression between 6.11 which is not VM install just a main install and  kernel 6.8  on VM over windows 
i found out that the card is working over this kernel very good if i am getting 30 network at monitor mode for this install 
infront of it with the same position at the standalone install for 6.11 kernel i get about 8 networks available which i dont understand why

wellitsmez, 2024-11-15:
@morrownr yes what you understood is right also i have tired kernel 6.1 and 6.8 stand alone installation and while on monitor mode it shows 8 networks only i tried that yesterday that's why i found it really weird when I get such a result with stand alone installation

morrownr, 2024-11-15:
@wellitsmez 

Hmmm...

If you could give me the commands you are using to scan, I could take a look here with different adapters and kernels.

wellitsmez, 2024-11-16:
@morrownr sudo airmon-ng start wlan0 
to start monitor mode using ur wlan wireless card after that lets make simple scan from channels 1 to 13 with airodumb
sudo airodumb-ng wlan0mon

wellitsmez, 2024-11-16:
@morrownr quick update i switched from usb 3 to usb2 for any kernel and its performing really good so far i will keep you updated if i found any crashes

morrownr, 2024-11-16:
@wellitsmez 

A couple of months ago I wrote a message that I pinned in Issues in USB-WiFi:

https://github.com/morrownr/USB-WiFi/issues/508

The reason I wrote the message was because problems with USB WiFi adapters are generally blamed on things in the following order:

1. driver
2. adapter
3. other things

USB falls into other things but it probably should be number 1 in reality. USB3 was not mankind's greatest creation. I think the engineers that developed it were pressured into too quick of a development cycle and inadequate testing was done and we are still feeling the consequences today and USB3.2 is creating additional issues. The best thing us USB WiFi adapter uses can do is be knowledgeable of the potential USB issues and know what to do to determine the solutions.

If you see anything in my message that needs more information or clarification, please let me know and I will try to make it better.

wellitsmez, 2024-11-17:
@morrownr i am not sure if the problem is from the USB or not as i mentioned before i have tried it out with usb3 over vm installation and was working fine but the problem lays behind working over usb3 under stand alone installation

morrownr, 2024-11-17:
@wellitsmez 

Your description of the situation suggests to me that neither the adapter or driver is at fault because things are working as you wish in one circumstance. In the case where the Linux USB stack is handling everything usb, things are not working right so that is where I would say the problem is. It is likely a system specific problem in that a particular usb driver is problematic on a specific usb chipset. I'm suspect the bug hunt should look at usb support, not the adapter or its driver. Could I be wrong? Sure.

On the system where the adapter is running bare metal, are you seeing anything in the system log related to usb?

wellitsmez, 2024-11-17:
@morrownr lets say its a system issue what i am going to do right now after i played with BIOS Settings like a lot i would go into the manufacture web site get a new bios update for the latest version after that if i didn't find any difference what i would like a suggest from to install a new Linux os from ur suggestion and i will carry the hunt one more time and about anything related to the system log regarding to the usb i can see it's the same however i used usb2 or 3

morrownr, 2024-11-17:
A bios update could help.

A newer kernel could help as it might have fixes to parts of the usb stack.

After the bios update and trying kernel 6.12, which will be released tomorrow, all you have left if things are not  better is to start looking for hints as to what is causing the problem and my bet is it is something in the usb stack in Linux. You can determine the usb3 hub chipset and search for reports related to it.

wellitsmez, 2024-11-18:
another sad update tested it with kernel 6.12 poor poor performance i had to deleted the BT firmware to make it work also had to
 
 echo "install btusb /bin/false" >> /etc/modprobe.d/local-dontload.conf

well i think i give up

morrownr, 2024-11-18:
Before you give up, try contacting Alfa to tell them what is happening. 

https://www.alfa.com.tw/

wellitsmez, 2024-11-18:
@morrownr i don't think that they will care don't they see what is going on with this chip since along time and the latest release for the firmware they have since early 2023 but alright in this case i will give it a shot i will get in touch with them

Ray78PM, 2024-11-18:
Hey thanks all for trying to solve this one. Picked up the AWUS036AXML over the weekend. Tried on two VM's running kali and parrot with kernels 6.11.2 and 6.10.11, respectively. I am using vbox for virtualisation. Managed to somehow get it working in parrot once. NFI. Have noticed issues with enabling bluetooth in the two VMs with this adapter. 

I also have Ubuntu on bare metal running kernel 6.8.0-48 and am having some success, but it seems with only certain USB cables. Not 3.0 cables - I have tried two 3.0 cables without success. The official ALFA cable will not work in either USB-A or USB-C ports and even trying it will cause the system to hang on reboot or restart. Using a 3.0 cable causes issues for Ubuntu and a simple logout/login can help. Note: I also have a Netgear A8000 (mt7921au) that is still works perfectly on all machines.

wellitsmez, 2024-11-18:
@Ray78PM hello ray i have quick update i managed to make it work at  kernel 6.11.2 with firmware for the adapter dated at 26-8-2024
if you are going to use kali use 2024.3 version when you install it over vm  

make sure that the usb port for the VM is usb3 (VM Settings) 
IMPORTANT NOTE :connect the alfa adapter to the vm while installing kali

then when you finish the installation do the following 

sudo apt update 

sudo apt update full-upgrade -y

that's only to get the new kernel 6.11.2 and some optimization

what you are going to find now at /lib/firmware/mediatek

supposed to be the firmware that is dated to 26-8-2024

then 

try to follow these steps unplug your adapter from the usb

modprobe -r btusb
then

sudo echo "blacklist btmtk \n blacklist btusb" >> /etc/modprobe.d/blacklist.conf

sudo reboot

install the firmware dated at 26-8-2024

then sudo reboot
when you login to your linux system plugin your alfa device
at this point i didn't get any crash so far

wellitsmez, 2024-11-19:
Quick update make sure you plugin your alfa adapter to usb2 interface but let it be for the vm settings as USB 3 have tried it everything for the past 10 hours didn't get any crash monitor mode and injection working perfect

wellitsmez, 2024-11-24:
Hello 
well i opened this issue due to something i noticed during making the AP work correctly final conclusion well be the following

1- USB3 is a big problem for ALFA AWUS036AXML if you tried to run the network adapter through USB3 you will get a huge 

downgrade at the performance  in general specially at packet injection and monitor mode 

2- second issue BT Won't work at all except from what i seen is working only at ubuntu 

3- how to make ALFA AWUS036AXML work well at kali linux as this was the topic of my research for stand alone installation , you can refer back to the old comment if you would like to make it work at VM 

3-1   plugin your alfa card before booting into installation phase to USB2  

3-2 install kali linux as stand alone installation , *** yes it works if you did dual boot also****

**************************************************IMPORTANT NOTICE FROM THIS PART  **********************************************

open your terminal 

3-3 sudo apt update

3-4 sudo apt update full-upgrade -y

3-4-1To install or update the firmware:

Go to the following site:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

Click on WIFI_MT7961_patch_mcu_1_2_hdr.bin

Click on plain

Save file

Click on WIFI_RAM_CODE_MT7961_1.bin

Click on plain

Save file

If your adapter/card has Bluetooth support:

Click on BT_RAM_CODE_MT7961_1_2_hdr.bin

Click on plain

Save file

Create the needed directory (if necessary):

sudo mkdir /lib/firmware/mediatek

Copy the files to the following locations:

sudo cp WIFI_MT7961_patch_mcu_1_2_hdr.bin /lib/firmware/mediatek

sudo cp WIFI_RAM_CODE_MT7961_1.bin /lib/firmware/mediatek

sudo cp BT_RAM_CODE_MT7961_1_2_hdr.bin /lib/firmware/mediatek 

3-5 unplug your alfa card 

3-6 sudo reboot

3-7 modprobe -r btusb

3-8 sudo su 

3-9 sudo echo "blacklist btmtk \n blacklist btusb" >> /etc/modprobe.d/blacklist.conf

3-10 sudo  echo "install btusb /bin/false" >> /etc/modprobe.d/local-dontload.conf

3-11 sudo reboot

3-12 plugin your alfa card to USB2 AND ENJOY

wellitsmez, 2024-11-24:
quick update : When doing evil twin attack , switching to VIF mode don't expect deauth will work at all ..... that's getting crazy i am about to pull my hair off

morrownr, 2024-11-24:
@wellitsmez 

> 3-12 plugin your alfa card to USB2 AND ENJOY

USB2 is much more stable than USB3.

> second issue BT Won't work at all except from what i seen is working only at ubuntu

I am actually in contact with Alfa since last week. I will report as able.

My AXML does not support BT but for example, if I plug my adapter into a clean installation of Ubuntu 24.10 and boot, a wireless interface never comes up without the workaround we have discussed. It seems the reason is that the system is detecting BT support but then hangs when the BT support is not actually there. Hopefully we can get to some kind of resolution.

If I have something important to report from Alfa, I will likely start a new issue because we have several AXML users.

> am about to pull my hair off

Take a deep breathe and maybe we can get to the bottom of this.

wellitsmez, 2024-11-25:
@morrownr with 6.11.2 every thing went smooth except the bt as you know also the big surprise after i thought everything was good it was while doing evil twin attack i found out when you change your card to VIF mode guess what the deuath packets not working anymore i didn't believe that i pulled Wireshark and yeah it's not working, the good news everything is working fine so far with kernel 6.0.0 except the bt part with the latest firmware also the sad part is the scanning part compared to 6.11.2 or any other kernel downgraded by half which means if i can get 50 network @6.11.2 i will get 25 network @ 6.0.0 kernel also seemes like we have a kernel issue when it comes to switching your adapter to VIF mode .......

wellitsmez, 2024-11-25:
i figured out what happened when changing the interface into VIF mode 
aireplay-ng  --deauth=0 -a XX:XX:XX:XX:XX mon1
09:06:02  Waiting for beacon frame (BSSID: XX:XX:XX:XX:XX) on channel -1
09:06:02  Couldn't determine current channel for mon1, you should either force the operation with --ignore-negative-one or apply a kernel patch
as you can see the channel switched to -1 i think we need to start from here

wellitsmez, 2024-11-30:
@morrownr is there any updates from alfa guys?

morrownr, 2024-11-30:
>  any updates from alfa guys?

Nothing to pass on yet.

I have been busy but I did read your other messages above. Some advice: There are people that stop by here at times that are a lot smarter on monitor mode than I am. The problem with your specific reports down in this thread is that almost nobody will see them. When you identify a new specific monitor mode issue, start a new thread.

My expertise is more in AP mode than monitor mode but I do know some really knowledgeable people.

wellitsmez, 2024-11-30:
@morrownr hello , i hope that you are doing well i managed to solve most of the problems above by getting kernel 6.0.0 and the old firmware at ALFA site except the bluetooth keep crashing the kernel from 6.0.0 till 6.12.1

closed 2024-11-30 by wellitsmez