morrownr/USB-WiFi #540: Please help me choose a good card

Zyluxar, 2024-11-17:
- Looking for a WiFi Adapter
	- Laptop does not support Monitor mode
	- Want to listen to WiFi packets on network etc and analyze to find out information and also want to potentially use this to learn how to crack WiFi passwords using my own WiFi network aswell as potentially learn ARP Spoofing.
	- Want to know what Active Monitor Mode does as Netgear A8000 that I am looking at has a [3] indicating that Active Monitor Mode is broken. Does this affect my use case? What does it do?
	- Does the Netgear A8000 work for my use cases? If not what would?
	- I am on Pop OS! so I would prefer compatibility with Linux and such
	- My range needs aren't high I just need it to be work in my home
	- I would be traveling with this adapter when I go to my families so I would prefer if it was portable and I could bring it through airport security etc (not really a requirement since I might not even bring it)

morrownr, 2024-11-17:
Hi @Zyluxar 

I modified the title because it looks like you are looking for a USB WiFi adapter, not an internal card. I hope you do not mind.

> use this to learn how to crack WiFi passwords using my own WiFi network as well as potentially learn ARP Spoofing.

I'm assuming this is a white hat activity as you say "your own network" as black hat activity is not supported here.

> Want to know what Active Monitor Mode does as Netgear A8000 that I am looking at has a [3] indicating that Active Monitor Mode is broken. Does this affect my use case? What does it do?

Let me suggest that you also ask questions at the following site that is run by @ZerBea :

https://github.com/ZerBea/hcxdumptool

I had one recent report that Active Monitor Mode is working again on very recent kernels but I have not seen a confirmation. Keep in mind that Active Monitor Mode is not Monitor Mode. It is simply a feature of Monitor Mode that a few use cases require.

> Does the Netgear A8000 work for my use cases? If not what would?

Given that you are not looking for maximum range, I think the Netgear A8000 would work well as the users of that adapter have never reported any problematic issues. It seems to be a really solid adapter.

> I am on Pop OS! so I would prefer compatibility with Linux and such

Linux support is what this site is about so you have a lot of Linux users stopping by. Individual distros may, at times, break something but your more popular Mainstream distros like PoP OS! fix problems fairly quickly  so I don't think this would be an issue.

I think any of the adapters in the `Recommended Adapters for Kali Linux` list will work for you but since you may travel with it and long range does not seem to be an issue, you might also consider some of the smaller adapters. All of the adapters in the Kali recommended list are also in the Plug and Play List which provides links to products so that you can see each product.

Let me know you have additional questions.

Zyluxar, 2024-11-17:
Thanks for your reply! Yeah it's white hat activity mostly just wanting to see how insecure my network is and just general curiosity on how these things are done. I think I'll be going with the Netgear A8000 based on the additional information you provided me. I really appreciate your help and the information that you provided me with!

closed 2024-11-17 by Zyluxar