morrownr/USB-WiFi #544: [Help]: WOWLAN for RTL8188EE

ch3fdan, 2024-11-26:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

uname -a Linux radio-shack 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 003: ID 2001:331e D-Link Corp. 802.11ac NIC Bus 001 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 005: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse Bus 002 Device 004: ID 152d:1561 JMicron Technology Corp. / JMicron USA Technology Corp. JMS561U two ports SATA 6Gb/s bridge Bus 002 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub Bus 002 Device 002: ID 0408:5220 Quanta Computer, Inc. HP Webcam Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

sudo rfkill list all 0: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no 1: phy1: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
rtl88x2bu/5.13.1, 5.15.0-126-generic, x86_64: installed
rtl88x2bu/5.13.1, 6.8.0-48-generic, x86_64: installed
rtl88x2bu/5.13.1, 6.8.0-49-generic, x86_64: installed
```

### iw

```text
sudo iw dev
phy#1
        Interface wlx4086cb22d6d2
                ifindex 4
                wdev 0x100000001
                addr 40:86:cb:22:d6:d2
                ssid The_ASUS_Project
                type managed
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 15.00 dBm
phy#0
        Interface wlo1
                ifindex 3
                wdev 0x1
                addr 80:c5:f2:12:7b:ef
                type managed
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcoltx-bytes tx-packets
                        0       0       0       0       0       0       0      00
```

### What happened?

i am looking to enable WOWLAN. I poked around in my system, checked the internet and couldn't find the answers I was lookimg for. Is it possible to enable this feature? Am I missing something completely obvious? Thanks in advance for any help/advice!

dubhater, 2024-11-27:
To me it looks like the rtl8188ee module doesn't implement wowlan.

ch3fdan, 2024-11-27:
when i run 'iw list' i see this:
 iw list
Wiphy phy1
        wiphy index: 1
'...'
        WoWLAN support:
                 * wake up on anything (device continues operating normally)
        software interface modes (can always be added):
      
so then i tried :
 > iw phy phy1 wowlan enable
command failed: Operation not permitted (-1)

> iw phy phy1 wowlan show  
WoWLAN is disabled.
                        
So I guess I am asking, Is there no way to enable this currently? And I should just get a new usb wifi adapter?

dubhater, 2024-11-27:
So actually you want to use wowlan with a RTL8812BU, not RTL8188EE?

Maybe this driver can do it if you enable wowlan in the makefile: https://github.com/morrownr/88x2bu-20210702

morrownr, 2024-11-27:
Confirm what device it is your are asking about. If rtl8812bu, then the repo @dubhater pointed you to could work. Use a test editor and open the file called Makefile. Search for the following line:

CONFIG_WOWLAN = n

You may want to change it to:

CONFIG_WOWLAN = y

Save the file and recompile/reinstall the driver.

There are some other wowlan setting but you will need to do the research to see what they do.

ch3fdan, 2024-11-27:
Apologies, yes rtl88x2bu is what i am talking about. I confused the onboard with the usb.
i edited makefile, recompiled, reinstalled.
ran i couple of commands:
> iw phy | grep -i wake
                 * wake up on anything (device continues operating normally)
 
> modinfo 88x2bu | grep -i wake
parm:           rtw_wakeup_event:uint

> sudo modprobe -r 88x2bu
sudo modprobe 88x2bu rtw_wakeup_event=1

and that is as far as I have gone because I was doing it on ssh and it stopped responding. 
I'm guessing it didn't work. :)

morrownr, 2024-11-27:
@dubhater 

Have you noticed wowlan support in rtw88?

ch3fdan, 2024-11-27:
No, not at all. i had a rtl881cu that worked on windows but not on linux so i was just curious and wanted to see what the experts thought. thanks for your time.

closed 2024-11-27 by ch3fdan

morrownr, 2024-11-27:
@ch3fdan 

The question that I asked was to Dubhater. He has been doing a lot of work on the rtw88 in-kernel series of drivers this year and it would be faster to ask him than to go digging into the code.

> i had a rtl8811cu that worked on windows but not on linux

If you still have that adapter, you might want to try it again. A lot of progress has been made with rtw88 this year.

ch3fdan, 2024-11-27:
i will give it a go and i will keep messing with this one too.

dubhater, 2024-11-28:
Yes, rtw88 supports wowlan with the RTL8822CE.