morrownr/USB-WiFi #545: [Help]: MT7925 no multi-link operation???

fhteagle, 2024-11-27:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux pluto 6.12.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 22 Nov 2024 16:04:27 +0000 x86_64 GNU/Linux

### lsusb

lspci:  ... aa:00.0 Network controller: MEDIATEK Corp. Device 7925 (rev 01) ....

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
nvidia/565.57.01, 6.11.10-hardened1-1-hardened, x86_64: installed
nvidia/565.57.01, 6.12.1-arch1-1, x86_64: installed
nvidia/565.57.01, 6.6.63-1-lts, x86_64: installed
```

### iw

```text
phy#0
	Interface wlan0
		ifindex 2
		wdev 0x1
		addr 8c:53:e6:d6:5d:78
		ssid GreatCircle
		type managed
		channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
p
```

### What happened?

I have been testing both an Intel BE200 and a Mediatek MT7925 (each in m.2 card form factor, STA mode) with a TP-Link BE3600 dual band AP. BTW, if anyone wants any specific A/B testing between these two cards, let me know.

With the BE200, I get the MLO lines at the bottom of the output of ```iw dev```. With the MT7925, no MLO output lines. See iw dev output above. I have forced the STA to associate with each of the AP's BSSIDs to make sure it really is trying to connect to the MLO activated SSID, but no change.

```
ethtool -i wlan0
driver: mt7925e
version: 6.12.1-arch1-1
firmware-version: ____000000-20241104133053
expansion-rom-version:
bus-info: 0000:aa:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

This is really not a big deal for me as I can go back to the BE200 (both are giving pretty similar real world iperf3 results throughout my house). Even the BE200 seems to be only using enhanced multi-link single radio (EMSLR) anyway, so no boost in real throughput or latency with the BE200. But I like Mediatek better in general, so if anyone has any tricks to get MLO working on MT7925, please let me know.  Or if the answer is just a "wait for it to make it into the firmware", that's fine too.

morrownr, 2024-11-27:
@fhteagle 

My memory may be off a little but it seems like the mt7925 driver was first merged into kernel 6.7 without MLO support but then around July/August of this year the MLO support flowed. Not able to test MLO support here yet.

fhteagle, 2024-12-05:
I switched back to the BE200.

Example output of iw dev:

```
phy#0
	Unnamed/non-netdev interface
		wdev 0x4
		addr dc:97:ba:e6:56:XX
		type P2P-device
	Interface wlan0
		ifindex 2
		wdev 0x1
		addr dc:97:ba:e6:56:XX
		ssid XXXXXXXXX
		type managed
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
		MLD with links:
		 - link ID  0 link addr 06:c5:7a:cd:49:61
		   channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		 - link ID  1 link addr aa:2a:ad:e8:8b:a9
```

morrownr, 2024-12-13:
@fhteagle 

There was a series of 17 patches that came in from a Mediatek dev earlier this week. It looks like several have to do with MLO on the mt7925.

fhteagle, 2024-12-13:
I threw the MT7925 card in my custom build router to play with hostapd on it at some point. I will have a look to see if it gets the MLO lines and capability at some point. Holidays are bearing down on me though....

morrownr, 2024-12-13:
I'd like to hear an AP mode report on the mt7925 as I have only used mine in managed mode so far.

fhteagle, 2024-12-13:
Definitely on my "play time to do list". Just got a TPLink BE9300 AP that I picked up for $100 on black friday deal. Having fun shaking that down at the moment.

morrownr, 2024-12-13:
Right now I'm working a couple of adapter makers as far as them producing a USB adapter version of the mt7925. It is normal for the card version of the chip to be available several months before the USB version but we are at about that point so I'm hoping we see some adapters soon. The desire is there but the USB chips have not started flowing yet.

Mad-Dad, 2025-01-26:
I have the 7925 as built into the x870 asus board, it does not advertise bluetooth, but I cannot seem to add the controller to a bridge.  I am new to this so not sure if I am doing something wrong
```
sudo brctl addif br0 wlp9s0 
can't add wlp9s0 to bridge br0: Operation not supported

sudo lshw -C network
  *-network UNCLAIMED       
       description: Ethernet controller
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:08:00.0
       version: 0c
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd cap_list
       configuration: latency=0
       resources: ioport:e000(size=256) memory:dd000000-dd00ffff memory:dd010000-dd013fff
  *-network
       description: Wireless interface
       product: MEDIATEK Corp.
       vendor: MEDIATEK Corp.
       physical id: 0
       bus info: pci@0000:09:00.0
       logical name: wlp9s0
       version: 00
       serial: 28:d0:43:5d:1e:08
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=mt7925e driverversion=6.8.0-51-generic firmware=____000000-20231227093232 ip=192.168.1.20 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:140 memory:dca00000-dcbfffff memory:dcc00000-dcc07fff
```

morrownr, 2025-01-26:
Hi @Mad-Dad 

> it does not advertise bluetooth...

What kernel are you using?

$ uname -r

Also, post the results of:

$ lsusb

>  I cannot seem to add the controller to a bridge.

What are you trying to do?