morrownr/USB-WiFi #548: [Help]: mt7921au (EDUP AX3000) dual-band/tri-band AP mode on OpenWRT

paewie, 2024-12-03:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux OpenWrt 5.15.167 #0 SMP Mon Sep 23 12:34:46 2024 aarch64 GNU/Linux

### lsusb

Bus 001 Device 005: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 001 Device 003: ID 0424:ec00 Bus 001 Device 004: ID 2357:0601 TP-Link USB 10/100/1000 LAN Bus 001 Device 002: ID 0424:9514 Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.167 dwc_otg_hcd DWC OTG Controller

### rfkill

Not applicable to OpenWRT(?)

### dkms

```text
Not applicable to OpenWRT(?)
```

### iw

```text
phy#0
        Interface phy0-ap0
                ifindex 27
                wdev 0x13
                addr e8:4e:06:ad:99:4b
                ssid OpenWRT
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       70      0       0       0       0       8502            71
phy#1
        Interface phy1-ap0
                ifindex 5
                wdev 0x100000001
                addr ba:27:eb:6b:3c:66
                ssid OpenWRT
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 31.00 dBm
```

### What happened?

Hello @morrownr and contributors,

I ran into an issue and hope you guys can help point me in the right direction.

I'm trying to replace the ISP-provided (rented) router with a Raspberry Pi 3B running OpenWRT. The limited I/O (e.g. 100 MBit/s) does not pose a problem as the internet speed does not surpass these values. So far, almost everything runs fine.

To enable WiFi I consulted your plug and play list, read about some other caveats you pointed out around this repo, and finally chose the relatively newly added EDUP AX3000(*) for its WiFi 6e support (mt7921au chipset). Now I ran into a problem setting up the Tri-Band configuration however. Currently I am not sure where exactly the problem lies and I only have limited knowledge about Linux's (and therefore OpenWRT's) hardware interactions. It might be an OpenWRT issue but I could not find any mentioning of this online or in their guides except a pointer at the chipset's capabilities.

To go into more detail, every Frequency band for itself works perfectly fine (2.4 / 5 / 6 GHz), however as soon as I try to enable - say - the 2.4 and 5 GHz simultaneously, none of both can accept any client connections with only one broadcasting its SSID (or sending anything at all). In the OpenWRT interface, only the first configured SSID shows to have a BSSID, any subsequent ones report "Wireless is not associated" [image] In addition I noticed that I can only set up one "Operating Frequency" for all SSIDs that would use the same radio.

Investigating a bit, I found the hint towards checking VAPs support, which it reportedly does not
```
root@OpenWRT:~# iwinfo
phy0-ap0  ESSID: "OpenWRT"
          Access Point: E8:4E:06:AD:99:4B
          Mode: Master  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/g/n
          Hardware: USB 0E8D:7961 [MediaTek MT7921AU]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: no  PHY name: phy0
```
Before I dig deeper into this with e.g. a firmware update, I wanted to ask you guys (since you likely are more knowledgable than me) if the Stick/Chipset even supports running Tri-Band operation. I assumed it would from this entry on the plug and play list ("* AP/VLAN")
```
Note: Single-state (no windows driver onboard, wifi only adapter.
Note: Uses the mt7921au chipset.
Note: Uses the standard Mediatek device ID (VID/PID) for the mt7921au chipset: ID 0e8d:7961
Note: Oldest kernel that supports this adapter: 5.18
Note: Oldest LTS kernel that supports this adapter: kernel 6.1
Note: Recommended kernel: 6.6 or later
Note: Supported interface modes with recommended kernel:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
Note: Removable antennas.
```

On a side note, I found something that sounds related in #107 (Bug no. 8), even though I am not sure if this is related. For now, I let the EDUP only provide the 5 GHz range and the builtin Antenna a 2.4 GHz radio, but I would like to be able to return the Stick and buy another one if it actually ends up not supporting my use-case.

(*) I noticed you refered to it as "EDUP EP-AX1672", however your provided links point to the AX3000(M) with the same chipset

Please let me know any ideas you have, thanks in advance!

morrownr, 2024-12-03:
Hi @paewie 

Welcome. 

> however as soon as I try to enable - say - the 2.4 and 5 GHz simultaneously,

There is the problem. USB WiFi adapters do not do "simultaneously". USB WiFi adapters have one radio inside. That radio may be capable of 2 or more bands but not more than one at a time. For you to run dual band, you need two radios. You can use your Edup adapter for 5 Ghz and the internal for 2.4. If you do not like the internal radio, I always turn if off as soon as I burn a new sd, then you need 2 usb wifi adapters and if you want tri-band, you guessed it, you need 3 usb wifi adapters.

Commercial wifi router have multiple radios inside.

> (*) I noticed you refered to it as "EDUP EP-AX1672", however your provided links point to the AX3000(M) with the same chipset

AX3000(M) = class of the adapter
EDUP EP-AX1672 = model number of the adapter

Yes, there is a lot of inconsistency in how adapters are shown in ads. One of the big reasons that I established this site several years ago.

My opinion: The Edup adapter that you have should be about as good as it gets. I certainly enjoy using mine with a RasPi4B. It is somewhat future proof in that it will provide fast service for many years even if you upgrade your internet service and move to a RasPi5B.

paewie, 2024-12-03:
Thanks for your quick response!

> USB WiFi adapters do not do "simultaneously".

I was afraid this was where this would be going. I do appreciate the clarification, I sure misunderstood some of the terminology around this.

> If you do not like the internal radio, I always turn if off as soon as I burn a new sd

That was the original plan. I have only rarely used the WiFi on any Pis and always preferred wired ethernet. Hopefully the internal radio holds up to the few 2.4 GHz-only devices I have. I might need to reconsider getting a cheap dedicated AP otherwise.

> there is a lot of inconsistency in how adapters are shown in ads

I see. It did just occur to me while typing this that this might be a different model, which is why I pointed it out.

Also, I fully agree on your opinion regarding the Pi 4 or 5 with faster service. That actually was one of the reasons why I chose this particular stick. I'll give this situation another thought.

Well, thanks again for the detailled response. I appreciate it.

closed 2024-12-03 by paewie

morrownr, 2024-12-03:
>  I might need to reconsider getting a cheap dedicated AP otherwise.

OpenWRT One is available these days. I posted about it the other day here in Issues. It might not be as cheap as you are looking for though.

> Well, thanks again for the detailled response. I appreciate it.

Sure. Good luck.

jfargen, 2025-04-22:
Apologies for coming into this thread late. The RPI3b has comes with a Wifi Chipset built-in. Why not use the built-in for 2.4Ghz and the EDUP adapter for 5Ghz?

morrownr, 2025-04-22:
Hi @jfargen 

> The RPI3b has comes with a Wifi Chipset built-in.

What you propose could work. The onboard wifi is Broadcom and they have a well deserved reputation for poor Linux support. I always turn off the onboard wifi on PI's so I don't even know if the RasPi wifi drivers have gotten better.

paewie, 2025-04-23:
Hi @jfargen 
> Apologies for coming into this thread late. 
Thanks for your reply anyway.

> The RPI3b has comes with a Wifi Chipset built-in. Why not use the built-in for 2.4Ghz and the EDUP adapter for 5Ghz?
I actually did that as a temporary solution. However, there are multiple issues as @morrownr hinted towards. The first one being the stated bad Linux support. You can only operate the onboard WiFi as AP on Channel 6 (all others crash with no conclusive error message). Additionally, only legacy operations modes (802.11b iirc) are available. Secondly, the builtin antenna is relatively weak. Up to the point where I could only reach less than 1 MBit/s transmission speeds in the corners of my appartment, the router being roughly in the middle.

In the meantime I bough a cheap, second WiFi Stick to provide the 2.4 GHz radio. I missed the reports about not using the same chipset twice, nevertheless I got this to work very reliably with the new OpenWRT 24.10 and a firmware update.

morrownr, 2025-04-23:
@paewie 

> I missed the reports about not using the same chipset twice...

Not using 2 USB WiFi adapters with the same chipset on the same system just applies to Realtek chips if using the Realtek out-of-kernel drivers... unless I have missed something.

paewie, 2025-04-24:
@morrownr that might be true, I can't currently find this piece of advice again (and how carefully I read the context).

I was observing issues with two mt7921au-based Sticks concurrently, but that might have been due to a number of different resons. This combination had the tendency to crash either instantly or (with a firmware update) after about a week on the older OpenWRT with Kernel 5.15. One of them (always the same one iirc) got into some "corrupt" state where the Kernel could not set AP Parameters after a failure to reset the device. Replugging, rebooting, or manually resetting the device in Luci worked for a limited time. It got fixed with the newer OpenWRT and same hardware. I do not believe it was a power issue, as I was/am using a powered USB-Hub on both OS versions.

The crash included some kernel stackstrace mentioning the mt79** kmod, which I did not observe with only one stick and the builtin adapter. However, as I said, I'm not 100% positive what that issue was in the end.

morrownr, 2025-04-24:
@paewie 

I do not doubt your experience at all. With the most recent stable OpenWRT, that may not be the case anymore. The mt7921u driver only went into the kernel with kernel 5.18 and AP mode did not happen until kernel 5.19. So it had to be backported to 5.15 for OpenWRT. That can be problematic when we are talking about new technology such as WiFi 6 and 6 GHz. Heck, the wifi stack has had so much work on it over the last 3 years and it continues with WiFi 7. 

My testing with 2 mt7921 and 2 mt7612u based adapters with hostapd on modern versions of RasPiOS show no problems but then my testing is with code where many patches have gone in. I would bet that if you tried again on the recently released OpenWRT, you might not see that problem.

paewie, 2025-04-24:
@morrownr I suspected there might be issues with the backport as well, especially since the firmware update did improve the situation without completely solving it. I made no efforts to test the hardware on a Raspberry OS since each individual stick worked fine in each frequency band. However, as you suspected, using the 24.10 release of OpenWRT with Linux kernel version 6.6 has seemingly solved these behaviours. It's been running for quite some time now without any crash or hangup I could recall.

It's indeed interesting to see how much work there has been on the new WiFi technologies. It almost felt as if WiFi 6 was just about getting adopted finally.