morrownr/USB-WiFi #549: [News] Expect numerous changes to this site regarding Realtek WiFi 5 support over the next few months...

morrownr, 2024-12-06:
Hello everyone,

About one year ago, Larry Finger, Bitterblue Smith and I started a project to improve Realtek WiFi 5 support on Linux. Unfortunately Larry passed away in May but others jumped in to help with one thing and then another. Bitterblue ( @dubhater ) has carried the majority of the load and I thank him for that.

What does this mean as far as changes to this site? The Plug and Play List will slowly be updated starting in 2025 to include Realtek WiFi 5 adapters that now have good support in the kernel. Realtek WiFi 5 adapters are still available on the market and likely will be for many years. Additions with be made slowly because it takes time to test and determine if any specific adapter is likely to be trouble-free on Linux. I will call for your suggestions when the time is right. No multi-state (windows driver onboard) adapters will be considered. Multi-function (wifi and bluetooth) will on be considered for adapters that are USB2. I already have a small amount of adapters to add to the list but I look forward to your input.

Warning: This message is about Realtek WiFi 5 support, not WiFi 6 support. My messages warning Linux users to AVOID USB WiFi adapters that use Realtek WiFi 6 chips is still applicable. Good support for Realtek WiFi 6 USB chips could be years away if it happens at all. Please heed this WARNING and buy adapters with Mediatek chips for WiFi 6 and 7. I will do my best to keep a variety of WiFi 6 and 7 USB WiFi adapters that use Mediatek chips in the Plug and Play List so that you are aware of the better options.

Adapters will the following WiFi 5 chips are now supported in rtw88 (in-kernel):

USB : RTW8812AU, RTW8821AU, RTW8811AU, RTW8822BU, RTW8812BU, RTW8822CU, RTW8812CU, RTW8821CU, RTW8811CU, RTW8723DU

The rtl8812au and rtl8821/11au drivers just went into kernel 6.13. rtl8814au is in development and not ready at this time. Remember, if you upgrade to kernel 6.13 to check the drivers out:

- your distro has to add the firmware to their distro for the driver to work.
- your distro has to turn on compiling of these new drivers. Ubuntu, I'm looking at you.

If you want to try the drivers on kernels older than 6.12, I recommend going to the development repo and installing from there for the time being.

https://github.com/lwfinger/rtw88

If you have questions, please post below.

@morrownr

jaearr, 2024-12-11:
>No multi-state (windows driver onboard) adapters will be considered. Multi-function (wifi and bluetooth) adapters will only be considered for adapters that are USB2.

I have two multifunction single state Realtek WiFi 5 adapters (rtl8822bu) from Aliexpress - the [Fenvi AC1300](https://www.aliexpress.us/item/1005005061612054.html) and the [Comfast CF727B](https://www.aliexpress.us/item/1005001279319429.html). My setup isn't as noise free as the testing you've done in the [speed comparison test section](https://github.com/morrownr/USB-WiFi/blob/main/home/Speed_Comparison_Test.md), but it's worth noting that the CF727B (which is USB2 only) has extremely bad 2.4GHz performance: 3-6Mbps if you're streaming audio over bluetooth, 14-25Mbps if bluetooth is rfkill blocked. The Fenvi, which will do USB3 with the modprobe flag (tested in Debian 12 with 88x2bu-20210702), was doing 17-23 Mbps on 2.4GHz with bluetooth transmitting, 80-87Mbps with bluetooth blocked.* They have similar 5GHz performance (USB2: 207-326Mbps, USB3: 220-408Mbps).

I could not get the rtl8822bu to connect to more than 1 bluetooth device at the same time in Linux (I wanted to test with iperf3 while playing tuxracer with a bluetooth gamepad and headset connected; could pair both but not connect both), so that would be another thing worth mentioning if your tests come up with a similar result.

Is there a way to switch from USB2 to USB3 with the in-kernel drivers? 

My setup was a WRT-1900AC v1 running OpenWRT 23.05.5 with iperf3. 
*For reference: all numbers are the iperf3 10 second average as a client. The highest 2.4GHz/20MHz average I got with iperf3 was 99.8Mbps with an rtl8812au (8812au-20210820, Debian 12) on USB3. The highest 2.4GHz/40MHz (forced) average was 143Mbps with an mt7921au (OpenSuSE Tumbleweed 6.11.8-1) on USB3. The absolute worst was 2.4GHz/40MHz (forced) with the Comfast CF727B with bluetooth active: 310Kbps-1.98Mbps. Chipsets I tested were rtl8812au, rtl8822bu, mt7921au, Intel 7260 mPCIe, and Intel BE200 ngff. 

tl;dr: the correct USB answer for me is still a mt7921au with a separate CSR bluetooth adapter, but now with numbers to support that conclusion.

morrownr, 2024-12-11:
@ZerBea 

Thanks for the post. dubhater has worked tirelessly for over 1 year to make this happen. I think things will be stabilized and distros will have done the things they need to do by maybe around April 2025 (time for Ubuntu 25.04). USB wifi support for Linux is becoming good and is getting better. One big sore spot that I continue to warn Linux users about is to avoid adapters based on the Realtek WiFi 6 chips. The drivers are out-of-kernel and are not that good plus almost all adapters based on the Realtek WiFi 6 chips are multi-state and I can speak first hand that some are very problematic in this regard. The rtl8852/32cu chips are very bad when it comes to problems with onboard storage for the Windows driver. I am working on a vendor driver for the rtl8852/32cu chips right now and I'm pulling my hair out trying to figure out some problems.

morrownr, 2024-12-11:
@ZerBea 

> Both drivers have received extensive improvements.

Yes, it took many hours of testing by many testers as well as patches and retesting and more patches... and not everything is in mainline yet. There are patches waiting to get in and I'm sure there will need to be some clean up patches for things that were missed. I hope that users understand that this effort was not by Realtek but rather by Linux community members. And I hope Linux users understand that they have a choice. Mediatek developments and supports their drivers the right way and it is a way that works well for desktop/laptop users.

Keep us posted with your testing. I mostly test AP and managed modes so input on monitor mode is good.

jaearr, 2024-12-29:
@morrownr 
>USB3 support for the rtl8822cu will show up in kernel 6.14

[On line 20 of usb.c](https://github.com/torvalds/linux/commit/315c23a64e99552502dd4d18d6ddc073fad9a7c3#diff-76aab974876d8a05c1eb7a968cab49d3d045e4ee25fabb3a87b2a5bba349181b) it looks it will default to USB3. When I tested and enabled USB3 on the backported driver (88x2bu-20210702) the Comfast adapter wouldn't start but the Fenvi would work fine. I'll test when 6.14 makes it into OpenSuSE Tumbleweed to see how it behaves.

>locked to USB2

[I acknowledge that Realtek says it's a USB2 interface](https://www.realtek.com/Product/Index?id=585&cate_id=194), although B-Link's filing to the FCC [shows USB superspeed pins in section 5.2](https://fcc.report/FCC-ID/2AL6K-R8822BU1-A/4540816.pdf), just like Mediatek's equivalent (but no design wins to test) [MT7663BUN](https://mediatek-marketing.files.svdcdn.com/production/documents/MT7663BUN-Datasheet-Brief_v1.3.pdf?dm=1696869430) which also says it's limited to USB high-speed yet shows superspeed pins in section 3.1. I'll need to make a better test bench so I can exceed USB high-speed on 5GHz/80MHz. `lsusb -t` output for the Fenvi:
`/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M`
`    |__ Port 2: Dev 2, If 0, Class=Wireless, Driver=btusb, 5000M`
`    |__ Port 2: Dev 2, If 1, Class=Wireless, Driver=btusb, 5000M`
`    |__ Port 2: Dev 2, If 2, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M`

It may not be supported by the chipmaker even if it's capable (e.g. like [the AMD PCIe 4.0 thing](https://www.techpowerup.com/258044/amds-latest-agesa-update-removes-pcie-4-0-support-from-pre-x570-motherboards)).

>vendor driver

I presume OpenSuSE uses the same kernel bluetooth firmware as everyone else. I tested bluetooth+wifi with the kernel 6.11.8 driver when I found the issue with headset+gamepad not connecting at the same time. I did not test multiple bluetooth connections using the backported driver as I assumed it would be the same as upstream. I did not make an attempt to use the vendor driver. I could try if there is a recommended OEM realtek driver source to confirm it has always been an issue.

dubhater, 2024-12-29:
> I presume OpenSuSE uses the same kernel bluetooth firmware as everyone else. I tested bluetooth+wifi with the kernel 6.11.8 driver when I found the issue with headset+gamepad not connecting at the [...]

Isn't that a Bluetooth 5.0 feature? RTL8822B has Bluetooth 4.2.

morrownr, 2024-12-30:
@jaearr 

> I acknowledge that Realtek says it's a USB2 interface, although B-Link's filing to the FCC shows USB superspeed pins in section 5.2, just like Mediatek's equivalent (but no design wins to test) MT7663BUN which also says it's limited to USB high-speed yet shows superspeed pins in section 3.1.

In several places on this site, I make statements strongly recommending that users buy only single-function (no bluetooth) wifi adapters. I recommend this because it is a bad idea to buy multi-function adapters. I understand that it seems to be something that would work well to those that are not informed. USB3 is not mankind's greatest invention. USB2 is pretty solid but not USB3. Below is a White Paper by Intel from 2012. The paper is a report about EMI in the 2.4 GHz range caused by USB3 cables and connections. 

https://www.usb.org/sites/default/files/327216.pdf

Yes, USB3 cables and connections create interference (undesired radio signals) in the 2.4 GHz range. Bluetooth uses 2.4 GHZ frequencies as does 2.4 GHz wifi. An extreme example, that is actually common, is when a user has an external SSD on a USB3 cable. That type of setup can take out your ability to use an external bluetooth adapter if the bluetooth adapter is in the wrong place. Another example that is fairly common is USB3 WiFi adapters in USB3 mode interfering with its own 2.4 GHz wifi signal. @dubhater has done a good job with the rtw88 usb3 drivers in that he makes a module parameter (driver options) so that users can keep their USB3 wifi adapter in USB2 mode.

I could talk about this subject until you are bored to tears but I'll to wrap it up quickly as I think I have presented enough information for you to be able to research the issue more if you are interested. USB WiFi adapter makers mess things up fairly often. There are several filters that I use when making The Plug and Play List... and there are good reasons for each filter. There no multi-state adapters in the Plug and Play List and there likely will never be.

I hope have a better understanding of the issue.

alen0038, 2025-02-02:
@morrownr 
Thanks for the good news!
I have a TP-Link Archer T4U Rev 3.2 with an RTL88x2BU chipset. It had been collecting dust in a drawer for almost two years. While it worked fine under Windows, it always connected as a USB 2.0 device on Linux. I could compile the driver, but it wasn’t truly plug-and-play.
Thankfully, this issue was finally fixed in kernel 6.12! :-)
According to this [commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=315c23a64e99552502dd4d18d6ddc073fad9a7c3) it's actually a Realtek feature - initially, the adapter connects as a USB 2.0 device, and then the driver resets it to USB 3.0.
Until kernel 6.12, my adapter always remained in USB 2.0 mode, and I had to manually load the driver, at least up to version 6.8. It will take some time for kernel 6.12+ to become widely available across distributions (I use Mint), but out of curiosity, I tested it in a VM, and it now works fine.
In the kernel log, the device was initially recognized as USB 2.0. However, once rtw_8822bu and the firmware were loaded, it reconnected as USB 3.0. Performance was solid, reaching around 600-630 Mbps up/down in iperf3 in the same room as the access point - comparable to the Windows driver.
I also have a small Comfast adapter with the 8821CU chipset (USB 2.0), and I didn't notice any changes with kernel 6.12 - it behaves the same as before.
I switched to MTK-based adapters a long time ago, but it's nice to see Realtek making improvements too :-)