morrownr/USB-WiFi #554: Enable GitHub Pages?

webknjaz, 2024-12-26:
$sbj. It should be a one-click thing in the repository settings that would make https://morrownr.github.io/USB-WiFi serve the markdown content, with no additional configuration. The viewing experience might be better this way.

morrownr, 2025-01-08:
Hi @webknjaz 

Can you explain further. I do not understand.

webknjaz, 2025-01-08:
You have a bunch Markdown files viewable via https://github.com/morrownr/USB-WiFi. You can make them viewable at https://morrownr.github.io/USB-WiFi, additionally at zero overhead.

https://docs.github.com/en/pages

kylefmohr, 2025-05-12:
Might not be a bad idea - you can see what it would look like here: https://kylefmohr.github.io/USB-WiFi/ (but it would use your username if you enabled it). It's just a more user-friendly way to display the markdown documents. 

If you want to enable it, just go here: https://github.com/morrownr/USB-WiFi/settings/pages under `branch`, choose `main`, click `save`, and about a minute or so later it will be accessible at https://morrownr.github.io/USB-WiFi

morrownr, 2025-05-18:
@webknjaz @kylefmohr 

Sorry for the delay. I have been having health problems that have delayed me.

Would one or both of you be interested in being given admin rights to the USB-WiFi repo so that you can help me maintain it? I could certainly use the help.

kylefmohr, 2025-05-18:
> @webknjaz @kylefmohr 
> 
> Sorry for the delay. I have been having health problems that have delayed me.
> 
> Would one or both of you be interested in being given admin rights to the USB-WiFi repo so that you can help me maintain it? I could certainly use the help.

@morrownr 

Yes, absolutely. I'd be happy to help with the USB-WiFi repo in any way I can. I know how much work you put into it and the whole community benefits immensely from your dedication.

I'm very sorry to hear about your health problems and am wishing you all the best for a speedy and full recovery. Please prioritize your well-being.

While I don't have your depth of knowledge, I'm willing to do what I can to help. 

Thank you for everything you do.

-Kyle

morrownr, 2025-05-19:
Kyle,

> I'm willing to do what I can to help.

That is what is needed. I hope it can be as much fun for you as it has been for me. To start I am going to send you my email address in the next message and then delete the message. Please contact me via email.

@morrownr

morrownr, 2025-05-19:
morrownr@gmail.com

morrownr, 2025-05-31:
@kylefmohr 

Where have you been? I thought you were going to send me an email.

Here is my email address again: morrownr@gmail.com

This time I will leave it up until I get an email from you.

@morrownr

morrownr, 2025-06-03:
@webknjaz 

I can appreciate what you are saying. Hopefully @kylefmohr does answer. I do need someone that can help out with the site.