morrownr/USB-WiFi #560: Help Needed: Full 5GHz Spectrum (including DFS) Support for AWUS036AXML and AWUS036ACHM?

Sentello, 2025-01-10:
Hi everyone,

I’m in the process of upgrading my Wi-Fi adapter and could use some guidance. I currently have an Alfa AWUS051NH (chipset: RT2770), which has been a solid performer over the years but doesn’t seem to support the complete 5GHz spectrum, particularly DFS channels. Despite setting the regulatory domain to 00 and US, I still can't scan or use those channels. I’ve tested this on both modern Linux distros (Kali, Arch, Parrot) and older ones like BackTrack 5, with no success. It’s possible the chipset itself doesn’t support these frequencies fully.

I’m now considering two Alfa cards:

- AWUS036AXML
- AWUS036ACHM

I’m open to recommendations for other Wi-Fi adapters
My main requirements are:
Full support for the entire 5GHz spectrum, including DFS channels.
The ability to enable monitor mode and capture packets reliably.
I’m planning to use the adapter with modern Linux distributions (Kali Linux and Parrot OS). If anyone has experience with these cards, I’d greatly appreciate your input on:

Whether they support the full 2.4GHz and 5GHz spectrum, including DFS channels.
How well they perform in monitor mode and packet capture.
Alternatively, if there are other adapters you’d recommend for this use case, I’m open to suggestions. I’d love to hear from anyone who has used these models and can confirm their compatibility and features.

Thanks in advance for your help!

morrownr, 2025-01-11:
Hi @Sentello 

Maybe I can help.

> I currently have an Alfa AWUS051NH ( RT2770), which has been a solid performer over the years but doesn’t seem to support the complete 5GHz spectrum, particularly DFS channels.

That is a good adapter. I still have adapters based on rt2870 and rt3070 chips. They were similar to the rt2770. Those are single band chips. Of the readily available chips, I think the mt7610u, ac600, chip would work well based on your requirements. It is the chip in the Alfa ACHM. The ACHM is a favorite of those doing pen testing/security analysis. It has really good range and is included in The Plug and Play List as well as the Recommended for Kali List here on the Main Menu.

I cannot currently recommend the AXML. During 2024 a problem came about for 3 adapters that I had to remove from The Plug and Play List. They are the Alfa AXML, Alfa AXM and Comfast CF-953AX. The problem is strange and has to do with bluetooth. A problem with bluetooth is blocking wifi. Hopefully this problem is fixed soon but until it is resolved I cannot recommend any of those 3 adapters.

> Full support for the entire 5GHz spectrum, including DFS channels

I assume you are not talking about AP mode and DFS channels. Chips with mt7610u, mt7612u and mt7921au should support all 5 GHz channels that are allowed in your country, including managed mode DFS channels.

> The ability to enable monitor mode and capture packets reliably, wifi hacking

The same 3 chips as shown above should work here as well. The Plug and Play List has details of many good adapters with those 3 chips. See Main Menu item 2. Here is the Main Menu:

https://github.com/morrownr/USB-WiFi

> I’m planning to use the adapter with modern Linux distributions (Kali Linux or Parrot OS).

The ACHM is a favorite with Kali users. It is not the lastest, fastest adapter available but it certainly does what pen testers want. Other adapters to consider:

Alfa ACM
Edup EP-AX1672
Panda PAU0B
Panda PAU0D
Panda PAU0F

All of the above are in The Plug and Play List as well as the Recommended List for Kali.

Here is a direct link to the Recommended for Kali List:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

I have several of the adapters that I posted above including the Alfa ACHM, Alfa ACM and Edup EP-AX1672 so if you want me to do some testing, just ask.

Sentello, 2025-01-12:
Hi @morrownr, 

I really appreciate the insight into the different options.

> That is a good adapter. I still have adapters based on rt2870 and rt3070 chips. They were similar to the rt2770. Those are single band chips

If you have the same or a similar card as mine (RT2770), could you try checking which channels are available on it? Specifically, could you run iwlist wlan0 freq? I’m curious to see if you can detect the DFS channels as well.

> I assume you are not talking about AP mode and DFS channels. Chips with mt7610u, mt7612u and mt7921au should support all 5 GHz channels that are allowed in your country, including managed mode DFS channels.

I need an adapter that can also work on DFS channels, especially all outdoor frequencies – because those are exactly the channels where internet providers operate, and they are the ones I’m interested in.

> The ACHM is a favorite with Kali users.

Otherwise, I agree with you that the best option for me is probably the AWUS036ACHM

morrownr, 2025-01-12:
> If you have the same or a similar card as mine (RT2770), could you try checking which channels are available on it?

Sure. I'll use the following adapter. It seems to have the same specs as the rt2770:

ID 148f:2870 Ralink Technology, Corp. RT2870 Wireless Adapter

Here is what `iw list` says about frequencies:

		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)

This makes sense as it is a single band, 2.4 GHz capable adapter. It does not support 5 or 6 GHz. DFS only happens in the 5 GHz band.

The  AWUS036ACHM adapter is dual band and supports DFS channels in managed mode. I can recheck this is if you want.

Sentello, 2025-01-21:
@morrownr 
First off, thank you again for all the input in this thread—it has been incredibly insightful. I wanted to provide an update on my situation. After purchasing the AWUS036ACHM,  I’ve conducted extensive testing to detect and scan DFS, but the adapter has failed to perform as expected. 

Here are my key findings:

> ┌──(root㉿kali)-[/home/kali]
└─# iwlist wlan0 freq
wlan0     32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Channel 144 : 5.72 GHz
          Channel 149 : 5.745 GHz
                                                                                                                                                                                                                                            
> ┌──(root㉿kali)-[/home/kali]
└─# iw reg get       
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

1. Despite being technically capable of operating on 5 GHz frequencies, the adapter cannot scan or detect APs operating on DFS channels like 5605 MHz. Commands like: 
 `sudo iw dev wlan0 scan freq 5605`
result in:
`command failed: Operation not supported (-95)`
2. Running airodump-ng to capture on specific DFS channels, such as:
`sudo airodump-ng -C 5605 wlan0` results in: `No valid frequency given.`
3.  A Ubiquiti 5GHz client antenna in the same environment successfully detects the AP broadcasting on channel 121 (5605 MHz). This confirms the issue lies with the adapter or its driver, not the AP.
4.  Even after setting the regulatory domain to both US and DE, the issue persists. The `iw reg get`output shows DFS support, but scanning and capturing remain problematic.

Questions:
is the current mt7610u driver missing functionality for DFS channel scanning, particularly passive scanning? If so, is there an alternative driver or configuration that could enable full DFS support? Could this issue be tied to the hardware capabilities of the AWUS036ACHM itself, or is it solely a driver issue?

could you please test and share results for the following, scanning DFS Channels:
`iwlist <interface> freq
iw dev <interface> scan freq 5605
`
and try the passive monitoring?
`sudo airodump-ng -C 5605 <interface>`

morrownr, 2025-01-22:
Hi @Sentello 

Many questions. Let's unpack this and see what we get.

> After purchasing the AWUS036ACHM...

One of my favorite adapters.

> but the adapter has failed to perform as expected.

I see.

> # iwlist wlan0 freq

Instead of that command, which is depreciated, let's try the following command:

$ iw list

I get this for 5 GHz on a mt7921au based adapter that is connected at the moment:

```
		Frequencies:
			* 5180 MHz [36] (23.0 dBm)
			* 5200 MHz [40] (23.0 dBm)
			* 5220 MHz [44] (23.0 dBm)
			* 5240 MHz [48] (23.0 dBm)
			* 5260 MHz [52] (24.0 dBm) (radar detection)
			* 5280 MHz [56] (24.0 dBm) (radar detection)
			* 5300 MHz [60] (24.0 dBm) (radar detection)
			* 5320 MHz [64] (24.0 dBm) (radar detection)
			* 5500 MHz [100] (24.0 dBm) (radar detection)
			* 5520 MHz [104] (24.0 dBm) (radar detection)
			* 5540 MHz [108] (24.0 dBm) (radar detection)
			* 5560 MHz [112] (24.0 dBm) (radar detection)
			* 5580 MHz [116] (24.0 dBm) (radar detection)
			* 5600 MHz [120] (24.0 dBm) (radar detection)
			* 5620 MHz [124] (24.0 dBm) (radar detection)
			* 5640 MHz [128] (24.0 dBm) (radar detection)
			* 5660 MHz [132] (24.0 dBm) (radar detection)
			* 5680 MHz [136] (24.0 dBm) (radar detection)
			* 5700 MHz [140] (24.0 dBm) (radar detection)
			* 5720 MHz [144] (24.0 dBm) (radar detection)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (27.0 dBm) (no IR)
			* 5865 MHz [173] (27.0 dBm) (no IR)
			* 5885 MHz [177] (27.0 dBm) (no IR)

```

Notice how channels 52-144 say (radar detection). Those are your DFS channel if in the US. This varies a lot.

Radar detection means that before a radio can initiate a radio signal, it must scan for conflicting signals on the DFS channels. This takes about 1 minute. However, if the device is not capable of scanning for conflicting signals, it can't initiate a signal on DFS channels. I'm not aware of any USB WiFi adapters that are capable of doing the scans for conflicts. I have two wifi routers that are capable.

That does not mean that you can't use USB WiFi adapter on DFS channels. If fact I do in the normal course of business here. My primary wifi router is set to channel 100, 160 MHz channel width. When I turn the router on, 2.4 GHz comes up really fast but not 5 GHz. Since I am trying to a DFS channel in AP mode, the router has to first scan to see if any conflicting signals are detected. This usually takes around 1 minute. If no conflicts are detected then it starts operating as an AP and I can connect my ACHM to it on channel 100 and use it.

> Despite being technically capable of operating on 5 GHz frequencies, the adapter cannot scan or detect APs operating on DFS channels like 5605 MHz.

Try it for a non-DFS channel. It should work. You are trying to initiate a signal on a device that is not allowed to do it on DFS channels. Before you say, what is this? Well, I'm not so sure that it matters. DFS channels, in my area, and other areas I have lived, are almost never used. My router using channel 100 is the ONLY AP/router on a DFS channel. Period. If you are passively scanning, there may not be anything to scan. Yes, I regularly see transfer rates of over 1 Gbps. 

> A Ubiquiti 5GHz client antenna in the same environment successfully detects the AP broadcasting on channel 121 (5605 MHz). This confirms the issue lies with the adapter or its driver, not the AP.

Then the ACHM should be able to detect it. It will have to get a beacon from the AP. ... I just went and grabbed my ACHM and plugged it into my dev box. Within 20 seconds I was connected to my AP on channel 100. This message is going through my ACHM. Can you set a DFS channel in your AP/router? Not all support DFS.

> Is this behavior normal, or could it indicate an issue with the adapter?

This is normal behavior as the driver has never supported LED's. That can change but then we would need a way to turn it off.

I'm not sure if I helped or caused more confusion. Let me know.

Sentello, 2025-01-22:
Hi @morrownr,

Thank you for your response, I really appreciate the insights.

I’ve dedicated significant time to testing and experimenting to get it working with DFS channels, specifically on **5605 MHz (channel 121)**. Here’s a summary of what I’ve tried and the results so far:

---

1. **Regulatory Domain switchig**:
   - I tested multiple regulatory domains (`US`, `JP`, `SG`, `KR`, `CA`, and `CN`) to see if any allowed unrestricted access to DFS frequencies.
   - After switching regulatory domains using `iw reg set`, I confirmed the available frequencies with `iw list | grep -A 15 Frequencies`. Unfortunately, Channel 121 (5605 MHz) remained unavailable or non-functional on all tested domains.

2. **Scanning with Airodump-ng**:
   - Using `sudo airodump-ng wlan0 --band a --channel 121`, I attempted to detect my home MikroTik router broadcasting on **5605 MHz (Channel 121)**. Despite the AP being correctly configured, the Alfa adapter could not detect it.
   - I successfully scanned non-DFS channels and some DFS channels, such as **Channel 104 (5520 MHz)**, when using the `US` regulatory domain. However, I was unable to scan higher DFS channels, including **Channel 121**.

3. **Driver and firmware**:
   - Verified the adapter’s driver (`mt76x0u`) and firmware (`mediatek/mt7610u.bin`) using `modinfo mt76x0u`.
   - The driver theoretically supports DFS, my testing suggests that there are limitations or enforcement issues when attempting to access higher DFS channels.

4. **DFS Testing across domains**:
   - Testing across different regulatory domains showed mixed results:
     - **US**: Detected AP on Channel 104 (5520 MHz), but not on Channel 121 (5605 MHz).
     - **JP, KR, SG**: No significant improvement; Channel 121 remained undetected.
   - This indicates either a driver or firmware limitation rather than purely a regulatory enforcement issue.

---

### Notes:

- **No DFS Channels Scanned Above 116**: Despite trying multiple configurations, **Channel 121 (5605 MHz)** and other high DFS channels could not be scanned or detected. Lower DFS channels, such as Channel 104 (5520 MHz), were partially successful.
- **Driver?**: The **mt76x0u** driver appears to have incomplete support for DFS functionality on higher channels.
- **Non-DFS Channels**: Scanning and connecting to non-DFS channels, including those in both 2.4 GHz and lower 5 GHz bands, worked without any issues.

---

If you have the ability and time to replicate this by configuring a router on DFS channels (like channel 121), it would be great to confirm if you experience the same behavior with the Alfa AWUS036ACHM.

Thank you again for your time!

morrownr, 2025-01-22:
@Sentello 

What country do you live in if you don't mind telling me? I need to know to understand what you should be seeing.

> So why do I care so much about DFS? The reason is simple: in the area where I live, most internet service providers (ISPs) operate in the 5 GHz DFS spectrum. 

Interesting. I've have never seen an IPS using DFS channels. The big reason is that there are a lot of wireless devices that cannot use DFS channels. An example would be Roku devices for TV's. If ISPs tried to use DFS channels, they will end up with customers complaining.  

> However, it’s now evident that this adapter doesn’t support DFS, which is a big disappointment.

Yes, it does. I am connected to a DFS channel right now with my ACHM. This message will go through a DFS channel. I think you have a misunderstanding of how DFS channels work.

> Why doesn’t the Alfa adapter support DFS?

It does.

> If you have the ability and time to replicate this by configuring a router on DFS channels (like channel 121), it would be great to confirm if you experience the same behavior with the Alfa AWUS036ACHM.

My router is almost always on a DFS channel. I cannot set channel 121. I can set channels 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140 and 144. My country code is US. This can and does vary considerably depending on your location. Would you like for me to set one of the channels that I can set? It is on channel 100 currently and it works well.

> They create high-performance hardware with great potential, yet fail to provide support for outdoor DFS channels.

Here is a link to the Wikipedia page that goes over wifi channels. In my country, there is no differentiation between indoor and outdoor for DFS channels so I really need to know your country. I cannot address some of these issues without knowing country. If you have said what it is then I missed it and would for you to tell me again.

 > No DFS Channels Scanned Above 116: Despite trying multiple configurations, Channel 121 (5605 MHz) and other high DFS channels could not be scanned or detected. Lower DFS channels, such as Channel 104 (5520 MHz), were partially successful.

That can be normal. I'm am still confused by channel 121. I have seen any AP that was capable of DFS channels that could set that channel.Maybe it is possible as there is variability in the coding of equipment.

> To give you some perspective, when I run a scan with my Ubiquiti outdoor unit, I can detect at least 12 access points operating on DFS channels, all of which are used by local ISPs.

Like I said earlier, I have never seen this in my country. DFS channels are avoided like the plague by ISPs. What channels are the those ISPs using?

Let's review some information about DFS channels:

DFS (Dynamic Frequency Selection) channels in WiFi refer to a set of 5GHz channels that are shared with radar systems, allowing Wi-Fi devices to utilize these frequencies while ensuring they don't interfere with active radar signals by constantly monitoring the channel and switching if necessary; essentially providing access to more 5GHz channels by dynamically avoiding radar interference, which is crucial in areas with high radar activity. 

My comment: Remember that setting a DFS channel is not a guarantee that you will get to use the channel that you set. You might use it for a while and then get booted. The AP/router has to continuously scan for radar's that have priority and if a signal is detected, your AP/router must immediately switch you off of the DFS channel you are using. This is another reason that ISPs don't like DFS channels. 

 Radar priority:

The primary function of DFS is to prioritize radar systems, meaning a Wi-Fi device operating on a DFS channel must vacate the channel immediately if it detects a radar signal.

 My comment: Just because you want to use a DFS channel does not mean that you will be able to do so. I happen to live in an area where there are no conflicting radar signals.

Channel availability check:

Before using a DFS channel, a Wi-Fi device performs a "channel availability check" to ensure no active radar is present on that frequency. 

Potential for disruption:

While offering more available channels, using DFS channels can lead to occasional brief interruptions in connection if a radar signal is detected and the device needs to switch channels. 

Regulation-based:

The specific DFS channels available vary depending on the region due to different radar regulations in different countries. 

My comment: I wrote out the DFS channels that are available to me in my country. Your available channels may be different.

One last thing:

> After switching regulatory domains using iw reg set...

Depending on the age of your Linux distro, that may not work. Here are some links that may help:

https://www.marcusfolkesson.se/blog/linux-wireless-regulatory/
https://docs.kernel.org/networking/regulatory.html
https://wireless.docs.kernel.org/en/latest/en/developers/regulatory/wireless-regdb.html

I hope this information was helpful but much more information is available.

Regards

Sentello, 2025-02-04:
Hi @morrownr 
> Interesting. I've have never seen an IPS using DFS channels. The big reason is that there are a lot of wireless devices that cannot use DFS channels. An example would be Roku devices for TV's. If ISPs tried to use DFS channels, they will end up with customers complaining.

[image]
I am currently connected to my ISP on this channel, so I am trying to get the DSF channels working because I have the opportunity to test it directly with my provider, capture my own packets, and see if it works.

dubhater, 2025-02-05:
Who chose channel 121? It doesn't look like a standard channel: https://en.m.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax/be)

Can you change it to 120?

Sentello, 2025-02-06:
No, I can't change the channel. It is set on the AP by the ISP, which is installed on a rooftop. My client antenna, mounted on my roof, is pointed at this AP and connects using the provided IP details.

dubhater, 2025-02-06:
As far as I can tell none of the wifi drivers included in the kernel advertise support for channel 121. You should ask the [linux-wireless](https://wireless.docs.kernel.org/en/latest/en/developers/mailinglists.html#stats-and-requirements-of-linux-wireless-mailing-list) mailing list how to work with channel 121. There are many knowledgeable people there.