morrownr/USB-WiFi #561: [Help]: mt7921u AX3000 unusable as a AP with more than 16 stations/clients

kbmetz, 2025-01-12:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux pi 6.8.0-1017-raspi #19-Ubuntu SMP PREEMPT_DYNAMIC Fri Dec  6 20:45:12 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 003: ID 1997:2433 Shenzhen Riitek Technology Co., Ltd wireless mini keyboard with touchpad Bus 001 Device 004: ID 0781:5583 SanDisk Corp. Ultra Fit Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 3: phy2: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
there is no dkms output
```

### iw

```text
phy#2
	Interface wlxe84e06ad13fa
		ifindex 6
		wdev 0x200000001
		addr e8:4e:06:ad:13:fa
		type AP
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

I'm attempting to run an AP based off hostapd. Everything is configured and working but once 16-20 clients have joined i start getting the following error
hostapd: wlxe84e06ad13fa: STA xx:xx:xx:xx:xx:xx IEEE 802.11: Could not add STA to kernel driver

Restarting resolves the issue. And if I keep it under 10 stations there is also no issue.

If this is a known issue that only mediatek can resolve then I recommend this adapter description on this repo be amended to make note of this limitation. 

This issue is also referenced here:
https://github.com/morrownr/USB-WiFi/issues/369
https://forum.openwrt.org/t/reference-for-number-of-supported-stations-in-different-wifi-hardware/156535

Is is possible to recompile the module with the limits increased? I like this adapter but the specific use case is for an AP

Also, I'm able to run hostapd with the same configuration on a rtl8812au and it supports up to 30 stations without issue.

morrownr, 2025-01-12:
@kbmetz 

> If this is a known issue that only mediatek can resolve ...

I cannot say that this is a known issue. I have used my Edup EP-AX1672 (mt7921au chip) as an AP a lot. I probably don't get over maybe 12-16 clients so maybe I could have avoided the issue but I cannot recall anyone else bringing this up. I took a look at the OpenWRT link.

You might report this to

https://bugzilla.kernel.org/

and

https://github.com/openwrt/mt76

I'd be interesting to know what you find out.

ExcaliburUBW, 2025-01-13:
My EDUP AX3000 doesn't work at all.

morrownr, 2025-01-13:
Hi @ExcaliburUBW 

> My EDUP AX3000 doesn't work at all.

Not much to go on. Is your adapter an EDUP EP-AX1672 or EDUP EP-AX1673?

What distro and what kernel are we talking about?

Describe the problem?

morrownr, 2025-01-13:
> The irony is that I bought this specifically so I wouldn't have to recompile anything as I've had to do with the rtl8812au.

Depending on what we find out:

- I will add a note about a 16 client limit in AP mode for the mt7921au.

- And we could figure out what the best patch would be and submit a patch to mainline to see what happens.

Also, a new mac80211 driver (in-kernel) for the rtl8812au just went into kernel 6.13. It should work well in 6.13 but there are some additional patches going into 6.14 so I'd say best performance with 6.14 and later. If you want to try it now:

https://github.com/lwfinger/rtw88

kbmetz, 2025-01-13:
> > The irony is that I bought this specifically so I wouldn't have to recompile anything as I've had to do with the rtl8812au.
> 
> Depending on what we find out:
> 
>     * I will add a note about a 16 client limit in AP mode for the mt7921au.

The AX3000 that I have is loading the mt7921u. Should it be using the mt7921au?

>     * And we could figure out what the best patch would be and submit a patch to mainline to see what happens.

Agreed. I'm going to read up and see about recompiling it with those options

> 
> 
> Also, a new mac80211 driver (in-kernel) for the rtl8812au just went into kernel 6.13. It should work well in 6.13 but there are some additional patches going into 6.14 so I'd say best performance with 6.14 and later. If you want to try it now:
> 
> https://github.com/lwfinger/rtw88

That would be great! Thanks for the mention

morrownr, 2025-01-14:
> The AX3000 that I have is loading the mt7921u. Should it be using the mt7921au?

No, the driver for the mt7921au chip is the mt7921u.

VVirusib, 2025-02-12:
@kbmetz 

New user here. 

I have an EDUP AX3000 wifi6E Usb adapter with driver mt7921u, which could potentially be the same hardware/firmware as yours.  

I've been killing myself for 2+ weeks to create an AP/hotspot on the wifi6 802.11ax band however, although the device is capable of this on my win11 pc, it will only create a Wifi5 (band  bg) hotspot on raspberry pi 5 os (brand new 16gb device, fully updated 64bit OS).

To clarify, the device runs up to frequencies in the 6000MHz (highest active is 6415 MHz; 23dBm). 
(iwlist wlan0 scan)

lsusb | grep -i wireless
gives:
Bus 003 Device 002: ID 8e8d:7961 Mediatek Inc. Wireless_Device

Is your AP WIFI6? If so, can you please share your settings?

Thanks!

morrownr, 2025-02-12:
Hi @VVirusib 

Evidently, there was a bug introduced:

https://bugzilla.kernel.org/show_bug.cgi?id=218731

It appears the bug was introduced with this patch:

https://kernel.googlesource.com/pub/scm/linux/kernel/git/stable/linux/+/09382d8f8641bc12fffc41a93eb9b37be0e653c0

This report went into linux-wireless recently and is being looked at by a Mediatek dev.

While we are waiting for a fix, you might try a couple things from the Frequently Asked Questions document:

Go to:

https://github.com/morrownr/USB-WiFi

Scroll down to menu item 12 and click on it.

Try no. 2 and no. 3, one at a time, to see if either helps.

Let me know.

By the way, what is your country code?

VVirusib, 2025-02-15:
Hi @morrownr
Thanks for your feedback. It explains why my efforts led to nothing.
I tried both 2 and 3, one after the other as per your instructions. Both gave me this:
$ sudo nmcli connection modify <connection> 802-11-wireless.band ax 802-11-wireless.mode apError: failed to modify 802-11-wireless.band: 'ax' not among [a, bg].
My country is set correctly. However, I do get this:
$ iw reg getglobalcountry <my correct country code>: DFS-ETSI (2400 - 2483 @ 40), (N/A, 20), (N/A) (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS (5725 - 5875 @ 80), (N/A, 13), (N/A) (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR (57000 - 66000 @ 2160), (N/A, 40), (N/A)
phy#0country 99: DFS-UNSET (2402 - 2482 @ 40), (6, 20), (N/A) (2474 - 2494 @ 20), (6, 20), (N/A) (5140 - 5360 @ 160), (6, 20), (N/A) (5460 - 5860 @ 160), (6, 20), (N/A)
I tried so many options to change the country of phy#0 but with no result, which seems strange to me. However, I dare believing it wouldn't make any difference anyway as it refers to my built-in wlan, which is not where I am setting my hotspot anyway. Nonetheless, if you think this might be the reason for not being able to run my hotspot in Wifi6, please let me know, suggesting some solution(s).
One thing that really worries me is the lack of knowledge of this OS environment and perhaps I made some modifications, with all my trials, that now block any possibility of having a Wifi6 hotspot. I should try reinstalling the OS at some point and trying from scratch, but I did that several times already anyway and now I have other projects piling up.
With the above, I assume I need to wait for a fix and meanwhile run the hotspot in Wifi5. I just don't know if I'll be able to notice the fix is out and update my settings :D
Thanks again.

 
  

https://bugzilla.kernel.org/show_bug.cgi?id=218731

https://kernel.googlesource.com/pub/scm/linux/kernel/git/stable/linux/+/09382d8f8641bc12fffc41a93eb9b37be0e653c0

https://github.com/morrownr/USB-WiFi

https://bugzilla.kernel.org/show_bug.cgi?id=218731

https://kernel.googlesource.com/pub/scm/linux/kernel/git/stable/linux/+/09382d8f8641bc12fffc41a93eb9b37be0e653c0

https://github.com/morrownr/USB-WiFi

morrownr, 2025-02-15:
@VVirusib 

> One thing that really worries me is the lack of knowledge of this OS environment and perhaps I made some modifications...

You cannot worry about inexperience. Could mods that you have not reverted be an issue? Sure. I just did a relook at this issue.

Part of your problems are timing and part are the distro and kernel you are using.

> Linux pi 6.8.0-1017-raspi #19-Ubuntu SMP PREEMPT_DYNAMIC Fri Dec 6 20:45:12 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

From this, it appears you are using Ubuntu for RasPi on a RasPi. Two things I would change about this:

1. The best distro for RasPi hardware is RasPiOS. You should use the 64 bit version.
2. Kernel 6.8 was the worst of the worst kernels last year as far as wifi is concerned.

If you burn a sd with a fresh RasPiOS 64 bit iso, that will fix several things. RasPiOS currently uses kernel 6.6 so the bugs I am talking about do not apply. 6.6 is a good LTS kernel.

Don't get frustrated. Take some time away from this project. Think about and do what I recommended above when you feel like it.

VVirusib, 2025-02-26:
@morrownr
Hi again. Still here so thanks for your feedback. Although I installed multiboot PINN and have already installed several OSs (which I did for the future, just in case) I am pretty confident that I installed the latest RaspberryPI OS as per this: https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit
 
Raspberry Pi OS with desktop and recommended software
   
   - Release date: November 19th 2024
   - System: 64-bit
   - Kernel version: 6.6
   - Debian version: 12 (bookworm)
   - Size: 2,955MB

Show SHA256 file integrity hash:
Release notes
and that's the OS that loads when I work on my project. Maybe with whatever updates/upgrades something was modified. I will verify this as soon as I have a moment, but this will probably be only in April. I still want this WIFI6 hotspot to work so I will definitely come back to it. Thanks for your encouragements.
To clarify, the last thing I did before putting the entire thing to a halt was to reinstall the OS (again using PINN and the above OS), so I intend to start working with a fresh OS when time becomes available. If you tell me what command to run so you get whatever info about my system you need, I'll send you the result when I get to it. Thanks.

morrownr, 2025-05-18:
> I am pretty confident that I installed the latest RaspberryPI OS as per this:

As of very recently, the new RasPiOS uses kernel 6.12 which should be better.

> If you tell me what command to run so you get whatever info about my system you need, I'll send you the result when I get to it.

I have been having health problems so I am behind on everything. Take your time and let know if you run into problems again.

patrikfh, 2025-05-29:
@kbmetz @morrownr Any Updates regarding this Issue?
I have the same Problem and also tried to increase the MT792x_WTBL_SIZE, as it is also way higher for other chipsets.
Unfortunately so far there was no improvement and it is impossible to connect more than 16 clients to the mt7921u...

patrikfh, 2025-05-29:
As for my setup, I’m using a Raspberry Pi 5 with an EDUP AX3000 WiFi adapter, running the latest 6.12-y kernel on Debian 12.

My hostapd settings are pretty similar to your [`hostapd-WiFi4.conf`](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi4.conf).

So far i increased 
`#define` MT792x_WTBL_SIZE	20` to 64 in [mt792x.h)](https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/net/wireless/mediatek/mt76/mt792x.h)

And also added some logging in [mt7921/main.c](https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/net/wireless/mediatek/mt76/mt7921/main.c).

I then rebuilt the kernel with these changes and installed it on my Raspberry Pi.

I’d very much appreciate any ideas or suggestions!

morrownr, 2025-05-29:
@deren

If you have time to provide some guidance on this issue, it would be greatly appreciated.

The issue of the limit of how many clients can be supported in AP mode by the mt7921 driver has come up several times over the last 2 years or so. It is not obvious in the driver if it is possible to try different settings. Maybe this is locked in firmware?

Many people report the mt7921 driver not handling over 16 clients in AP mode. The desire seems to be 32 or more. If you can point us in the right direction, there are several people in this issue that could provide testing.

Thanks for your time.