morrownr/USB-WiFi #563: [Help]: trouble shooting

itsyoboi09, 2025-01-14:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

f

### lsusb

f

### rfkill

b

### dkms

```text
g
```

### iw

```text
B
```

### What happened?

So I brought an Alfa awus036amxl can i be having issues with it on virtual box and with my host windows computer. I’ve download the the driver on my host which was fine i had no problems with it but trying to connect it with my virtual machine is where im stuck at I’ve download Kali Linux on VM and change the settings, I’ve disabled the network settings and add it on the usb 3.0 medatke and run some commands “lsusb” ifconfig” and iwconfig” but my VM wont take, I’ve tried to download the extension pack but that didn’t work, deleted the driver on my main host but that didn’t work,download the driver “sudo Install Realtek-rtl88xxau-dkms” can anyone help me?

itsyoboi09, 2025-01-14:
Sorry with no commands I didn’t have a computer with me im typing with my phone

morrownr, 2025-01-30:
@itsyoboi09 

Sorry for the delay. Been busy:

The Main Menu has some help:

https://github.com/morrownr/USB-WiFi

Look under menu item 9 for the following:

Using a USB Adapter in VirtualBox