morrownr/USB-WiFi #569: [Help]: MT7921AU - EDUP EP-AX1672 - Deauthentication Every 24 Hours with Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT

IamRPDev, 2025-01-20:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux admin01 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub Bus 001 Device 003: ID 04e6:5a11 SCM Microsystems, Inc. uTrust FIDO2 Security Key Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub Bus 002 Device 003: ID 17ef:1003 Lenovo Integrated Smart Card Reader Bus 002 Device 004: ID 072f:b307 Advanced Card Systems, Ltd AFD FIDO Bus 002 Device 005: ID 0a5c:21e6 Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad] Bus 002 Device 006: ID 5986:02d5 Bison Electronics Inc. Integrated Camera Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 006 Device 002: ID 0781:55b1 SanDisk Corp. SanDisk 3.2 Gen1 Bus 006 Device 003: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

0: tpacpi_bluetooth_sw: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 2: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
virtualbox/7.0.16, 6.8.0-51-generic, x86_64: installed
```

### iw

```text
phy#0
	Interface wlxe84e06ad59f9
		ifindex 3
		wdev 0x1
		addr e8:4e:06:ad:59:f9
		ssid TheInternet
		type managed
		channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
```

### What happened?

Hello morrownr,

I find myself plagued with this issue every 24 hours, the timing depends on when my AP was restarted. It also deactivates my wireless card completely until I reboot.

The only thing I see in syslog is kernel: wlxe84e06ad59f9: deauthenticated from 94:83:c4:a7:23:c2 (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT)

I have tried unplugging the usb, restarting the NetworkManager service, it seems only rebooting restores function.

Based on recommendations here I purchased 2 [EDUP EP-AX1672](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L) Which have provided fantastic connectivity besides this one issue.

I'm running a fresh install on both machines on Linux Mint 22, Kernel 6.8.0-51

My Router is a GL-MT6000 running Openwrt Snapshot custom build by PESA1234. Although I did experience the same issue with the standard snapshot.

I will do my best to rerun a few of the above commands when the issue presents itself tomorrow. 

I'd appreciate any help you could offer. Thanks!

morrownr, 2025-01-20:
Hi @IamRPDev 

Let's try some things. I run OpenWRT also.

Luci > Network > Wireless > Wireless Overview > Edit (the radio with the problem) > Interface Configuration > Advanced Settings >
Time interval for rekeying GTK

Try setting 0. I think that stops rekeying. See what happens.

If that does work, there is still something amiss but at least it won't bug you. There probably is a bug somewhere. I'll do some digging.

IamRPDev, 2025-01-20:
Thanks @morrownr 

I put the setting in place. I'm guessing if it didn't help I'll see the problem around this time tomorrow night. But I'm happy for a workaround for the time being.

I'm more than happy to test anything you come across, the only limit being the 24 hour time to test.

Thanks again for the assistance!

morrownr, 2025-01-24:
The verdict is?

IamRPDev, 2025-01-27:
Hey sorry for the late response, after finding out disabling the group key worked. I decided to try the new RC5. This solved my issue completely even with default settings. My best guess is something with the hostapd was causing the issue and not the linux driver for the card.

Thanks for helping out, and thanks for the excellent recommendations on hardware to purchase, I've been happily receiving 650mbps when testing with iperf3.

morrownr, 2025-01-27:
Glad it worked out.

> Thanks for helping out, and thanks for the excellent recommendations on hardware to purchase, I've been happily receiving 650mbps when testing with iperf3.

I get a lot of help with the Plug and Play List. It is actively managed with adapters being added and removed regularly.

Also, you do realize that your Edup adapters work with OpenWRT? There is a KMOD for the driver so if you your router has a USB port, you can add another radio.