morrownr/USB-WiFi #571: [Help]: Install driver on Kali Linux VM which install on Apple M1 Silicon Macbook

1PingSun, 2025-01-27:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux kali 6.11.2-arm64 #1 SMP Kali 6.11.2-1kali1 (2024-10-15) aarch64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet Bus 001 Device 003: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet Bus 001 Device 004: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet Bus 001 Device 005: ID 0409:55aa NEC Corp. Hub Bus 001 Device 006: ID 46f4:0001 QEMU QEMU USB HARDDRIVE Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano] Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

none (not thing)

### dkms

```text
rtl8812au/5.13.6-23, 6.11.2-arm64, aarch64: installed
```

### iw

```text
none (not thing)
```

### What happened?

I have tried for more than 24 hours, but it still can't work.
Does anyone know how to install the AC600 driver on Kali VM which is installed on Apple M1 Silicon Macbook?

Thanks.

## Device info
Laptop: Macbook Pro 14-inch, 2021 (M1 Pro chip)
VM Application: UTM
OS in VM: Linux Kali 6.11.2-arm64
USB WiFi Adapter: Archer T2U Nano (AC600)

morrownr, 2025-01-27:
Hi @1PingSun 

> rtl8812au/5.13.6-23, 6.11.2-arm64, aarch64: installed

> ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU

You have installed the wrong driver. You need the rtl8821au driver. You have options:

https://github.com/morrownr/8821au-20210708

or

https://github.com/lwfinger/rtw88

rtw88 is used as a development and test repo for upstreaming driver into the Linux kernel. In fact, when the distro you are using includes kernel 5.14 or later, you won't have to look for a driver anymore.

1PingSun, 2025-01-28:
Hi, thanks for your suggestion.

I am trying to use the lwfinger/rtw88 repo. After following the README file and installing the program I tried to load the driver with the command `sudo modprobe rtw_8811au`, but the output showed "modprobe: FATAL: Module RTW8811AU not found in directory /lib/modules/6.11.2-arm64". I don't know what wrong it is. And I don't know how to fix it. Help!

Thanks!

1PingSun, 2025-01-28:
I got it done!!

Finally, I used this repo: [https://github.com/morrownr/8821au-20210708](https://github.com/morrownr/8821au-20210708), and successfully install the driver for my WiFi Adapter smoothly!

Thank you for your kind help.

closed 2025-01-28 by 1PingSun

morrownr, 2025-01-28:
Great. Once you get to kernel 6.14, you won't want to install my morrownr driver as there will be a driver in the kernel. Your adapter should be plug and play at that point. Good luck.