morrownr/USB-WiFi #573: AX900 device id

firepower9966, 2025-02-06:
DriverFree Wifi6 AX900 + BT5.3

Bus 001 Device 004: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

morrownr, 2025-02-06:
Hi @firepower9966 

I am familiar with the concept. I assume you have a problem. What is the problem?

firepower9966, 2025-02-06:
no issue just id info as it was not in your  id list.
does your drivers also include bluetooth drivers?

morrownr, 2025-02-06:
> Bus 001 Device 004: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

That ID is specific to the internal storage. Your adapter is a multi-state adapter meaning that it has internal storage. That ID is not used by wifi drivers.

> Bus 001 Device 004: ID 0bda:b851 Realtek Semiconductor Corp. 802.11ax WLAN Adapter

I don't have a Linux driver up for that chip. Is that a 2.4 GHz only chip?

firepower9966, 2025-02-07:
Its Realtek RTL8851bu 2.4Ghz and 5Ghz  + BT5.3

dubhater, 2025-02-08:
I got one of these for 1.23 USD thanks to the AliExpress "welcome deal". 

The Bluetooth should just work if the kernel is new enough and you have the firmware.