morrownr/USB-WiFi #574: [Help]: ALFA AWUS036ACM hotspot status asleep after resume

crackforceone, 2025-02-07:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux dash 5.10.198-odroid-arm64 #1 SMP Thu Feb 6 19:40:54 CET 2025 aarch64 aarch64 aarch64 GNU/Linux

### lsusb

Bus 008 Device 009: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 006 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 010: ID 05ac:110a Apple, Inc. USB-C to 3.5mm Headphone Jack Adapter Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 009: ID 222a:0001 ILI Technology Corp. Multi-Touch Screen Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 001 Device 026: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) Bus 001 Device 025: ID 0c45:6369 Microdia UGREEN Camera Bus 001 Device 024: ID 24ae:9db6 Shenzhen Rapoo Technology Co., Ltd. 2.4G Receiver Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

14: phy7: Wireless LAN         Soft blocked: no         Hard blocked: no 15: hci0: Bluetooth         Soft blocked: no         Hard blocked: no

### dkms

```text
sudo: dkms: command not found
```

### iw

```text
phy#7
        Interface wlan0
                ifindex 10
                wdev 0x700000001
                addr xx:xx:xx:xx:xx:xx
                ssid dash
                type AP
                channel 36 (5180 MHz), width: 20 MHz, center1: 5180 MHz
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       85      0       0       0       0       19155           88
```

### What happened?

I am running Ubuntu 22.04 with Kernel 5.10 on my odroid m1 device. ALFA AWUS036ACM - MT7612U as the wifi device.
Currently i have set up a hotpot using nmcli and it's working well on a fresh boot. 
However when the device goes into suspend and then resumes then it takes quiet a while for the hotspot to recover. 

nmcli g after resume show:
`$ nmcli g
STATE   CONNECTIVITY  WIFI-HW  WIFI     WWAN-HW  WWAN    
asleep  unknown       enabled  enabled  enabled  enabled `

Also after resume wlan0 is missing from ifconfig.

This will recover, however it takes minutes to start the hotspot back up. For now I only found this command to restart the hotspot quicker:
`sudo systemctl NetworkManager restart`

Is there a proper way to get the wifi device to wake up from being "asleep"? Best case without having to restart the netwerk manager?

morrownr, 2025-02-09:
Hi @crackforceone 

I have an Alfa ACM also. Nice adapter. I run it in AP mode when I am not having to use the hw for testing something else.

I've never used Network Manager to setup an AP (hotspot). I use hostapd and tell NM to leave the adapter alone. 
i'm kinda scratching my head here as I think about this. I've never had a problem with sleep or suspend because I turn it off in the systems where I have an AP... I'm trying to figure out how the system can work as an AP if it is asleep.

crackforceone, 2025-02-09:
Hi @morrownr 
I have a bit of a weird usecase. Maybe I should expand on that.
So basically I am running OpenDash (https://github.com/openDsh/dash) on the Odroid M1 which is usually not connected with the internet. OpenDash is a wrapper around openAuto which is an AndroidAuto headunit emulator. 
So the AP on the Odroid is used to create a wireless connection from a phone to the host just to get the AndroidAuto video stream going.
So maybe there is a bit of a misunderstanding because when the system sleeps, then everything sleep - as is expected.
But when the system resumes from suspend then the AP stays in this STATE = "asleep" state for about a minute or so.
`$ nmcli g`
`STATE CONNECTIVITY WIFI-HW WIFI WWAN-HW WWAN`
`asleep unknown enabled enabled enabled enabled`

I have suspected that maybe Network Manager could introduce some weirdness.
So I tried to setup an AP with hostapd on a fresh Ubuntu server image to see if that just happens with NM. Now I can see the AP show up on other devices and with `isc-dhcp-server` I can also connect to it. But it disconnects after a couple of seconds and connects again until it finally disconnects.
I'm kind of at a loss right now, because I don't know how the setup the hostapd+dhcp server properly. Maybe use something else as the dhcp server?
Any advice on how fix the NM or hostapd situation would be highly appreciated.

morrownr, 2025-02-09:
> Any advice on how fix the NM or hostapd situation would be highly appreciated.

I'll do my best to help. This is an interesting use case.

> So I tried to setup an AP with hostapd on a fresh Ubuntu server image to see if that just happens with NM.

You can tell NM to disregard specific interfaces so that it does not interfere with a hostapd setup. netplan can also cause problems with a hostapd setup. Here is my AP guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

Go down to the section with the following title:

How to handle Network Manager

Personally, I find it easier to use a regular desktop that uses NM and tell NM not to disturb the interface you are using for the AP.

You can also post your hostapd.conf file so I can see your setup.

Let me know if this helps.

crackforceone, 2025-02-11:
Oof, just figured out why NM keeps the connection asleep after resume. I had a script running on resume that searches for a BT device and reconnects it. This script prevented the AP connection from waking up. So a basic user error.

But I am still interested in testing other options like hostapd. I have followed your guide, just skipping the rpi specific parts, and setup an AP. I used the hostapd-WiFi4.conf example and have only changed ssid, password and interface to wlan0.
The ssid shows up on other devices. However I can't connect to to it. 
I feel like I'm missing something really obvious, but I just can't find what it is.

crackforceone, 2025-02-13:
This is the .conf file that I have used so far:

````
# /etc/hostapd/hostapd-WiFi4.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2022-08-08

# SSID
ssid=dash
# PASSPHRASE
wpa_passphrase=xxxxxxxx
# Band: a = 5Ghz (a/n/ac), g = 2Ghz (b/g/n)
hw_mode=g
# Channel
channel=6
# Country code
country_code=DE

# Bridge interface
bridge=br0
# WiFi interface
interface=wlan0

# nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers
driver=nl80211
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

beacon_int=100
dtim_period=2
max_num_sta=32
rts_threshold=2347
fragm_threshold=2346
#send_probe_response=1

# security
# auth_algs=3 is required for WPA3-SAE and WPA3-SAE Transition mode
auth_algs=1
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
wpa_pairwise=CCMP
# WPA2-AES
wpa_key_mgmt=WPA-PSK
# WPA3-SAE
#wpa_key_mgmt=SAE
#wpa_group_rekey=1800
rsn_pairwise=CCMP
# ieee80211w=2 is required for WPA3-SAE
#ieee80211w=2
# If parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
#sae_require_mfp=1
# If parameter is not 9 set, 5 is the default value.
#sae_anti_clogging_threshold=10

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# Note: Only one ht_capab= line should be active. The content of these lines is
# determined by the capabilities of your adapter.
#
# generic 20 MHz setting
ht_capab=[SHORT-GI-20]
#
# generic 40 MHz setting
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# RasPi4B internal wifi
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
#
# rt5370 - rt3070
#ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][RX-STBC1]
#
# ar9271
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1][DSSS_CCK-40]
#
# mt7612u - mt7610u
#ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
#
# rtl8812au - rtl8811au -  rtl8812bu - rtl8811cu
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8814au
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

# End of hostapd-WiFi4.conf

````

morrownr, 2025-02-14:
> channel 36 (5180 MHz), width: 20 MHz, center1: 5180 MHz

I went back to look at your original post and I saw the above line. However, your hostapd.conf above is set to channel 6. I just need you to clarify which is correct as it makes a difference.

crackforceone, 2025-02-15:
Yea, with NM i can configure and use 5ghz.
That doesn't work with hostapd though. I think this is because the country code is set to 00 global.
Couldn't figure out how to change that yet. The solutions I found online didn't work for me and I guess I have to figure that one out with the board vendor. 
So that's why I was stuck with 2.4ghz on hostapd

morrownr, 2025-02-15:
> Yea, with NM i can configure and use 5ghz. That doesn't work with hostapd though.

It should. Let's work on the basis that you want 5 GHz and go that direction.

The above hostapd.conf you posted was made for WiFi 4, not WiFi 5. Let me dig something up:

```
# /etc/hostapd/hostapd-WiFi5.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf

# SSID
ssid=myPI-WiFi5
# PASSPHRASE
wpa_passphrase=myPW1234
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# Country code
country_code=US
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
#ieee80211h=1

# Bridge interface
bridge=br0
# WiFi interface
interface=wlan0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

# Various settings
#beacon_int=100
#dtim_period=2
#max_num_sta=32
macaddr_acl=0

# Security
# auth_algs=3 is required for WPA3-SAE and WPA3-SAE Transition mode
auth_algs=1
ignore_broadcast_ssid=0
# wpa=2 is required for WPA2 and WPA3 (read the docs)
wpa=2
wpa_pairwise=CCMP
rsn_pairwise=CCMP
# only one wpa_key_mgmt= line should be active.
# wpa_key_mgmt=WPA-PSK is required for WPA2-AES
wpa_key_mgmt=WPA-PSK
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-SAE Transition mode
#wpa_key_mgmt=SAE WPA-PSK
# wpa_key_mgmt=SAE is required for WPA3-SAE
#wpa_key_mgmt=SAE
#wpa_group_rekey=1800
# ieee80211w=1 is required for WPA3-SAE Transition mode
# ieee80211w=2 is required for WPA3-SAE
#ieee80211w=1

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1

# mt7612u - mt7610u
#ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]

# IEEE 802.11ac
ieee80211ac=1

# mt7612u
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

# end of hostapd-WiFi5.conf

```

That should work but you may need to change the following lines:

ssid=myPI-WiFi5
wpa_passphrase=myPW1234
country_code=US
bridge=br0
interface=wlan0

The rest should be fine but watch out for the below line:

vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

Make the copy and paste does not break the line. It should be one continuous line.

Lastly,

You could try shutting off power_save. See the Main Menu:

https://github.com/morrownr/USB-WiFi

Scroll down to `Frequently Asked Questions`.

See No.5

morrownr, 2025-02-21:
I noticed that you were having problems with country code as well. In the FAQ, take a look at No. 3.

Let me know what your status is.

crackforceone, 2025-02-23:
Sorry for the delay
I've tried No.3 and No.5 from the FAQ in combination with the conf that you posted.  But it just doesn't want to let me change the country code for some reason. 
 ```
$ sudo iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)`
```

 ```
$ sudo systemctl status hostapd
○ hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2025-02-23 14:50:04 UTC; 33min ago
       Docs: man:hostapd(8)
    Process: 1246 ExecStartPre=/bin/sleep 6 (code=exited, status=0/SUCCESS)
    Process: 1394 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS $DAEMON_CONF (code=exited, status=0/SUCCESS)
   Main PID: 1395 (code=exited, status=0/SUCCESS)
        CPU: 34ms

Feb 23 14:49:52 dash systemd[1]: Starting Access point and authentication server for Wi-Fi and Ethernet...
Feb 23 14:49:58 dash systemd[1]: Started Access point and authentication server for Wi-Fi and Ethernet.
Feb 23 14:50:03 dash hostapd[1395]: wlan0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=0) not found from the channel list of the current mode (2) IEEE 802.11a
Feb 23 14:50:03 dash hostapd[1395]: wlan0: IEEE 802.11 Hardware does not support configured channel
Feb 23 14:50:04 dash systemd[1]: hostapd.service: Deactivated successfully.

morrownr, 2025-02-23:
The thing that needs to be fixed before we can make progress is the setting of the country code. It is not clear to me why this is not working.

> Linux dash 5.10.198-odroid-arm64 

I am not familiar with this distro or hardware. I'm not sure what that line is telling me. Can you be specific about the Linux distro and hardware you are using?

That is a fairly old kernel. Do you have the ability to move to a newer kernel? 

This problem with setting the country code is something that might see more help in a support forum for your hardware/distro.