morrownr/USB-WiFi #575: Does the Mediatek mt7601u allow packet injection?

skerika, 2025-02-08:
I know that it only supports managed and monitor mode, but i was wondering if it supports packet injection (especially the EDUP MS8551).

I was looking to get into wireless stuff, but i need an adapter that supports monitor mode and seeing the EDUP MS8551 I was amazed at the long range it offers and its low price.

Is the EDUP MS8551 a good option for a begginner (i want to try some aircrack-ng and use it as a spare) or should i get something else?

Have a nice day
skerika

morrownr, 2025-02-08:
Hi @skerika 

> ... i was wondering if it supports packet injection (especially the EDUP MS8551).

I have one of those Edup ms8551 adapters. It certainly does have long range and is handy if you need something in your toolbox to get you connected when troubleshooting. However, I do not think it supports packet injection. If you want, I can test.

If you are looking for long range and full spectrum of ability to support aircrack-ng, then you might consider the Panda PAU0B or the ALFA AWUS036ACHM. Here is the full list of adapters that I recommend for security analysis:

 https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

You can also take a look at The Plug and Play List as all adapters in the list, with the exception of the Edup ms8551, should provide a good experience with your use case.

skerika, 2025-02-08:
Thanks for the fast response.

There is no need to test, I'll take a look at the other adapters.

It was a pleasure talking with you.
skerika

closed 2025-02-08 by skerika