morrownr/USB-WiFi #576: [News] wifi: mt76: fix returned txpower for mt7921 and mt7925

morrownr, 2025-02-11:
Greetings,

It appears that we are finally going to see a fix for txpower 3.00 dBm on various Mediatek drivers. The fix is in a series of 5 patches with the following title:

wifi: mt76: fix returned txpower for mt7921 and mt7925

You can view the patches at:  https://patchwork.kernel.org/project/linux-wireless/list/

I think we have had at least one report of this problem per week for the last 2 years. I have no idea why Mediatek did not correct this problem.

@morrownr

awaldram, 2025-03-21:
I think probably for marketing reasons. The mt792x.h limits all  mt792x devices to 15 client max in AP, the bandwidth is also limited in some drivers ie on the 7922 series you can only achieve 80Mhz in AP mode despite its ability to support 160Mhz in Client mode.

Ive had to revert to a mt7612u on my iot network as I expect each Ap to support approx 20 client in use and 40+ in failure mode.

Probably worth making a note in your excellent 'usb' guides as its going to become more and more relevant as WiFi device multiply.

So for the popular client Mediatek chip-sets USB,SDIO or M2 you are limited to approx 15 clients  in AP mode

via

#define MT792x_MAX_INTERFACES	4
#define MT792x_WTBL_SIZE	20
#define MT792x_WTBL_RESERVED	(MT792x_WTBL_SIZE - 1) = 19
#define MT792x_WTBL_STA		(MT792x_WTBL_RESERVED - MT792x_MAX_INTERFACES) 19-4 =15

so max Stations = 15 

in https://github.com/openwrt/mt76/blob/master/mt792x.h

morrownr, 2025-03-21:
Hi @awaldram 

I have no problem adding this information to the appropriate documents here. I would prefer anything we add to be complete with information about max interfaces for other chips as well. 

I see a couple of things to do. We can document it here and, probably as important, a well written message can be sent to linux-wireless in an effort to get the number changed or get an explanation for the current setting. Do you know how to send messages to linux-wireless?

awaldram, 2025-03-21:
HI,

I don't but would be surprised if the table can be altered as it is surely
a hardware limitation rather than software.

These devices are primarily designed as  'clients' and have many 'AP'
features missing such as ACS and DFS adding additional memory to the
devices to allow a larger Station table would increase the price and make
them uncommercial besides gives you a few more clients than raspberry pis
can deliver (10).

I think it is more the user's requirement to see if their 'use' case falls
inside the capability of the device rather than try and get the
manufacturer to remediate hardware limits.

for example  I use a mt7925bm NanoPi R5C on my perimeter because I know
i'll never have more then 5 'Wifi 7' clients here , my shed has a 7921u
providing 2.4gb iot connectivity for the pir and door sensors on the alarm
system.

Whilst my main 'AP's' bpi4 and openwrt1 I utilise DFS to bring up 160Mhz
5ghz @ch128 .. the openwrt1 supports 'zero-wait-dfs' so brings it up
immediately as a 2x2 (2402Mbs capability) whereas the bpi comes up as 3x3
(3603Mbs)) no 'client' designed adapter could hope to approach this but
even more perverse the bpi4 is not even approaching it max capability for
single channel bandwidth (6Gha,802.11Be 3x3 = 8647 Mbs capability)

My point is Cost for any solution capability is a compromise VS cost.
802.11be supports 16 spatial streams but I doubt any manufacturer will ever
deliver this same as we don't see any 8x8 802.11ax devices.

In performance I can;t saturate either the openwrt1 or bpi4 but can
easily flatline the mt7925 or mt7921u with multiple clients connected so I
think 15 clients is reasonable for the processing capability of the
devices, I find the mt7921u are much more stable on USB2 and in the  2.4Gb.
Start pushing them to perform and they get a bit flaky.

morrownr, 2025-03-21:
I wish this topic was in a new issue with an appropriate title as this topic comes up at times... not much but it does come up. 

> I don't but would be surprised if the table can be altered as it is surely a hardware limitation rather than software.

Not necessarily,

> These devices are primarily designed as  'clients' and have many 'AP' features missing such as ACS and DFS...

I agree that USB WiFi adapters are primary designed as client devices. However, I have tested some mods temporarily to see if the hardware is capable of DFS AP mode and handling larger amounts of connections. All of the Mediatek and Realtek adapters that I have tested can support both. Now, these adapters do not have the ability to detect conflicts as required by DFS AP mode so we would wonder off into illegal territory so I won't go there. On the other hand, we are no violating any laws by upping the number of connections supported so I am all for testing this. If you want to test, I'll try to help. I don't have more than 15 clients to test so I am no help with doing it myself.

A couple of years ago, I remember a guy that stopped by here and was in more or the less same situation you were in.  After some research and testing mine, I recommended a PAU08. I can check but I think that is a RT3070 chip. WiFi 4 only. He got one and reported back that things were working fine with over 32 IoT clients.  

> In performance I can;t saturate either the openwrt1 or bpi4 but can easily flatline the mt7925 or mt7921u with multiple clients connected so I think 15 clients is reasonable for the processing capability of the devices,

It, of course, depends on the throughput you need but the mt7921 and mt7925 have many times the p[rocessing power that the rt3070 does so that seems worth investigating.

Exactly which mt7921 adapter are you using and what computer is it plugged into?

And are you interested doing some testing?

awaldram, 2025-03-22:
I didn't mean USB explicitly were client devices but mt792x irrelevant of
interface. But by design.

As to what I have in this series it a long list due to the excellent
stability.

At on stage I had 3 mt7921 I hung off a raspberry pi (2 off a usb3 hub
unpowered).

But in general usb power is an issue the above was fine on a 4mb pi4 but
would stay up on an 8mb pi4.

Currently I have 2 mt7921u and an m.2  mt7925 on a Nanopi R5c (openwrt
24.10rc5)

You have to be careful with clean build running multiple adapters, I had
kernel panics and all sort of issues with an mt7921u solution that turned
out to be RTL drivers that if previously installed even though no RTL
hardware was in use.

My selection of mt7921u
Edup
Comfast
Fenvri
And some no-name clones

I find they all perform the same with performance related to the ae gain
not   manufacturer, the 2dbi aerials give great coverage at close to medium
distances whereas the 5dbi trials give worse close performance but better
range .

I'm not really in a position to do testing at the moment due to personal
projects in flight. But if that changes I'll try recompiling a pi kernel
with the max sta altered.

morrownr, 2025-03-26:
@awaldram 

> I'm not really in a position to do testing at the moment due to personal projects in flight. But if that changes I'll try recompiling a pi kernel with the max sta altered.

If you find that modifying max sta works, I can send in a patch to see if Mediatek will accept it. No hurry.

morrownr, 2025-03-26:
@goraj 

> There seems to be an additional attempt to fix some of the issue (openwrt/mt76#954) that has not been merged to linux-firmware

That is the same effort. 

> Distinct from some other mt7921 where people only see 3 dBM / 7 dBM I can retrieve the full txpowerlist but setting it above 3 dBm is not working. mt7921 eeprom

Please understand that this 3 dBm issue is cosmetic only. Your adapters are actually running at proper twpower settings. The code that needs work is the code that allows `iw` to read the settings properly and to allow `iw` to set txpower properly.

goraj, 2025-03-26:
Thank you @morrownr 
> Please understand that this 3 dBm issue is cosmetic only. Your adapters are actually running at proper twpower settings. The code that needs work is the code that allows iw to read the settings properly and to allow iw to set txpower properly.
Curious why `iwinfo phy1-ap0 txpowerlist` would return 20 dBm after setting tx power but after spinning the device up a subsequent call shows it reverted back to 3 dBm? If it's cosmetic I'd assume both cases should return 3 dBm?

morrownr, 2025-03-26:
> Curious why iwinfo phy1-ap0 txpowerlist would return 20 dBm after setting tx power...

I think `iwinfo` is designed to work with wireless extentions (WEXT) which is long since depreciated so it it best to use the modern `iw` utility.

 The code in the driver does not yet actually allow `iw` to set txpower if memory serves me correctly. It has been a while but there is an issue here where a deep dive was made and an individual used debug to go in and see what was actually going on. What was confirmed was that the driver is setting txpower at the max txpower allowed by regdb, which can vary depending on your country. I contend that there are situations where we want to reduce txpower to less than max so it would be nice to be able to reduce txpower. Hopefully the code for `iw` to show the proper txpower and the ability to change txpower is coming. Three patches to get ready for this went into kernel 6.15 this week. Until the remaining patches go in, we need to be patient.

Are you seeing a problem with range?

goraj, 2025-03-27:
> Are you seeing a problem with range?

I've ran an experiment setting txpower from 100 to 2500 while monitoring dBm with my mobile phone and see absolutely no change.

#### eg:
```
ifconfig  phy1-ap0 down
iw dev phy1-ap0 set txpower fixed {100, ..., 2500}
ifconfig  phy1-ap0 up
```

morrownr, 2025-03-27:
> ifconfig  phy1-ap0 down

`ifconfig` is depreciated. Please use `ip` and `iw` as they are the modern tools to use.

> iw dev phy1-ap0 set txpower fixed {100, ..., 2500}

I maybe have not been clear earlier. You are not setting txpower. We are trying to get that capability in the driver.

> I've ran an experiment setting txpower from 100 to 2500 while monitoring dBm with my mobile phone and see absolutely no change.

This is normal for now. You have no capability to set txpower.

If you think there is a problem with range, we can work that. I'd need to know the details of your setup.

morrownr, 2025-03-27:
Here is the documentation for `iw`:

https://wireless.docs.kernel.org/en/latest/en/users/documentation/iw.html

I can to be out of my office this afternoon but if you have trouble finding the documentation for `ip`, let me know and I will help.

goraj, 2025-03-29:
Thank you, I'll experiment a bit more.
Maybe the prior TP-Link devices (which worked much better) was due to their antenna. 
I've ordered some different adapters with better gain and will see if this resolves the issues.

morrownr, 2025-03-29:
> I've ordered some different adapters with better gain and will see if this resolves the issues.

There are a lot of things that can affect wireless range, speed and stability. 

- 6 GHz is just not to have the range that 5 GHz does and 5 GHz does not have the range that 2.4 GHz does.
- Using bluetooth can cause problems with 2.4 GHz. If fact, a lot of things can cause problems with 2.4 GHz. I am careful to keep 2.4 GHz wifi and bluetooth away from each other. I will only use/activate bluetooth in a specific location and the same goes with 2.4 wifi. 
- I recommend that testing is used to determine which bands and channels lead to the best results.
- What txpower is available with your equipment depends on your country.
- I find that DFS channels are wonderful for my location. Nobody uses DFS, there are no conflicts and my equipment supports 160 MHz channel width...well, the equipment that I want yo go fast does. I see wireless at over 1 Gbps... usually around 1.2-1.3 Gbps.
- Location of you equipment and the antennas is very important.
- Use the appropriate band depending on the needs of the client.
- Use the appropriate channels depending on the channels that you have that have little sharing going on.
- If things are not good with wifi, turning up txpower is not even something that I consider. I do turn down txpower if possible on the devices that support turning it down. There are good reasons for this.

Glad to help you with any of the above.

morrownr, 2025-06-05:
@awaldram 

> ...on the 7922 series you can only achieve 80Mhz in AP mode despite its ability to support 160Mhz in Client mode.

https://patchwork.kernel.org/project/linux-wireless/patch/20250511005316.1118961-1-sam8641@gmail.com/

So, the patch was accepted. I haven't tested it but the guy that submitted the patch did. I suspect you are correct about many things you said, such adapters having no ability to do DFS in AP mode. These adapters most likely do not have the capability to detect conflicting signals so DFS just cannot legally be done. On the other hand, I think there are situations that can be modified to what we want. The patch that went in was accepted without a discussion. I suspect the 80 MHz limit was a mistake. The patch is in kernel 6.16.

I also think we should be able to get the number of clients up above 15. Have you tried increasing the number:

#define MT792x_WTBL_SIZE 20

I have not tested it because I would have a hard time getting over 15 clients setup. If I had a home full of IoT, it would not be a problem but I have very limited IoT. I would recommend increasing the number in a limited way so that we can find out if there is a hardware or firmware limit.

awaldram, 2025-06-05:
@morrownr/USB-WiFi ***@***.***>

Thanks for the info , I'm no longer using any 7922 devices having moved
over to 7925 for BE capabilities.
I have two operating on my borders (as wifi fillin)  on 6ghz 160Mhz so
cannot test 15+ clients but they work very well
You may be interested that one 7925  is running in an openwrt1 off the NVME
m connector with an A+e adapter (without the BT/USB connected) the other is
in a nano R5C

[    3.631366] pci 0000:01:00.0: [14c3:7925] type 00 class 0x028000
[    3.879237] mtk-pcie-gen3 11280000.pcie: msi#0xd address_hi 0x0
address_lo 0x11280c00 data 13
[    4.159792] FIT: Detected U-Boot 2024.10-OpenWrt-r28161-ea17e958b9
[   15.146844] mt7925e 0000:01:00.0: assign IRQ: got 84
[   15.151866] mt7925e 0000:01:00.0: enabling device (0000 -> 0002)
[   15.157910] mt7925e 0000:01:00.0: enabling bus mastering
[   15.481943] mt7925e 0000:01:00.0: ASIC revision: 79250000
[   15.639045] mt7925e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time:
20231227093012a
[   15.984566] mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build
Time: 20231227093232
[   24.317976] mt7925e 0000:01:00.0 phy2-ap0: entered allmulticast mode
[   24.324540] mt7925e 0000:01:00.0 phy2-ap0: entered promiscuous mode

I use many 7921USB to cover the %ghz and 2.4Ghz and find them pretty much
rock solid provided you make sure they have plenty of steady power .
(plugging three into any raspberry will give you a world of pain (USB
devces locking/vanishing). the friendly elec dievices will generally run
three but every 4-6 weeks you'll need to reboot as the same issue as the pi
occurs just less frequently . .

I find 1 :M.2 NGFF mtk7925 and 2 MTK7921 gives me all channels and plenty
of bandwidth rock solid.