morrownr/USB-WiFi #579: mt7921 crash on 6.12.13

Simon566, 2025-02-12:
Alfa adapter:

Feb 12 11:43:58 odroid-h3 kernel: BUG: kernel NULL pointer dereference, address: 0000000000000400
Feb 12 11:43:58 odroid-h3 kernel: #PF: supervisor read access in kernel mode
Feb 12 11:43:58 odroid-h3 kernel: #PF: error_code(0x0000) - not-present page
Feb 12 11:43:58 odroid-h3 kernel: PGD 0 P4D 0
Feb 12 11:43:58 odroid-h3 kernel: Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
Feb 12 11:43:58 odroid-h3 kernel: CPU: 3 UID: 0 PID: 235904 Comm: mt76-usb-rx phy Tainted: G     U  W  OE      6.12.13 #14
Feb 12 11:43:58 odroid-h3 kernel: Tainted: [U]=USER, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Feb 12 11:43:58 odroid-h3 kernel: Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000020f222000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: Call Trace:
Feb 12 11:43:58 odroid-h3 kernel:  <TASK>
Feb 12 11:43:58 odroid-h3 kernel:  ? __die+0x23/0x80
Feb 12 11:43:58 odroid-h3 kernel:  ? page_fault_oops+0x173/0x5d0
Feb 12 11:43:58 odroid-h3 kernel:  ? exc_page_fault+0x76/0x1b0
Feb 12 11:43:58 odroid-h3 kernel:  ? asm_exc_page_fault+0x26/0x30
Feb 12 11:43:58 odroid-h3 kernel:  ? mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel:  mt7921_queue_rx_skb+0x1df/0x9a0 [mt7921_common]
Feb 12 11:43:58 odroid-h3 kernel:  mt76u_rx_worker+0x5be/0x5f0 [mt76_usb]
Feb 12 11:43:58 odroid-h3 kernel:  ? __pfx___mt76_worker_fn+0x10/0x10 [mt76]
Feb 12 11:43:58 odroid-h3 kernel:  __mt76_worker_fn+0x4c/0x80 [mt76]
Feb 12 11:43:58 odroid-h3 kernel:  kthread+0xde/0x110
Feb 12 11:43:58 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 12 11:43:58 odroid-h3 kernel:  ret_from_fork+0x31/0x50
Feb 12 11:43:58 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 12 11:43:58 odroid-h3 kernel:  ret_from_fork_asm+0x1a/0x30
Feb 12 11:43:58 odroid-h3 kernel:  </TASK>
Feb 12 11:43:58 odroid-h3 kernel: Modules linked in: rtw_8812au(OE) ftdi_sio mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_us>
Feb 12 11:43:58 odroid-h3 kernel:  snd_soc_acpi snd_soc_core x86_pkg_temp_thermal nls_ascii snd_compress nls_cp437 soundwire_bus intel_powerclam>
Feb 12 11:43:58 odroid-h3 kernel:  sunrpc garp stp mrp loop dm_mod llc efi_pstore configfs efivarfs ip_tables x_tables autofs4 hid_logitech_hidp>
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400
Feb 12 11:43:58 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000018337c000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: note: mt76-usb-rx phy[235904] exited with irqs disabled

and

Feb 12 11:43:58 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000018337c000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: note: mt76-usb-rx phy[235904] exited with irqs disabled
Feb 12 11:43:58 odroid-h3 dnsmasq-dhcp[1572]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar

no patches , vanilla kernel on Odroid H3 ( X86-64) with the USB Alfa adapter

morrownr, 2025-02-12:
This is interesting. Two basically identical reports within the last 12 hours with the mt7921u on kernel 6.12.13.

See: https://github.com/morrownr/USB-WiFi/issues/577

What kernel version were you using prior to 6.12.13?

Was it stable?

Simon566, 2025-02-12:
it was never stable , im going all the way from 5.19 over 6.6 to 6.12 ......

6.12.13 looks better then any version before , it was running a day before crashing , in AP mode

morrownr, 2025-02-12:
Never stable. Well, that is not good.

Tell which adapter you have?

What have you been seeing that makes it less than stable? (before the crash with 6.12.13)

Simon566, 2025-02-12:
which kind of details would you expect ?

i can do a downgrade on 6.12.12, will take some hours to compile, and repeat

morrownr, 2025-02-12:
> which kind of details would you expect ?

You can post the lines you posted above plus:

AP setup. Are you using hostapd? Or what?

If using hostapd, post the contents of your hostapd.conf file.

What distro and version are you using?

Does the distro use Network Manager (NM)?

If NM is being used, have you told NM to leave the wireless interface of your AXML alone? It will interfere.

> i can do a downgrade on 6.12.12, will take some hours to compile, and repeat

I would appreciate it if you could do this. I like to have at least 2 confirming reports before reporting something.