morrownr/USB-WiFi #580: ALFA AWUS036AXML - unable to run a stable AP

Simon566, 2025-02-13:
Alfa adapter:

Feb 12 11:43:58 odroid-h3 kernel: BUG: kernel NULL pointer dereference, address: 0000000000000400
Feb 12 11:43:58 odroid-h3 kernel: #PF: supervisor read access in kernel mode
Feb 12 11:43:58 odroid-h3 kernel: #PF: error_code(0x0000) - not-present page
Feb 12 11:43:58 odroid-h3 kernel: PGD 0 P4D 0
Feb 12 11:43:58 odroid-h3 kernel: Oops: Oops: 0000 [https://github.com/morrownr/USB-WiFi/issues/1] PREEMPT SMP NOPTI
Feb 12 11:43:58 odroid-h3 kernel: CPU: 3 UID: 0 PID: 235904 Comm: mt76-usb-rx phy Tainted: G U W OE 6.12.13 #14
Feb 12 11:43:58 odroid-h3 kernel: Tainted: [U]=USER, [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Feb 12 11:43:58 odroid-h3 kernel: Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS: 0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000020f222000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: Call Trace:
Feb 12 11:43:58 odroid-h3 kernel:
Feb 12 11:43:58 odroid-h3 kernel: ? __die+0x23/0x80
Feb 12 11:43:58 odroid-h3 kernel: ? page_fault_oops+0x173/0x5d0
Feb 12 11:43:58 odroid-h3 kernel: ? exc_page_fault+0x76/0x1b0
Feb 12 11:43:58 odroid-h3 kernel: ? asm_exc_page_fault+0x26/0x30
Feb 12 11:43:58 odroid-h3 kernel: ? mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: mt7921_queue_rx_skb+0x1df/0x9a0 [mt7921_common]
Feb 12 11:43:58 odroid-h3 kernel: mt76u_rx_worker+0x5be/0x5f0 [mt76_usb]
Feb 12 11:43:58 odroid-h3 kernel: ? __pfx___mt76_worker_fn+0x10/0x10 [mt76]
Feb 12 11:43:58 odroid-h3 kernel: __mt76_worker_fn+0x4c/0x80 [mt76]
Feb 12 11:43:58 odroid-h3 kernel: kthread+0xde/0x110
Feb 12 11:43:58 odroid-h3 kernel: ? __pfx_kthread+0x10/0x10
Feb 12 11:43:58 odroid-h3 kernel: ret_from_fork+0x31/0x50
Feb 12 11:43:58 odroid-h3 kernel: ? __pfx_kthread+0x10/0x10
Feb 12 11:43:58 odroid-h3 kernel: ret_from_fork_asm+0x1a/0x30
Feb 12 11:43:58 odroid-h3 kernel:
Feb 12 11:43:58 odroid-h3 kernel: Modules linked in: rtw_8812au(OE) ftdi_sio mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_us>
Feb 12 11:43:58 odroid-h3 kernel: snd_soc_acpi snd_soc_core x86_pkg_temp_thermal nls_ascii snd_compress nls_cp437 soundwire_bus intel_powerclam>
Feb 12 11:43:58 odroid-h3 kernel: sunrpc garp stp mrp loop dm_mod llc efi_pstore configfs efivarfs ip_tables x_tables autofs4 hid_logitech_hidp>
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400
Feb 12 11:43:58 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS: 0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000018337c000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: note: mt76-usb-rx phy[235904] exited with irqs disabled

and

Feb 12 11:43:58 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 12 11:43:58 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 12 11:43:58 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27>
Feb 12 11:43:58 odroid-h3 kernel: RSP: 0018:ffffa8e68334bdb8 EFLAGS: 00010202
Feb 12 11:43:58 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8d05e3b83d00 RCX: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: RDX: 0000000000000000 RSI: 000000000000068a RDI: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: RBP: ffff8d05cad9c800 R08: 0000000000000000 R09: 0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: R10: ffff8d05e3b83d00 R11: 0000000000000000 R12: ffff8d05c5591000
Feb 12 11:43:58 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843806 R15: ffff8d05ce022000
Feb 12 11:43:58 odroid-h3 kernel: FS: 0000000000000000(0000) GS:ffff8d092ff80000(0000) knlGS:0000000000000000
Feb 12 11:43:58 odroid-h3 kernel: CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 12 11:43:58 odroid-h3 kernel: CR2: 0000000000000400 CR3: 000000018337c000 CR4: 0000000000352ef0
Feb 12 11:43:58 odroid-h3 kernel: note: mt76-usb-rx phy[235904] exited with irqs disabled
Feb 12 11:43:58 odroid-h3 dnsmasq-dhcp[1572]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar

----

- no patches , vanilla kernel 6.12.13 on Odroid H3 ( X86-64) with the USB Alfa adapter
- running Debian Bookworm , no network manager

    it was never able to run a stable AP
        tx urb failed issues
        null ptr accesses
        device is unusable since i bought it ( one year ) passing over 5.19 , 6.6 and 6.12 kernel versions
    bluetooth crash is avoided by your FAQ and blacklisting

[hostapd_alfa.conf.txt](https://github.com/user-attachments/files/18779981/hostapd_alfa.conf.txt)

morrownr, 2025-02-13:
For others that may read this issue:

> odroid-h3 kernel: BUG: kernel NULL pointer dereference, address: 0000000000000400

This is an issue that just popped up with the recent release of 6.12.13. One reporter has reported that backing off to 6.12.12 removes the problem making it looking like a patch that went into 6.12.13 is the cause. I have tested with 6.14 rc1 and do not see this problem. My intent is to report this to linux-wireless when I have enough information.

Hi @Simon566 

> ...never able to run a stable AP... with Alfa AXML

Let's see what we can do.

I am currently reading `hostapd_alfa.conf.txt`. I am not finished reading it but I have some questions:

- Is that the hostapd.conf file you are actually using?

- Is there a reason that you are taking an adapter that is capable of AXE3000 class performance and running it is what amounts to G72. I mean, it is not even running in N150 (WiFi 4) mode, let along the WiFi 6 mode that it is capable of in all 3 bands.

- I need to know what your intent is:

What band do you want to use? (2.4 GHz, 5 GHz or 6 GHz)

What channel do you want to use?

What throughput are you looking for?

What devices are connecting to this AP and what are their capabilities?

- I am seeing some problems in the hostapd.conf. I understand that hostapd.conf can be a real challenge. Been there, done that. Let me help you. Answer my questions about what you want to do and I'll see what I can do.

Simon566, 2025-02-13:
Im running it in 2,4ghz mode with 11ax Options, 20mhz Channel width. Im able to reach 100Mbit netto Transfer....

Do you think the config for this is wrong?

morrownr, 2025-02-13:
Opps...

I had to correct myself, the following will work for N150:

hw_mode=g

> Do you think the config for this is wrong?

Let me continue reading. Setting up AX on band 1 (2.4 GHz) is not as simple as it could be. Even the guide I have post for the mt7921 defaults to 5 GHz. Let me work this and let you know what I figure out.

> Im running it in 2,4ghz mode with 11ax Options, 20mhz Channel width.

Let me work with this.

morrownr, 2025-02-13:
@Simon566 

Your hostapd.conf is really pretty good. My old eyes had a hard time going through the file as it is due to all of the documentation. I'm not saying using the documentation is a bad idea. What I did is run the following to get rid of the comments and the document came to life for me:

$ sed '/^#/ d' < inputFile.txt > outputFile.txt

I did leave the lines in the order they were in. I have used `#` to show my comments:

```

interface=wlx00c0cab38118

bridge=br-guest

driver=nl80211

logger_syslog=-1
logger_syslog_level=3
logger_stdout=-1
logger_stdout_level=3

ctrl_interface=/var/run/hostapd

ctrl_interface_group=0

ssid=xxxxxxxxxxxxxxx

country_code=DE

ieee80211d=1

hw_mode=g

channel=6

max_num_sta=14

macaddr_acl=0

auth_algs=1

ignore_broadcast_ssid=0

wmm_enabled=1

skip_inactivity_poll=1

ieee80211n=1

Change
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][SU-BEAMFORMEE][MU-BEAMFORMEE]
to
ht_capab=[LDPC][SHORT-GI-20][TX-STBC][RX-STBC1][MAX-AMSDU-7935][SU-BEAMFORMEE][MU-BEAMFORMEE]

# Not needed, 2.4 GHz cannot do AC
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-2][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

ieee80211ax=1

he_su_beamformer=1
he_su_beamformee=1
# The below line is a capability the adapter does not have 
he_mu_beamformer=1
he_bss_color=37

# Not needed
he_oper_chwidth=0

# Not needed
eapol_key_index_workaround=0

# Not needed
eap_server=0

# Is this line needed?
own_ip_addr=127.0.0.1

wpa=2

wpa_passphrase=xxxxxxxxxxxxxxxxxxxx

wpa_pairwise=CCMP
rsn_pairwise=CCMP

```

Next:

It appears that the Odroid H3 has two USB2 ports. Make sure you are plugging the AXML into a USB2 port, not a USB3 port. Why? We want to avoid the interference caused by USB3 in the 2.4 GHz band.

Next:

Does the back of your AXML have a line like this?  80211ax 2x2+BT5.2

If so, do you use the bluetooth? If yes, it would probably be best not to do so since you are on the 2.4 GHz band. Why? bluetooth works on the 2.4 GHz band and will interfere with wifi on the 2.4 GHz band. I have advocated for many year not to buy USB WiFi adapters that have bluetooth support.

Next:

There is no good reason for scatter-gather to be active in your setup and I have seen problems with it on some small board systems depending on the chip used in the USB3 hub so:

To shut down scatter-gather support:

Open a terminal (Ctrl+Alt+t)

```
sudo -i
echo "options mt76_usb disable_usb_sg=1" >> /etc/modprobe.d/mt76_usb.conf
exit
```

Final word for now: Let me know if any of the above help your AXML become stable in AP mode.
sudo reboot

Simon566, 2025-02-13:
Hi,

- i did the adaptions in the hostapd config file
- yes its a bluetooth capable adapter , but i blacklisted it , is it enough
-  its plugged into USB2

scatter gather was on , will turn it off

will report back

thank you

Simon566, 2025-02-14:
here is another crash , with your adjustments made

[Fr Feb 14 14:41:25 2025] br-lan: port 2(enp1s0) entered disabled state
[Fr Feb 14 14:49:33 2025] BUG: kernel NULL pointer dereference, address: 0000000000000400
[Fr Feb 14 14:49:33 2025] #PF: supervisor read access in kernel mode
[Fr Feb 14 14:49:33 2025] #PF: error_code(0x0000) - not-present page
[Fr Feb 14 14:49:33 2025] PGD 1078b6067 P4D 1078b6067 PUD 0
[Fr Feb 14 14:49:33 2025] Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
[Fr Feb 14 14:49:33 2025] CPU: 1 UID: 0 PID: 697 Comm: mt76-usb-rx phy Tainted: G     U     OE      6.12.13 #14
[Fr Feb 14 14:49:33 2025] Tainted: [U]=USER, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[Fr Feb 14 14:49:33 2025] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[Fr Feb 14 14:49:33 2025] RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
[Fr Feb 14 14:49:33 2025] Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27 48 8b 80 78 03 00 00 <48> 8b 80 00 04 00 00 48 8d 90 e0 00 00 00 48 85 c0 48 0f 45 c2 31
[Fr Feb 14 14:49:33 2025] RSP: 0018:ffffbfb880757db8 EFLAGS: 00010202
[Fr Feb 14 14:49:33 2025] RAX: 0000000000000000 RBX: ffffa0f660fb2300 RCX: 0000000000000000
[Fr Feb 14 14:49:33 2025] RDX: 0000000000000000 RSI: 000000000000068e RDI: ffffa0f64b9b2000
[Fr Feb 14 14:49:33 2025] RBP: ffffa0f6a7d3fb40 R08: 0000000000000000 R09: 0000000000000000
[Fr Feb 14 14:49:33 2025] R10: ffffa0f660fb2300 R11: 0000000000000000 R12: ffffa0f6a7e54000
[Fr Feb 14 14:49:33 2025] R13: 0000000038010148 R14: 000000008084380a R15: ffffa0f64b9b2000
[Fr Feb 14 14:49:33 2025] FS:  0000000000000000(0000) GS:ffffa0f9afe80000(0000) knlGS:0000000000000000
[Fr Feb 14 14:49:33 2025] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Fr Feb 14 14:49:33 2025] CR2: 0000000000000400 CR3: 00000001507bc000 CR4: 0000000000352ef0
[Fr Feb 14 14:49:33 2025] Call Trace:
[Fr Feb 14 14:49:33 2025]  <TASK>
[Fr Feb 14 14:49:33 2025]  ? __die+0x23/0x80
[Fr Feb 14 14:49:33 2025]  ? page_fault_oops+0x173/0x5d0
[Fr Feb 14 14:49:33 2025]  ? exc_page_fault+0x76/0x1b0
[Fr Feb 14 14:49:33 2025]  ? asm_exc_page_fault+0x26/0x30
[Fr Feb 14 14:49:33 2025]  ? mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
[Fr Feb 14 14:49:33 2025]  mt7921_queue_rx_skb+0x1df/0x9a0 [mt7921_common]
[Fr Feb 14 14:49:33 2025]  mt76u_rx_worker+0x5be/0x5f0 [mt76_usb]
[Fr Feb 14 14:49:33 2025]  ? __pfx___mt76_worker_fn+0x10/0x10 [mt76]
[Fr Feb 14 14:49:33 2025]  __mt76_worker_fn+0x4c/0x80 [mt76]
[Fr Feb 14 14:49:33 2025]  kthread+0xde/0x110
[Fr Feb 14 14:49:33 2025]  ? __pfx_kthread+0x10/0x10
[Fr Feb 14 14:49:33 2025]  ret_from_fork+0x31/0x50
[Fr Feb 14 14:49:33 2025]  ? __pfx_kthread+0x10/0x10
[Fr Feb 14 14:49:33 2025]  ret_from_fork_asm+0x1a/0x30
[Fr Feb 14 14:49:33 2025]  </TASK>
[Fr Feb 14 14:49:33 2025] Modules linked in: cls_matchall sch_ingress sch_cake act_mirred ifb nft_masq nft_nat nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 cmac ccm veth wireguard libchacha20poly1305 chacha_x86_64 poly1305_x86_64 curve25519_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel pppoe pppox nf_tables ppp_generic crc32c_generic nfnetlink slhc nvme_fabrics bluetooth crc16 msr bridge binfmt_misc snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component snd_sof_pci_intel_icl i915 snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda mt7921u snd_hda_ext_core mt792x_usb snd_sof_pci mt7921_common snd_sof_xtensa_dsp snd_sof mt792x_lib snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi mt76_connac_lib snd_soc_core mt76_usb snd_compress mt76 x86_pkg_temp_thermal rtw_8812au(OE) rtw_8812a(OE) rtw_88xxa(OE) rtw_usb(OE)
[Fr Feb 14 14:49:33 2025]  soundwire_bus rtw_core(OE) intel_powerclamp nls_ascii kvm_intel snd_hda_intel cec nls_cp437 mac80211 snd_intel_dspcfg rc_core vfat snd_intel_sdw_acpi kvm drm_buddy snd_hda_codec fat drm_display_helper crct10dif_pclmul snd_hda_core libarc4 processor_thermal_device_pci_legacy processor_thermal_device ghash_clmulni_intel processor_thermal_wt_hint sha512_ssse3 intel_rapl_msr processor_thermal_rfim mei_hdcp ttm mei_pxp snd_hwdep cfg80211 processor_thermal_rapl sha256_ssse3 snd_pcm sha1_ssse3 intel_rapl_common aesni_intel snd_timer drm_kms_helper iTCO_wdt gf128mul crypto_simd cryptd intel_pmc_bxt snd processor_thermal_wt_req pcspkr iTCO_vendor_support intel_cstate mei_me processor_thermal_power_floor watchdog soundcore ftdi_sio processor_thermal_mbox intel_pmc_core mei int340x_thermal_zone wmi_bmof i2c_algo_bit ee1004 rfkill usbserial intel_soc_dts_iosf nfsd intel_vsec pmt_telemetry acpi_pad pmt_class button acpi_tad evdev joydev sg auth_rpcgss it87(OE) hwmon_vid emc2103 nfs_acl lockd coretemp drm 8021q grace
[Fr Feb 14 14:49:33 2025]  sunrpc dm_mod garp stp mrp loop llc fuse efi_pstore configfs efivarfs ip_tables x_tables autofs4 btrfs xor raid6_pq libcrc32c hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid sd_mod ahci libahci xhci_pci libata xhci_hcd r8169 nvme realtek mdio_devres nvme_core usbcore crc32_pclmul libphy scsi_mod crc32c_intel i2c_i801 intel_lpss_pci i2c_smbus intel_lpss scsi_common idma64 usb_common video wmi
[Fr Feb 14 14:49:33 2025] CR2: 0000000000000400
[Fr Feb 14 14:49:33 2025] ---[ end trace 0000000000000000 ]---
[... cut ...]
[Fr Feb 14 14:49:37 2025] R10: ffffa0f660fb2300 R11: 0000000000000000 R12: ffffa0f6a7e54000
[Fr Feb 14 14:49:37 2025] R13: 0000000038010148 R14: 000000008084380a R15: ffffa0f64b9b2000
[Fr Feb 14 14:49:37 2025] FS:  0000000000000000(0000) GS:ffffa0f9afe00000(0000) knlGS:0000000000000000
[Fr Feb 14 14:49:37 2025] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Fr Feb 14 14:49:37 2025] CR2: 0000000000000000 CR3: 0000000195f58000 CR4: 0000000000352ef0
[Fr Feb 14 14:49:37 2025] note: kworker/u16:2[26114] exited with irqs disabled

morrownr, 2025-02-14:
> 6.12.13

The problem you posted about above is specific to kernel 6.12.13. You need to move back to kernel 6.12.12 or earlier while the problem is fixed.

Simon566, 2025-02-14:
here is a crash from 6.12.12

Feb 14 18:11:51 odroid-h3 kernel: mt7921u 1-4:1.3: Message 00020003 (seq 10) timeout
Feb 14 18:11:52 odroid-h3 kernel: mt7921u 1-4:1.3: timed out waiting for pending tx
Feb 14 18:11:52 odroid-h3 kernel: ------------[ cut here ]------------
Feb 14 18:11:52 odroid-h3 kernel: WARNING: CPU: 3 PID: 64063 at kernel/kthread.c:662 kthread_park+0x89/0xa0
Feb 14 18:11:52 odroid-h3 kernel: Modules linked in: cmac rtw_8812au(OE) rtw_8812a(OE) rtw_88xxa(OE) rtw_usb(OE) rtw_core(OE) it87(OE) cls_matchall sch_ingress sch_cake act_mirred ifb nft>
Feb 14 18:11:52 odroid-h3 kernel:  mt76_usb snd_soc_core kvm_intel mt76 snd_compress mac80211 soundwire_bus cec kvm rc_core snd_hda_intel libarc4 drm_buddy snd_intel_dspcfg snd_intel_sdw_>
Feb 14 18:11:52 odroid-h3 kernel:  grace stp sunrpc fuse mrp dm_mod llc loop efi_pstore configfs efivarfs ip_tables x_tables autofs4 hid_logitech_hidpp btrfs hid_logitech_dj hid_generic u>
Feb 14 18:11:52 odroid-h3 kernel: CPU: 3 UID: 0 PID: 64063 Comm: kworker/u16:6 Tainted: G     U     OE      6.12.12 #17
Feb 14 18:11:52 odroid-h3 kernel: Tainted: [U]=USER, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Feb 14 18:11:52 odroid-h3 kernel: Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
Feb 14 18:11:52 odroid-h3 kernel: Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Feb 14 18:11:52 odroid-h3 kernel: RIP: 0010:kthread_park+0x89/0xa0
Feb 14 18:11:52 odroid-h3 kernel: Code: 5b 5d 31 f6 31 ff c3 cc cc cc cc 0f 0b 48 8b ab 58 0a 00 00 a8 04 74 ac 0f 0b b8 da ff ff ff 5b 5d 31 f6 31 ff c3 cc cc cc cc <0f> 0b b8 f0 ff ff f>
Feb 14 18:11:52 odroid-h3 kernel: RSP: 0018:ffffb70d4620bd60 EFLAGS: 00010202
Feb 14 18:11:52 odroid-h3 kernel: RAX: 0000000000000004 RBX: ffff8a4ed7ca1880 RCX: 0000000000000000
Feb 14 18:11:52 odroid-h3 kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8a4ed7ca1880
Feb 14 18:11:52 odroid-h3 kernel: RBP: ffff8a4e83dbf580 R08: 0000000000000000 R09: 0000000000000000
Feb 14 18:11:52 odroid-h3 kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff8a4e9b3e2060
Feb 14 18:11:52 odroid-h3 kernel: R13: ffff8a4e9b3e2060 R14: ffff8a4e80d52b68 R15: 0000000000000100
Feb 14 18:11:52 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8a51eff80000(0000) knlGS:0000000000000000
Feb 14 18:11:52 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 14 18:11:52 odroid-h3 kernel: CR2: 00007f4f5ca77000 CR3: 000000017d5ee000 CR4: 0000000000352ef0
Feb 14 18:11:52 odroid-h3 kernel: Call Trace:
Feb 14 18:11:52 odroid-h3 kernel:  <TASK>
Feb 14 18:11:52 odroid-h3 kernel:  ? __warn+0x89/0x130
Feb 14 18:11:52 odroid-h3 kernel:  ? kthread_park+0x89/0xa0
Feb 14 18:11:52 odroid-h3 kernel:  ? report_bug+0x1a2/0x1d0
Feb 14 18:11:52 odroid-h3 kernel:  ? handle_bug+0x66/0xa0
Feb 14 18:11:52 odroid-h3 kernel:  ? exc_invalid_op+0x17/0x80
Feb 14 18:11:52 odroid-h3 kernel:  ? asm_exc_invalid_op+0x1a/0x20
Feb 14 18:11:52 odroid-h3 kernel:  ? kthread_park+0x89/0xa0
Feb 14 18:11:52 odroid-h3 kernel:  mt76u_stop_tx+0x224/0x310 [mt76_usb]
Feb 14 18:11:52 odroid-h3 kernel:  ? __pfx_autoremove_wake_function+0x10/0x10
Feb 14 18:11:52 odroid-h3 kernel:  mt7921u_mac_reset+0x72/0x1b0 [mt7921u]
Feb 14 18:11:52 odroid-h3 kernel:  mt7921_mac_reset_work+0x9f/0x190 [mt7921_common]
Feb 14 18:11:52 odroid-h3 kernel:  ? __schedule+0x400/0xca0
Feb 14 18:11:52 odroid-h3 kernel:  process_one_work+0x179/0x390
Feb 14 18:11:52 odroid-h3 kernel:  worker_thread+0x262/0x370
Feb 14 18:11:52 odroid-h3 kernel:  ? _raw_spin_lock_irqsave+0x27/0x80
Feb 14 18:11:52 odroid-h3 kernel:  ? __pfx_worker_thread+0x10/0x10
Feb 14 18:11:52 odroid-h3 kernel:  kthread+0xde/0x110
Feb 14 18:11:52 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 14 18:11:52 odroid-h3 kernel:  ret_from_fork+0x31/0x50
Feb 14 18:11:52 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 14 18:11:52 odroid-h3 kernel:  ret_from_fork_asm+0x1a/0x30
Feb 14 18:11:52 odroid-h3 kernel:  </TASK>
Feb 14 18:11:52 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 14 18:11:53 odroid-h3 kernel: mt7921u 1-4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a
Feb 14 18:11:53 odroid-h3 kernel: mt7921u 1-4:1.3: WM Firmware Version: ____010000, Build Time: 20241106151045

morrownr, 2025-02-14:
Wow! Have you tried all of the usb ports on that h3?

Take a look at this message:

https://github.com/morrownr/USB-WiFi/issues/508

Do you have another system to test with?

The more we talk, the more it seems to me that the is something more serious than a driver bug going on here. USB can be a challenge in some cases. Is your adapter plugged into a powered hub or directly into a port on the system?

Simon566, 2025-02-15:
the Odroid H3 is a intel Atom based system with a widespread chip. i wouldnt expect issues with USB ports , especially the USB2 which im using . is xHCI handoff and things like that not only USB3 ?

i tried different Ports , USB3 / USB2 but the results are the same. So far i havent got a different test system , maybe it s going to be a odroid h4 some day

morrownr, 2025-02-15:
@Simon566 

> i wouldnt expect issues with USB ports

Famous last words.

Does your bios have any settings for USB other than on and off?

What I am seeing are things that I'm not used to seeing. Now that you went back to 6.12.12 it appears somewhat different but the only times I've seen ugliness like this is when a user has several usb devices plugged in and the usb bus is not providing adequate power for what is requested. That is a common problem on RasPis because their usb subsystem does not provide anywhere near spec power. Any thoughts on whether that could be a problem?

Simon566, 2025-02-15:
As im using a x86 Board with a 80w or so PSU, Power is No issue

Yes,i do have BIOS settings for USB, does xhci handoff also Count for usb2? Im Happy to Test suggested settings

morrownr, 2025-02-15:
> As im using a x86 Board with a 80w or so PSU, Power is No issue

Well, I have one x86 system with a 350 w PSU and a usb subsystem that has a power problem. 

> does xhci handoff also Count for usb2?

Unknown but xhci handoff is a known problem so I would recommend changing the setting to test.

Simon566, 2025-02-17:
more crashes , this time with 6.12.14 and the changed USB settings:

Feb 17 16:01:38 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:01:39 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 255.255.255.255: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:01:39 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.12.162: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:01:39 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.196: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:01:40 odroid-h3 kernel: BUG: kernel NULL pointer dereference, address: 0000000000000400
Feb 17 16:01:40 odroid-h3 kernel: #PF: supervisor read access in kernel mode
Feb 17 16:01:40 odroid-h3 kernel: #PF: error_code(0x0000) - not-present page
Feb 17 16:01:40 odroid-h3 kernel: PGD 0 P4D 0
Feb 17 16:01:41 odroid-h3 kernel: Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
Feb 17 16:01:41 odroid-h3 kernel: CPU: 2 UID: 0 PID: 34483 Comm: mt76-usb-rx phy Tainted: G     U     OE      6.12.14 #20
Feb 17 16:01:41 odroid-h3 kernel: Tainted: [U]=USER, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Feb 17 16:01:41 odroid-h3 kernel: Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
Feb 17 16:01:41 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 17 16:01:41 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27 48 8b 80 78 03 00 00 <48> 8b 80 00 >
Feb 17 16:01:41 odroid-h3 kernel: RSP: 0018:ffffae968acffdb8 EFLAGS: 00010202
Feb 17 16:01:41 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8fceae5e8f00 RCX: 0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: RDX: 0000000000000000 RSI: 0000000000000686 RDI: ffff8fcb7b602000
Feb 17 16:01:41 odroid-h3 kernel: RBP: ffff8fcb71bd93c0 R08: 0000000000000000 R09: 0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: R10: ffff8fceae5e8f00 R11: ffffffffaf3f7ed0 R12: ffff8fcb7357a000
Feb 17 16:01:41 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843802 R15: ffff8fcb7b602000
Feb 17 16:01:41 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8fceaff00000(0000) knlGS:0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 17 16:01:41 odroid-h3 kernel: CR2: 0000000000000400 CR3: 00000002cae22000 CR4: 0000000000352ef0
Feb 17 16:01:41 odroid-h3 kernel: Call Trace:
Feb 17 16:01:41 odroid-h3 kernel:  <TASK>
Feb 17 16:01:41 odroid-h3 kernel:  ? __die+0x23/0x80
Feb 17 16:01:41 odroid-h3 kernel:  ? page_fault_oops+0x173/0x5d0
Feb 17 16:01:41 odroid-h3 kernel:  ? exc_page_fault+0x76/0x1b0
Feb 17 16:01:41 odroid-h3 kernel:  ? asm_exc_page_fault+0x26/0x30
Feb 17 16:01:41 odroid-h3 kernel:  ? kmem_cache_alloc_noprof+0x250/0x2c0
Feb 17 16:01:41 odroid-h3 kernel:  ? mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 17 16:01:41 odroid-h3 kernel:  mt7921_queue_rx_skb+0x1df/0x9a0 [mt7921_common]
Feb 17 16:01:41 odroid-h3 kernel:  mt76u_rx_worker+0x5be/0x5f0 [mt76_usb]
Feb 17 16:01:41 odroid-h3 kernel:  ? __pfx___mt76_worker_fn+0x10/0x10 [mt76]
Feb 17 16:01:41 odroid-h3 kernel:  __mt76_worker_fn+0x4c/0x80 [mt76]
Feb 17 16:01:41 odroid-h3 kernel:  kthread+0xde/0x110
Feb 17 16:01:41 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 17 16:01:41 odroid-h3 kernel:  ret_from_fork+0x31/0x50
Feb 17 16:01:41 odroid-h3 kernel:  ? __pfx_kthread+0x10/0x10
Feb 17 16:01:41 odroid-h3 kernel:  ret_from_fork_asm+0x1a/0x30
Feb 17 16:01:41 odroid-h3 kernel:  </TASK>
Feb 17 16:01:41 odroid-h3 kernel: Modules linked in: mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_usb mt76 cmac ccm rtw_8812au(OE) rtw_8812a(OE) rtw_88xxa(OE) r>
Feb 17 16:01:41 odroid-h3 kernel:  hid_logitech_dj snd_sof x86_pkg_temp_thermal snd_sof_utils intel_powerclamp snd_soc_acpi_intel_match snd_soc_acpi kvm_intel snd_soc_core snd_comp>
Feb 17 16:01:41 odroid-h3 kernel:  nfs_acl lockd drm grace hwmon_vid fuse emc2103 coretemp 8021q sunrpc garp stp dm_mod mrp loop llc efi_pstore configfs efivarfs ip_tables x_tables>
Feb 17 16:01:41 odroid-h3 kernel: CR2: 0000000000000400
Feb 17 16:01:41 odroid-h3 kernel: ---[ end trace 0000000000000000 ]---
Feb 17 16:01:41 odroid-h3 kernel: RIP: 0010:mt792x_rx_get_wcid+0x38/0x70 [mt792x_lib]
Feb 17 16:01:41 odroid-h3 kernel: Code: 77 48 0f b7 f6 48 81 c6 84 06 00 00 48 8b 44 f7 08 48 85 c0 74 36 84 d2 75 32 f6 80 bc 00 00 00 01 74 27 48 8b 80 78 03 00 00 <48> 8b 80 00 >
Feb 17 16:01:41 odroid-h3 kernel: RSP: 0018:ffffae968acffdb8 EFLAGS: 00010202
Feb 17 16:01:41 odroid-h3 kernel: RAX: 0000000000000000 RBX: ffff8fceae5e8f00 RCX: 0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: RDX: 0000000000000000 RSI: 0000000000000686 RDI: ffff8fcb7b602000
Feb 17 16:01:41 odroid-h3 kernel: RBP: ffff8fcb71bd93c0 R08: 0000000000000000 R09: 0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: R10: ffff8fceae5e8f00 R11: ffffffffaf3f7ed0 R12: ffff8fcb7357a000
Feb 17 16:01:41 odroid-h3 kernel: R13: 0000000038010148 R14: 0000000080843802 R15: ffff8fcb7b602000
Feb 17 16:01:41 odroid-h3 kernel: FS:  0000000000000000(0000) GS:ffff8fceaff00000(0000) knlGS:0000000000000000
Feb 17 16:01:41 odroid-h3 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 17 16:01:41 odroid-h3 kernel: CR2: 0000000000000400 CR3: 0000000177fae000 CR4: 0000000000352ef0
Feb 17 16:01:41 odroid-h3 kernel: note: mt76-usb-rx phy[34483] exited with irqs disabled
[... cut ...]
Feb 17 16:02:46 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:02:50 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:02:54 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:02:58 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Feb 17 16:03:02 odroid-h3 dnsmasq-dhcp[1563]: Fehler beim Se

morrownr, 2025-02-17:
@Simon566 

> BUG: kernel NULL pointer dereference, address: 0000000000000400

Your 6.12.14 kernel doesn't seem to have the one line patch from the other thread, does it?

Simon566, 2025-02-18:
shame on me , patch was overwritten by git , i will repeat the test 

sorry

morrownr, 2025-02-18:
It happens. We might as well laugh a little. I'll bet we figure this out.

Simon566, 2025-02-19:
nothing happened since 24h with the NULL ptr patch, it runs . BUT:

- i removed the rtw88 lwfinger driver adapter , the rtl8812au one . i recognized it corrupts the kernel memory/drivers somehow . f.e. i couldnt use DHCP on clients anymore ( on mt7921u ) , when the rtw_8812au was plugged in (had also a kernel warning about it). After unplugging the OTHER adapter (rtw_8812au) and shutting down hostapd for it , the mt7921 worked. Also there seem to be some action going on in the kernel SKB buffer allocation on rtw_8812au

i will keep testing it for at least a week, maybe the rtw_8812au is the one to blame

morrownr, 2025-02-19:
> maybe the rtw_8812au is the one to blame

That driver is new. We have tested as best we can but that driver simply does not have time on its side. As more people start using it, starting with kernel 6.14 and the release of Ubuntu 25.04 and other distros with kernel 6.14, much more testing will take place.

I assume you are using lwfinger/rtw88, right? If so, you could just do a quick uninstall:

$ sudo make uninstall

And then install the following driver:

https://github.com/morrownr/8812au-20210820

So as to do a comparison. That repo is pretty darn stable. I have been maintaining it for a long time and it does AP mode really well. You should be able to do a comparison.

> there seem to be some action going on in the kernel SKB buffer allocation on rtw_8812au

When you have something to report, please do at:

https://github.com/lwfinger/rtw88

Simon566, 2025-02-21:
it looks pretty stable

6.12.14 with the NULL ptr patch + the repo for the 8812au adapter here
https://github.com/morrownr/8812au-20210820

no dmesg warnings none , just runs .....

will report back in 2-3 weeks

morrownr, 2025-02-21:
> no dmesg warnings none , just runs .....

That is what we wanted. It looks like we had 2 different things going wrong. That can make it hard to figure out. The problem with the mt7921u driver was a result of a recent patch and the dev fixed that fast. Now that you are running the out-of-kernel driver and not seeing any problems, it would be good to fix the rtw88 driver but enough fixing for now. Enjoy the setup that works and when you have time, let me know and we can get some info to the primary dev of the rtw88 driver so that it can be solid also.

Simon566, 2025-02-27:
reporting back:

- it was stable until now (around a week) , but i recognized times with no auth / no transfer, got worse over time
-  no dmesg nothing , until i tried to restart the IF with ifdown/ifup

then this dmesg appeared , but only after bringdown:

[Mi Feb 26 17:17:08 2025] mt7921u 1-4:1.3 wlx00c0cab38118: left promiscuous mode
[Mi Feb 26 17:17:08 2025] br-guest: port 2(wlx00c0cab38118) entered disabled state
[Mi Feb 26 17:17:11 2025] mt7921u 1-4:1.3: Message 00020002 (seq 5) timeout
[Mi Feb 26 17:17:12 2025] mt7921u 1-4:1.3: timed out waiting for pending tx
[Mi Feb 26 17:17:12 2025] ------------[ cut here ]------------
[Mi Feb 26 17:17:12 2025] WARNING: CPU: 1 PID: 1388904 at kernel/kthread.c:662 kthread_park+0x89/0xa0
[Mi Feb 26 17:17:12 2025] Modules linked in: cls_matchall sch_ingress sch_cake act_mirred ifb nft_masq nft_nat nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 veth wireguard libchacha20poly1305 chacha_x86_64 poly1305_x86_64 curve25519_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel pppoe nf_tables pppox crc32c_generic ppp_generic slhc nfnetlink nvme_fabrics ccm bluetooth crc16 msr bridge snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component binfmt_misc i915 snd_sof_pci_intel_icl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel mt7921u(O) soundwire_generic_allocation mt792x_usb(O) soundwire_cadence mt7921_common(O) snd_sof_intel_hda_common mt792x_lib(O) snd_soc_hdac_hda snd_sof_intel_hda_mlink mt76_connac_lib(O) snd_sof_intel_hda snd_hda_ext_core snd_sof_pci mt76_usb(O) snd_sof_xtensa_dsp mt76(O) snd_sof snd_sof_utils mac80211 snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core x86_pkg_temp_thermal snd_compress nls_ascii soundwire_bus
[Mi Feb 26 17:17:12 2025]  intel_powerclamp nls_cp437 libarc4 kvm_intel 8812au(OE) vfat snd_hda_intel cfg80211 snd_intel_dspcfg fat snd_intel_sdw_acpi kvm snd_hda_codec cec rc_core drm_buddy snd_hda_core crct10dif_pclmul drm_display_helper ghash_clmulni_intel snd_hwdep processor_thermal_device_pci_legacy processor_thermal_device ttm sha512_ssse3 mei_hdcp mei_pxp snd_pcm intel_rapl_msr processor_thermal_wt_hint sha256_ssse3 processor_thermal_rfim sha1_ssse3 processor_thermal_rapl snd_timer iTCO_wdt snd intel_rapl_common mei_me aesni_intel processor_thermal_wt_req drm_kms_helper intel_pmc_bxt gf128mul crypto_simd processor_thermal_power_floor cryptd iTCO_vendor_support mei soundcore watchdog intel_cstate processor_thermal_mbox nfsd ftdi_sio wmi_bmof pcspkr ee1004 usbserial int340x_thermal_zone rfkill i2c_algo_bit intel_soc_dts_iosf intel_pmc_core intel_vsec pmt_telemetry evdev joydev acpi_pad button pmt_class acpi_tad auth_rpcgss sg nfs_acl lockd it87(OE) hwmon_vid grace emc2103 coretemp drm 8021q sunrpc fuse garp stp mrp llc dm_mod
[Mi Feb 26 17:17:12 2025]  efi_pstore configfs loop efivarfs ip_tables x_tables autofs4 hid_logitech_hidpp hid_logitech_dj btrfs hid_generic usbhid hid xor raid6_pq libcrc32c sd_mod ahci libahci xhci_pci libata xhci_hcd r8169 nvme scsi_mod realtek usbcore mdio_devres nvme_core crc32_pclmul intel_lpss_pci crc32c_intel libphy i2c_i801 i2c_smbus intel_lpss scsi_common idma64 usb_common video wmi
[Mi Feb 26 17:17:12 2025] CPU: 1 UID: 0 PID: 1388904 Comm: kworker/u16:1 Tainted: G     U     OE      6.12.16 #21
[Mi Feb 26 17:17:12 2025] Tainted: [U]=USER, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[Mi Feb 26 17:17:12 2025] Hardware name: HARDKERNEL ODROID-H3/ODROID-H3, BIOS 5.19 07/19/2023
[Mi Feb 26 17:17:12 2025] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[Mi Feb 26 17:17:12 2025] RIP: 0010:kthread_park+0x89/0xa0
[Mi Feb 26 17:17:12 2025] Code: 5b 5d 31 f6 31 ff c3 cc cc cc cc 0f 0b 48 8b ab 58 0a 00 00 a8 04 74 ac 0f 0b b8 da ff ff ff 5b 5d 31 f6 31 ff c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb cd 0f 0b eb c7 66 2e 0f 1f 84 00 00 00 00
[Mi Feb 26 17:17:12 2025] RSP: 0018:ffffb8000642fd60 EFLAGS: 00010202
[Mi Feb 26 17:17:12 2025] RAX: 0000000000000004 RBX: ffff8fdcd9a61880 RCX: 0000000000000000
[Mi Feb 26 17:17:12 2025] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8fdcd9a61880
[Mi Feb 26 17:17:12 2025] RBP: ffff8fdd154e8d00 R08: 0000000000000000 R09: 0000000000000000
[Mi Feb 26 17:17:12 2025] R10: 0000000000000000 R11: 0000000000000000 R12: ffff8fdcc98a2060
[Mi Feb 26 17:17:12 2025] R13: ffff8fdcc98a2060 R14: ffff8fdd18ed6aa8 R15: 0000000000000100
[Mi Feb 26 17:17:12 2025] FS:  0000000000000000(0000) GS:ffff8fe02fe80000(0000) knlGS:0000000000000000
[Mi Feb 26 17:17:12 2025] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Mi Feb 26 17:17:12 2025] CR2: 00007f779d409000 CR3: 00000001322f2000 CR4: 0000000000352ef0
[Mi Feb 26 17:17:12 2025] Call Trace:
[Mi Feb 26 17:17:12 2025]  <TASK>
[Mi Feb 26 17:17:12 2025]  ? __warn+0x89/0x130
[Mi Feb 26 17:17:12 2025]  ? kthread_park+0x89/0xa0
[Mi Feb 26 17:17:12 2025]  ? report_bug+0x1a2/0x1d0
[Mi Feb 26 17:17:12 2025]  ? handle_bug+0x66/0xa0
[Mi Feb 26 17:17:12 2025]  ? exc_invalid_op+0x17/0x80
[Mi Feb 26 17:17:12 2025]  ? asm_exc_invalid_op+0x1a/0x20
[Mi Feb 26 17:17:12 2025]  ? kthread_park+0x89/0xa0
[Mi Feb 26 17:17:12 2025]  mt76u_stop_tx+0x224/0x310 [mt76_usb]
[Mi Feb 26 17:17:12 2025]  ? __pfx_autoremove_wake_function+0x10/0x10
[Mi Feb 26 17:17:12 2025]  mt7921u_mac_reset+0x72/0x1b0 [mt7921u]
[Mi Feb 26 17:17:12 2025]  mt7921_mac_reset_work+0x9f/0x190 [mt7921_common]
[Mi Feb 26 17:17:12 2025]  process_one_work+0x179/0x390
[Mi Feb 26 17:17:12 2025]  worker_thread+0x262/0x370
[... cut ...]
[Mi Feb 26 17:17:12 2025]  ? __pfx_kthread+0x10/0x10
[Mi Feb 26 17:17:12 2025]  ret_from_fork+0x31/0x50
[Mi Feb 26 17:17:12 2025]  ? __pfx_kthread+0x10/0x10
[Mi Feb 26 17:17:12 2025]  ret_from_fork_asm+0x1a/0x30
[Mi Feb 26 17:17:12 2025]  </TASK>
[Mi Feb 26 17:17:12 2025] ---[ end trace 0000000000000000 ]---
[Mi Feb 26 17:17:13 2025] mt7921u 1-4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a

[Mi Feb 26 17:17:13 2025] mt7921u 1-4:1.3: WM Firmware Version: ____010000, Build Time: 202411061510

morrownr, 2025-06-21:
Hi @mikaelhg 

It appears that you have a `NULL pointer dereference` in the file `mt792x_lib` in `mt792x_rx_get_wcid`. 

I have had trouble duplicating the problems in this issue. The things that contribute to this are:

- Network configuration
- Method used to setup AP mode

There are a lot of ways to setup the network configuration and AP mode. Without information about these topics, I do not have a chance to figure out what is causing this.