morrownr/USB-WiFi #584: [Help]: TPLINK TX50UH not working (endeavour os; Cachy os 6.13.2)

Pebrd, 2025-02-18:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux pedro-a520mhdv 6.13.2-2-cachyos #1 SMP PREEMPT_DYNAMIC Sat, 08 Feb 2025 17:46:21 +0000 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 2357:0604 TP-Link TP-Link UB500 Adapter Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 005 Device 002: ID 1d57:fa60 Xenta 2.4G Wireless Device Bus 005 Device 003: ID 28bd:0928 XP-Pen ACD02 Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 007 Device 003: ID 1397:0508 BEHRINGER International GmbH UMC204HD 192k Bus 007 Device 004: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode) Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no

### dkms

```text
nvidia/570.86.16, 6.13.2-2-cachyos, x86_64: installed
nvidia/570.86.16, 6.13.2-arch1-1, x86_64: installed
nvidia/570.86.16, 6.13.2-zen1-1-zen, x86_64: installed
zenpower/0.1.12, 6.13.2-2-cachyos, x86_64: built
zenpower/0.1.12, 6.13.2-arch1-1, x86_64: installed (Differences between built and installed modules)
zenpower/0.1.12, 6.13.2-zen1-1-zen, x86_64: built
zenpower3/0.2.0, 6.13.2-2-cachyos, x86_64: installed
zenpower3/0.2.0, 6.13.2-arch1-1, x86_64: installed (Original modules exist)
zenpower3/0.2.0, 6.13.2-zen1-1-zen, x86_64: installed
```

### iw

```text
None
```

### What happened?

I bought the TX50UH since it said that it worked by creating the udev rules, tried it and It didnt work, I am a begginer linux user so dont really know how to fix it.  what I do think its related to the problem its that it says it on Driver cdrom mode `"Bus 007 Device 004: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)" `

Pebrd, 2025-02-18:
Also, then I turn on the pc with the adapter plugged in it tries to boot on it i think because systemd-boot doesnt show my kernel options util i unplug it or close to a minute passes.

bjlockie, 2025-02-18:
The adapter defaults to cdrom mode.
Turn off booting from cdrom in your computer bios.

Pebrd, 2025-02-18:
> The adapter defaults to cdrom mode.
> Turn off booting from cdrom in your computer bios. 

to solve the booting issue or thw adapter not working?

I already solved the booting problem by changing the boot options on my bios

morrownr, 2025-02-18:
Hi @Pebrd 

Let's see what is wrong. I hope I have not made a mistake in The Plug and Play List.

The Plug and Play List shows the following adapter: TP-Link TXE50UH

You posted that you have the TPLINK TX50UH

Those are two different adapters. Notice the TXE in the name of the one in the list. Here is an article that I found:

https://www.sindastra.de/p/3313/triband-wi-fi-hacking-on-linux-with-tp-link-tx50uh

It appears the TX50UH uses a Realtek rtl8852/32cu chip. While I do have a driver for that chip up here, my recommendation is for you to send the adapter back to get your money back and then search for a TP-Link TXE50UH.

Here is the link to the driver for the adapter that you have:

https://github.com/morrownr/rtl8852cu-20240510

Unfortunately it is an out-of-kernel driver and, like most adapters with the rtl8852/32cu chip, your adapter is a multi-state (windows driver onboard) variety which can be problematic. You likely will not be happy with this TX50UH adapter over the long run as it will be problematic.

I am sorry this happened to you and I have posted a WARNING about this situation in the listing in The Plug and Play List.

@morrownr

Pebrd, 2025-02-18:
> Hi @Pebrd
>
> Let's see what is wrong. I hope I have not made a mistake in The Plug and Play List.
>
> The Plug and Play List shows the following adapter: TP-Link TXE50UH
>
> You posted that you have the [...]

Hi! thanks for answering!

damn, its indeed a TX50UH, my bad for not reading well hahaha

Thanks for leaving the link for the driver! ill see how it works and if I cant make it work i'll send it back sadly. :)

morrownr, 2025-02-18:
> and if I cant make it work i'll send it back sadly. :)

There is nothing wrong with playing with the TX50UH, however, I make it clear in the README of the driver that I do not recommend that users take the availability of the out-of-kernel driver to be an invitation to buy adapters with the chip. It may be a long time before good support is available, if ever. Realtek does USB Linux support poorly. Here is a post that I keep up for users:

https://github.com/morrownr/USB-WiFi/issues/449

I think you will be much more happy with a TXE50UH. Good luck.

Pebrd, 2025-02-18:
> 
> I think you will be much more happy with a TXE50UH. Good luck.

Sadly I cant find it in my country so I'll just refund.

Thanks for your time! :)

morrownr, 2025-02-18:
I have lived in several countries so I understand. I can help you find a good adapter if you want to give me a link to where you shop. Are you looking for a WiFi 6e adapter?

Pebrd, 2025-02-20:
> I have lived in several countries so I understand. I can help you find a good adapter if you want to give me a link to where you shop. Are you looking for a WiFi 6e adapter?

Hi! yes ofc I'd be amazing if you can helpe

https://www.hardgamers.com.ar/search?text=wifi

here it compiles the catalog of most of the pc stores on my country so if it doesnt show there its because probably nobody sells it here.

im mainly looking for an adapter with 5.0ghz because im literally above the router but cant use an ethernet.

morrownr, 2025-02-20:
@Pebrd 

I used to live in BA.

I'm looking at the list right now.

morrownr, 2025-02-20:
Let me ask this:

Your WiFi router, what is the capability?

AC600, AC1200, AC1900, AC2600, AX3000 ?

Also, what Linux distro do you use?

The adapters that I am looking at right now are:

1. TP-LINK ARCHER T2U - https://www.hardgamers.com.ar/product/1056135760 - AC600 dual band - either rtl8811au or mt7610u
2. TP-Link Archer T4U - https://www.hardgamers.com.ar/product/-31850789 - AC1200 dual band - rtl8812bu
3. TP-LINK ARCHER T9UH - https://www.hardgamers.com.ar/product/-1765572853 - AC1900 dual band - rtl8814au

I generally try to avoid TP-Link as they make it hard for Linux users to determine the chips in their adapters. However, it looks like you do not have a lot of options. The 3 options above are adapters that support the 5 GHz band and I am reasonably sure that we can make sure you have good drivers. They use Realtek chips instead of Mediatek but it is what is. 

1 and 2 will have good in-kernel Linux drivers with kernel 6.14 and later. 3 will likely have good in-kernel driver with kernel 6.15 or later. In the meantime, if you are using an earlier kernel, which you probably are, I can give you a link to drivers that you can install until you are using kernel 6.14 or later which will be in Ubuntu 25.04.

Pebrd, 2025-02-20:
> Let me ask this:
>
> Your WiFi router, what is the capability?
>
> AC600, AC1200, AC1900, AC2600, AX3000 ?
>
> Also, what Linux distro do you use?
>
> The adapters that I am looking at right now are:
>
> 1. [...]

Im running endeavour os, the kernel is 6.13, i dont really know my routers capabilities, anyways out of those 3 im going to buy the one you recommend me the most/is the strongest, there arent much difference between them price-wise sook

morrownr, 2025-02-20:
Okay, so you are looking to buy the T9UH. That adapter uses the rtl8814au chip.

The in-kernel driver is just now going in so you will probably want to install and use the following driver until the kernel you are using has the in-kernel driver:

https://github.com/lwfinger/rtw88

Pebrd, 2025-02-21:
> Okay, so you are looking to buy the T9UH. That adapter uses the rtl8814au chip.
> 
> The in-kernel driver is just now going in so you will probably want to install and use the following driver until the kernel you are using has the in-kernel driver:
> 
> https://github.com/lwfinger/rtw88

Amazing! Tysm

I'll send back the other one and buy that, the store where I bought it has it too.

ps: if u can write a warnig in the TX50UH section for people to check that its trully TX and no TXE lmao

Amazing project!

morrownr, 2025-02-21:
>  if u can write a warnig in the TX50UH

Yes, I wrote the following warning:

WARNING WARNING WARNING: TP-Link makes an adapter with a very similar name, TX50UH, but it uses a rtl8852/32cu chip and is a multi-state (windows driver onboard) adapter. That is not what you want. Use care to ensure that you order a TP-Link TXE50UH

> Amazing project!

USB WiFi needed help. I am not the only one doing things to help it along on Linux. If fact, USB WiFi support on Linux is good these days. The only areas to stay away from are the adapters based on Realtek WiFi 6 and 7.

You are welcome to stop to see what is going on whenever you want. The direct link to the Main Menu is:

https://github.com/morrownr/USB-WiFi

morrownr, 2025-02-25:
@Pebrd 

https://github.com/lwfinger/rtw88

I need to give you the story. That repo is what we have been using over the last more than a year to update and add Realtek WiFi 5 drivers to the linux kernel. The T4U uses the rtl8812bu chip which is a AC1200 class chip. The above repo actually has a lot of drivers in it and the one for the rtl8812bu is one of them.

Once you are using kernel 6.14 or later, you should not need to the above driver. The Linux kernel will have the driver inside so the adapter will be plug and play. Ubuntu 25.04, for example, will ship with the 6.14 kernel in April. Let me know if you have any problems.

Pebrd, 2025-02-25:
@morrownr 

Hi! sorry to bother once again but if I dont ask you I dont know how to solve all of this :)

It arrived an hour ago, I installed the driver, it works like a charm, I moved my wifi mouse and drawing tablet to the 2.0 ports since I read that plugin them in 3.0 ports can create interference, should I do the same with my bt dongle?

I did have problem tho, it disconnected idk why, happened a few times and the pc feels a bit laggy at first but then it gets working right, can it be related to the driver? also, the connecting/disconnecting (the adapter kinds of stop wirking) its normal? or is it related to my pc?

morrownr, 2025-02-25:
@Pebrd 

> It arrived an hour ago, I installed the driver, it works like a charm, I moved my wifi mouse and drawing tablet to the 2.0 ports since I read that plugin them in 3.0 ports can create interference, should I do the same with my bt dongle?

Bluetooth is inherently a USB 2 technology. It is also a 2.4 GHz technology. Therefore I recommend:

Plug your new USB WiFi adapter in a USB 3 port but keep it and the USB BT adapter far away from each other, especially if using 2.4 GHz with WiFi. I actually use an extension cable to make sure they are far apart. You may have to test to see what works best.

Here is a pinned issue where I explain how USB problems can be solved. 

 https://github.com/morrownr/USB-WiFi/issues/508

I'd say your problem is USB not WiFi. Let me know how it goes.

Pebrd, 2025-02-26:
> @Pebrd
>
>
> Bluetooth is inherently a USB 2 technology. It is also a 2.4 GHz technology. Therefore I recommend:
>
> Plug your new USB WiFi adapter in a USB 3 port but keep it and the USB BT adapter far [...]

I tried unplugging everything but the adapter and my sound card, it didnt work, the adapter keeps connecting and disconnecting itself, im trying to rebuild the driver and after that tryin in various kernels, also the computer keeps running slow even when there is no process running

Regarding the issue 508 I couldnt find any of those options on my bios (Ive an Asrock A520M-HDV)

How should I proceed? (Im trying to get an usb extender tomorrow too)

morrownr, 2025-02-26:
@Pebrd 

>  (Im trying to get an usb extender tomorrow too)

Recommend against this. We can discuss this later.

Regarding the less than great performance:

Install an app called `wavemon"

In Debian, Ubuntu or other Debian based distros:

$ sudo apt install wavemon

Then, while still in the terminal, start `wavemon`

$ wavemon

There should be two things that you can post here:

link quality
signal level

That information will help us see what to do.

Pebrd, 2025-02-26:
> @Pebrd 
> 
> >  (Im trying to get an usb extender tomorrow too)
> 
> Recommend against this. We can discuss this later.

im using one rigth now and for some reason kind of fixed the connection/disconnection, it appears like the adapter was loose when plugged in the motherboard usb (idk why because the mother is less than 6 months new

> 
> Regarding the less than great performance:
> 
> Install an app called `wavemon"
> 
> In Debian, Ubuntu or other Debian based distros:

im running arch based, I'll search for it tho

> 
> $ sudo apt install wavemon
> 
> Then, while still in the terminal, start `wavemon`
> 
> $ wavemon
> 
> There should be two things that you can post here:
> 
> link quality

above 95% all the time

> signal level

between -40 and -43
> 
> That information will help us see what to do.

Regarding the cachyos kernel update, now it works ok

might be that the extender its not usb 3.0? anyways when I plugged the adapter in a usb 2.0 it had the same problems but now it doesnt with the extender on 

I disabled powersaving for the usb adapter, also I ran sudo modprobe rtw_8822bu

morrownr, 2025-02-27:
> im using one rigth now and for some reason kind of fixed the connection/disconnection

That appears to indicate a less than optimal situation straight line between the adapter and the wifi router.

I guess I will be quiet and not mess with success but extenders tend to cut down on bandwidth. Can you send me a link to the extender?

> im running arch based

Arch should have `wavemon` available for installation.

> between -40 and -43
-50 on 5.0

-40 to -43 is good. -50 is not bad but not as good. What does `on 5.0` mean?

> 5.0 still doesnt work right

What is this?

Pebrd, 2025-02-28:
> That appears to indicate a less than optimal situation straight line between the adapter and the wifi router.
>
> I guess I will be quiet and not mess with success but extenders tend to cut down on [...]

now im using one i had, suoer generic lmao
> 
> > im running arch based
> 
> Arch should have `wavemon` available for installation.
> 
> > between -40 and -43
> -50 on 5.0
> 
> -40 to -43 is good. -50 is not bad but not as good. What does `on 5.0` mean?
> 
> > 5.0 still doesnt work right
> 
> What is this?
5.0 ghz

morrownr, 2025-02-28:
> 5.0 ghz

What country are you in?

What do you get with the following command?

$ iw reg get