morrownr/USB-WiFi #585: [Help]: RTL8812AU Kali 2024

nonobike, 2025-02-24:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux kali 6.1.92-v8+ #1 SMP PREEMPT Tue Jun  4 14:30:43 UTC 2024 aarch64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz] Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub                                                               

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8: added
```

### iw

```text
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr ba:27:eb:74:f2:6c
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:74:f2:6c
		type managed
		channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz
		txpower 31.00 dBm
```

### What happened?

the https://docs.alfa.com.tw/Support/Linux/RTL8812AU/ are not compatible on kali 2024

morrownr, 2025-02-25:
Hi @nonobike 

> the https://docs.alfa.com.tw/Support/Linux/RTL8812AU/ are not compatible on kali 2024

I'll contact Alfa and help them update their page. In the meantime, let's get you going.

You have arm64 hardware. Can you be more specific about what hardware you are using?

It appears that you are using Kali. This is Kali for RasPi? Or what?

You appear to have a mess on your hands. We need to clean that up before making any progress so please answer the above questions. Oh, and your adapter is based on the rtl8811au chip, not the rtl8812au. Which Alfa adapter are you using?

nonobike, 2025-03-07:
@morrownr have you a solutions ?

thanks

morrownr, 2025-03-07:
> have you a solutions ?

Yes

> awus036acs

I have one of those adapter. Good little adapter.

The first thing that needs to happen is that you need to remove any previous drivers that you may have installed. You may need to refer to the instructions for each driver you have tried to install.

You can post the results of the following command to help me help you:

$ dkms status

Once we have all previous drivers purged, please go to the following repo and follow the installation instructions:

https://github.com/lwfinger/rtw88

The instructions for the RasPi are for the RasPiOS so if you are using Kali Arm64 on a RasPi5B, please tell me if this is the case so I can try to help as the instructions will not work for Kali. I'll need to know exactly what version of Kali for RasPi you are using.

FYI: The rtw88 driver i pointed you to above is a work and test repo where the results go into the Linux kernel. In fact, the new driver for the rtl8811au went into kernel 6.14 so you should be plug and play once you are using kernel 6.14 or later.

nonobike, 2025-03-09:
> You can post the results of the following command to help me help you:
> 
> $ dkms status

/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8: added
rtl8821cu/5.12.0.4: added

> 
> FYI: The rtw88 driver i pointed you to above is a work and test repo where the results go into the Linux kernel. In fact, the new driver for the rtl8811au went into kernel 6.14 so you should be plug and play once you are using kernel 6.14 or later.

in lsusb the adapter is seen but it is not mounted as a network card. if config returns me lo0 and eth0 only

morrownr, 2025-03-12:
> /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory

> realtek-rtl8814au/5.8.5.1~git20240527.d8208c8: added

> rtl8821cu/5.12.0.4: added

There are 3 things that need to be cleaned up. Do you know how to clean them up?

> Linux kali 6.1.92-v8+ #1 SMP PREEMPT Tue Jun 4 14:30:43 UTC 2024 aarch64 GNU/Linux

Kali for RasPi is one of my biggest nightmares around here. Do you have to use Kali on your RasPi?

> if config returns me lo0 and eth0 only

Please discontinue using `iwconfig` as it has been superseded by `iw` and `ip`.

My recommendation is to burn a new sd card with the latest 64 bit RasPiOS. Once the hardware is setup, you can install the apps you were going to use on Kali. Be glad to help if you go that direction.

nonobike, 2025-03-12:
> > /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
> 
> > realtek-rtl8814au/5.8.5.1~git20240527.d8208c8: added
> 
> > rtl8821cu/5.12.0.4: added
> 
> There are 3 things that need to be cleaned up. Do you know how to clean them up?

I installed a new Kali without any packages

> 
> > Linux kali 6.1.92-v8+ [#1](https://github.com/morrownr/USB-WiFi/issues/1) SMP PREEMPT Tue Jun 4 14:30:43 UTC 2024 aarch64 GNU/Linux
> 
> Kali for RasPi is one of my biggest nightmares around here. Do you have to use Kali on your RasPi?

yes i have a https://kali.download/arm-images/kali-2024.4/kali-linux-2024.4-raspberry-pi5-arm64.img.xz
> Please discontinue using `iwconfig` as it has been superseded by `iw` and `ip`.
>
> My recommendation is to burn a new sd card with the latest 64 bit RasPiOS. Once the hardware is setup, you can [...]

I installed a new raspi os 64 (6.6.74) but the adapter is seen in lsusb but does not work (same on kali)

morrownr, 2025-03-12:
> I installed a new raspi os 64 (6.6.74) but the adapter is seen in lsusb but does not work (same on kali)

Okay. For now, let's work with an sd that has a clean RasPiOS 64 bit installation. As we learn and get the adapter going, you can also try on Kali. I regularly use RasPiOS.

I just reread this thread for details. One detail that is not clear to me is which driver that you are trying to install. Can you tell me and be specific?

You say that the doe not work. What driver did you install?

FYI: You can see devices with `lsusb` without a driver installed. If a device is plugged in, it should show with `lsusb`.

nonobike, 2025-03-13:
> > I installed a new raspi os 64 (6.6.74) but the adapter is seen in lsusb but does not work (same on kali)
> 
> Okay. For now, let's work with an sd that has a clean RasPiOS 64 bit installation. As we learn and get the adapter going, you can also try on Kali. I regularly use RasPiOS.
yes :) 

> 
> I just reread this thread for details. One detail that is not clear to me is which driver that you are trying to install. Can you tell me and be specific?

none I am on a new installation

> 
> You say that the doe not work. What driver did you install?
> 
> FYI: You can see devices with `lsusb` without a driver installed. If a device is plugged in, it should show with `lsusb`.

it's work ! 
Bus 001 Device 002: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]

morrownr, 2025-03-13:
G'day @nonobike 

> Bus 001 Device 002: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]

That is not the driver. That is the output of `lsusb` and that is totally different than a driver.

An example of telling me what driver you are using would be:

https://github.com/lwfinger/rtw88

or

https://github.com/morrownr/8821au-20210708

You can check to see if the driver is working by running:

$ iw dev

You should get a wireless interface if the driver is working.

nonobike, 2025-03-14:
> FYI: The rtw88 driver i pointed you to above is a work and test repo where the results go into the Linux kernel. In fact, the new driver for the rtl8811au went into kernel 6.14 so you should be plug and play once you are using kernel 6.14 or later.

I did not install a driver following the previous message

iw dev = just wlan0 ( wifi on pi ) 

install : https://github.com/lwfinger/rtw88

it's WORKS !! 

I try on kali, I'll come back to you

morrownr, 2025-03-14:
Good. We seem to be making some progress.

nonobike, 2025-03-15:
On KALI : 
- Installation Guide
- Prerequisites : 

- Ubuntu => sudo apt install linux-headers-$(uname -r) build-essential git
Result : Error: Unable to locate package linux-headers-6.6.63-v8
Error: Couldn't find any package by glob 'linux-headers-6.6.63-v8'

- Raspberry Pi OS => sudo apt install -y raspberrypi-kernel-headers build-essential git
Result : Error: Unable to locate package raspberrypi-kernel-headers

morrownr, 2025-03-15:
That message is a little confusing.

You say `On KALI` but then you show me errors for Ubuntu and RasPiOS. I thought you had a clean RasPiOS installation and you had success installing the rtw88 driver. What is the purpose of the above message?

nonobike, 2025-03-15:
> That message is a little confusing.
> 
> You say `On KALI` but then you show me errors for Ubuntu and RasPiOS. I thought you had a clean RasPiOS installation and you had success installing the rtw88 driver. What is the purpose of the above message?

There's no Kali build on the rtw88 GitHub.
So I tried Ubuntu and raspi OS, but neither worked.

morrownr, 2025-03-15:
Okay, so, you looked for KALI installation information in:

rtw88 > README > Installation Guide > Prerequisites

Right?

Before we continue, let me warn you about what you did:

SUDO is a weapon on mass destruction on Linux. Do not use SUDO with a command unless you know exactly what you are doing. SUDO gives the command ADMIN rights and it can do anything it wants... including the destruction of your installed distro. More often it messes something up and you are left not knowing why something does not work correctly.

We can find the correct way to do this on Kali but it may be different depending on exactly what version of Kali that you are using. You are using a RasPi5B system, right? There is a Kali for RasPi and a Kali for RasPi5B. Are you using Kali for RasPi5B or Kali for RasPi?

Andy-45, 2025-03-15:
Hey nonobike, 
if you want to continue with KALI for Raspi 5, please read this: https://www.kali.org/docs/arm/raspberry-pi-5/

nonobike, 2025-03-17:
> Hey [@nonobike](https://github.com/nonobike) if you want to continue with KALI for Raspi 5, please read this: https://www.kali.org/docs/arm/raspberry-pi-5/

hi, i try 

 sudo apt install --no-install-recommends realtek-rtl88xxau-dkms

but i have 

dpkg: error processing package realtek-rtl88xxau-dkms (--configure):
 dependency problems - leaving unconfigured
Processing triggers for kali-menu (2024.4.0) ...
Processing triggers for man-db (2.13.0-1) ...
Errors were encountered while processing:
 realtek-rtl8814au-dkms
 realtek-rtl88xxau-dkms
Error: Sub-process /usr/bin/dpkg returned an error code (1)

nonobike, 2025-03-17:
> Okay, so, you looked for KALI installation information in:
> 
> rtw88 > README > Installation Guide > Prerequisites
> 
> Right?

Yes 
> Before we continue, let me warn you about what you did:
>
> SUDO is a weapon on mass destruction on Linux. Do not use SUDO with a command unless you know exactly what you are doing. SUDO gives the [...]

I use a raspi 5 and i download an .img for raspi 5
He dont propose an .img for pi5b

morrownr, 2025-03-17:
Okay, so you went to the following page and downloaded the .img for RasPi5.

https://www.kali.org/get-kali/#kali-arm

You may not understand why I am pushing to find out exactly which .img you installed but it has to do with the installation of the header files. How to do that may be different on each .img. You may need to ask in a Kali forum how to install the header files but you have to be very specific which Pi you have and exactly which Kali .img you are using.

nonobike, 2025-03-20:
> Ok, vous êtes donc allé sur la page suivante et avez téléchargé le .img pour RasPi5.
> 
> https://www.kali.org/get-kali/#kali-arm
> 
> Vous ne comprenez peut-être pas pourquoi je vous demande de préciser quel fichier .img vous avez installé, mais cela concerne l'installation des fichiers d'en-tête. La procédure peut varier d'un fichier .img à l'autre. Vous devrez peut-être demander sur un forum Kali comment installer les fichiers d'en-tête, mais il est important de préciser le type de Pi que vous possédez et le fichier .img Kali que vous utilisez.

Hello 

It's exactly

Andy-45, 2025-03-21:
@nonobike 

I have the same adapter - and meanwhile I can confirm it does not work with Kali for RPi5 

but if you go back to Raspi OS then it is easy to get the adapter working, this is your driver:

   https://github.com/morrownr/8821au-20210708

just follow the instructions

-- the result on my Pi 5 :

> Linux rpi5-1 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
> 
> root@rpi5-1:/home/pi# lsusb
> 
> 
> > Bus 003 Device 002: ID 0bda:0811 Realtek Semiconductor Corp. Realtek 8812AU/8821AU 802.11ac WLAN Adapter [USB Wireless Dual-Band Adapter 2.4/5Ghz]
> 
> 
> > /:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
> > /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
> >     |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl8821au, 480M
> > /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
> > /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
> > 
> 
>