morrownr/USB-WiFi #586: USB subsystem crash with Edup EP-1673

tom709, 2025-02-25:
I have the Edup EP-1673 for 2 months now and in that time have seen two complete crashes of the USB subsystem; disabling keyboard, mouse, everything. A reboot was not enough to fix the issue, I had to power down the machine. This does not have to be caused by the Edup EP-1673, but I have had the machine for a year before and never had issues.
Has anyone seen anything similar?
My Kernel is 6.12 on Debian Bookworm.

morrownr, 2025-02-25:
Hi @tom709 

This does appear to be USB related but then it could be one of several things. Here is a link to a pinned issue in:

USB-WiFi / issues

https://github.com/morrownr/USB-WiFi/issues/508

I am still working on the post as more information about the BIOS settings need to be added but it may give you some things to try.

Also, if I could get you to move back to kernel 6.6 to see if the problem is there also, that could help narrow down specifically what your problem is.

Regards

tom709, 2025-12-27:
Hi @morrownr,

Thanks for the advice. I tried following things during the past year, but the USB subsystem still crashes often when using the Edup EP-1673:
- disabled USB legacy support
- disabled xhci handoff
- disabled: CPU integ gfx
- upgraded to Debian Trixie/kernel 6.12
- various changes in cabling, connecting the WIFI stick in different ways as well as recabling keyboard/mouse/USB hub

Now my guess is, this could be related to electrostatic discharge. It seems to happen when I get up from the computer; then when I come back sometimes just 1 min later, all USB equipment is dead, but the computer still continues to work, e.g. playing music.

What could I possibly do against electrostatic discharge? Ground the antenna? :)

morrownr, 2025-12-27:
Hi @tom709 

> Now my guess is, this could be related to electrostatic discharge.

This could be the case. Not easy to troubleshoot. Some electrical system is some older homes/apartments do not wire the ground. It could also be in the system. I have equipment to check grounding. Do you have a friend that is an electrician?

Have you tried the USB adapter with a different system? Maybe with a laptop.

tom709, 2026-01-06:
There is a pair of USB 3.0 ports at the front of the PC chassis and they frequently get into physical contact with synthetic fabric during operation. I found out that the shields of these USB 3.0 ports is not grounded, while the shields of all other USB ports, including those on the mainboard itself, are in fact grounded. Online research showed that there is a controversy among hardware folks on whether or not to ground USB port shields. I now added grounding to those front port shields and on first inspection all seems to be working fine.