morrownr/USB-WiFi #588: [News] Alfa AXML and AXM appear to be on the way to good health again...

morrownr, 2025-03-15:
G'day all,

For a considerable part of 2024, we saw a very annoying bug that affected the Alfa AXML and AXM. The bug would simply stop wifi from working. There have been some workarounds but I removed both adapters from The Plug and Play List because it was not clear when a fix would be available. To complicate things, it appeared that a problem in bluetooth was the cause.

I have monitored this problem and even investigated it myself. I was aware of Mediatek devs working the issue. It was not an easy fix. I knew some fixes were coming.

I downloaded Ubuntu 25.04 (in testing) yesterday to test some things and my AXML and AXM just worked. I had not added any workarounds. They just worked. For others that maybe interested, Ubuntu moved to the 6.14 kernel about a week ago and it looks like kernel 6.14 rc6 is currently in use. I plan to continue testing and will post any additional observations in this issue. If you have either of these adapters, please post your experiences as well.

Regards,

@morrownr

stealth27374, 2025-04-03:
tumbleweed just got  6.14  ,  i removed  modprobe.blacklist=btusb from grub.  
detected and running fine as client. "so far".    txpower still reporting 3.

Ray78PM, 2025-04-05:
Hi all. 

I have done a couple of tests. Installed Ubuntu 25.04 (kernel 6.14) as both a VM and then a rpi4 install. Installed aircrack-ng on both.  AXML on kernel 6.14 is capable of going into monitor mode and packet injection is working. However...

1) For the VM test, I believe I am having issues getting AXML working due to the host machine which is an Ubuntu 24.10 install (kernel 6.11) running virtualbox (current version including extension pack). VM will not recognise AXML nor will the host machine (as expected). Will probably just wait a couple of weeks for the official release of 25.04 and upgrade the host machine.

2) rpi4: AXML is working on the rpi4, plug and play. There were no issues starting AXML, bit of a delay, but had issues with changing from AXML to another usb adapter. rpi4 did not recognise the next usb adapter I tried. A reboot was done (solved :) ). 

On a side note, kernel 6.14 was tested with a Netgear A8000 (mt7921au) and Alfa AWUS036ACH (rtl8812au) plug and play. mt7921au worked flawlessly as expected. rtl8812au worked perfectly on the rpi4, but was buggy on the VM probably due to host (as above). Frequent adapter disconnects/reconnects on the VM side on startup. Monitor mode and packet injection worked for both adapters once they were up. Great to see the AWUS036ACH working on the current kernel and many thanks for all your work!

Cheers

Ray

morrownr, 2025-04-05:
@stealth27374 @Ray78PM 

Thanks for the information.

> txpower still reporting 3.

Disregard that for now. The report is not correct information. The numbers in use, that we can't see, are correct. There is work to fix that but it is cosmetic.

> Installed aircrack-ng on both. AXML on kernel 6.14 is capable of going into monitor mode and packet injection is working.

Side note so that you know: We should have WiFi 7 USB WiFi adapters reasonably soon. aircrack-ng won't work with WiFi 7 adapters or cards. The Linux kernel has code in it these days to stop the function of any WiFi 7 wireless device if it asks for WEXT support. That is the old depreciated wireless api. Many parts of aircrack-ng use WEXT. Some other wireless apps and utilities do as well.

> the host machine which is an Ubuntu 24.10 install (kernel 6.11) running virtualbox (current version including extension pack).

6.11 likely still has the bug so counting on it to work is probably a futile effort.

> mt7921au worked flawlessly as expected.

The problematic bug is only found in the AXML and AXM. The Netgear adapter has never been a problem.

> rtl8812au worked perfectly on the rpi4, but was buggy on the VM probably due to host (as above).

The new rtl9912au driver was not in kernel 6.11. It is new with kernel 6.14.

If you need or want to install the new driver on older kernels go to:

https://github.com/lwfinger/rtw88

> Monitor mode and packet injection worked for both adapters once they were up.

Good to hear.

Keep me posted with addition information as you continue.

morrownr, 2025-04-23:
@Ray78PM 

Question: Does your Alfa AXML have bluetooth support? If so, is bluetooth working on your RasPi4B?

I have been testing the AXML and AXM on RasPiOS 64 bit with kernels 6.6, 6.12 and 6.14. WiFi works fine but bluetooth is not. I also tested both adapters with Ubuntu 24.10 with kernels 6.11, 6.12 and 6.14 and bluetooth works fine. I'm not much of a bluetooth expert but I am having a hard time figuring out why bluetooth is working on Ubuntu but not on RasPiOS. Any thoughts?

Ray78PM, 2025-04-24:
@morrownr I am running Ubuntu 25.04 development branch headless/server on my rpi4 (kernel 6.14.0-1005-raspi). It seems stable, but in saying that I was not attempting to use the AXML's bluetooth in an earlier test and only installed bluez today to test this. 

On today's boot with AXML attached there were two errors at login regarding this device (hci1) that may or may not help your case: "Bluetooth hci1: Failed to read MSFT supported features (-100)" followed by "Bluetooth hci1: AOSP get vendor capabilities (-100)". 

I ran a full-upgrade and installed bluez. hci1 came up as hoped, scanned with bluetoothctl ok and there appears to be no conflicts/issues with the rpi4's inbuilt bluetooth adapter either (hci0). Rebooted and only one error at login: "Bluetooth hci1: AOSP get vendor capabilities (-100)". AXML bluetooth is working after reboot.

The wifi adapter seems to be functioning fine as well with bluetooth up and is going into monitor mode. Airodump-ng seems stable. Might do some more testing over the weekend and report back if there are any issues. I also upgraded to Ubuntu 25.04 on my main machine and appears to be working well for both wifi and bluetooth. I would usually  turn to you for advice haha. I hope this helps...

morrownr, 2025-04-24:
Good day @Ray78PM 

> only installed bluez today to test this.

Thanks for doing this. I am working on a project where bluetooth was required and it was not working on the AXM and AXML. This was on the RasPiOS 64 bit on a RasPi4B. I generally just work on wifi and not bluetooth so I was pulling my hair out. I found out yesterday after contacting a couple of folks that RasPiOS does not compile CONFIG_BT_HCIBTUSB_MTK ( .config ) so it could not work until I compiled a custom kernel. 

> The wifi adapter seems to be functioning fine as well with bluetooth up and is going into monitor mode. 

With the patches that went in earlier this year, I am not seeing any problems with either the AXML or AXM. I've been testing to see if it is a good idea to put both adapters back in The Plug and Play List. What do you think?

> Airodump-ng seems stable.

Something to keep in the back of your mind, since we will have mt7925 based WiFi 7 adapters soon, Aircrack-ng uses WEXT (the old Wireless Extensions) and the linux kernel has code to keep WEXT from working with WiFi 7. I was helping a guy with a mt7925 based M.2 card recently and he was complaining about the driver not supporting something. I have a m.2 card based on the mt7925 and when I tested, boom, right in demsg, it said no no no. So either Aircrack-ng will have to be updated or ? It is not a problem with modern supported tools like Wireshark.

> I would usually turn to you for advice haha.

I think we all end up helping each other. Thanks for your help. 

Regards

Ray78PM, 2025-04-25:
_I am not seeing any problems with either the AXML or AXM. I've been testing to see if it is a good idea to put both adapters back in The Plug and Play List. What do you think?_

Seems promising to include these two in the list. With Kali 2025.2 just around the corner it might be worth holding off until then. My 2c anyway.

Cheers

Ray

morrownr, 2025-04-25:
@Ray78PM 

> With Kali 2025.2 just around the corner it might be worth holding off until then. 

That is reasonable. I'm trying to make sure everything is good to go before adding them to the list. Would you mind testing with Kali 2025.2 when it is released? Then you could let me know how it goes and what your final recommendation is.

Thanks

Ray78PM, 2025-06-27:
> [@Ray78PM](https://github.com/Ray78PM)
>
>
> That is reasonable. I'm trying to make sure everything is good to go before adding them to the list. Would you mind testing with Kali 2025.2 when it is [...]

@morrownr have not forgotten your request. 2025.2 released a couple of weeks ago with kernel 6.12.25. My AXML adapter is not stable with Kali but still working fine with Ubuntu (6.14.0-22). Hard to recommend yet as a plug-and-play pentesting adapter until the Kali devs update to the current kernel I guess. I doubt Parrot will be any different. Will keep you posted.

ManuHry, 2025-06-27:
Hi and thanks for these comments! As latest version of OpenWrt (24.10.2) ships kernel 6.6.93, I suppose Alfa AXM(L) adapters are still not a good choice for this purpose?
I would be glad to publish iperf outputs and my feelings about one of these adapters once they will be considered enough stable to buy.

ManuHry, 2025-06-27:
Oh thank you @morrownr I thought drivers were in-kernel for OpenWrt. If you have some spare time to test I would be interested!

To be honest I am considering to buy the cheapest one for my current needs (and because of upcoming Wi-Fi 7 adapters as you mentionned).
In Europe AXM costs around $45 including shipping. And $70 for AXML. I will let you know once received.

morrownr, 2025-06-28:
@ManuHry 

> I thought drivers were in-kernel for OpenWrt. 

OpenWRT is a very specialized Linux distro. It has to work with extremely limited resources so it cannot be made like the general purpose desktop/laptop distros. There is no USB WiFi drivers in the basic distro for any hardware. That is because no routers/boards/systems come with USB WiFi adapters as standard equipment. However, for Mediatek based adapters, the drivers simply have to be installed via the package manager or Luci which is easy to do.

> If you have some spare time to test I would be interested!

Please give me the details of what you want me to test. Please be specific.

> To be honest I am considering to buy the cheapest one for my current needs (and because of upcoming Wi-Fi 7 adapters as you mentionned).

This is a good strategy. The Plug and Play List has some good WiFi 5 and 6 adapters, some of which are very low cost. If you tell me what your use case is and what you want to spend then maybe I can point you to some adapters that may work for you.

ManuHry, 2025-06-29:
@morrownr this makes sense, after all images are stripped down to only few megabytes for low-compute devices.
Could you try to use your AXML adapter in AP mode with OpenWrt 24.10.2, and perform some IPv4 iPerf3 tests for a duration of about 5 minutes from a Wi-Fi 6/6E/7 client? I read [some old reports of failure under high load](https://forum.openwrt.org/t/alfa-awus036acm-mt7612u-stops-working-under-heavy-load-in-rpi-4/170441).

I have a spare Raspberry Pi 4 with an up-to-date OpenWrt already installed. I use it for now as a [dumb AP](https://openwrt.org/docs/guide-user/network/wifi/wifiextenders/bridgedap) to have a more reliable Wi-Fi connection than one provided by my old ISP router (note: newer routers will have Wi-Fi 7 antennas).
I am looking for a USB adapter, Wi-Fi 6 at least, to approach bandwidth of Gigabit Cat 5e Ethernet. I have only 2 Wi-Fi 6 clients, on which I regularly transfer large files to a local file server. For this poor usage, I do not want to invest more than $50.
--> Alfa AXM adapter has the advantage to have an L-shape USB connector already included. Meening no need to buy one from another shop, and enough space to not force over neighbor ports.

@bjlockie I agree, I already use [OpenWrt firmware selector](https://firmware-selector.openwrt.org/) to [remove package that I do not need](https://openwrt.org/docs/guide-user/additional-software/saving_space#excluding_packages) and add others like [kmod-tcp-bbr](https://cloud.google.com/blog/products/networking/tcp-bbr-congestion-control-comes-to-gcp-your-internet-just-got-faster?hl=en) (because of management via poor Internet connection) and iperf3. Package list and scheduled tasks can also be retained with [Attended SysUpgrade](https://openwrt.org/docs/guide-user/installation/attended.sysupgrade).

morrownr, 2025-06-30:
@ManuHry 

> but it is still too expensive for me. I prefer mid-end devices to producers I do not know well.

I have one of these Edup AX1672 adapters. I really like it:

https://www.aliexpress.us/item/3256807258105481.html

> Could you try to use your AXML adapter in AP mode with OpenWrt 24.10.2, and perform some IPv4 iPerf3 tests for a duration of about 5 minutes from a Wi-Fi 6/6E/7 client? 

Can do. Would you like for me to throw a Edup AX1672 into the mix as well?

ManuHry, 2025-06-30:
@morrownr yes sure, thanks! 🙏🏻
In this case, do you update your _USB WiFi adapters that are supported with Linux in-kernel drivers_ page? I saw that you wrote a little summary for Netgear A6210.

ManuHry, 2025-07-04:
Okay thank you @morrownr, I will look for local Edup resellers and wait your bandwidth measures :)