morrownr/USB-WiFi #589: [Help]: PCIe Wifi card compatible with Linux

Bucephalus-44, 2025-03-20:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux pop-os 6.12.10-76061203-generic #202412060638~1740154617~22.04~b4b3ebc SMP PREEMPT_DYNAMIC Fri F x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub              Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 010: ID 0424:704a Microchip Technology, Inc. (formerly SMSC) USB2 Controller Hub Bus 001 Device 007: ID 3384:0007 System76 Launch Heavy Configurable Keyboard (launch_heavy_1) Bus 001 Device 004: ID 3384:0003 System76 USB4006 Smart Hub Bus 001 Device 003: ID 1209:1776 Generic Io Bus 001 Device 008: ID 1c4f:0034 SiGma Micro XM102K Optical Wheel Mouse Bus 001 Device 005: ID 04f9:0061 Brother Industries, Ltd HL-L2300D series Bus 001 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub              Bus 001 Device 009: ID 8087:0033 Intel Corp.  Bus 001 Device 006: ID 048d:5702 Integrated Technology Express, Inc. ITE Device Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
system76/1.0.17~1735251933~22.04~f01bffd, 6.12.10-76061203-generic, x86_64: installed
system76/1.0.17~1735251933~22.04~f01bffd, 6.9.3-76060903-generic, x86_64: installed
system76_acpi/1.0.2~1719257749~22.04~7bae1af, 6.12.10-76061203-generic, x86_64: installed (original_module exists)
system76_acpi/1.0.2~1719257749~22.04~7bae1af, 6.9.3-76060903-generic, x86_64: installed (original_module exists)
system76-io/1.0.4~1732138800~22.04~fc71f15, 6.12.10-76061203-generic, x86_64: installed
system76-io/1.0.4~1732138800~22.04~fc71f15, 6.9.3-76060903-generic, x86_64: installed
```

### iw

```text
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr 08:8e:90:88:9a:42
		type P2P-device
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr 08:8e:90:88:9a:41
		type managed
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 22.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
```

### What happened?

I am think of getting the Wifi 7 & Bluetooth 5.4 BE6500Mbps PCIe x1 Tri-Band Wireless Network Adapter Card with MediaTek MT7925 chipset (Integrated Antennas) for my desktop with the POP_OS 22.04.  If I am correct this should work?  I am thinking of getting this to replace a malfunctioning built in Wifi card on my Motherboard.  This is made by HCX-eu.com.

Bucephalus-44, 2025-03-27:
Here is what I got:

total 176
-rw-r--r-- 1 root root 78941 Feb 21 11:16 mt7921-common.ko.zst
-rw-r--r-- 1 root root 35101 Feb 21 11:16 mt7921e.ko.zst
-rw-r--r-- 1 root root 31180 Feb 21 11:16 mt7921s.ko.zst
-rw-r--r-- 1 root root 27243 Feb 21 11:16 mt7921u.ko.zst

I have the Intel 6E AX210 (Gig+) module and my speed varies greatly.  I have ordered the Fenvi Wifi 6 Adapter AX1800 MT7921 Dual BAND Wireless USB 3.0 Dongle for PC and Desktop to see if that will help.  If not then I am considering removing the Intel and replacing it with the module I mentioned.  Here is the link for that:

https://www.hcx-eu.com/product/wifi-7-bluetooth-5-4-be6500mbps-pcie-x1-tri-band-wireless-network-adapter-card-with-mediatek-mt7925-chipset-direct-antennas/

Thanks,

Roland

morrownr, 2025-03-27:
Hi @Bucephalus-44 

I messed up. You are looking at buying a PCIe card based on the WiFi 7 mt7925 chip. Let me start over with the questions:

The following will show the mt7925 drivers on your system:

 $ ls -l /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76/mt7925

The following are the two drivers (modules) that you need to see:

mt7925-common.ko.zst
mt7925e.ko.zst

Now, we need to check for the required firmware files:

$ ls -l  /lib/firmware/mediatek/mt7925

The following are the 3 firmware files that you need to see:

WIFI_MT7925_PATCH_MCU_1_1_hdr.bin.zst
WIFI_RAM_CODE_MT7925_1_1.bin.zst
BT_RAM_CODE_MT7925_1_1_hdr.bin.zst

If any of the firmware files or either of the driver files are missing, we need to take action. I know you said that your kernel is 6.12, which is good, but I am curious as to the underlying version of Ubuntu that is used for this version of Pop OS. The mt7925 is kinda bleeding edge. The first version of Ubuntu that supported it was 24.04. I have used a mt7925 M.2 card since mid-2024. I have used it with Ubuntu 24.04 and later and it works well.

One thing to point out is that a recent bug is reducing the throughput of the mt7925 but the patch has already gone in so even if you see that, if should be fixed relatively quickly.

There is also the mt7922 chip that is available in PCIe and M.2 format. I run a mt7922 based PCIe card in my main dev box here. Extremely stable and has been supported going back a couple more years so should not be a problem. The mt7922 is WiFi 6e and supports 160 MHz channel width. It also is tri-band like the mt7925 but it is not WiFi 7. My speeds on my system with the mt7922 card runs around 1.1 Gbps so it is outperforming 1 Gb ethernet.

Bucephalus-44, 2025-03-28:
Here are the results of the output:
rolandtremblay@pop-os:~$ ls -l /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76/mt7925
total 180
-rw-r--r-- 1 root root 116437 Mar 20 17:04 mt7925-common.ko.zst
-rw-r--r-- 1 root root  34511 Mar 20 17:04 mt7925e.ko.zst
-rw-r--r-- 1 root root  27056 Mar 20 17:04 mt7925u.ko.zst
rolandtremblay@pop-os:~$ ls -l /lib/firmware/mediatek/mt7925
total 1708
-rw-r--r-- 1 root root  445935 Nov 21 10:00 BT_RAM_CODE_MT7925_1_1_hdr.bin
-rw-r--r-- 1 root root  210656 Nov 21 10:00 WIFI_MT7925_PATCH_MCU_1_1_hdr.bin
-rw-r--r-- 1 root root 1086064 Nov 21 10:00 WIFI_RAM_CODE_MT7925_1_1.bin
rolandtremblay@pop-os:~$ 

Hope this helps,

Roland

Bucephalus-44, 2025-03-28:
Here are the results of the output:

rolandtremblay@pop-os:~$ ls -l /usr/lib/modules/$(uname -r)/kernel/drivers/net/wireless/mediatek/mt76/mt7925
total 180
-rw-r--r-- 1 root root 116437 Mar 20 17:04 mt7925-common.ko.zst
-rw-r--r-- 1 root root  34511 Mar 20 17:04 mt7925e.ko.zst
-rw-r--r-- 1 root root  27056 Mar 20 17:04 mt7925u.ko.zst
rolandtremblay@pop-os:~$ ls -l /lib/firmware/mediatek/mt7925
total 1708
-rw-r--r-- 1 root root  445935 Nov 21 10:00 BT_RAM_CODE_MT7925_1_1_hdr.bin
-rw-r--r-- 1 root root  210656 Nov 21 10:00 WIFI_MT7925_PATCH_MCU_1_1_hdr.bin
-rw-r--r-- 1 root root 1086064 Nov 21 10:00 WIFI_RAM_CODE_MT7925_1_1.bin
rolandtremblay@pop-os:~$

Roland

morrownr, 2025-03-28:
Interesting that the drivers are compressed and the firmware files are not.

It is looks to me like you should be okay to buy and install a PCIe card based on the mt7925 chip.

Let me know how it goes. BYW, the card will have bluetooth capability and it should work.

Bucephalus-44, 2025-03-28:
Thanks.  The problem I have now though is trying to get the card.  If you noticed on the [www.HCX-eu.com](http://www.hcx-eu.com/) website is that their Tylogy page has just come up in the last two days.  Also, only a very few of their products are available through Amazon Belgium.  I am guessing that this will change slowly in the coming weeks.  I tried several times sending them emails asking when I can get this item in the US and have received no response from them.  I would at least like them to let me know that they don't if that is the case instead of saying nothing.  How typical of companies to do this.  They do have a list of places where their products will be available including Amazon here but I will just have to wait.

In the meantime, in order to solve the problem I have been having with my Internet speed varying so much I have ordered the Fenvi Wifi 6 USB Adapter AX1800 MT7921 Dual Band Wireless USB3.0 Dongle for PC Desktop which I noticed on your page.  If that does not solve the problem then I will get the card if and when it becomes available in the US.  This to hopefully not have to remove a faulty card with an Intel be AX210 (Gig +) Module.

Thanks for your help.  I will let you know what happens.

Bucephalus-44, 2025-03-28:
I think that this one should work though?

https://www.asus.com/us/networking-iot-servers/adapters/all-series/pce-be6500/

morrownr, 2025-03-28:
> I think that this one should work though?

No. I saw no indication of what chip the asus card uses. More research needs to be done.

If you are having a difficult time finding a PCIe card, what you could do is get a PCIe card that supports a M.2 card. You buy the PCIe card and then the mt7925 chip M.2 card and put it in the slot. Are you following? M.2 cards may be easier to find.

What country are you in? Maybe I can find something that might work for you.

Bucephalus-44, 2025-03-28:
I do not have the knowledge to follow you but will take what ever explanation or link to such which you can provide me.  I am in the US.

morrownr, 2025-03-28:
> I do not have the knowledge to follow you but will take what ever explanation or link to such which you can provide me.

Thanks for letting me know. I'll try to keep things simple and will concentrate on a simple card.

Your internal Intel wifi. Is it using a M.2 card? If it is, it should be simple to replace it and mt7925 M.2 cards are readily available from online stories here in the US. If you do not know, provide me with information about the brand and model of the computer you are using and I will try to find the information.

Bucephalus-44, 2025-03-28:
Here is the link to what I have:

https://www.intel.com/content/www/us/en/products/docs/wireless/wi-fi-6e-ax210-module-brief.html

Bucephalus-44, 2025-03-28:
Perhaps this is more helpful:

rolandtremblay@pop-os:~$ lspci
00:00.0 Host bridge: Intel Corporation Device 4610 (rev 05)
00:02.0 VGA compatible controller: Intel Corporation Device 4693 (rev 0c)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 05)
00:14.0 USB controller: Intel Corporation Device 7ae0 (rev 11)
00:14.2 RAM memory: Intel Corporation Device 7aa7 (rev 11)
00:14.3 Network controller: Intel Corporation Device 7af0 (rev 11)
00:15.0 Serial bus controller: Intel Corporation Device 7acc (rev 11)
00:15.1 Serial bus controller: Intel Corporation Device 7acd (rev 11)
00:15.2 Serial bus controller: Intel Corporation Device 7ace (rev 11)
00:15.3 Serial bus controller: Intel Corporation Device 7acf (rev 11)
00:16.0 Communication controller: Intel Corporation Device 7ae8 (rev 11)
00:17.0 SATA controller: Intel Corporation Device 7ae2 (rev 11)
00:19.0 Serial bus controller: Intel Corporation Device 7afc (rev 11)
00:19.1 Serial bus controller: Intel Corporation Device 7afd (rev 11)
00:1c.0 PCI bridge: Intel Corporation Device 7ab8 (rev 11)
00:1f.0 ISA bridge: Intel Corporation Device 7a86 (rev 11)
00:1f.3 Audio device: Intel Corporation Device 7ad0 (rev 11)
00:1f.4 SMBus: Intel Corporation Device 7aa3 (rev 11)
00:1f.5 Serial bus controller: Intel Corporation Device 7aa4 (rev 11)
01:00.0 Non-Volatile memory controller: Sandisk Corp Device 5017 (rev 01)
02:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 03)
rolandtremblay@pop-os:~$

Bucephalus-44, 2025-03-28:
This is the link to the motherboard I have.  I found it in an old email from System76:

https://www.gigabyte.com/Motherboard/B660I-AORUS-PRO-DDR4-rev-1x/sp#sp

morrownr, 2025-03-28:
The following is from the first link you sent:

Form Factors (M.2 2230 and 1216)

So, it sounds like your Intel card is a removable M.2 card. That is good but there are standard M.2 cards and then Intel has their own spin so we need to do more research before picking a card. Are you going to be comfortable opening your computer to remove and replace a card?

Bucephalus-44, 2025-03-28:
About 10 tears ago I and my sister built our own computer from scratch.  It was quite an experience and we used a server tower so we had more room but we are up to it.  I currently have the Thelio b-3 tower from System76 which they no longer use by they have variations on it.

morrownr, 2025-03-28:
Okay.

A quick look tells me something along these lines might work:

https://www.amazon.com/dp/B0DS4KBZNP

Note that it appears to be a PCIe card that is holding an M.2 card. This combines what I was talking about earlier as it would give you the flexibility to use it in PCIe or M.2 slots going into the future.

I also do not like the antennas on the back of the system so something like the following is what I would get with it:

https://www.amazon.com/Tri-Band-Magnetic-Omnidirectional-Extension-Computer/dp/B0DHJNGWMQ

If it were me, I would probably remove the Intel card and store it.

With that said, use caution. I do not have either of these products at this time. The PCIe card appears to be a white box product that many sellers have brought to the market lately. Very lately actually as I can find no reviews. We also do not know who the company is that makes this product, but it may be a good product. We just don't know so take what I am saying here to be an idea of a good solution where more research is likely required. 

Remember to use a can of air or equivalent to blow the dust out of your PCIe slot before putting the card in. Also, remember static electricity can kill components to ground yourself regular during installation and don't directly touch lines that provide connections.

Thoughts?

Bucephalus-44, 2025-03-29:
If I understand you correctly you recommend getting both products but use the antennas on the second product instead of what comes with first one?

Bucephalus-44, 2025-03-29:
What of you think of this card instead to just replace the card I have now.  I would not need the antennas would I?

https://www.ebay.com/itm/256521809091?norover=1&mkevt=1&mkrid=711-153677-346401-4&mkcid=2&mkscid=102&keyword=&crlp=651102044523_&MT_ID=&geo_id=&rlsatarget=dsa-1456167871416&adpos=&device=c&

Bucephalus-44, 2025-03-29:
I noticed that the wifi system in my apartment only provides up to wpa3.  Will this be a problem with a card which has wpa4?  They allow both personal and enterprise setting which we can use.  Which is better for securuty?

Bucephalus-44, 2025-03-29:
Here is the company who manufactured the card.  I ordered the two items this morning.  It is coming from China.

SHENZHENSHI YEHONG TECHNOLOGY COMPANY

morrownr, 2025-03-29:
> Will this be a problem with a card which has wpa4?

There is no such thing as WPA4. Some of the BS that you see in online ads is pretty far out there.

> They allow both personal and enterprise setting which we can use. Which is better for securuty?

If you are in an enterprise setting then you will have a system admin that can take care of the security. If at home, just use WPA3 personal. It is very good security. WPA2 personal is good also and you may have devices that cannot do WPA3 yet, so, in that case use WPA2. I use WPA2 for 2.4 GHz band and WPA3 for 5 and 6 GHz.

> If I understand you correctly you recommend getting both products but use the antennas on the second product instead of what comes with first one?

That is correct. I am not fond of internal antennas on a desktop system so I like the option of having the option to have good external antennas.

> What of you think of this card instead to just replace the card I have now. I would not need the antennas would I?

That will probably work. If it were me, I would check with the maker of the computer first or do more research. There are 2 types of M.2 wifi cards. One is NGFF. That is what my laptop and Mini PC use.

morrownr, 2025-03-29:
Oh. 

> Here is the company who manufactured the card. I ordered the two items this morning. It is coming from China.

Can you give me the links to what you ordered? I am curious.

Bucephalus-44, 2025-03-29:
The maker of my computer, System76, wanted me to stick with an Intel chip as they feel it would be more compatible with their system.  But at the same time they say it my my system to do with as I please.  They do not go into different aspects or equipment as you do.  They obviously have contracts for certain components which obviously must work for them.  But if it fails then I can do what I want.

Bucephalus-44, 2025-03-29:
I ordered the two items based on the links which you sent me.  The one was listed as Ciciglow but they were not the manufacturer just the seller.  Here is the storefront on Amazon for the seller calling themselves Ciciglow.  If you scroll down the page you will see who the actual manufacturer is.

https://www.amazon.com/sp?seller=A34XVXTBCHMT4Z&sshmPath=shipping-rates

morrownr, 2025-03-29:
Okay, so you ordered the following and the extension antenna:

https://www.amazon.com/dp/B0DS4KBZNP

I was thinking that this card, which appears to be a M.2 card in a PCIe card would allow for maximum flexibility going into the future. As long as you have an open PCIex1 slot, it should work.

I do recommend that you remove the Intel card as I have seen a component go bad and then take other things down with it over time. I'm sure you don't want your motherboard to be damaged. You should consider taping up the ends of the internal antennas as there is open metal that could cause shorts.

> The maker of my computer, System76, wanted me to stick with an Intel chip as they feel it would be more compatible with their system.

Yes, they see something they are not familiar with as a support problem. I get that. However, since you have a bad card anyway and they are not offering to replace it for free, why not get something that allows more capability in case you need it in the future. Intel has long provided good Linux drivers for its wireless cards for managed and P2P modes but there is far more you can do with wireless and this site is frequented by users that make use of monitor mode, AP mode and more. Once you go beyond band 1 (2.4 GHz) Intel's support for monitor mode and AP mode is just not where it needs to be.

As far as wireless chip and driver makers go, on Linux I would rate them in this order:

Mediatek (really good support for the maximum amount of modes and capabilities)
Qualcomm (not bad but not at the level of Mediatek) (they don't make USB chips)
Intel (really good support for managed and P2P mode) (they don't make USB chips)
Realtek (their drivers for chips for cards is improving greatly these days but their USB support is BAD)
Broadcom (junk as far as Linux is concerned)
Others (not worth considering)

Give me a report when able.

Bucephalus-44, 2025-03-29:
In the meantime I thought that I would try the Fenvi AX1800 USB adapter to see if that would do anything.  However, I am not exactly sure where to plug it in.  I put it into a red USB slot on my computer and shortly afterwards it knocked out my Wifi connection.  Somehow, I think that I did something wrong?

morrownr, 2025-03-30:
@Bucephalus-44 

Take a look at this:

https://github.com/morrownr/USB-WiFi/issues/508

Bucephalus-44, 2025-03-30:
I also tried a USB hub and nothing different happened and they have the same USB connection.  I hate to get into the BIOS situation.  Perhaps it just may be best to just remove the old card and put in the new one.  It will not be here for about a week yet.

morrownr, 2025-03-30:
> I put it into a red USB slot on my computer and shortly afterwards it knocked out my Wifi connection.

Tell me about the USB ports in your system. How many? What color?

Open a terminal and run the following with the new usb adapter plugged in:

$ iw dev

Post the result so that we can see how many wireless interfaces that you have.

Bucephalus-44, 2025-03-30:
rolandtremblay@pop-os:~$ iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x3
		addr 08:8e:90:88:9a:42
		type P2P-device
	Interface wlo1
		ifindex 3
		wdev 0x1
		addr 08:8e:90:88:9a:41
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 22.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00

olandtremblay@pop-os:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 20000M/x2
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 11, If 0, Class=Printer, Driver=usblp, 480M
        |__ Port 2: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 5: Dev 3, If 1, Class=CDC Data, Driver=system76-io, 12M
    |__ Port 5: Dev 3, If 0, Class=Communications, Driver=system76-io, 12M
    |__ Port 7: Dev 8, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 7: Dev 10, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 6: Dev 9, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 6: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 6: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 11: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 14: Dev 6, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M

Bucephalus-44, 2025-03-30:
file:///home/rolandtremblay/Desktop/Screenshot%202025-03-30%20at%2012-08-06%20Gigabyte%20B660I%20Aorus%20Pro%20DDR4%20user%20manual%20(English%20-%2036%20pages).png

Bucephalus-44, 2025-03-30:
[image]

Bucephalus-44, 2025-04-19:
I have received the card which you suggested to replace my built-in Wifi card.  I contacted System76 to ask for specific instructions on how to get to the motherboard in the best way possible.  Their client service is quite good but for some reason they have not answered my request on how to best access the motherboard on the Thelio b3.  Their site has a list of current and former models of their desktops-except for this one which I find very strange.  I do not know what is going one with them.  Would you happen to have information on this particular unit or can direct me to where I can find it?  If not I will definitely call them.

morrownr, 2025-04-19:
@Bucephalus-44 

If the card you got is

https://www.amazon.com/dp/B0DS4KBZNP

Then it goes in a pcie x1 slot on the motherboard. Do you have a link to the manual for your system. I'm looking for a manual that shows the slots that are available.

The biggest concern when doing this job is to not zap the card or motherboard with statis electricity when doing the job. We can talk about that.

Bucephalus-44, 2025-04-19:
That is the link in question.  The problem is that I have no manual and have been able to find any link on the System76 website which references that tower even in their previous version section.  We are prepared to do the work with an anti-static mat which has a cord for grounding and also a wrist band for the same purpose.  As far as I know the PCIe x16 slots are all available.  I just need to get to the motherboard to get to the card without taking the whole computer apart.

morrownr, 2025-04-19:
Okay, Sounds like you understand static electricity. This actually should be fairly straightforward. 

We need to find and remove the intel card and we need to install the new card.

It would be good if you have a can of air to blow dust out of your system and especially in the slot where you are going to add the new card.

What you can do is take pictures of the mobo and send them to me.

Bucephalus-44, 2025-04-20:
The problem we have is trying to take apart the computer to get to the motherboard to replace the card.  I can tell you though that the board is the B6601 Aorus Pro DDr4 (rev. 1.x).  We had taken off the case to get to the interior but were not sure how to proceed without taking the whole thing apart.

morrownr, 2025-04-20:
I've been looking at the limited information that is available. Is that a little itx mobo? 

I wish I could see what you are looking at. Even with the pictures, it is not clear to me where the Intel card is.

Bucephalus-44, 2025-04-20:
Yes, it is a mini itx.  I am including a shot of the ports on the back of the tower.  It is a side shot but the orientation would be with the two round circles on the bottom of the picture.  Those circles are the attachments for the Wifi antennas.

[output.pdf](https://github.com/user-attachments/files/19825148/output.pdf)

Bucephalus-44, 2025-04-20:
Would this help?  https://www.aorus.com/en-us/motherboards/B660I-AORUS-PRO-DDR4-rev-1x/Key-Features

morrownr, 2025-04-20:
> Those circles are the attachments for the Wifi antennas.

Yes, I did some digging. So far I cannot find an pictures that seem to show the little wifi card. There will be wires on the back of the panel that goes from the wifi antenna connections to the wifi chip. It is covered so we can't see. I have no idea how to take that cover off.

> Would this help?

I've already found that site. It is not much help. It might be time to search on something like:

how to change wifi card [name of mobo]

I'll keep you posted if I find anything that helps. This is harder than it should be.

morrownr, 2025-04-20:
I just did a Google search on the following:

`how to change the wifi card on a B6601 Aorus Pro DDr4 (rev. 1.x)`

Here is the reply from Google AI:

To replace the WiFi card on a Gigabyte B660I Aorus Pro DDR4 (rev. 1.x) motherboard, you'll need to carefully remove the I/O shield, the M.2 carrier, and then the card itself. Ensure you have the new card and a Phillips screwdriver. Then, carefully reinstall the new card and carrier, and the I/O shield. 

Here's a more detailed breakdown:

1. Power Down and Disconnect:
Turn off your computer and unplug the power cord to avoid electrical shocks. 

2. Access the M.2 Slot:
Locate the I/O shield and bracket on the back of the motherboard. 
Remove the screws that hold the shield and bracket in place. 
Carefully remove the I/O shield and bracket to expose the M.2 slot. 

3. Remove the M.2 Carrier:
Remove the screws holding the M.2 carrier in the slot. 
Carefully remove the carrier from the slot. You may need to tilt it slightly to disengage it from the slot. 

4. Remove the WiFi Card:
Remove the screw that holds the WiFi card in the carrier. 
Disconnect the antenna cables from the card. 
Remove the card from the carrier. 

5. Install the New Card:
Install the new WiFi card into the carrier, ensuring the antenna cables are properly connected. 
Secure the card with the screw. 

6. Reassemble:
Carefully slide the carrier back into the M.2 slot. 
Secure the carrier with the screws. 
Reinstall the I/O shield and bracket. 

7. Power On and Test:
Plug in the power cord and turn on the computer. 
Install the necessary drivers for the new WiFi card. 
Test the WiFi functionality to ensure it is working correctly. 

-----

It appears that you will need to remove the M.2 wifi card from the PCIe card that you bought and just use the little M.2 wifi card once you have removed the Intel M.2 card. The antennas with have to be connected. I am not there so be careful and go slow. Good luck.

Bucephalus-44, 2025-04-20:
Thanks for that information.  But instead of removing the old card and putting in the new one I was planning on putting the new card in the PCIe x16 slot removing the bracket from the new card and lining the antennas up with the holes on the back of the tower and then attaching the antennas which you recommend which I would place on top of the tower.  But before I do that I wanted to use the precise instructions for removing the motherboard from the Thelio b-3.  I am assuming that the reason I have  not heard from them for a week is that they no longer have this information.  They list previous units along with current ones but for some reason the b-3 is not included.  They are probably trying to find the information.  If this is true it would at least be good if they told me this instead of saying nothing.  I'll let you know what happens.

morrownr, 2025-04-20:
> But before I do that I wanted to use the precise instructions for removing the motherboard from the Thelio b-3.

That sure seems like a lot of work for something that should not be that hard. Is there a way to remove the I/O cover without removing the mobo?

Bucephalus-44, 2025-04-20:
We will have to take a closer look at it tomorrow to see if we can figure it out.  From one side the fan with it's cover is in the way.  From the side close to the motherboard it does not appear to be obvious.

Bucephalus-44, 2025-04-22:
I have some interesting news for you.  We took apart the tower and got to the mobo but could not really disconnect anything as the fan unit was there and the connections were really tight.  So we put it back together and tried something else which we had thought about but had not tried.  We connected the tower to those long antennas which you recommended I buy with the Wifi unit.  Upon doing so I tried an Internet speed test and reached a speed of 170 Mbps.  Most of the time I was lucky to reach 50 Mbps and most of the time not even that.  So it appears that maybe the short antennas were the issue.  Have to ever run into this issue before?

morrownr, 2025-04-22:
> Have to ever run into this issue before?

Yes. There is a lot of variability in the quality and capability of antennas. It appears that your new antenna is simply a better antenna. 

After studying the mobo in your system, that was not what I was expecting. I should have asked more questions before throwing out some ideas. ITX mobos can be a challenge and I did not even consider that. Sorry about that.

Bucephalus-44, 2025-04-23:
Not a problem.  If you had not suggested that antenna I would not have gotten the improvement.  After last contacting you I actually got a speed  as high as 230Mbps which was quite impressive.  That, of course, will fluctuate in a Wifi system but it is so much better than I had before.  Thanks, again for the help.  Perhaps we may meet again in this forum someday.

morrownr, 2025-04-23:
> Perhaps we may meet again in this forum someday.

Sure, stop by anytime. I like to work on wifi performance. Below is a quick iperf3 showing my wifi connection speed. That is WiFi 6, 5 GHz, 160 MHz channel width. Nothing special, that is the normal speed from my dev box to my wifi router. It is using a pcie card with a mt7922 chip. I have a M.2 card in another machine and it is based on the same mt7925 that you now have. It can do 1.2 Gbps connected to the same AP...so you have a really fast card there. I hope you are able to use it. 

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 35568 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   129 MBytes  1.09 Gbits/sec    0   1.25 MBytes       
[  5]   1.00-2.00   sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
[  5]   2.00-3.00   sec   126 MBytes  1.06 Gbits/sec    0   1.25 MBytes       
[  5]   3.00-4.00   sec   132 MBytes  1.11 Gbits/sec    0   1.25 MBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0   1.25 MBytes       
[  5]   5.00-6.00   sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
[  5]   6.00-7.00   sec   134 MBytes  1.12 Gbits/sec    0   1.25 MBytes       
[  5]   7.00-8.00   sec   132 MBytes  1.11 Gbits/sec    0   1.25 MBytes       
[  5]   8.00-9.00   sec   128 MBytes  1.07 Gbits/sec    0   1.25 MBytes       
[  5]   9.00-10.00  sec   131 MBytes  1.10 Gbits/sec    0   1.25 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.28 GBytes  1.10 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.27 GBytes  1.09 Gbits/sec                  receiver

```