morrownr/USB-WiFi #590: Should I update wifi firmware when I can't update wifi driver to the latest?

imkuang, 2025-03-27:
Hello,

Hi, I'm using OpenWrt 24.10 + mt7925e and I noticed that my current firmware-version is 20231227093232:

```
root@OpenWrt:~# ethtool -i mon0
driver: mt7925e
version: 6.6.83
firmware-version: ____000000-20231227093232
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

From https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek/mt7925, I see that the latest fw version is 20250305132908a. The version I am using is actually the first released mt7925 firmware version.

The firmware provided by openwrt should actually come from here: https://github.com/openwrt/mt76/tree/master/firmware/mt7925 , It really doesn't get updated.

I have read your article: https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md, but I have some questions, is there a version correspondence between the wifi firmware and the wifi driver? I can't update the mt7925e driver used in openwrt, so can I use the firmware with the updated version provided by linux-firmware.git?

morrownr, 2025-03-27:
Hi @imkuang 

> can I use the firmware with the updated version provided by linux-firmware.git?

It will work in theory. 

> It really doesn't get updated.

The main Mediatek wireless dev is part of OpenWRT. He makes sure Mediatek drivers and firmware are up to speed. You need to remember that OpenWRT is not a normal Linux distro. Due to the need for stability, the kernel that is used is usually around 1 LTS version behind what is available. The devs do pick and choose to modernize some of the supportings stacks such as the wifi stack as they see the need.

My advice is to not do anything unless you have a specific problem you are dealing with. Do you have a problem that you are trying to fix?

imkuang, 2025-03-28:
> It will work in theory.

I have used Mediatek's proprietary WIFI driver (mt_wifi) and firmware at work, and sometimes we encounter some problems that require updating both the driver and firmware from Mediatek to fix, so I think they may have some version correspondence (at least in some functions), but I am not sure whether the open source mt76 driver behaves the same.

> Do you have a problem that you are trying to fix?

I did run into some performance issues in AP mode, the throughput of the mt7925e (≈150Mbps) seemed to be much lower than my mt7921au (550Mbps+) in the same scenario (my environment has a lot of wireless interference, and the performance of the station is not guaranteed, so it cannot be regarded as peak data.).
But I don't pay attention to it anymore recently, I just built a tool box with OpenWrt to capture packets for analyze wireless related issues, at least it works well in monitor mode (I do have a small fix for openwrt: https://github.com/openwrt/openwrt/pull/18319).

morrownr, 2025-03-28:
> I am not sure whether the open source mt76 driver behaves the same.

The mac80211 drivers are not supposed to act like that. I say, in theory, because this stuff is so complicated these days that I don't see there cannot be a mistake at times.

> I did run into some performance issues in AP mode, the throughput of the mt7925e (≈150Mbps) seemed to be much lower than my mt7921au (550Mbps+)...

Yes, that. I came about because of a bad patch that worked its way back into the LTS kernels. A patch to fix the problem has gone in and should work its way to you at some point.

ichbinkenny, 2025-04-11:
I'm actually seeing this issue as well with my MT7925e on 6.14.1. Do you happen to know where this patch is @morrownr ? I'd love to install it locally if so.

morrownr, 2025-04-11:
@ichbinkenny 

> Do you happen to know where this patch is @morrownr ?

I don't have this patch bookmarked but you can search with this link:

https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=3&q=mt7925&archive=both&delegate=

The title is not about what we are seeing so you have to read the patches but I think that search will get it. I'm pretty sure it went into kernel 6.15.

ichbinkenny, 2025-04-12:
> [@ichbinkenny](https://github.com/ichbinkenny)
>
>
> I don't have this patch bookmarked but you can search with this link: [...]

Thank you for this! I'll mess around with the patch set this weekend!