morrownr/USB-WiFi #592: [Help]: Panda PAU0B (MT7610U)

RobyBaz, 2025-04-01:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux kali 6.12.13-amd64 #1 SMP PREEMPT_DYNAMIC Kali 6.12.13-1kali1 (2025-02-11) x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0e8d:7610 MediaTek Inc. WiFi Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub

### rfkill

0: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8, 6.11.2-amd64, x86_64: installed
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8, 6.12.13-amd64, x86_64: installed
realtek-rtl88xxau/5.6.4.2~git20240726.63cf0b4, 6.11.2-amd64, x86_64: installed
realtek-rtl88xxau/5.6.4.2~git20240726.63cf0b4, 6.12.13-amd64, x86_64: installed
rtl8821au/5.12.5.2, 6.12.13-amd64, x86_64: installed
```

### iw

```text
hy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 9e:8f:4b:5a:33:6b
                type managed
                txpower 11.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```

### What happened?

Hello, following morronwnr suggestion in recommended Kali Wi-Fi Cards, I got a PANDA PAU0B adapter that was listed and reported as a long range unit. 
Now...I don't know if i'm missing something but this adapter discover less network than my old ALFA AWUS036ACS and mainly it reports a TXPower=11db

[image]

Is there anyone who can help me clarify this and know if I have received a defective unit or if it is normal like this.

Thank you in advance

morrownr, 2025-04-02:
Hi @RobyBaz 

I've been on the road for a few days and just got home. If you will show me the commands you used to scan, I'll do some testing.

morrownr, 2025-04-02:
> IW REG GET reports country 00 (unset) but no way to change it.

What ways have you tried?

If you go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

The last menu item is a Frequently Asked Questions (FAQ) document. The first 2 questions may apply to your situation.

RobyBaz, 2025-04-02:
Tried both method in FAQ and nothing changed as you can see in the screenshot:

[image]

morrownr, 2025-04-02:
Well, remove anything that you did so that it does not cause problems in the future.

Do you have the capability to move back to kernel 6.12.12?

I have seen various problems with 6.12.13.

RobyBaz, 2025-04-02:
Thanks for your assistance...I tried booting from the previous kernel version but nothing changed:

[image]

morrownr, 2025-04-02:
Let me do some testing here with my PAU0B. Something is not right.

morrownr, 2025-04-03:
@RobyBaz 

I had some time to do some testing this afternoon. My testing was limited but seemed to show more APs than I thought would be in range so nothing obvious so far.

Regarding the txpower. Disregard that for now as it may not be an accurate reading. I have an Alfa ACS so I'll see about doing a comparison.

RobyBaz, 2025-04-04:
@morrownr 

thanks for your time, I was able to chage regional settings on a Ubuntu VM and the situation has improved but the Alfa ACS still shows many more APs than the Panda. I thought maybe I got a defective unit but when I tested both adapters in a Windows enviroment tha Panda was superior at reception, exactly how it was supposed to be for its specs.

May I ask where I can find info about the real advantage of the "Active Monitor" capability (BTW...the main reason I wanted to buy and test the Panda)?

RobyBaz, 2025-04-04:
Just to picture the proportion:

This is what I get scanning with Alfa ACS:

[image]

and with Panda PAU0B

[image]

morrownr, 2025-04-04:
> the Alfa ACS still shows many more APs than the Panda.

I have the Panda 0B and Alfa ACS so I'll try to find some time today to see what I can see,

> I thought maybe I got a defective unit but when I tested both adapters in a Windows enviroment tha Panda was superior at reception, exactly how it was supposed to be for its specs.

It is always to check things out to make sure you did not get a bad adapter but that makes it look like it is not the adapter.

I'll try to help so that we know what the problem is. If I find a problem with an adapter that is on one of the Lists, I will remove it until fixes are in place. I try to do a good job with the lists.

morrownr, 2025-04-04:
Did a head to head test with both adapters:

Alfa ACS - 24 APs
Panda PAU0B - 49 APs

Here are the details:

```
$ iw dev

phy#2
	Interface wlx00c0caac47xx
		ifindex 4
		wdev 0x200000001
		addr 00:c0:ca:ac:47:xx
		type managed
		txpower 30.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlx9cefd5f8f9xx
		ifindex 3
		wdev 0x1
		addr 9c:ef:d5:f8:f9:xx
		type managed
		txpower 11.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

```

Command that I used:

$ sudo iw wlx00c0caac47xx scan | grep wlx00c0caac47xx

Adapters were connect to the same system. Adapters were in managed mode and not connected to anything.

Kernel in use is 6.12.12 on Debian 12. The driver for the Panda is the in-kernel Mediatek mt7610u. The driver for the Alfa is:

https://github.com/lwfinger/rtw88

Once you get to kernel 6.14, mainline will have the above driver as it has been upstreamed.

So, maybe you can try to the tested that I performed to see what you see. I think the Alfa ACS is a really good little adapter, especially now that a good driver has gone into the kernel but it does not seem to have the range that the Panda does... or at least, not here.

RobyBaz, 2025-04-04:
@morrownr 

first of all, thank you for taking the time, which is far from obvious and greatly appreciated. I think I understand that the problem is with Kali (even with clean installation). I installed BackBox based on Ubuntu and the situation has definitely improved and the results are in line with yours. The Panda sees many more APs than the little Alfa which is perfectly logical and what I expected. I'm sorry for not thinking of this sooner and wasting your time.

morrownr, 2025-04-04:
You are welcome and I hope you have success.

> I think I understand that the problem is with Kali (even with clean installation).

I know that the people that work on Kali try to do a good job but I finally gave up on it. I used to keep a copy on one of my test systems here since many USB WiFi adapter users use it but it finally drove me crazy. I have little tolerance for more than just a sporadic bug. It is a bleeding edge rolling distro that seems to be in alpha testing at all times.

Stop by anytime.

morrownr, 2025-04-04:
If you have time, and since you have rtl8811au, rtl8812au and rtl8814au adapters, we have been doing a lot of work on upstreaming good drivers for those chips. If you have time to test, you don't have to wait for kernel 6.14 or 6.15 to use the new drivers, just to the following site:

https://github.com/lwfinger/rtw88

You can post reports on that site.

morrownr, 2025-04-04:
> could you briefly tell me the real benefit of having the Active Monitor available in WiFi adapters?

I test the basic monitor mode here, not active monitor mode. Active monitor mode is a just capability within monitor mode and it is used by a few applications. I forget what it does right now but users ask about it so I keep up with which drivers have the capability. Basically you need Mediatek chips, mt7610u, mt7612u, mt7921au or mt7925.

RobyBaz, 2025-04-04:
Once again...THANK YOU.

BTW...sent you an email.

closed 2025-04-04 by RobyBaz