morrownr/USB-WiFi #593: [Help]: MT7921 kernel NULL pointer error

M-y, 2025-04-04:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux bananapi 6.13.7-edge-sunxi #1 SMP Thu Mar 13 12:08:08 UTC 2025 armv7l GNU/Linux

### lsusb

Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 004 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
bash: dkms: command not found
```

### iw

```text
phy#0
        Interface wlxe84e06afb8f1
                ifindex 3
                wdev 0x1
                addr e8:4e:06:af:b8:f1
                ssid xxx
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        9868    164     555     0       0       0       0       48548           572
```

### What happened?

Hi,
I want to setup an access point using Banana Pi M1. Here are my specs:

- I have soldered [official raspberry pi 5V 3A power supply](https://www.raspberrypi.com/products/type-c-power-supply/) to Banana pi for supplying enough power. Not using micro usb to power the device. 
- Added a heatsink and running a fan using another power source. 
- EDUP EP-AX1672 is my usb wireless device, inserted directly to the usb port.
- Updated the firmware files for MT7921.
- Armbian 25.2.3 bookworm is my distro.
- Using wifi6 conf for hostapd provided here.
```
ssid=xxx
wpa_passphrase=passwrd
sae_password=passwrd
hw_mode=g
channel=6
country_code=TR
ieee80211d=1
interface=wlxe84e06afb8f1
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
beacon_int=100
dtim_period=2
skip_inactivity_poll=1
auth_algs=3
max_num_sta=16
macaddr_acl=1
accept_mac_file=/etc/hostapd/macs.allow
[... 88 lines cut ...]
2025-04-04T14:12:47.186733+03:00 bananapi kernel: [  837.203843] dec0: 0000f052 c02d9d7f f6755ca4 00000000 00000000 fbc669c0 c0000000 00000000
2025-04-04T14:12:47.186741+03:00 bananapi kernel: [  837.218981] dee0: c5034a98 00000000 c5034a80 00000003 00001000 fbc669c0 00001000 c9b60680
2025-04-04T14:12:47.186750+03:00 bananapi kernel: [  837.234091] df00: bfbca7cd 00000001 c5034a80 00001000 c535b000 c3c09860 c3c0b01c bfb8845f
2025-04-04T14:12:47.186759+03:00 bananapi kernel: [  837.249201] df20: c3c0b068 00000000 c3c0b020 00000146 00000000 c3c0b01c c3c0986c c25d1400
2025-04-04T14:12:47.186768+03:00 bananapi kernel: [  837.264279] df40: c3c0b52c c25d1400 c3c0b01c c160cb78 f083df68 00000000 c3c0cbe8 c25d1400
2025-04-04T14:12:47.186776+03:00 bananapi kernel: [  837.279336] df60: c3c0cbe0 00000001 f6755ca4 00000000 00000000 bfac74f7 c9ab2240 c25d1400
2025-04-04T14:12:47.186785+03:00 bananapi kernel: [  837.294393] df80: bfac74a9 c3c0cbe0 c9a77f80 c014a377 00000ffa c9ab2240 c014a2b9 00000000
2025-04-04T14:12:47.186793+03:00 bananapi kernel: [  837.309458] dfa0: 00000000 00000000 00000000 c0100155 00000000 00000000 00000000 00000000
2025-04-04T14:12:47.186802+03:00 bananapi kernel: [  837.324518] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
2025-04-04T14:12:47.186810+03:00 bananapi kernel: [  837.339543] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
2025-04-04T14:12:47.186819+03:00 bananapi kernel: [  837.354491] Call trace: 
2025-04-04T14:12:47.186828+03:00 bananapi kernel: [  837.354534]  mt792x_rx_get_wcid [mt792x_lib] from mt7921_queue_rx_skb+0xff/0x624 [mt7921_common]
2025-04-04T14:12:47.186836+03:00 bananapi kernel: [  837.379495]  mt7921_queue_rx_skb [mt7921_common] from mt76u_rx_worker+0x3a3/0x3ac [mt76_usb]
2025-04-04T14:12:47.186845+03:00 bananapi kernel: [  837.395221]  mt76u_rx_worker [mt76_usb] from __mt76_worker_fn+0x4f/0x74 [mt76]
2025-04-04T14:12:47.186853+03:00 bananapi kernel: [  837.409914]  __mt76_worker_fn [mt76] from kthread+0xbf/0xd4
2025-04-04T14:12:47.186861+03:00 bananapi kernel: [  837.422865]  kthread from ret_from_fork+0x11/0x1c
2025-04-04T14:12:47.186870+03:00 bananapi kernel: [  837.434933] Exception stack(0xf083dfb0 to 0xf083dff8)
2025-04-04T14:12:47.186878+03:00 bananapi kernel: [  837.447075] dfa0:                                     00000000 00000000 00000000 00000000
2025-04-04T14:12:47.186887+03:00 bananapi kernel: [  837.462397] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
2025-04-04T14:12:47.186896+03:00 bananapi kernel: [  837.477782] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
2025-04-04T14:12:47.186904+03:00 bananapi kernel: [  837.491410] Code: 0301 d006 f8d0 32e8 (f8d3) 032c 
2025-04-04T14:12:47.186913+03:00 bananapi kernel: [  837.503526] ---[ end trace 0000000000000000 ]---
```

I can provide any information or make any other tests. Also I can change the distro if you have any advice.

M-y, 2025-04-05:
In 6.6 and 6.12, I was experiencing pending tx errors. Because of that I decided to use 6.13. I have looked the 6.13 tags [in kernel repository](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/refs/tags) the patch you pointed out isn't added. So I have patched manually and compiled the kernel. Thank you very much, this patch solves the null pointer error. 
I have tested using 1 station makes iperf about 1 hour. AP is working solid. 
I have put the test a bit further. Started ookla speedtes from 4 stations simultaneously. This time AP goes down and these errors appers:

```
2025-04-05T13:46:52.330628+03:00 bananapi kernel: [  546.434583] mt7921u 2-1:1.0: vendor request req:63 off:8454 failed:-110
2025-04-05T13:46:55.882751+03:00 bananapi kernel: [  549.983909] mt7921u 2-1:1.0: vendor request req:63 off:8458 failed:-110
2025-04-05T13:46:59.402586+03:00 bananapi kernel: [  553.507220] mt7921u 2-1:1.0: vendor request req:63 off:845c failed:-110
2025-04-05T13:47:02.922677+03:00 bananapi kernel: [  557.024614] mt7921u 2-1:1.0: vendor request req:63 off:8460 failed:-110
2025-04-05T13:47:06.442637+03:00 bananapi kernel: [  560.547931] mt7921u 2-1:1.0: vendor request req:63 off:8464 failed:-110
2025-04-05T13:47:09.962683+03:00 bananapi kernel: [  564.065224] mt7921u 2-1:1.0: vendor request req:63 off:8468 failed:-110
2025-04-05T13:47:13.482590+03:00 bananapi kernel: [  567.588558] mt7921u 2-1:1.0: vendor request req:63 off:846c failed:-110
2025-04-05T13:47:17.002612+03:00 bananapi kernel: [  571.105777] mt7921u 2-1:1.0: vendor request req:63 off:841c failed:-110
2025-04-05T13:47:20.522631+03:00 bananapi kernel: [  574.629202] mt7921u 2-1:1.0: vendor request req:63 off:8478 failed:-110
2025-04-05T13:47:24.042645+03:00 bananapi kernel: [  578.149510] mt7921u 2-1:1.0: vendor request req:63 off:d02c failed:-110
2025-04-05T13:47:27.562616+03:00 bananapi kernel: [  581.666699] mt7921u 2-1:1.0: vendor request req:63 off:d054 failed:-110
2025-04-05T13:47:31.086703+03:00 bananapi kernel: [  585.187209] mt7921u 2-1:1.0: vendor request req:63 off:d058 failed:-110
2025-04-05T13:47:34.602604+03:00 bananapi kernel: [  588.709309] mt7921u 2-1:1.0: vendor request req:63 off:53b8 failed:-110
2025-04-05T13:47:38.122594+03:00 bananapi kernel: [  592.230702] mt7921u 2-1:1.0: vendor request req:63 off:53c4 failed:-110
2025-04-05T13:47:41.642612+03:00 bananapi kernel: [  595.751511] mt7921u 2-1:1.0: vendor request req:66 off:53c4 failed:-110
2025-04-05T13:47:45.170866+03:00 bananapi kernel: [  599.276312] mt7921u 2-1:1.0: vendor request req:63 off:d698 failed:-110
2025-04-05T13:47:48.682610+03:00 bananapi kernel: [  602.791549] mt7921u 2-1:1.0: vendor request req:63 off:d520 failed:-110
2025-04-05T13:47:52.202630+03:00 bananapi kernel: [  606.308725] mt7921u 2-1:1.0: vendor request req:63 off:d518 failed:-110
2025-04-05T13:47:55.754604+03:00 bananapi kernel: [  609.864126] mt7921u 2-1:1.0: vendor request req:63 off:d688 failed:-110
2025-04-05T13:47:59.274751+03:00 bananapi kernel: [  613.381691] mt7921u 2-1:1.0: vendor request req:63 off:d690 failed:-110
2025-04-05T13:48:02.794586+03:00 bananapi kernel: [  616.904608] mt7921u 2-1:1.0: vendor request req:63 off:d558 failed:-110
2025-04-05T13:48:06.314574+03:00 bananapi kernel: [  620.424882] mt7921u 2-1:1.0: vendor request req:63 off:d564 failed:-110
2025-04-05T13:48:09.838623+03:00 bananapi kernel: [  623.945408] mt7921u 2-1:1.0: vendor request req:63 off:d568 failed:-110
2025-04-05T13:48:13.358880+03:00 bananapi kernel: [  627.466476] mt7921u 2-1:1.0: vendor request req:63 off:d7a8 failed:-110
2025-04-05T13:48:16.874618+03:00 bananapi kernel: [  630.982669] mt7921u 2-1:1.0: vendor request req:63 off:a150 failed:-110
2025-04-05T13:48:20.394647+03:00 bananapi kernel: [  634.505926] mt7921u 2-1:1.0: vendor request req:63 off:a158 failed:-110
2025-04-05T13:48:23.914585+03:00 bananapi kernel: [  638.023061] mt7921u 2-1:1.0: vendor request req:63 off:d780 failed:-110
2025-04-05T13:48:27.434608+03:00 bananapi kernel: [  641.546485] mt7921u 2-1:1.0: vendor request req:63 off:d770 failed:-110
2025-04-05T13:48:30.954593+03:00 bananapi kernel: [  645.063561] mt7921u 2-1:1.0: vendor request req:63 off:d774 failed:-110
2025-04-05T13:48:34.570735+03:00 bananapi kernel: [  648.683148] mt7921u 2-1:1.0: vendor request req:63 off:d55c failed:-110
2025-04-05T13:48:38.090605+03:00 bananapi kernel: [  652.203123] mt7921u 2-1:1.0: vendor request req:63 off:10e0 failed:-110
2025-04-05T13:48:41.614757+03:00 bananapi kernel: [  655.724389] mt7921u 2-1:1.0: vendor request req:63 off:10e4 failed:-110
2025-04-05T13:48:45.130605+03:00 bananapi kernel: [  659.243592] mt7921u 2-1:1.0: vendor request req:63 off:10e8 failed:-110
2025-04-05T13:48:48.746742+03:00 bananapi kernel: [  662.856896] mt7921u 2-1:1.0: vendor request req:63 off:10ec failed:-110
2025-04-05T13:48:52.266730+03:00 bananapi kernel: [  666.377080] mt7921u 2-1:1.0: vendor request req:63 off:10f0 failed:-110
2025-04-05T13:48:55.786766+03:00 bananapi kernel: [  669.897396] mt7921u 2-1:1.0: vendor request req:63 off:10f4 failed:-110
2025-04-05T13:48:59.310752+03:00 bananapi kernel: [  673.421510] mt7921u 2-1:1.0: vendor request req:63 off:10f8 failed:-110
2025-04-05T13:49:02.826742+03:00 bananapi kernel: [  676.937811] mt7921u 2-1:1.0: vendor request req:63 off:10fc failed:-110
2025-04-05T13:49:06.426625+03:00 bananapi kernel: [  680.529938] mt7921u 2-1:1.0: vendor request req:63 off:d7dc failed:-110
[... cut ...]
2025-04-05T13:55:37.834572+03:00 bananapi kernel: [ 1071.969398] mt7921u 2-1:1.0: vendor request req:63 off:4230 failed:-110
2025-04-05T13:55:41.358698+03:00 bananapi kernel: [ 1075.490994] mt7921u 2-1:1.0: vendor request req:63 off:4230 failed:-110
2025-04-05T13:55:44.874576+03:00 bananapi kernel: [ 1079.010076] mt7921u 2-1:1.0: vendor request req:63 off:4230 failed:-110
2025-04-05T13:55:48.408402+03:00 bananapi kernel: [ 1082.531380] mt7921u 2-1:1.0: vendor request req:63 off:4230 failed:-110
``` 

Looks like `off:4230` will continue forever.

M-y, 2025-04-06:
This is another test result using kernel 6.13.7 with [this patch](https://patchwork.kernel.org/project/linux-wireless/patch/20250218033343.1999648-1-mingyen.hsieh@mediatek.com/).

```
2025-04-06T15:45:10.273898+03:00 bananapi kernel: [ 2946.005307] mt7921u 2-1.3:1.0: vendor request req:63 off:815c failed:-110
2025-04-06T15:45:13.793891+03:00 bananapi kernel: [ 2949.528661] mt7921u 2-1.3:1.0: vendor request req:63 off:8160 failed:-110
2025-04-06T15:45:17.313890+03:00 bananapi kernel: [ 2953.048905] mt7921u 2-1.3:1.0: vendor request req:63 off:8164 failed:-110
2025-04-06T15:45:20.834006+03:00 bananapi kernel: [ 2956.569074] mt7921u 2-1.3:1.0: vendor request req:63 off:8168 failed:-110
2025-04-06T15:45:24.354027+03:00 bananapi kernel: [ 2960.086412] mt7921u 2-1.3:1.0: vendor request req:63 off:816c failed:-110
2025-04-06T15:45:27.874083+03:00 bananapi kernel: [ 2963.606504] mt7921u 2-1.3:1.0: vendor request req:63 off:811c failed:-110
2025-04-06T15:45:31.393881+03:00 bananapi kernel: [ 2967.129630] mt7921u 2-1.3:1.0: vendor request req:63 off:8178 failed:-110
2025-04-06T15:45:34.913935+03:00 bananapi kernel: [ 2970.649954] mt7921u 2-1.3:1.0: vendor request req:63 off:8250 failed:-110
2025-04-06T15:45:38.433913+03:00 bananapi kernel: [ 2974.170056] mt7921u 2-1.3:1.0: vendor request req:63 off:8254 failed:-110
2025-04-06T15:45:41.958087+03:00 bananapi kernel: [ 2977.691331] mt7921u 2-1.3:1.0: vendor request req:63 off:8258 failed:-110
2025-04-06T15:45:45.473881+03:00 bananapi kernel: [ 2981.207579] mt7921u 2-1.3:1.0: vendor request req:63 off:825c failed:-110
2025-04-06T15:45:48.994002+03:00 bananapi kernel: [ 2984.727602] mt7921u 2-1.3:1.0: vendor request req:63 off:8260 failed:-110
2025-04-06T15:45:52.513934+03:00 bananapi kernel: [ 2988.251432] mt7921u 2-1.3:1.0: vendor request req:63 off:8264 failed:-110
2025-04-06T15:45:56.034061+03:00 bananapi kernel: [ 2991.768094] mt7921u 2-1.3:1.0: vendor request req:63 off:8268 failed:-110
2025-04-06T15:45:59.553873+03:00 bananapi kernel: [ 2995.288191] mt7921u 2-1.3:1.0: vendor request req:63 off:826c failed:-110
2025-04-06T15:46:03.073877+03:00 bananapi kernel: [ 2998.808395] mt7921u 2-1.3:1.0: vendor request req:63 off:821c failed:-110
2025-04-06T15:46:06.593864+03:00 bananapi kernel: [ 3002.331682] mt7921u 2-1.3:1.0: vendor request req:63 off:8278 failed:-110
2025-04-06T15:46:10.114375+03:00 bananapi kernel: [ 3005.848883] mt7921u 2-1.3:1.0: vendor request req:63 off:8550 failed:-110
2025-04-06T15:46:13.633874+03:00 bananapi kernel: [ 3009.372075] mt7921u 2-1.3:1.0: vendor request req:63 off:8554 failed:-110
2025-04-06T15:46:17.158041+03:00 bananapi kernel: [ 3012.893351] mt7921u 2-1.3:1.0: vendor request req:63 off:8558 failed:-110
2025-04-06T15:46:17.283483+03:00 bananapi kernel: [ 3013.017735] ------------[ cut here ]------------
2025-04-06T15:46:17.283631+03:00 bananapi kernel: [ 3013.017793] WARNING: CPU: 0 PID: 13040 at net/wireless/util.c:1521 cfg80211_calculate_bitrate_he+0x1cb/0x258 [cfg80211]
2025-04-06T15:46:17.283656+03:00 bananapi kernel: [ 3013.018398] Modules linked in: aes_arm_bs crypto_simd aes_arm ccm nft_chain_nat xt_MASQUERADE nf_nat nft_compat mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_usb mt76 mac80211 libarc4 lima gpu_sched sunxi_cir sun4i_gpadc_iio axp20x_adc sunxi_cedrus(C) sun4i_ts v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 cfg80211 videobuf2_common videodev mc rfkill drm_shmem_helper display_connector simplefb cpufreq_dt evdev uio_pdrv_genirq uio zram zsmalloc binfmt_misc dm_mod autofs4 ext4 mbcache jbd2 axp20x_usb_power industrialio pinctrl_axp209 sun4i_gpadc sunxi phy_generic realtek
2025-04-06T15:46:17.283670+03:00 bananapi kernel: [ 3013.018776] CPU: 0 UID: 0 PID: 13040 Comm: hostapd Tainted: G         C         6.13.7-edge-sunxi #1
2025-04-06T15:46:17.283682+03:00 bananapi kernel: [ 3013.018795] Tainted: [C]=CRAP
2025-04-06T15:46:17.283690+03:00 bananapi kernel: [ 3013.018800] Hardware name: Allwinner sun7i (A20) Family
2025-04-06T15:46:17.283698+03:00 bananapi kernel: [ 3013.018807] Call trace: 
2025-04-06T15:46:17.283707+03:00 bananapi kernel: [ 3013.018825]  unwind_backtrace from show_stack+0x11/0x14
2025-04-06T15:46:17.283717+03:00 bananapi kernel: [ 3013.018865]  show_stack from dump_stack_lvl+0x35/0x40
2025-04-06T15:46:17.283726+03:00 bananapi kernel: [ 3013.018906]  dump_stack_lvl from __warn+0xbd/0xe8
2025-04-06T15:46:17.283734+03:00 bananapi kernel: [ 3013.018933]  __warn from warn_slowpath_fmt+0xfb/0x104
2025-04-06T15:46:17.283743+03:00 bananapi kernel: [ 3013.018961]  warn_slowpath_fmt from cfg80211_calculate_bitrate_he+0x1cb/0x258 [cfg80211]
2025-04-06T15:46:17.283752+03:00 bananapi kernel: [ 3013.019485]  cfg80211_calculate_bitrate_he [cfg80211] from nl80211_put_sta_rate+0x39/0x2e0 [cfg80211]
2025-04-06T15:46:17.283762+03:00 bananapi kernel: [ 3013.020155]  nl80211_put_sta_rate [cfg80211] from nl80211_send_station+0x483/0xa7c [cfg80211]
2025-04-06T15:46:17.283779+03:00 bananapi kernel: [ 3013.020831]  nl80211_send_station [cfg80211] from nl80211_get_station+0xc5/0x140 [cfg80211]
2025-04-06T15:46:17.283798+03:00 bananapi kernel: [ 3013.021442]  nl80211_get_station [cfg80211] from genl_family_rcv_msg_doit+0x83/0xb4
2025-04-06T15:46:17.283811+03:00 bananapi kernel: [ 3013.021816]  genl_family_rcv_msg_doit from genl_rcv_msg+0x165/0x238
2025-04-06T15:46:17.283827+03:00 bananapi kernel: [ 3013.021839]  genl_rcv_msg from netlink_rcv_skb+0x75/0xb0
[... cut ...]
2025-04-06T15:47:16.993865+03:00 bananapi kernel: [ 3072.735728] mt7921u 2-1.3:1.0: vendor request req:63 off:8678 failed:-110
2025-04-06T15:47:20.514076+03:00 bananapi kernel: [ 3076.252836] mt7921u 2-1.3:1.0: vendor request req:63 off:d02c failed:-110
2025-04-06T15:47:24.033898+03:00 bananapi kernel: [ 3079.776050] mt7921u 2-1.3:1.0: vendor request req:63 off:d054 failed:-110
2025-04-06T15:47:27.586040+03:00 bananapi kernel: [ 3083.325245] mt7921u 2-1.3:1.0: vendor request req:63 off:d058 failed:-110
```

morrownr, 2025-04-07:
I don't have any experience with the bananapi distro. I know there were issues in and around kernel 6.13.6. If you have the options to use something like kernel 6.12.12, give that a try.

M-y, 2025-04-08:
Armbian is a Debian based distro. It has [a build script](https://github.com/armbian/build). This tool fetches Linux kernel from git repository and builds the sd card image for my banana pi device. Also I can select a specific version and add my own patches to kernel. So I built kernel 6.12.12 with this build tool and tested, these vendor request errors are appearing. 

```
Apr 08 00:29:54 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8260 failed:-110
Apr 08 00:29:57 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8264 failed:-110
Apr 08 00:30:01 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8268 failed:-110
Apr 08 00:30:04 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:826c failed:-110
Apr 08 00:30:08 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:821c failed:-110
Apr 08 00:30:12 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8278 failed:-110
```

Also they provide pre built linux images, I can install them using apt. Here are the packages for my device:
```
| Package Name                          | Kernel Version |
|--------------------------------------|----------------|
| linux-image-legacy-sunxi=25.2.3      | v6.1.104       |
| linux-image-legacy-sunxi=25.2.2      | v6.1.104       |
| linux-image-legacy-sunxi=24.11.1     | v6.1.104       |
| linux-image-legacy-sunxi=24.8.2      | v6.1.104       |
| linux-image-legacy-sunxi=24.5.1      | v6.1.92        |
| linux-image-legacy-sunxi=24.2.1      | v6.1.77        |
| linux-image-current-sunxi=25.2.3     | v6.6.75        |
| linux-image-current-sunxi=25.2.2     | v6.6.75        |
| linux-image-current-sunxi=24.11.1    | v6.6.62        |
| linux-image-current-sunxi=24.8.2     | v6.6.44        |
| linux-image-current-sunxi=24.5.1     | v6.6.31        |
| linux-image-current-sunxi=24.2.1     | v6.6.16        |
| linux-image-edge-sunxi=25.2.3        | v6.13.7        |
| linux-image-edge-sunxi=25.2.2        | v6.12.16       |
| linux-image-edge-sunxi=24.11.1       | v6.11.9        |
| linux-image-edge-sunxi=24.8.3        | v6.10.9        |
| linux-image-edge-sunxi=24.5.1        | v6.7.12        |
| linux-image-edge-sunxi=24.2.1        | v6.7.4         |
```

I have installed and tested v6.6.75 and v6.12.16. In these kernel versions I see same vender request errors.
Any ideas?

morrownr, 2025-04-08:
I'd say that this is not a driver issue. I am not seeing this here so I did a little looking around.

Let's look at the firmware version and hostapd version:

$ ethtool -i `interface name`

$ hostapd --version

M-y, 2025-04-08:
OK, I was in 6..6
```
$ uname -a
Linux bananapi 6.6.75-current-sunxi #1 SMP Sat Feb  1 17:37:57 UTC 2025 armv7l GNU/Linux
```

```
$ ethtool -i wlxe84e06afb8f1 
driver: mt7921u
version: 6.6.75-current-sunxi
firmware-version: ____010000-20241106151045
expansion-rom-version: 
bus-info: 2-1:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
$ hostapd -v
hostapd v2.10
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2022, Jouni Malinen <j@w1.fi> and contributors
```

morrownr, 2025-04-08:
> OK, I was in 6..6

Firmware and apps are part of the distro and do not change when kernel is changed.

> firmware-version: ____010000-20241106151045

That is the most up to date firmware available for the mt7921u driver.

> hostapd v2.10

This is not the latest release. Based on my research, I'd say getting hostapd up to date is likely the issue. I'm running hostapd v2.11 here and am not seeing your problem. If a packaged update is not available, I have a guide to compiling hostapd if you are interested. It is easy.

M-y, 2025-04-08:
In what platform are you running hostapd?

I compiled the hostapd.
```
$ hostapd -v
hostapd v2.12-devel-hostap_2_11-806-gfe5b88e78
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2024, Jouni Malinen <j@w1.fi> and contributors
```

In first try it looks like restarted the driver with a  

> timed out waiting for pending tx

 and when I see this I restarted hostapd and vendor request errors came again. This time hostapd became unresponsive, I can't restart it. 
```
Apr 08 22:25:48 bananapi kernel: mt7921u 3-1:1.0: Message 00020003 (seq 5) timeout
Apr 08 22:25:49 bananapi kernel: mt7921u 3-1:1.0: timed out waiting for pending tx
Apr 08 22:25:50 bananapi kernel: ------------[ cut here ]------------
Apr 08 22:25:50 bananapi kernel: WARNING: CPU: 1 PID: 25393 at kernel/kthread.c:661 kthread_park+0xad/0xdc
Apr 08 22:25:50 bananapi kernel: Modules linked in: aes_arm_bs crypto_simd ccm nft_chain_nat xt_MASQUERADE nf_nat nft_compat mt7921u mt792x_usb mt7921_common mt792x_lib mt76_connac_lib mt76_usb mt76 mac80211 libarc4 lima gpu_sched sunxi_cedrus(C) sun4i_gpadc_iio v4l2_mem2mem videobuf2_dma_contig axp20x_adc sunxi_cir videobuf2_memops videobuf2_v4l2 sun4i_ts videodev videobuf2_common lz4hc lz4 mc cfg80211 display_connector drm_shmem_helper simplefb cpufreq_dt rfkill evdev zram zsmalloc uio_pdrv_genirq uio binfmt_misc dm_mod autofs4 ext4 mbcache jbd2 axp20x_usb_power industrialio pinctrl_axp209 sun4i_gpadc sunxi phy_generic realtek
Apr 08 22:25:50 bananapi kernel: CPU: 1 PID: 25393 Comm: kworker/u4:3 Tainted: G         C         6.6.75-current-sunxi #1
Apr 08 22:25:50 bananapi kernel: Hardware name: Allwinner sun7i (A20) Family
Apr 08 22:25:50 bananapi kernel: Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
Apr 08 22:25:50 bananapi kernel:  unwind_backtrace from show_stack+0x11/0x14
Apr 08 22:25:50 bananapi kernel:  show_stack from dump_stack_lvl+0x2b/0x34
Apr 08 22:25:50 bananapi kernel:  dump_stack_lvl from __warn+0xb5/0xe0
Apr 08 22:25:50 bananapi kernel:  __warn from warn_slowpath_fmt+0xfb/0x104
Apr 08 22:25:50 bananapi kernel:  warn_slowpath_fmt from kthread_park+0xad/0xdc
Apr 08 22:25:50 bananapi kernel:  kthread_park from mt76u_stop_tx+0x1a9/0x1f8 [mt76_usb]
Apr 08 22:25:50 bananapi kernel:  mt76u_stop_tx [mt76_usb] from mt7921u_mac_reset+0x5b/0x138 [mt7921u]
Apr 08 22:25:50 bananapi kernel:  mt7921u_mac_reset [mt7921u] from mt7921_mac_reset_work+0x93/0x168 [mt7921_common]
Apr 08 22:25:50 bananapi kernel:  mt7921_mac_reset_work [mt7921_common] from process_one_work+0x11b/0x2d0
Apr 08 22:25:50 bananapi kernel:  process_one_work from worker_thread+0x21f/0x358
Apr 08 22:25:50 bananapi kernel:  worker_thread from kthread+0xbf/0xd4
Apr 08 22:25:50 bananapi kernel:  kthread from ret_from_fork+0x11/0x1c
Apr 08 22:25:50 bananapi kernel: Exception stack(0xf1d91fb0 to 0xf1d91ff8)
Apr 08 22:25:50 bananapi kernel: 1fa0:                                     00000000 00000000 00000000 00000000
Apr 08 22:25:50 bananapi kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Apr 08 22:25:50 bananapi kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Apr 08 22:25:50 bananapi kernel: ---[ end trace 0000000000000000 ]---
Apr 08 22:25:50 bananapi kernel: mt7921u 3-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a
Apr 08 22:25:50 bananapi kernel: mt7921u 3-1:1.0: WM Firmware Version: ____010000, Build Time: 20241106151045

######### I killed and restarted hostapd here #######

Apr 08 22:26:41 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Lost carrier               
Apr 08 22:26:41 bananapi systemd-networkd[241]: wlxe84e06afb8f1: DHCPv6 lease lost          
Apr 08 22:26:41 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Link DOWN                  
Apr 08 22:26:41 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Link UP                    
Apr 08 22:26:44 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Link DOWN                  
Apr 08 22:26:44 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Link UP                    
Apr 08 22:26:50 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Gained carrier             
Apr 08 22:26:52 bananapi systemd-networkd[241]: wlxe84e06afb8f1: Gained IPv6LL
Apr 08 22:27:15 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8150 failed:-110
Apr 08 22:27:19 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8154 failed:-110
Apr 08 22:27:22 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8158 failed:-110
Apr 08 22:27:26 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:815c failed:-110
Apr 08 22:27:29 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8160 failed:-110
Apr 08 22:27:33 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8164 failed:-110
Apr 08 22:27:37 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8168 failed:-110
Apr 08 22:27:40 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:816c failed:-110
Apr 08 22:27:44 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:811c failed:-110
Apr 08 22:27:47 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8178 failed:-110
Apr 08 22:27:51 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8250 failed:-110
Apr 08 22:27:54 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8254 failed:-110
Apr 08 22:27:58 bananapi kernel: mt7921u 3-1:1.0: vendor request req:63 off:8258 failed:-110

```

morrownr, 2025-04-08:
> In what platform are you running hostapd?

RasPiOS 64 bit, kernel 6.6 which is what it ships with. hostapd 2.11 which it also ships with.

Did you check the hostapd ,config to make sure everything you need is active?

A lot of people that stop by here have APs based on hostapd and adapters with mt7921 chips. I'm not seeing any reports similar to this. Is your hardware 64 bit capable? Do you have options for the distro that you use?

M-y, 2025-04-08:
> Did you check the hostapd ,config to make sure everything you need is active?

hostapd config is same all of the time, it is in my first message.

> Is your hardware 64 bit capable? Do you have options for the distro that you use?

No, this is not an arm64 device. This is 32 bit. You can check the device in here: https://wiki.banana-pi.org/Banana_Pi_BPI-M1
This page also shows a variety of distros. I can test anyone. Vanilla Debian? Openwrt?

morrownr, 2025-04-09:
> hostapd config is same all of the time, it is in my first message.

I am not talking about hostapd.conf that is used to set runtime configuration. I am talking about `.config` that is used to set compile time configuration. Let me paste my hostapd compile guide at the end of the message.

> I can test anyone. Vanilla Debian? Openwrt?

Vanilla Debian is too old at this point. OpenWRT might be the best option since Mediatek devs spend time at OpenWRT. Is 24.10 available? If it is, give it a try. Do you know how to install the mt7921u driver?

morrownr, 2025-04-09:
Here is my How to Compile hostapd Guide. I have not needed to use it for about a year so I might need to look at the recommendations for `.config` active lines but it should work for a WiFi 6e adapter.

```
2024-05-05 (cinco de mayo)

Compile Latest hostapd

Purpose: If you need the very latest capabilities, you may need to
compile hostapd.

This guide was tested on a Raspberry Pi 4B using Raspberry Pi OS.

Install hostapd (if not already installed):

```
$ sudo apt install hostapd
```

Activate hostapd (if not already activated):

```
$ sudo systemctl unmask hostapd
$ sudo systemctl enable hostapd
```

Install git (if not already installed):

```
$ sudo apt install git
```

Clone hostapd:

Note: I use a directory call `src` in my home directory.

```
$ mkdir -p ~/src
```

Move to `src` directory:

```
$ cd ~/src
```

Download hostap source:

```
$ git clone git://w1.fi/srv/git/hostap.git
```

Move to hostapd directory:

```
$ cd hostap/hostapd
```

Copy default settings to .config:

```
$ cp defconfig .config
```

Edit .config:

```
$ nano .config
or
$ geany .config
or
  use your favorite Linux text editor
```

Activate the appropriate lines: (uncomment, if not already uncommented)

Note: * means that the entry was commented out in the default .config.

CONFIG_DRIVER_HOSTAP=y
CONFIG_DRIVER_NL80211=y
CONFIG_LIBNL32=y
CONFIG_RSN_PREAUTH=y
CONFIG_OCV=y
CONFIG_EAP=y
CONFIG_ERP=y
CONFIG_EAP_MD5=y
CONFIG_EAP_TLS=y
CONFIG_EAP_MSCHAPV2=y
CONFIG_EAP_PEAP=y
CONFIG_EAP_GTC=y
CONFIG_EAP_TTLS=y
CONFIG_IPV6=y
CONFIG_IEEE80211R=y
CONFIG_WNM=y
CONFIG_IEEE80211AC=y
CONFIG_IEEE80211AX=y
CONFIG_IEEE80211BE=y
CONFIG_SAE=y
CONFIG_SAE_PK=y
CONFIG_NO_RANDOM_POOL=y
CONFIG_GETRANDOM=y
CONFIG_ELOOP_EPOLL=y
CONFIG_ACS=y
CONFIG_MBO=y
CONFIG_NO_TKIP=y

Save the file

Get and install dependencies

```
$ apt show hostapd
```

Pre-Depends: init-system-helpers (>= 1.54~)
Depends: libc6 (>= 2.34), libnl-3-200 (>= 3.2.7), libnl-genl-3-200 (>= 3.2.7), libnl-route-3-200 (>= 3.2.7), libssl3 (>= 3.0.0~~alpha1), lsb-base

Note: Some of the below packages may already be installed.

```
$ sudo apt install -y build-essential init-system-helpers libc6 lsb-base

$ sudo apt install -y libnl-3-200 libnl-genl-3-200 libnl-route-3-200

$ sudo apt install -y libnl-3-dev libnl-genl-3-dev libssl-dev

```

Compile hostapd:

```
$ make
```

Copy hostapd executable over installed executable:

```
$ sudo cp hostapd_cli /usr/sbin/hostapd_cli

$ sudo systemctl disable hostapd

$ sudo reboot

$ cd ~/src/hostap/hostapd

$ sudo cp hostapd /usr/sbin/hostapd

$ sudo systemctl enable hostapd

$ sudo reboot
```

Check:

```
$ hostapd -v
```

-----

M-y, 2025-04-09:
I was used the .config from your guide [here](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md). But looks like your above comment is a bit different. These are the differences:
```
CONFIG_OCV=y
CONFIG_IEEE80211BE=y
CONFIG_MBO=y
```

M-y, 2025-04-09:
> Vanilla Debian is too old at this point. OpenWRT might be the best option since Mediatek devs spend time at OpenWRT. Is 24.10 available? If it is, give it a try. Do you know how to install the mt7921u driver?

Looks like there is a [banana pi page](https://openwrt.org/toh/lemaker/banana_pi). I am investigating OpenWRT now. If you have any information about installing mt7921u driver I would appreciate.

morrownr, 2025-04-09:
>  If you have any information about installing mt7921u driver I would appreciate.

Sure. I have been using OpenWRT for years. 

OpenWRT has to maintain a very small footprint due to resource constraints with most WiFi routers so drivers for many things are not included in the image for your hardware. If you want to use an adapter with the mt7921au chip, use Luci (web interface) or the command line (SSH) to install the following package. It will include the driver and firmware.

kmod-mt7921u

If you are new to OpenWRT and do not understand that, get back to me when you need to do it.

morrownr, 2025-04-09:
OpenWRT > Network > Wireless > Country Code

Make sure you set your country code.

>  I left all the options default other than these:

Well, there is more that needs to be set. How is internet coming into your Pi? Ethernet?

morrownr, 2025-04-09:
> Yes ethernet

Okay, so connectivity is via your ethernet port and then you have a USB WiFi adapter acting as an AP.

How are you working with OpenWRT? Luci? I need to know so as to know how to help.

OpenWRT > Network > Interfaces

How many interfaces are showing and what are the names?

morrownr, 2025-04-09:
> hostapd

Don't install that as OpenWRT already uses a limited version of hostapd. Not sure how to clean this up. You may need to start clean again.

Yes, we are learning. OpenWRT is really good and it is fun but it is not a standard Linux distro. It is designed to run wifi routers. It is small and takes a little time to understand.

M-y, 2025-04-09:
> Make sure you set your country code.

OK. set

> How are you working with OpenWRT? Luci? I need to know so as to know how to help.

Using Luci, have ssh access if you need.

> OpenWRT > Network > Interfaces
> How many interfaces are showing and what are the names?

[image]

[image]

> Don't install that as OpenWRT already uses a limited version of hostapd. Not sure how to clean this up. You may need to start clean again.

I can't select WPA2-PSK or any other encryption. There is a warning in Luci to install hostapd if you want to select an encrpyption. Now I deleted the hostapd package from System > Software. There is hostapd-common package also, but I can't delete it:

```
 * print_dependents_warning: Package hostapd-common is depended upon by packages:
 * print_dependents_warning: 	kmod-mac80211
 * print_dependents_warning: These might cease to work if package hostapd-common is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
```

Is this enough or start over?

morrownr, 2025-04-09:
> I can't select WPA2-PSK or any other encryption. There is a warning in Luci to install hostapd if you want to select an encrpyption.

That is a new one on me. Not sure at this moment what is going on.Go ahead and reinstall hostapd for now.

As far as Interfaces go, you need a `wan` interface and a `wan6` interface. Both should be set to device `wan`.

M-y, 2025-04-09:
> As far as Interfaces go, you need a wan interface and a wan6 interface. Both should be set to device wan.

How can I create these interfaces. I don't have a device named wan in the list.

morrownr, 2025-04-09:
OpenWRT > Network > Interfaces > Add new interfaces

M-y, 2025-04-09:
Selected protocol as DHCP Client and device as eth0. Is this what you want? What to do with this interface?

[image]

morrownr, 2025-04-09:
[image]

Device needs to be `wan`

M-y, 2025-04-09:
I don't have a device named wan. 

[image]

morrownr, 2025-04-09:
Oh, rats. You are correct. I was pulling from one of my real wifi routers. Let me see...

morrownr, 2025-04-09:
The issue is that the pi's don't have a switch like a router. What we need is a bridged, not routed, setup. Basically we need to change the fixed ip address of the pi to one that will be part of the network that we are bridging to:

Main router 192.168.1.1 >> Pi (default is 192.168.1.1) but we need to change that to something like 192.168.1.4 and connect the pi to the main router via ethernet.

I need to be out my office for the rest of the day so maybe I can continue tomorrow. I'll look at setting my Raspi4b up so I can follow along.

M-y, 2025-04-09:
Yes I have already done that by giving a static IP to pi. It is 192.168.1.120. I am reaching the LuCI interface by this IP.

M-y, 2025-04-14:
I have tested the usb wifi adapter as client using wpa_supplicant. These vendor request failed errors also appears when some iperf tests made. So this error is not corresponding to AP mode using hostapd.

morrownr, 2025-04-14:
I'm a little behind due to illness but I am working on a couple of other things that will set me up to do more testing on this issue. What USB WiFi adapter do you have?

M-y, 2025-04-15:
@morrownr I hope you get well soon. I am continuing tests also. 
My device is EDUP EP-AX1672

morrownr, 2025-05-08:
@M-y 

As I have been testing my AXML, AXM and EP-AX1672, I have not run into this. Have you made a recent progress?