morrownr/USB-WiFi #595: Advice: Realtek chipset with BLE & wifi coexistence

mgonellabolduc, 2025-04-15:
Hi,

I'm looking at cheap usb dongles based on realtek chipsets.
The thing is, I need one that has bluetooth and wifi coexistence (meaning, no mode-switching both functions need to be working at the same time).

Is there one chipset model that I should be looking at with a reliable driver, j in-kernel?
I'm not sure the latest RTL8821AU in-kernel driver supports wifi and bluetooth at the same time and I could not find any info about this.

Thank you for the support.
Mikael

morrownr, 2025-04-15:
Hi @mgonellabolduc 

>  I need one that has bluetooth and wifi coexistence (meaning, no mode-switching both functions need to be working at the same time).

Can you elaborate? I am not one to recommend combo adapters for a lot of reasons but... if there is a good reason, I'd like to know.

> Is there one chipset model that I should be looking at with a reliable driver, preferably in-kernel?

https://store.rokland.com/collections/802-11ac-wi-fi-clients-receivers/products/alfa-awus036eacs-low-profile-802-11ac-ac600-dual-band-wifi-bt-usb-adapter

That adapter is based on the rtl8821cu chip. I used one to test while working on rtw88 during 2024. If you are using kernel 6.12 or later, you should find it to be a plug and play experience as well as stable. Please understand that I am not recommending any adapter with that chipset, only that specific adapter. I have seen many problems that users have experienced with that chipset over the years but Alfa seems to have done a great job with this specific adapter.

Cheers

mgonellabolduc, 2025-04-15:
Hi @morrownr ,

Thank you for the quick reply.

> Can you elaborate? I am not one to recommend combo adapters for a lot of reasons but... if there is a good reason, I'd like to know.

It's for cost savings on an embedded product. If I need to buy 2 adapters well I end up doubling the cost for this specific function.

> That adapter is based on the rtl8821cu chip

Just be sure that I'm clear, with this specific adapter, you can at the same time connect to a wifi network and get a reasonable bandwith and use bluetooth as well?

morrownr, 2025-04-15:
> Just be sure that I'm clear, with this specific adapter, you can at the same time connect to a wifi network and get a reasonable bandwith and use bluetooth as well?

Give me test you want to run and I'll run it.

I understand cost savings. I also understand the various problems with combo adapters. I could write novel with the various problems I've seen user have. Remember, wifi band 1 (2.4 GHz) and bluetooth use the same band. It can slow things down if you intend to use band 1. I would only use band 2 (5 GHz) if bluetooth is going to be in regular use or good throughput is needed. Also, make sure the adapter is far away from any USB3 cables or connections as they will emit interference.

Many adapter makers have made mistakes when they loaded the eeprom. They would get the antenna configuration wrong and it would cause problems. So, I am only recommending this specific adapter... and things go downhill from there. If it were me, I would look elsewhere for cost savings but if you want me to do some testing as I have time, I will.

mgonellabolduc, 2025-04-15:
> Give me test you want to run and I'll run it.

Running a kernel 6.14, the test I intend to do with it is as follows:

- Wifi router 25 feets away, with clear view.
- Cellphone 8 feets away, with clear view
- Adapter connects to wifi using 2.4 ghz and cellphone using bluetooth
- Cellphone streams BL classic audio music to the device
- Device downloads a file using wifi with wget --limit-switch at 50KB/s
- Device pings the wifi router with fping with 1s period and 500ms timeout

If the audio does not skip, the wifi does not disconnect and the fping does not timeout it's good enough even if the download rate is low while streaming the music. 

It's a complicated setup for a quick test but since you've been working this adapter for a while if you think it might work well it's worthwhile for me to try it out.

morrownr, 2025-04-16:
> It's a complicated setup for a quick test but since you've been working with this adapter for a while if you think it might work well it's worthwhile for me to buy it and try it out.

That would take some time to set up. 

> Adapter connects to wifi using 2.4 ghz and cellphone using bluetooth

Ouch. That is a killer. I'll give you maybe a 25% chance of the suggested adapter meeting your expectations. 

From Google's AI:

Low Wi-Fi speed with a combo USB Wi-Fi adapter can be caused by several factors, including the USB port, the frequency band, and driver issues. Ensure the adapter is plugged into a USB 3.0 port, try connecting to a 5GHz band if supported, and update or roll back the drivers if necessary. 

It goes on to mention bluetooth and wifi both using 2.4 GHz and that the interference can cause low throughput. 

Is this adapter for a laptop or desktop? I might be able to make some suggestions for things for you to explore if I know what type of system you have.

mgonellabolduc, 2025-04-17:
> Is this adapter for a laptop or desktop? I might be able to make some suggestions for things for you to explore if I know what type of system you have.

It's for an arm embedded computer, like a raspberry pi but custom design. From what I'm reading here and your answers to this topic this looks like a dead end that might cause a ton of problems even if the bandwith required is low.

morrownr, 2025-04-17:
Okay, so it is for a sbc. For my own use, I would not consider this setup if using band 1 (2.4 GHz) for wifi.

I assume the sbc will be more or less stationary while in use. If cost is your main concern, then let me think about I would do if I wanted to add good bluetooth and wifi capability to an sba... what kernels are available for the distro to be used with the sbc?

morrownr, 2025-04-17:
> Both of these will cost way more than a single adapter.

The following 2 adapters cost no more than the combo adapter that I recommended. Both are plug and play.

USB WiFi adapter based on the mt7921au chip. Cost $9.35
https://www.aliexpress.us/item/3256805749323751.html?gatewayAdapt=glo2usa4itemAdapt

Edup Bluetooth adapter. Cost $10
https://www.amazon.com/dp/B09KG7QQ5V

I have the bluetooth adapter. It has longer range than most bluetooth adapters. It works well. The wifi adapter is a cheapy but gets good reports and is shown in The Plug and Play List.

mgonellabolduc, 2025-04-17:
Alright, I will look into getting 2x adapters instead. Thank you for the advice.

closed 2025-04-17 by mgonellabolduc

morrownr, 2025-04-17:
Let me know how it goes.

Remember what I said about using 5 GHz instead of 2.4 Ghz for the wifi. That would be best and it can also help, if you get a USB3 adapter, like the one above, that you run it in a USB2 port. USB3 connections emit interference in the 2.4 GHz band and that can interfere with bluetooth. These issues are not specific to Linux. There are a lot of things to know to have a smooth running wireless system when you add bluetooth to the mix.

Good luck.