morrownr/USB-WiFi #596: [Help]: mt7921u AP not working

jfargen, 2025-04-18:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

# uname -a Linux rpi5-8g-0 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

### lsusb

# sudo lsusb Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 002: ID 0781:5572 SanDisk Corp. Cruzer Switch Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

# sudo rfkill list all 0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy1: Wireless LAN 	Soft blocked: no 	Hard blocked: no 2: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
# sudo dkms status
zfs/2.2.3, 6.1.0-32-arm64, aarch64: installed
zfs/2.2.3, 6.1.21-v8+, aarch64: installed
zfs/2.2.3, 6.6.74+rpt-rpi-2712, aarch64: installed
zfs/2.2.3, 6.6.74+rpt-rpi-v8, aarch64: installed
```

### iw

```text
# sudo iw dev
phy#0
	Interface wlan1
		ifindex 4
		wdev 0x1
		addr e8:4e:06:ae:d7:43
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	54	0	0	0	0	6936		61
phy#1
	Interface wlan0
		ifindex 3
		wdev 0x100000001
		addr d8:3a:dd:d2:16:75
		type managed
		channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz
		txpower 31.00 dBm
```

### What happened?

Grabbed the latest version of https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6e.conf and set the interface=wlan1 to try out the new EDUP AX3000. When starting hostapd it is not working and presents these errors.

# hostapd -d /etc/hostapd/hostapd.conf 
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
Line 62: unknown configuration item 'he_6ghz_reg_pwr_type'
ctrl_interface_group=0
Line 104: unknown configuration item 'he_6ghz_rx_ant_pat'
Line 202: unknown configuration item 'ieee80211ax'
Line 205: unknown configuration item 'he_oper_chwidth'
Line 208: unknown configuration item 'he_oper_centr_freq_seg0_idx'
Line 212: unknown configuration item 'he_bss_color'
Line 214: unknown configuration item 'he_mu_beamformer'
Line 215: unknown configuration item 'he_su_beamformee'
Line 216: unknown configuration item 'he_su_beamformer'
SAE: Enabling SAE H2E on 6 GHz
9 errors found in configuration file '/etc/hostapd/hostapd.conf'
Failed to set up interface with /etc/hostapd/hostapd.conf
hostapd_init: free iface 0x555659cd0e70
Failed to initialize interface

# hostapd -v
hostapd v2.12-devel-hostap_2_11-835-g9bc3fd041

This version of the config does work:
# cat /etc/hostapd/hostapd-working.conf 
interface=wlan1
driver=nl80211
ssid=redacted
hw_mode=g
channel=0
#country=DO
ieee80211n=1
wmm_enabled=1
auth_algs=1
wpa=2
wpa_passphrase=redacted
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

morrownr, 2025-04-19:
Hi @jfargen 

The example 7921u hostapd.conf file on the main menu should work fine:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

The file you are using has not had adequate testing as it is a real challenge to test band 4 (6 GHz) in the US because of our laws. I think it should work but I need users from countries that have reasonable laws, like the EU, to test and report.

The problem I am seeing in your post appears to be that the version of hostapd that you are using was not compiled with 6 GHz support. Can you expound on where you got your hostapd and what country you are in?

jfargen, 2025-04-22:
First I tried hostapd that comes with Bookworm 12.10 and was seeing a similar error and I am in the US. Then pulled down the latest source code from https://w1.fi/hostap.git and compiled hostapd.