morrownr/USB-WiFi #598: How do I make a patch for a new device ID?

vincegre, 2025-04-21:
Hi

I have an USB stick for Wifi 7 that is a new product soon to be released. It uses a chipset already supported by linux if I add it's USB IDs manually in my system.

The device shows up like that:
`Bus 002 Device 003: ID 0846:9072`

How can I add it in kernel so it's automatically recognised and supported by linux in future ?

Thanks

Vincèn

morrownr, 2025-04-21:
Hi @vincegre 

Great question. We have time before the next window opens up for new patches so let's step through this in a deliberate manner and maybe we can make a guide for others to follow while we are at it.

When new capability or changes or deletions to the Linux mainline kernel are needed, one or more patches are required. In this case, one short patch should get the job done. Now we need to know how to make the patch and where to send it.

Where do we send wireless patches? The answer is that wireless patches go to the `linux-wireless` list. Here is a link to instructions on how to subscribe to the linux-wireless list:

https://wireless.docs.kernel.org/en/latest/en/developers/mailinglists.html

I recommend that you subscribe so that you can learn how things work. There are two things to know before trying to subscribe:

1. Your email app MUST BE in PLAIN TEXT mode. Your email will be rejected if you are not using PLAIN TEXT mode.
2. It is a good idea for your email app to support filtering to folders/tabs so that you can set a filter to send all emails from this list to a specific folder/tab. There is a lot of traffic on the list.

Once you are subscribed and are becoming familiar with linux-wireless, let me know and we can continue with how to make the patch.

vincegre, 2025-07-03:
Hi @morrownr 

Thanks for your great help but unhappy I really don't have time to go through all the cumbersome process in this moment :(

Vincèn