morrownr/USB-WiFi #600: [Help]: MT7922e Wi-Fi6 speed not ideal

Williamclouf, 2025-04-27:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux 6.1.99 #32 SMP Mon Jan 20 15:50:32 CST 2025 aarch64 aarch64 aarch64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 006 Device 002: ID 13d3:3568 IMC Networks Wireless_Device

### rfkill

0: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
/
```

### iw

```text
phy#0
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr 14:13:33:6e:ee:15
		ssid mt7922
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	11304	0	0	0	60	4760957		17923
```

### What happened?

I am using an MT7922 PCIe card, serving as an AP, everything is going well.
But I heard that the mt79 driver has a 'txpower 3.00 dBm' issue. In the link https://github.com/openwrt/mt76/issues/935#issuecomment-2688143334, your comment indicates that iperf3 can achieve a speed of 1.24 Gbits/sec. I want to improve my bitrate, but I don't know if it's a txpower issue or a problem with my configuration. Below is some output.

iperf3 test, done through 1 wall with about 6 m/20 feet of distance.
Even when I conduct an iperf3 download test about 1 feet away from the antenna, the maximum speed is only around 400 Mbps. 
```
$ iperf3 -c 192.168.0.1
Connecting to host 192.168.0.1, port 5201
[  5] local 192.168.0.177 port 54220 connected to 192.168.0.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  38.2 MBytes   319 Mbits/sec                  
[  5]   1.00-2.00   sec  38.4 MBytes   323 Mbits/sec                  
[  5]   2.00-3.00   sec  37.6 MBytes   315 Mbits/sec                  
[  5]   3.00-4.00   sec  37.8 MBytes   318 Mbits/sec                  
[  5]   4.00-5.00   sec  36.0 MBytes   302 Mbits/sec                  
[  5]   5.00-6.00   sec  38.2 MBytes   320 Mbits/sec                  
[  5]   6.00-7.00   sec  38.6 MBytes   323 Mbits/sec                  
[  5]   7.00-8.00   sec  36.1 MBytes   303 Mbits/sec                  
[  5]   8.00-9.00   sec  38.5 MBytes   324 Mbits/sec                  
[  5]   9.00-10.01  sec  36.8 MBytes   307 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.01  sec   376 MBytes   315 Mbits/sec                  sender
[  5]   0.00-10.03  sec   374 MBytes   313 Mbits/sec                  receiver
```

Current Network Information on the macOS client, operates in 802.11ax mode.

```
mt7922:
  PHY Mode:	802.11ax
  Channel:	36
  Network Type:	Infrastructure
  Security:	WPA2 Personal
  Signal / Noise:	-65 dBm / -91 dBm
  Transmit Rate:	576
  MCS Index:	4
```

hostapd has been upgraded to the latest following `AP_Mode/Upgrade_hostapd.md`

```
$ hostapd -v
hostapd v2.12-devel-hostap_2_11-838-g9fa657a73
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2024, Jouni Malinen <j@w1.fi> and contributors
```

Here is my hostapd-wifi6.conf

```conf
interface=wlan0
driver=nl80211
ssid=mt7922
wpa_passphrase=MyPassword
hw_mode=a
channel=36
country_code=US
ieee80211d=1
logger_syslog=-1
logger_syslog_level=2
beacon_int=100
dtim_period=2
skip_inactivity_poll=1
auth_algs=1
max_num_sta=16
macaddr_acl=0
ignore_broadcast_ssid=0
okc=1
wpa=2
wpa_pairwise=CCMP
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42
```

morrownr, 2025-04-27:
Hi @Williamclouf 

> I am using an MT7922 PCIe card...

I have one of those as well.

> But I heard that the mt79 driver has a 'txpower 3.00 dBm' issue.

It is a cosmetic issue. The driver in use is the mt7921e driver. The driver is using the max allowed txpower based on country and band. The output with `iw dev` is wrong. There was some recent work correct this but I don't think it is complete.

> ...your comment indicates that iperf3 can achieve a speed of 1.24 Gbits/sec.

It can perform like that in managed mode as it can do 160 MHz channel width in managed mode. AP mode is limited to 80 MHz channel width so some around 600 Mbps should be around the max we should see. Don't ask, I don't know why. The mt7925 chip should 160 Mhz channel width in managed and AP mode. I have a M.2 card with the mt7925 and it is fast but I have not tested it in AP mode yet.

> Even when I conduct an iperf3 download test about 1 feet away from the antenna, the maximum speed is only around 400 Mbps.

I wouldn't test that close as being too close can cause issues. Maybe no closer than about 3 feet.

> Signal / Noise:	-65 dBm / -91 dBm

-65 dBm is not good at all. Try installing `wavemon` on one of your client systems. It helps you watch the signal while you are doing things... including running iperf3. What kind of antenna are you using?

I'll take a look at hostapd.conf.

morrownr, 2025-04-29:
@Williamclouf 

I took a look at the following lines and posted what I think the lines should be:

# mt7921 and mt7922  (HT capabilities 0x9ff)
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

# mt7922   (VHT capabilities 0x339071f6)
#vht_capab=[MAX-MPDU-11454][VHT160][RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7]

I have not tested them so your mileage could vary.

In looking things over again, that signal level is the main thing I would investigate. I don't have a bird's eye view of your setup so I would need to know more details to offer more.

sam8641, 2025-05-05:
Try different WiFi antennas. The antenna that came with Asus Wifi motherboards appears to work nicely for my MT7922 card.

I recently got MT7922 PCI-E add on card in Desktop tower, Ryzen 5700x, Asrock steel legend X570, Linux kernel 6.12.22, Debian Trixie.

> AP mode is limited to 80 MHz

I may have found a possible reason for 80 MHz AP limitation on MT7922. looks like 160 MHz is missing in Linux kernel driver.
drivers/net/wireless/mediatek/mt76/mt7921/main.c, line 86
.. mt7921_init_he_caps, case NL80211_IFTYPE_AP:
```
if (is_mt7922(phy->mt76->dev)) {
	he_cap_elem->phy_cap_info[0] |=
		IEEE80211_HE_PHY_CAP0_CHANNEL_WIDTH_SET_160MHZ_IN_5G;
}
```
I added this for myself and recompiled, then this allows 160MHz to work in AP mode.

/etc/modprobe.d/options.conf: options mt7921-common disable_clc=1
With CLC disabled, The 6 GHz signal becomes slightly stronger, and 5 GHz channels 169, 173, 177 now becomes available in US region, with "NO-IR" issue.

Client device is Android Pixel 6a.
Using 6 GHz, channel 133 (143 at 160 MHz), 2 feet distance.
iperf3 -P8, Pixel 6a to MT7922: 1.14 Gbit/sec
iperf3 -P8, MT7922 to Pixel 6a: 1.47 Gbit/sec. At 10 feet, 0.8 Gbit/sec.

In US region, the "NO-IR" problem for the entire 6 GHz, and radar channels don't work on MT7922 AP mode, makes hostapd unable to find sufficient channels for 160 MHz, unless the wireless regulatory or wireless region is changed. Channel 149 to 177 (163 at 160 MHz), with vht_capab=[VHT160], do work for me, but only if "NO-IR" was bypassed.

morrownr, 2025-05-05:
> In US region, the "NO-IR" problem for the entire 6 GHz, and radar channels don't work on MT7922 AP mode, makes hostapd unable to find sufficient channels for 160 MHz, unless the wireless regulatory or wireless region is changed.

https://patchwork.kernel.org/project/linux-wireless/patch/000201db8822$98f28da0$cad7a8e0$@gmail.com/

That stinks. Hopefully this can be changed at some point.

> I may have found a possible reason for 80 MHz AP limitation on MT7922. looks like 160 MHz is missing in Linux kernel driver.
drivers/net/wireless/mediatek/mt76/mt7921/main.c, line 86
.. mt7921_init_he_caps, case NL80211_IFTYPE_AP:

if (is_mt7922(phy->mt76->dev)) {
	he_cap_elem->phy_cap_info[0] |=
		IEEE80211_HE_PHY_CAP0_CHANNEL_WIDTH_SET_160MHZ_IN_5G;
}
I added this for myself and recompiled, then this allows 160MHz to work in AP mode.

Have you considered sending a patch to linux-wireless?

> With CLC disabled, The 6 GHz signal becomes slightly stronger, and 5 GHz channels 169, 173, 177 now becomes available in US region, with "NO-IR" issue.

I have this clc setting in my Frequently Asked Questions on the Main Menu. Maybe I should add your results?

sam8641, 2025-05-06:
Just tried different wifi channels, and I can see a speed problem on my MT7922 on channel 7 (11 at 80 MHz, 15 at 160 MHz).

5 GHz, MT7922 to client:
36 (42 at 80 MHz): 0.5 to 0.64 Gbps, 10 nearby AP.
149 (155 at 80 MHz): 0.3 to 0.6 Gbps, 8 nearby AP.
165 (171 at 80 MHz): 0.68 Gbps.
149 (163 at 160 MHz): 0.5 to 1.3 Gbps.
165 (163 at 160 MHz): 0.8 to 1.3 Gbps, 10% of time: 80 Mhz transmit.

6 Ghz, MT7922 to client:
5 (11 at 80 MHz): 0.3 Gbps, stuck at transmit 40 MHz, receive 80 MHz
5 (15 at 160 Mhz): 0.3 Gbps, stuck at transmit 40 MHz, receive 160 MHz.
21 (15 at 160 Mhz): 0.65 Gbps, stuck at transmit 80 MHz, receive 160 MHz.
37 (47 at 160 MHz): 1.35 Gbps.
69 (79 at 160 MHz): 1.36 Gbps.
133 (143 at 160 MHz): 1.35 Gbps.

6 Ghz, MT7921u to client:
5 (11 at 80 MHz): 0.64 Gbps.

Looks like my MT7922 have a bad channel, and my MT7921u USB wifi on the same channel works fine at 80 MHz speed. Other channels on both my WiFi devices works fine as expected.

> Have you considered sending a patch to linux-wireless?

I may think about it. Will likely take some time for me as I have a history of only 1 Linux patch to a different driver about 8 years ago. I may look at what to do on how to to send a patch.

> Maybe I should add your results?

If you want to, yes.

eddiegb, 2025-05-06:
> /etc/modprobe.d/options.conf: options mt7921-common disable_clc=1 With CLC disabled, The 6 GHz signal becomes slightly stronger, and 5 GHz channels 169, 173, 177 now becomes available in US region, with "NO-IR" issue.

@sam8641 thank you SO much for mentioning about this option. I've been struggling for days to figure out why the range was so abysmal with my MT7922 M.2 card as an AP.  After seeing your comment I tried disabling CLC for the mt7921_common module and this immediately resulted in a significant increase in range for me in the 5Ghz band. I was finally able to associate clients 1 level downstairs from where the AP is.

Previously I was seeing between -80dBm and -90dBm for these clients, so naturally they refused to associate.  By simply disabling CLC this reduced to between -60dBm and -70dBm which was enough for them to consistently be able to associate and achieve a usable connection. And on the same floor level as the AP signal strength is very good everywhere that is within a reasonable range (for 5Ghz, won't be using 6Ghz for the time being).

This is in the UK ("gb" regulatory domain), kernel 6.6 on x86_64, with latest firmware for the MT7922.

For anyone unfamiliar with kernel matters wanting to disable CLC you need to find out how to set options for kernel modules for your Linux distro. Typically this will be by adding the following line to a file in /etc/modprobe.d/:

options mt7921_common disable_clc=1

then run: depmod -a
and either unload/load all the mt7921 related modules or reboot

fs30000, 2025-05-31:
I have a MT7922B22M, that works better than mt7925 for AP. My conf:

````
#
interface=wlP1p1s0
bridge=lanbr0

driver=nl80211
country_code=YOUR-CONTRY-CODE-HERE

ssid=1
utf8_ssid=1
multi_ap=0
ignore_broadcast_ssid=0
stationary_ap=1

# Auth
wpa_passphrase=password
sae_require_mfp=1
sae_pwe=2
auth_algs=1
wpa=2
wpa_pairwise=CCMP
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=SAE
okc=1

# misc
ieee80211w=2

ctrl_interface=/var/run/hostapd
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
uapsd_advertisement_enabled=1
group_mgmt_cipher=AES-128-CMAC
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
tx_queue_data2_burst=2.0

# 5Ghz
hw_mode=a

ieee80211d=1
ieee80211h=1
beacon_int=100
channel=36
chanlist=36

# N
ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

# AC
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

# AX
ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42
####he_bss_color=128
he_spr_sr_control=3
he_default_pe_duration=4
he_rts_threshold=1023
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_be_ecwmin=9
he_mu_edca_ac_be_ecwmax=10
he_mu_edca_ac_be_timer=255
he_mu_edca_ac_bk_aifsn=15
he_mu_edca_ac_bk_aci=1
he_mu_edca_ac_bk_ecwmin=9
he_mu_edca_ac_bk_ecwmax=10
he_mu_edca_ac_bk_timer=255
he_mu_edca_ac_vi_ecwmin=5
he_mu_edca_ac_vi_ecwmax=7
he_mu_edca_ac_vi_aifsn=5
he_mu_edca_ac_vi_aci=2
he_mu_edca_ac_vi_timer=255
he_mu_edca_ac_vo_aifsn=5
he_mu_edca_ac_vo_aci=3
he_mu_edca_ac_vo_ecwmin=5
he_mu_edca_ac_vo_ecwmax=7
he_mu_edca_ac_vo_timer=255
````

sam8641, 2025-12-03:
@guderf More information is needed.
Which kernel, which OpenWRT? `uname -a; cat /etc/openwrt_release`
Which PCI, USB device you have? `lspci -nn; lsusb`, with pciutils, usbutils installed.

Linux Kernel 6.6.104, found in OpenWRT 24.10.3, included my MT7922 160 MHz AP mode patch. No manual patch neccessary for newer Kernel versions.
- https://github.com/gregkh/linux/blob/v6.6.104/drivers/net/wireless/mediatek/mt76/mt7921/main.c Switch branch/tag, then click on history.

I have: "MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter [14c3:0616]", 160 MHz on 6 GHz works ok. This card do not support radar detection, so no DFS channels in AP mode.

guderf, 2025-12-04:
# uname -a; cat /etc/openwrt_release
Linux OpenWrt 6.12.59 #0 SMP Fri Nov 28 09:27:53 2025 ppc GNU/Linux
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r32139-1f879b8839'
DISTRIB_TARGET='mpc85xx/p1020'
DISTRIB_ARCH='powerpc_8548'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r32139-1f879b8839'
DISTRIB_TAINTS=''

# lspci -nn; lsusb
9000:00:00.0 PCI bridge [0604]: Freescale Semiconductor Inc P1020E [1957:0100] (rev 11)
9000:01:00.0 Network controller [0280]: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter [14c3:7922]
a000:02:00.0 PCI bridge [0604]: Freescale Semiconductor Inc P1020E [1957:0100] (rev 11)
a000:03:00.0 Network controller [0280]: Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter [168c:003c]

******* Well basically 5 GHz at 160 MHz is out of the question. It would be amazing if I could get 6 GHz to work. These outputs may help too. Please let me know if you want me to run more commands ******

# iw list
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
[... 681 lines cut ...]
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xb0 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ POWERED_ADDR_CHANGE ]: can change MAC address while up