morrownr/USB-WiFi #601: [Help]: CANNOT GET WIFI 6G TO CONNECT ON LINUX MINT

Ynot70, 2025-04-29:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux odell-XPS-8900 6.8.0-58-generic #60-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 14 18:29:48 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 413c:301a Dell Computer Corp. Dell MS116 Optical Mouse Bus 001 Device 003: ID 413c:2113 Dell Computer Corp. KB216 Wired Keyboard Bus 001 Device 004: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 002: ID 0846:9060 NetGear, Inc. Wireless_Device Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

0: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
nvidia/550.120, 6.8.0-38-generic, x86_64: installed
nvidia/550.120, 6.8.0-58-generic, x86_64: installed
```

### iw

```text
phy#0
	Interface wlx54077d99f58c
		ifindex 3
		wdev 0x1
		addr 54:07:7d:99:f5:8c
		ssid ASMITH6E_5G
		type managed
		channel 48 (5240 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

I'm running Linux Mint kernel 6.8.0-58 using a Netgear AXE3000 A8000 wireless 3.0 USB adapter.  Linux sees adapter and lists 2.4G, 5G, and 6G bands.  I'm able to connect using the 5G band and am getting a 77-80% signal.  Whenever I try to connect on 6G, system will not connect.  Is there no Linux support for 6G?  The computer I'm using is an older Dell XPS 8900 that would not support Windows 11, so I decided to go with Linux instead of discarding a really good, if old, computer.  Internal WYLAN and bluetooth have been disabled in BIOS.

Ynot70, 2025-05-02:
When I run wavemon, I get link quality 76% and Signal level from -53 to as high as -68 dBm

morrownr, 2025-05-02:
> I get link quality 76% and Signal level from -53 to as high as -68 dBm

Is that on 5 GHz?

Is this with a laptop that you are moving around? A little change in signal level is to be expected during use but that is big change. -53 is not bad but -68 is bad. maybe not usable or able to stay connected.

Are you using a PCIe or M.2 card? In a laptop or desktop? Tell me about your antennas. External or internal?

Ynot70, 2025-05-03:
It just dawned on me that signal level is measured in negative numbers.  This is an old Dell XPS 8900 using a netgear AXE3000 A800 USB 3.0 network adapter.  The internal network card does not even support 5G, let alone 6G, so it is disabled in BIOS.  The netgear has a cradle that it mounts in with a three foot cord.  I've added a six foot usb 3.0 extension cord and relocated the cradle and fiddled with the antenas on my router.  Now I'm getting link quality at  99 to100% and signal level from -40 to -41dBm running on 5G.  6G band shows up in network connections but still fails to connect.  If the drivers were wrong the adapter would not work at all, right?

morrownr, 2025-05-03:
> and signal level from -40 to -41dBm

You are smoking now! That is a great signal.

> If the drivers were wrong the adapter would not work at all, right?

No, it is possible for the driver to work perfectly on one band and not work on another but I suspect this is not a driver problem.

Ubuntu has a HWE 6.11 available for you install if you wish. You might give that a try.

Have you updated the firmware?

It might be time to talk about router and the settings. What 6 GHz channel is in use and what country are you in?

Ynot70, 2025-05-03:
Okay, I installed sudo linssid and ran it.  The channel I'm on at 5G is channel 48 at -41 signal.  The bgnac listing showed up and I think that means 802.11ac.  Only 2.4Ghz and 5Ghz channels are listed in the graph and no bgnax.  My 6G channel shows up in wireless connection list and as I said before it refuses to connect.  Does this app support 6g?  I'm located in North Carolina USA.  I have not updated the firmware because I don't have half a clue as to how to do it in Linux.  As far as Netgear is concerned, Linux does not exist.  As you can tell, I'm not the brightest bulb on the Christmas tree and I'm old.

morrownr, 2025-05-03:
> I installed sudo linssid and ran it. Does sudo linssid support 6G?

No, linssid has not been updated for WiFi 6, 6e or 7. It is pretty much useless for this problem.

> I have not updated the USB network adapter firmware because I don't have half a clue as to how to do it in Linux. 

I have a guide for that. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Look for menu item 7:

Firmware - How to install or update firmware for Mediatek and RTW88 based USB WiFi adapters

You have the Netgear adapter with the mt7921au chip so section 3 should be for you. If there is anything you do not understand, let me know.

> Ubuntu has a HWE 6.11 kernel available for you install if you wish. You might give that a try.

You did not address this. Did you install the 6.11 kernel?

> I'm not the brightest bulb on the Christmas tree and I'm old.

That pretty much describes me as well.

>  I'm using a TPLink Archer AXE 75 Tri-Band 6-Stream Wi-Fi 6E router. 

Is the firmware updated in your router?

Ynot70, 2025-05-05:
Updated to 6.11 kernel and nothing changed as far as wifi.  It did pick up my Brother printer and it works.  I've spent hours trying to update the firmware using the exact instructions that I printed off from your link.  Terminal will not create the directories as they are listed.  Says file does not exist.  I'm just about fed up with messing with this. Are you certain that this plug in USB adapter has firmware to update?  I plugged it into my windows desktop and installed the latest software and 6G channel worked like a charm. The LED even lit up on it which it fails to do on Linux, even though it's getting the 5G channel. There is no mention of any firmware update feature on the Netgear website or router login . Linux mint is not an easy windows substitute, it's like screwing around with MSDOS. It still has a LONG way to go.  I'm going to keep using the 5G . It's like using only four cylinders on a V8.

morrownr, 2025-05-05:
> Are you certain that this plug in USB adapter has firmware to update?

Let's see. Run:

$ iw dev

... to get the interface name and then run:

ethtool -i <interface name> 

Make your you replace <interface name> with your interface name. Post the result.

> Linux mint is not an easy windows substitute, it's like screwing around with MSDOS.

Give it a little time. A little familiarity will work wonders. Opening up a terminal is required for very little in this day in time but doing this type of support is easier for me to tell you what to do from a terminal/command line. 

> It's like using only four cylinders on a V8.

You may need to trim your expectations. My experience is that while 6 GHz has a bigger pipeline, it also has seriously reduced range. What kind of speed are you seeing with 5 GHz?

I may have a fix for the 6 GHz. Run the following:

$ iw list

Find the section that looks like the following and post it here:

```
		Frequencies:
			* 5955 MHz [1] (23.0 dBm)
			* 5975 MHz [5] (23.0 dBm)
			* 5995 MHz [9] (23.0 dBm)
			* 6015 MHz [13] (23.0 dBm)
			* 6035 MHz [17] (23.0 dBm)
			* 6055 MHz [21] (23.0 dBm)
			* 6075 MHz [25] (23.0 dBm)
			* 6095 MHz [29] (23.0 dBm)
			* 6115 MHz [33] (23.0 dBm)
			* 6135 MHz [37] (23.0 dBm)
			* 6155 MHz [41] (23.0 dBm)
			* 6175 MHz [45] (23.0 dBm)
			* 6195 MHz [49] (23.0 dBm)
			* 6215 MHz [53] (23.0 dBm)
			* 6235 MHz [57] (23.0 dBm)
			* 6255 MHz [61] (23.0 dBm)
			* 6275 MHz [65] (23.0 dBm)
			* 6295 MHz [69] (23.0 dBm)
			* 6315 MHz [73] (23.0 dBm)
			* 6335 MHz [77] (23.0 dBm)
			* 6355 MHz [81] (23.0 dBm)
			* 6375 MHz [85] (23.0 dBm)
			* 6395 MHz [89] (23.0 dBm)
			* 6415 MHz [93] (23.0 dBm)
			* 6435 MHz [97] (disabled)
			* 6455 MHz [101] (disabled)
			* 6475 MHz [105] (disabled)
			* 6495 MHz [109] (disabled)
			* 6515 MHz [113] (disabled)
			* 6535 MHz [117] (disabled)
			* 6555 MHz [121] (disabled)
			* 6575 MHz [125] (disabled)
			* 6595 MHz [129] (disabled)
			* 6615 MHz [133] (disabled)
			* 6635 MHz [137] (disabled)
			* 6655 MHz [141] (disabled)
			* 6675 MHz [145] (disabled)
			* 6695 MHz [149] (disabled)
			* 6715 MHz [153] (disabled)
			* 6735 MHz [157] (disabled)
			* 6755 MHz [161] (disabled)
			* 6775 MHz [165] (disabled)
			* 6795 MHz [169] (disabled)
			* 6815 MHz [173] (disabled)
			* 6835 MHz [177] (disabled)
			* 6855 MHz [181] (disabled)
			* 6875 MHz [185] (disabled)
			* 6895 MHz [189] (disabled)
			* 6915 MHz [193] (disabled)
			* 6935 MHz [197] (disabled)
			* 6955 MHz [201] (disabled)
			* 6975 MHz [205] (disabled)
			* 6995 MHz [209] (disabled)
			* 7015 MHz [213] (disabled)
			* 7035 MHz [217] (disabled)
			* 7055 MHz [221] (disabled)
			* 7075 MHz [225] (disabled)
			* 7095 MHz [229] (disabled)
			* 7115 MHz [233] (disabled)

```

Ynot70, 2025-05-05:
Interface wlx54077d99f58c
		ifindex 3
		wdev 0x1
		addr 54:07:7d:99:f5:8c
		ssid ASMITH6E_5G
		type managed
		channel 48 (5240 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
                                                                                                              Frequencies:
			* 5955.0 MHz [1] (12.0 dBm) (no IR)
			* 5975.0 MHz [5] (12.0 dBm) (no IR)
			* 5995.0 MHz [9] (12.0 dBm) (no IR)
			* 6015.0 MHz [13] (12.0 dBm) (no IR)
			* 6035.0 MHz [17] (12.0 dBm) (no IR)
			* 6055.0 MHz [21] (12.0 dBm) (no IR)
			* 6075.0 MHz [25] (12.0 dBm) (no IR)
			* 6095.0 MHz [29] (12.0 dBm) (no IR)
			* 6115.0 MHz [33] (12.0 dBm) (no IR)
			* 6135.0 MHz [37] (12.0 dBm) (no IR)
			* 6155.0 MHz [41] (12.0 dBm) (no IR)
			* 6175.0 MHz [45] (12.0 dBm) (no IR)
			* 6195.0 MHz [49] (12.0 dBm) (no IR)
			* 6215.0 MHz [53] (12.0 dBm) (no IR)
			* 6235.0 MHz [57] (12.0 dBm) (no IR)
			* 6255.0 MHz [61] (12.0 dBm) (no IR)
			* 6275.0 MHz [65] (12.0 dBm) (no IR)
			* 6295.0 MHz [69] (12.0 dBm) (no IR)
			* 6315.0 MHz [73] (12.0 dBm) (no IR)
			* 6335.0 MHz [77] (12.0 dBm) (no IR)
			* 6355.0 MHz [81] (12.0 dBm) (no IR)
			* 6375.0 MHz [85] (12.0 dBm) (no IR)
			* 6395.0 MHz [89] (12.0 dBm) (no IR)
			* 6415.0 MHz [93] (12.0 dBm) (no IR)
			* 6435.0 MHz [97] (12.0 dBm) (no IR)
			* 6455.0 MHz [101] (12.0 dBm) (no IR)
			* 6475.0 MHz [105] (12.0 dBm) (no IR)
			* 6495.0 MHz [109] (12.0 dBm) (no IR)
			* 6515.0 MHz [113] (12.0 dBm) (no IR)
			* 6535.0 MHz [117] (12.0 dBm) (no IR)
			* 6555.0 MHz [121] (12.0 dBm) (no IR)
			* 6575.0 MHz [125] (12.0 dBm) (no IR)
			* 6595.0 MHz [129] (12.0 dBm) (no IR)
			* 6615.0 MHz [133] (12.0 dBm) (no IR)
			* 6635.0 MHz [137] (12.0 dBm) (no IR)
			* 6655.0 MHz [141] (12.0 dBm) (no IR)
			* 6675.0 MHz [145] (12.0 dBm) (no IR)
			* 6695.0 MHz [149] (12.0 dBm) (no IR)
			* 6715.0 MHz [153] (12.0 dBm) (no IR)
			* 6735.0 MHz [157] (12.0 dBm) (no IR)
			* 6755.0 MHz [161] (12.0 dBm) (no IR)
			* 6775.0 MHz [165] (12.0 dBm) (no IR)
			* 6795.0 MHz [169] (12.0 dBm) (no IR)
			* 6815.0 MHz [173] (12.0 dBm) (no IR)
			* 6835.0 MHz [177] (12.0 dBm) (no IR)
			* 6855.0 MHz [181] (12.0 dBm) (no IR)
			* 6875.0 MHz [185] (12.0 dBm) (no IR)
			* 6895.0 MHz [189] (12.0 dBm) (no IR)
			* 6915.0 MHz [193] (12.0 dBm) (no IR)
			* 6935.0 MHz [197] (12.0 dBm) (no IR)
			* 6955.0 MHz [201] (12.0 dBm) (no IR)
			* 6975.0 MHz [205] (12.0 dBm) (no IR)
			* 6995.0 MHz [209] (12.0 dBm) (no IR)
			* 7015.0 MHz [213] (12.0 dBm) (no IR)
			* 7035.0 MHz [217] (12.0 dBm) (no IR)
			* 7055.0 MHz [221] (12.0 dBm) (no IR)
			* 7075.0 MHz [225] (12.0 dBm) (no IR)
			* 7095.0 MHz [229] (12.0 dBm) (no IR)
			* 7115.0 MHz [233] (12.0 dBm) (no IR)

morrownr, 2025-05-05:
My bad on what I posted:

To get the firmware date:

$ ethtool -i  wlx54077d99f58c

The frequencies post indicates that your adapter should be working in 6 GHz. There are other things we can look at if you want. I run several distros here. Linux Mint is a solid distro. Sorry that I am wearing you out with terminal commands but this type of tech support is a challenge. If you want to use 5 GHz for some time while you familiarize yourself with Linux Mint, you can always stop by here in the future and we can pick up where we left off.

Ynot70, 2025-05-05:
odell@odell-XPS-8900:~$ sudo ethtool <wlx54077d99f58c>
bash: syntax error near unexpected token `newline'
odell@odell-XPS-8900:~$ sudo ethtool wlx54077d99f58c
[sudo] password for odell:              
Settings for wlx54077d99f58c:
	Link detected: yes
odell@odell-XPS-8900:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: enp0s31f6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 64:00:6a:89:41:4e brd ff:ff:ff:ff:ff:ff
3: wlx54077d99f58c: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 54:07:7d:99:f5:8c brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.196/24 brd 192.168.0.255 scope global dynamic noprefixroute wlx54077d99f58c
       valid_lft 7194sec preferred_lft 7194sec
    inet6 fe80::ebd4:a77:c8b2:a3bc/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
odell@odell-XPS-8900:~$ ethtool wlx54077d99f58c
netlink error: Operation not permitted
Settings for wlx54077d99f58c:
	Link detected: yes
odell@odell-XPS-8900:~$ sudo ethtool wlx54077d99f58c
Settings for wlx54077d99f58c:
	Link detected: yes
odell@odell-XPS-8900:~$ sudo ethtool enp0s31f6
Settings for enp0s31f6:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Unknown! (255)
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 2
	Transceiver: internal
	MDI-X: Unknown (auto)
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
	Link detected: no
$ ethtool -i wlx54077d99f58c
driver: mt7921u
version: 6.11.0-25-generic
firmware-version: ____010000-20240219111038
expansion-rom-version: 
bus-info: 2-3:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

Ynot70, 2025-05-06:
Guess what?  This "Linux Expert" tried another one of your postings on updating the mt7921au firmware and completely screwed up the whole f-ing works.  Now the driver for this chip apparently went to live with Jesus.  Things were still working on the 5G band when I shut down after last night's episode, but still no 6G.  I came in here this afternoon and on startup the system was completely blind to the existence of the  Netgear AXE3000 A8000 wireless usb adapter with the mt7921au chip.  I tried unplugging and re-plugging and absolutely nothing.  I thought I'd somehow fried the Netgear, but when i stick it in my windows 11 machine it fires right up on all cylinders.  The network settings on the Linux machine decided I was using ethernet and that my cord was unplugged.  I'm using a $3 amazon special usb adapter that barely gets a 5G signal to post this, but at least it works.  What do I have to do to get the driver back?  Is there a way to reset the OS?

Ynot70, 2025-05-06:
Thanks to the person who put System Snapshot in Linux.  I rolled things back to 5/4/25 and it fixed the driver.  Still no luck with updating firmware.  I get free tech support from Netgear  for 30 days.  Just for a hoot I emailed their support team and got some totally useless info as I suspected I would.  They seem to think if I install the Win 11 software from the Netgear site it updates the firmware.  I did that when I hooked the adapter to my win 11 machine days ago.

morrownr, 2025-05-07:
Been sick today. Eck!

> Thanks to the person who put System Snapshot in Linux. 

The Linux Mint folks do a good job. However, if we try really hard, we can break it.

Given the nature of this site, I have one main dev box and a couple of other desktops along with a laptop that do normal work with. I also have 3 test systems where I keep multiple distros installed so as to do testing. My main dev box runs LMDE (Linux Mint Debian Edition). It is simply Mint that is directly based on Debian instead of Ubuntu. It is really hard to break.

> They seem to think if I install the Win 11 software from the Netgear site it updates the firmware. I did that when I hooked the adapter to my win 11 machine days ago.

You can get Linux support from Panda, Alfa and Brostrend for their USB WiFi adapters. The Netgear A8000 is on The Plug and Play list not because of Linux support from Netgear but rather because it meets the criteria I established for adapters to be considered very reliable. I think that would be the case for you except for the 6 GHz thing. I've been watching patches come in and being backported for some things that might be a cause of what you are seeing. Work in continuous. This is really technical and I think maybe using 5 GHz while you become more familiar with Linux is a good idea. You can stop by anytime. Just remember, you are learning the most widely used operating system in the world now. On the 500 Most Powerful Supercomputers list, Linux has 100% market. Linux is dominant on smartphones given Android. It is basically the backbone of the internet and I could go on. It is gaining market share in the desktop/laptop market. Anytime you need help, even if it is not for USB WiFi adapters, please stop by.

Regards

Ynot70, 2025-05-09:
Once again I'm trying to get 6G to work.  Now I have yet another usb wireless adapter.  I have a EDUP EP-AX1672.  I chose that because alas, it has Linux support.  I have downloaded from the EDUP site the exact same compressed folder of drivers as for the Netgear.  What luck!  One or the other of these is going back where it came from.   All of the files are unzipped and are in Bin format.  How do I get them out of the DOWNLOADS folder into terminal?  In your instructions you say "Create a folder/directory to hold the downloaded firmware files" mkdir  -p -/firmware.  When I do that I get nothing.  If I type mkdir firmware I get /firmware like it's there. Another set of instructions says to mkdir /lib/firmware/mediatek.  If I somehow manage to get these files loaded, which directory do I need to create?  How do I get the files to download to either the /firmware folder or the /lib/firmware/mediatek instead of DOWNLOADS folder?  I've tried copy and paste  from downloads folder(PERMISSION DENIED) and every thing I know to do. I need the WIFI patch files and the RAM code files.  All I need is a complete set of commands.

sam8641, 2025-05-12:
Try `/sbin/iw reg get` to see which wireless country regulation is set to.

If this shows "country 00", it is not set. Try `iw reg set US` or your country code, requires root permission. This may possibly allow connection to 6 GHz.
* Can be done later: To set on startup, try `options cfg80211 ieee80211_regdom=US`, or your country code if not in United States, to /etc/modprobe.d/custom.conf. Some process like hostapd with configuration file can change country code.
* When client regulatory is not set, the region or frequency information of AP may be temporary used, but may have a problem. If this Wifi name and password is only on 6 GHz, and Wireless regulatory region is not set, then client may not connect nor able to see 6 GHz.

Some linux distribution like Debian have firmware-mediatek package to try to install. Debian will need "non-free-firmware" to be added in /etc/apt/sources.list. However, some firmwares may be old.

Newer firmwares can be manually copied. Will need to know linux commands. Best to search the internet with "beginner linux terminal file operations" or similar.

Some commands require root permissions. Use `su` with root passord, or if sudo is set up, try `sudo su` with user password. Using commands as root can avoid permission denied problems, with a risk of incorrect or dangerous commands may destroy data.

For MT7921, these 2 files may be good enough. May need to provide the full path to where the file is downloaded to.
```
cp WIFI_RAM_CODE_MT7961_1.bin /lib/firmware/mediatek/
cp WIFI_MT7961_patch_mcu_1_2_hdr.bin /lib/firmware/mediatek/
```
* Linux kernel programmers can see Linux driver firmware filenames at drivers/net/wireless/mediatek/mt76/mt792x.h

> For compressed files, you will first need to compress ...

Looks like the word "compress" may be a typo, the correct word is decompress.

> Make sure the firmware files are owned by root after moving them to the correct directory before rebooting.

This can be important. May also needs to make sure file permissions is not writable by groups and others. An easy way is, after all the needed firmware files was placed there, try these commands as root:
```
chown root:root /lib/firmware/mediatek/*.bin /lib/firmware/mediatek/*/*.bin
chmod og-w /lib/firmware/mediatek/*.bin /lib/firmware/mediatek/*/*.bin
```

Then either restart computer, or do a reload with:
`rmmod mt7921u; modprobe mt7921u`

Check Linux kernel messages with a root command, `dmesg`, to see if firmware and other details do works or if some errors show up.

If all else fails, can try other distribution like Ubuntu live CD or image to see if 6 GHz works there. Be sure to set region with `iw reg set US` or your country code.

morrownr, 2025-05-12:
@Ynot70 

> Do these files need to be compressed or uncompressed?

I am as confused as you are. The guide instructions are to download .bin firmware files that can be compressed, if necessary, depending on your distro. I am sure you are having trouble with the guide but I do not have vision on what you are doing so I do not know how to help.

> If files are compressed, why would they need to be compressed again? 

If you are downloading compressed files, please STOP. My guide takes you to the Linux official site for firmware and the files are not compressed. Show me what you are doing. You can show me the files in your Downloads folder with the following command:

$ ls -l ~/Downloads

You can move the files with the graphical file manager or you can work on them in your Downloads directory.

> In your instructions you say "Create a folder/directory to hold the downloaded firmware files" mkdir -p -/firmware.

My instructions do not say `mkdir -p -/firmware`. They say `mkdir -p ~/firmware`. Note that the character before `/firmware` is not a dash. Look under the esc key for the correct character. ~ means home in Linux so that command means to create a new directory under your home directory call `/firmware`.

> I've tried copy and paste from downloads folder(PERMISSION DENIED)

You should copy the firmware files, the ones from where my guide says to download them, to the ~/firmware folder. You can use the graphical files manager as no root access is required.

Root access will be required once your have properly compressed the files, if necessary. You can invoke root capability buy using:

$ sudo 

> If I somehow manage to get these files loaded, which directory do I need to create?

Probably none as the directory will likely be there in the distro you are using. The guide tries to cover all possibilities and server distros often do not have the correct directory to hold the firmware files. If you do run the command to create the firmware directory, no big deal if it exists.

> All I need is a complete set of commands. This is like trying to decipher hieroglyphics.

We are trying to help. Rome wasn't built in a day.

morrownr, 2025-05-12:
Hi @sam8641 

> Looks like the word "compress" may be a typo, the correct word is decompress.

No, the word "compress" is exactly what I meant to say. Numerous Linux distros compress their firmware files in this day in time. If your firmware files are compressed on your distro, then you need to compress them with the correct compression software before copying the files to their final resting place or you can end up with more than one of each firmware files and you may not upgrade a system that you thought that you did.

Here is the link to the guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Follow the guide and you will see the need. 

You provided some very good information for @sam8641 . Let's remember he is a Windows 10 refugee and has no prior Linux experience. I've been using Linux since '94 and have never been a regular Windows user so we are having to learn to communicate first. Hopefully we get to where he wants to go before he throws his computer in a recycling bin.

sam8641, 2025-05-12:
Output of: `lsb_release -a` or `cat /etc/os-release` from a user having problem, as well as `/sbin/wpa_supplicant -v` or `/sbin/hostapd -v`, may help determine if the problem is outdated software.
* Very old versions of wpa-supplicant, hostapd, network manager, wireless regulatory database, and more, may prevent successful 6 GHz.
* To get up to date software, either follow instruction from your current distribution on how to dist-upgrade, or just find a more up to date Linux distribution.

Although I don't have problems, I will show some of my output.
`lsb_release -a` Description:    Debian GNU/Linux trixie/sid
`/sbin/wpa_supplicant -v` wpa_supplicant v2.10
`uname -a` ... 6.12.22 ... x86_64 GNU/Linux.
`/sbin/iw reg get` country US: DFS-FCC

As for Linux firmware: I use Debian, and firmware files appears as no compression. Some other distros do use compressions. Did a small edit to my earlier comment about making sure firmware files is owned by root.

Some guides can maybe use some simplifying or less repeats.
* The guide on how to update Mediatek looks like repeating most of the same instructions 7 times with different filenames. Could simplify or have less repeats on to how to download and update all Mediatek WiFi firmwares at the same time, for users with multiple different WiFi devices.

morrownr, 2025-05-14:
@sam8641 

> The guide on how to update Mediatek looks like repeating most of the same instructions 7 times with different filenames. Could simplify or have less repeats on to how to download and update all Mediatek WiFi firmwares at the same time, for users with multiple different WiFi devices.

I'm sure there are things that can be done to make the Firmware guide better. You are correct that many steps in each section do use the same steps but it is that way because I have seen problems with a more consolidated look. Maybe I should make the section numbers larger so they are easier to see.

I've been trying to ease off some things with @Ynot70 since he has 5 GHz working fine. Starting with a new operating system is going to be frustrating for anyone if they are not familiar with it. A couple of months from now, he will understand more and will be ready for more detailed things. I have never been a regular user of Windows. On small systems, I went from CP/M to PCDOS to DRDOS, to OS/2 and then to Linux. I have one partition on one system that has Windows 11. I boot it once every 3-4 months to update it or test a driver for a user here but I find Windows to be rather frustrating.

Ynot70, 2025-05-14:
Well kids, I've successfully bricked both a Netgear and an EDUP trying to get 6G to work in this Micky Mouse operating system.  The EDUP went back to Amazon, and the Netgear was past BestBuy return date, so it got shipped back to Netgear for an exchange.  The fun doesn't stop there.  I've come to the conclusion that 6G wireless in Linux Mint is a myth.  Since the cheap little $5 Amazon special was actually working on 5G running on a Realtek chip, I decided to take a chance on a BrosTrend AX1800 AX4.  Plugged it in and NOTHING! I can't say I was surprised.  The AX4 is advertised as having Linux Mint support out of the box.  Another bald- faced lie.  After spending a couple of hours searching, I found a link to a Linux driver supposedly from BrosTrend.  Downloaded and installed driver package and still nothing.  After several restarts, the driver manager in Linux suddenly found a new driver.  Funny thing is, I don't even think it was the same driver from BrosTrend, but it worked!  Everything was just wonderful! I was getting 100% signal level at -37 db. and a really good download speed.  Unfortunately, this also turned into a disaster.  The next time I started the machine, it would not boot.  Had to go through recovery mode and managed to get things up and working again.  Restarted twice to make sure things were good.  Next day same no boot issue.  I figured the driver must be screwed up.  After digging on internet again, I found someone on Github that said this was the wrong driver and provided another package for the correct driver.  Copied and pasted everything in terminal and installation started.  When install was almost complete, low disk space warning started to appear and install aborted.  Since then, linux would barely boot.  LOW DISK SPACE.  Not even enough space left on a 256G solid state drive for re-installing Linux from installation USB stick.  I thought the install feature would wipe the drive and start all over.  Apparently, it needs some disk space to work, and Micky Mouse has somehow managed to gobble EVERYTHING up! I'm going to dig the SSD out of the old XPS and hook it to my newer XPS windows machine and format the drive using a Kingston kit the SSD fits into and connects through usb.  If this doesn't work, the old XPS is going to the dump and to Hell with Linux Mint.

morrownr, 2025-05-14:
@Ynot70 

Seriously, do you work for Microsoft? Your approach to learning Linux is flawed. If you were to take the option of understanding that you need to go slow and learn Linux, you would be more successful than this seemingly brute force, it needs to work my way approach. 

You certainly can install Linux Mint and by having the Linux Mint installer format the disk. There is a lot of help available if you ask. Calling Linux Mint a "Mickey Mouse" operating system is not the way to encourage those that will help to do so.

Ynot70, 2025-05-14:
No, I do not work for Microsoft.  As far as Linux help goes, you get a hundred different answers from a hundred different people.  Never a clear solution.  It’s like a big guessing game.