morrownr/USB-WiFi #603: mt7925e crashes occasionally in AP mode

memfissoftware, 2025-05-02:
in ap mode mt7925e crashes occasionally.
and there is not much output from dmesg.

i tried to disable ASPM, fastboot but result is same.
trying older and newer fw's did not changed situtations.

when crash occurs i get some output like that from dmesg:
[ 14.010930] mt7925e 0000:01:00.0: enabling device (0000 -> 0002)
[ 14.011227] mt7925e 0000:01:00.0: disabling ASPM L0s L1
[ 14.020518] mt7925e 0000:01:00.0: ASIC revision: 79250000
[ 14.106534] mt7925e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20231227093012a
[ 14.445882] mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20231227093232
[ 17.414227] mt7925e 0000:01:00.0 wlan0: entered allmulticast mode
[ 17.414415] mt7925e 0000:01:00.0 wlan0: entered promiscuous mode
[40042.476822] mt7925e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20231227093012a
[40042.818661] mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20231227093232

lspci output for this card:
[lspci.txt](https://github.com/user-attachments/files/20006974/lspci.txt)

time to time i see also a timeout error like but not always:
mt7925e 0000:04:00.0: Message 00020003 (seq 15) timeout

interesting thing is that when it crashed i saw always on 'iw dev' normal but in reality card crashed and ssid lost.

morrownr, 2025-05-02:
Hi @memfissoftware 

> trying older and newer fw's did not changed situtations.

> mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20231227093232

That is a very old version of the firmware files. Did you try new files and then put the old ones back in?

FYI: New wifi firmware files for the mt7925 just arrived yesterday:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

You need to be really careful when installing new firmware files. The firmware files for mt7925 do not go in the same place as the mt7921 files. Also, if you system uses compressed firmware, you need to compress the new files so that they write over the old ones or you may not be upgrading anything. The guide goes into those details... take it slow and following the instructions carefully.

What kernel are you using?

memfissoftware, 2025-05-02:
i have updated with latest given for merge and now it is no crashing but still suddenly ssid is disappearing on clients also connection.
restarting hostapd is solving problem for a while.
now i have this from kernel output:
[   14.154644] mt7925e 0000:01:00.0: enabling device (0000 -> 0002)
[   14.156887] mt7925e 0000:01:00.0: disabling ASPM L0s L1
[   14.181715] mt7925e 0000:01:00.0: ASIC revision: 79250000
[   14.261070] mt7925e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20250425072955a
[   14.603254] mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20250425073109
[   17.413039] mt7925e 0000:01:00.0 wlan0: entered allmulticast mode
[   17.413239] mt7925e 0000:01:00.0 wlan0: entered promiscuous mode
[23789.516706] mt7925e 0000:01:00.0 wlan0: left allmulticast mode
[23789.516731] mt7925e 0000:01:00.0 wlan0: left promiscuous mode
[23793.055911] mt7925e 0000:01:00.0 wlan0: entered allmulticast mode
[23793.056104] mt7925e 0000:01:00.0 wlan0: entered promiscuous mode
[23972.081788] mt7925e 0000:01:00.0 wlan0: left allmulticast mode
[23972.081804] mt7925e 0000:01:00.0 wlan0: left promiscuous mode
[23975.553406] mt7925e 0000:01:00.0 wlan0: entered allmulticast mode
[23975.553598] mt7925e 0000:01:00.0 wlan0: entered promiscuous mode
[25363.308909] mt7925e 0000:01:00.0 wlan0: left allmulticast mode
[25363.308920] mt7925e 0000:01:00.0 wlan0: left promiscuous mode
[25366.780601] mt7925e 0000:01:00.0 wlan0: entered allmulticast mode
[25366.780774] mt7925e 0000:01:00.0 wlan0: entered promiscuous mode

morrownr, 2025-05-02:
Well, a little progress.

I have a M.2 mt7925 card but have only been using it in managed mode. It would be a pain to reconfigure to test right now.

The current issue with clients disconnecting could possibly be an issue with setup. Can you post your hostapd.conf?

Also, what kernel and distro are you using?

memfissoftware, 2025-05-02:
basicly it is crashing when clients try to reconnect:
I have added my hostapd conf.

[hostapd.txt](https://github.com/user-attachments/files/20015297/hostapd.txt)
also i use latest hostapd from git and ubuntu 24.04
hostapd v2.12-devel-hostap_2_11-835-g9bc3fd041

Linux n150 6.11.0-24-generic #24~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Mar 25 20:14:34 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: New station 60:57:18:2b:dd:dd
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Drv Event 19 (NL80211_CMD_NEW_STATION) received for wlan0
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Event message available
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Delete station 60:57:18:2b:dd:dd
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Drv Event 20 (NL80211_CMD_DEL_STATION) received for wlan0
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Event message available
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Frame TX command accepted; cookie 0x13
May 02 17:35:48 n150 start-ap.sh[1343]: CMD_FRAME - hexdump(len=30): b0 00 00 00 60 57 18 2b 26 a9 fc 70 2e 88 60 d5 fc 70 2e 88 60 d5 00 00 00 00 02 00 00 00
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: CMD_FRAME freq=5500 wait=0 no_cck=0 no_ack=0 offchanok=0
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: send_mlme -> send_frame_cmd
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: send_mlme - Use bss->freq=5500
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: send_mlme - da=60:57:18:2b:dd:dd sa=dd:dd:2e:88:60:d5 bssid=dd:dd:2e:88:60:d5 noack=0 freq=0 no_cck=0 offchanok=0 wait_time=0 no_encrypt=0 fc=0xb0 (WLAN_FC_STYPE_AUTH) nlmode=3
May 02 17:35:48 n150 start-ap.sh[1343]: authentication reply: STA=60:57:18:2b:dd:dd auth_alg=0 auth_transaction=2 resp=0 (IE len=0) (dbg=handle-auth)
May 02 17:35:48 n150 start-ap.sh[1343]: wlan0: STA 60:57:18:2b:dd:dd MLME: MLME-DELETEKEYS.request(60:57:18:2b:dd:dd)
May 02 17:35:48 n150 start-ap.sh[1343]: wlan0: STA 60:57:18:2b:dd:dd MLME: MLME-AUTHENTICATE.indication(60:57:18:2b:dd:dd, OPEN_SYSTEM)
May 02 17:35:48 n150 start-ap.sh[1343]: wlan0: STA 60:57:18:2b:dd:dd IEEE 802.11: authentication OK (open system)
May 02 17:35:48 n150 start-ap.sh[1343]:   * flags set=0x0 mask=0xa0
May 02 17:35:48 n150 start-ap.sh[1343]:   * listen_interval=0
May 02 17:35:48 n150 start-ap.sh[1343]:   * aid=1 (UNASSOC_STA workaround)
May 02 17:35:48 n150 start-ap.sh[1343]:   * capability=0x0
May 02 17:35:48 n150 start-ap.sh[1343]:   * supported rates - hexdump(len=3): 0c 18 30
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: NL80211_CMD_NEW_STATION STA 60:57:18:2b:dd:dd
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: bridge FDB entry delete for 60:57:18:2b:dd:dd ifindex=4 ifname wlan0 failed: Object not found
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: sta_remove -> DEL_STATION wlan0 60:57:18:2b:dd:dd --> -2 (No such file or directory)
May 02 17:35:48 n150 start-ap.sh[1343]: ap_sta_add: register ap_handle_timer timeout for 60:57:18:2b:dd:dd (300 seconds - ap_max_inactivity)
May 02 17:35:48 n150 start-ap.sh[1343]:   New STA
May 02 17:35:48 n150 start-ap.sh[1343]: authentication: STA=60:57:18:2b:dd:dd auth_alg=0 auth_transaction=1 status_code=0 wep=0 seq_ctrl=0xbc0
May 02 17:35:48 n150 start-ap.sh[1343]: mgmt::auth
May 02 17:35:48 n150 start-ap.sh[1343]: wlan0: Event RX_MGMT (18) received
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: RX frame da=dd:dd:2e:88:60:d5 sa=60:57:18:2b:dd:dd bssid=dd:dd:2e:88:60:d5 freq=5500 ssi_signal=-53 fc=0xb0 seq_ctrl=0xbc0 stype=11 (WLAN_FC_STYPE_AUTH) len=30
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Frame event
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: MLME event frame - hexdump(len=30): b0 00 3c 00 fc 70 2e 88 60 d5 60 57 18 2b 26 a9 fc 70 2e 88 60 d5 c0 0b 00 00 01 00 00 00
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: MLME event 59 (NL80211_CMD_FRAME) on wlan0(dd:dd:2e:88:60:d5) A1=dd:dd:2e:88:60:d5 A2=60:57:18:2b:dd:dd on link_id=-1
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan0
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Event message available
May 02 17:35:48 n150 start-ap.sh[1343]: ap_free_sta: cancel ap_handle_timer for 60:57:18:2b:dd:dd
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: multicast to unicast disabled on interface wlan0
May 02 17:35:48 n150 start-ap.sh[1343]:   * center_freq2=0
May 02 17:35:48 n150 start-ap.sh[1343]:   * center_freq1=5570
May 02 17:35:48 n150 start-ap.sh[1343]:   * channel_width=5
May 02 17:35:48 n150 start-ap.sh[1343]:   * bandwidth=160
May 02 17:35:48 n150 start-ap.sh[1343]:   * radar_background=0
May 02 17:35:48 n150 start-ap.sh[1343]:   * ht_enabled=1
May 02 17:35:48 n150 start-ap.sh[1343]:   * vht_enabled=1
May 02 17:35:48 n150 start-ap.sh[1343]:   * he_enabled=1
May 02 17:35:48 n150 start-ap.sh[1343]:   * eht_enabled=1
May 02 17:35:48 n150 start-ap.sh[1343]:   * freq=5500
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: assocresp_ies - hexdump(len=10): 7f 08 04 00 00 02 00 00 00 40
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: proberesp_ies - hexdump(len=10): 7f 08 04 00 00 02 00 00 00 40
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: beacon_ies - hexdump(len=10): 7f 08 04 00 00 02 00 00 00 40
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: group_cipher=0x10
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: pairwise_ciphers=0x50
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: key_mgmt_suites=0x2
[... cut ...]
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Beacon head - hexdump(len=54): 80 00 00 00 ff ff ff ff ff ff fc 70 2e 88 60 d5 fc 70 2e 88 60 d5 00 00 00 00 00 00 00 00 00 00 64 00 11 00 00 06 4d 65 6d 66 69 73 01 08 8c 12 98 24 b0 48 60 6c
May 02 17:35:48 n150 start-ap.sh[1343]: nl80211: Set beacon (beacon_set=1)
May 02 17:35:48 n150 start-ap.sh[1343]: Management frame protection is required in 6 GHz
May 02 17:35:48 n150 start-ap.sh[1343]: Management frame protection is required in 6 GHz
May 02 17:35:48 n150 start-ap.sh[1343]: hostapd_ht_operation_update new operation mode=0x0 changes=1

morrownr, 2025-05-03:
Per the wireless reg.db, the rules in your country are:

	# 6 GHz band
	(5945 - 6425 @ 320), (23), NO-OUTDOOR, wmmrule=ETSI

> channel=100

That presents a problem as channel 100 on band 4 (6 GHz) is not allowed in your country. You might want to try another channel like maybe 65.

memfissoftware, 2025-05-04:
@morrownr 
it is not about channel i use a test reg. db. for just testing.
```
: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 40), (N/A)
	(5170 - 5835 @ 320), (N/A, 40), (N/A)
	(5925 - 7125 @ 320), (N/A, 40), (N/A)
```

that issue can occurs even at 2.4 or any channel indeed.
i removed beamformings and now much more stable but it is even worse with new fw and new driver from mainline kernel.
it is easy to reproduce bug.
especially with old intel mac. just keep trying disconnect and connect then occurs. it even crashes with whole kernel's network stack and even lan etc all is gone.

it seems that there is a terrible memory leak in mt76 driver even i checked same with openwrt's mt76.
ap mode of mt7925 is total failure if my card does not have a hw issue i cant find.

memfissoftware, 2025-05-04:
it seems that issue is adding new device. that is the reason we saw only deauth because adding new sta crashes afterwards.
finally i reproduce with kernel panic and get the crash log
```
<4>[  222.120550] Oops: Oops: 0000 [#1] PREEMPT SMP NOPTI
<4>[  222.120585] CPU: 1 UID: 0 PID: 1072 Comm: hostapd Tainted: G        W  O       6.11.0-24-generic #24~24.04.1-Ubuntu
<4>[  222.120643] Tainted: [W]=WARN, [O]=OOT_MODULE
<4>[  222.120667] Hardware name: Default string Default string/Default string, BIOS M6_MAX V0.06 02/19/2025
<4>[  222.120712] RIP: 0010:mt7925_sta_set_decap_offload+0xd3/0x180 [mt7925_common]
<4>[  222.120778] Code: 00 00 00 b8 01 00 00 00 f3 48 0f bc c0 41 89 c6 3c 0e 77 b5 49 8d 87 30 02 00 00 48 89 45 b8 49 8b 87 18 06 00 00 41 0f b6 ce <66> 83 78 98 00 74 6d 48 63 c1 80 f9 0e 77 7b 49 8b 84 c7 a0 05 00
<4>[  222.120858] RSP: 0018:ffff9cb5810f73a8 EFLAGS: 00010293
<4>[  222.120890] RAX: 0000000000000000 RBX: ffff8918508c2020 RCX: 0000000000000000
<4>[  222.120927] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
<4>[  222.120963] RBP: ffff9cb5810f7400 R08: 0000000000000000 R09: 0000000000000000
<4>[  222.120999] R10: 0000000000000000 R11: 0000000000000000 R12: ffff89185feb9d38
<4>[  222.121034] R13: 0000000000000001 R14: 0000000000000000 R15: ffff8918421b8a98
<4>[  222.121070] FS:  00007d8ff3952740(0000) GS:ffff891bafa80000(0000) knlGS:0000000000000000
Oops#1 Part4
<4>[  222.121116] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
<4>[  222.121147] CR2: ffffffffffffff98 CR3: 00000001013aa000 CR4: 0000000000f50ef0
<4>[  222.121185] PKRU: 55555554
<4>[  222.121205] Call Trace:
<4>[  222.121224]  <TASK>
<4>[  222.121245]  ? show_regs+0x6c/0x80
<4>[  222.121278]  ? __die+0x24/0x80
<4>[  222.121304]  ? page_fault_oops+0x96/0x1b0
<4>[  222.121336]  ? kernelmode_fixup_or_oops.isra.0+0x69/0x90
<4>[  222.121373]  ? __bad_area_nosemaphore+0x1a1/0x2d0
<4>[  222.121404]  ? radix_tree_lookup+0xd/0x20
<4>[  222.121434]  ? start_flush_work+0x227/0x2e0
<4>[  222.121468]  ? bad_area_nosemaphore+0x16/0x30
<4>[  222.121496]  ? do_kern_addr_fault+0x78/0xa0
<4>[  222.121524]  ? exc_page_fault+0x1b0/0x1c0
<4>[  222.121557]  ? asm_exc_page_fault+0x27/0x30
<4>[  222.121590]  ? mt7925_sta_set_decap_offload+0xd3/0x180 [mt7925_common]
<4>[  222.121647]  ? mt7925_sta_set_decap_offload+0x50/0x180 [mt7925_common]
<4>[  222.121706]  drv_sta_set_decap_offload+0x98/0x1e0 [mac80211]
<4>[  222.122015]  ieee80211_check_fast_rx+0x315/0x420 [mac80211]
<4>[  222.122301]  _sta_info_move_state+0x38e/0x3f0 [mac80211]
<4>[  222.122551]  sta_info_move_state+0x13/0x20 [mac80211]
<4>[  222.122798]  sta_apply_auth_flags.isra.0+0x5a/0x1e0 [mac80211]
<4>[  222.123082]  sta_apply_parameters+0x26c/0x350 [mac80211]
<4>[  222.123362]  ieee80211_add_station+0xde/0x1a0 [mac80211]
<4>[  222.123615]  nl80211_new_station+0x4e3/0x780 [cfg80211]
<4>[  222.123839]  genl_family_rcv_msg_doit+0xf7/0x160
<4>[  222.123873]  genl_family_rcv_msg+0x182/0x250
<4>[  222.123901]  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
<4>[  222.124107]  ? __pfx_nl80211_new_station+0x10/0x10 [cfg80211]
Oops#1 Part3
<4>[  222.124314]  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
<4>[  222.124517]  genl_rcv_msg+0x4c/0xb0
<4>[  222.124538]  ? __pfx_genl_rcv_msg+0x10/0x10
<4>[  222.124561]  netlink_rcv_skb+0x5a/0x110
<4>[  222.124588]  genl_rcv+0x28/0x50
<4>[  222.124606]  netlink_unicast+0x245/0x390
<4>[  222.124633]  netlink_sendmsg+0x213/0x470
<4>[  222.124661]  ____sys_sendmsg+0x3a8/0x410
<4>[  222.124688]  ___sys_sendmsg+0x9a/0xf0
<4>[  222.124718]  __sys_sendmsg+0x89/0xf0
<4>[  222.124742]  __x64_sys_sendmsg+0x1d/0x30
<4>[  222.124765]  x64_sys_call+0x912/0x25f0
<4>[  222.124791]  do_syscall_64+0x7e/0x170
<4>[  222.124816]  ? __sys_setsockopt+0x76/0xe0
<4>[  222.124842]  ? aa_sk_perm+0x46/0x240
<4>[  222.124866]  ? syscall_exit_to_user_mode+0x4e/0x250
<4>[  222.124895]  ? copy_from_sockptr_offset.constprop.0+0x24/0x30
<4>[  222.124924]  ? do_sock_setsockopt+0xbe/0x190
<4>[  222.124950]  ? __sys_setsockopt+0x76/0xe0
<4>[  222.124975]  ? syscall_exit_to_user_mode+0x4e/0x250
<4>[  222.125003]  ? do_syscall_64+0x8a/0x170
<4>[  222.125026]  ? syscall_exit_to_user_mode+0x18d/0x250
[... 11 lines cut ...]
<4>[  222.132006] RDX: 0000000000000000 RSI: 00007ffcb51821c0 RDI: 0000000000000005
<4>[  222.132881] RBP: 00007ffcb51821b0 R08: 0000000000000004 R09: 00000000000000f0
<4>[  222.133745] R10: 00007ffcb51822cc R11: 0000000000000202 R12: 00005632b6ae94c0
<4>[  222.134602] R13: 00005632b6a8b3f0 R14: 00007ffcb51821c0 R15: 00007ffcb51822cc
<4>[  222.135452]  </TASK>
<4>[  222.136280] Modules linked in: cmac ccm snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_cadence qrtr snd_sof_intel_hda_common snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi snd_sof_utils snd_soc_hdac_hda aic8800_fdrv(O) snd_soc_acpi_intel_match snd_hda_codec_realtek soundwire_generic_allocation snd_hda_codec_generic snd_soc_acpi snd_hda_scodec_component soundwire_bus snd_soc_avs snd_soc_hda_codec snd_hda_ext_core snd_soc_core coretemp snd_compress ac97_bus snd_pcm_dmaengine kvm_intel snd_hda_intel kvm snd_intel_dspcfg bridge snd_intel_sdw_acpi stp snd_hda_codec crct10dif_pclmul llc polyval_clmulni snd_hda_core polyval_generic snd_hwdep ghash_clmulni_intel sha256_ssse3 mt7925e sha1_ssse3 snd_pcm aesni_intel mt7925_common snd_seq_midi crypto_simd binfmt_ [... cut ...]
it seems that there is a null pointer somewhere most possibly in mt7925_sta_set_decap_offload()

but it seems a bug that was not fixed for a very long time :S.
https://github.com/morrownr/USB-WiFi/issues/359
https://forum.banana-pi.org/t/random-kernel-crashes-kernel-panic-log-attached/22117

first report is 11/24
https://bugzilla.kernel.org/show_bug.cgi?id=219505
 even from 08/24
https://github.com/openwrt/openwrt/issues/16273

around a year no fix :) then ap mode is broken maybe not much people use that ap mode. my chinese aic8800 works like a charm meanwhile from 2.5 usd from china :D.

morrownr, 2025-05-08:
@memfissoftware 

I've been sick for a few days.

I am researching this issue. I just posted a message in: https://github.com/openwrt/openwrt/issues/16273

If you read that message, you will see that I posted a link to a patch that could be the problem. What actually needs to happen is a bisection needs to happen in order to pinpoint what patch broke things. With that information we can go directly to linux-wireless with the information. Results would likely come very rapidly. Are you able to bisect and pinpoint the bad patch?

memfissoftware, 2025-05-09:
ho @morrownr 
i hope you get well.

what do you mean exactly?
you want me to find when this driver broken with which changes?

i may able to do it with compiling openwrt mt76 from git and try to go back on code.

morrownr, 2025-05-14:
> i hope you get well.

It has taken a while and now I am behind.

I hate it when this happens because then I have to read the entire thread again to know what is up. I scanned this thread and saw openwrt, ubuntu 24.04 and AP mode. 

What I have noticed lately is a lot of patches for the mt7925. Many went into kernel 6.15 and are in the process of being backported to 6.14 and 6.12. A quick check today on kernel 6.15 rc6 on Ubuntu 25.04 and a couple of bugs I had noticed are gone. I think it may be a matter of letting the fixes catch up with us and things will be fine. If not, we can try to chase it down.

morrownr, 2025-05-14:
What problems are you seeing with the mt7921? 

2024 was a difficult year for wifi in general. All makers had problems with their tri-band devices. I've never seen the frequency of firmware updates. The tremendous amount of updates to the wifi stack has contributed but what do we do? The stack has to be updated and backwards compatibility needs to be there but the thing with these WiFi 6 and later devices is that the drivers are big and complicated. For me, my mt7921 devices have been very stable in all modes for some time with the exception of the Alfa AXML and AXM which have bluetooth turned on. That was a can of worms but with Ubuntu 25.04 that is fixed. My mt7925 device was stable in managed mode until recently but that is fixed now. I only run it in managed mode for now.

FYI: Ubuntu is not my daily driver. I have test systems here and Ubuntu is one of the distros that I use for testing. I have tried to love the gnome interface but I simply do not like it. My daily driver is Mint (LMDE). Mint based on Debian.

fs30000, 2025-05-15:
Oh, i'm just using these cards on a gateway and dump AP.  I understand they get a lot of updates, but my experience has been not so good. Quite probably because i am running them now on a Banana pi r4, arm64 arch.  When i was running the mt7922 on a x86 mini pc, it was rock solid.

Oh, yeah my mistake, i have a mt7922 not 21. With the 22, some sites/apps take a bit to load. 

Other than that, the banana router has been ok. Even if, the software took a while to get stable. I run debian 12 (frank-w's repo) on the main gateway and openwrt on the dump/transparent AP. On the openwrt, SSH connections get slow  sometimes.

Always some kind of shit happening with this banana. On a way, the banana is very atractive, because it has a lot of features and stuff, on the other hand, the software/drivers suck on one way or another.

For desktop, i don't use linux. Ah, linux for desktop still has a lot to go. Last time i tried it, and i always try it once in a while in these last 20 years, always some kind of annoying stuff. Like in gnome, to create a shortcut on the desktop, i needed to install an "extension". What?!... I also tried KDE and its endless settings. They never learn, some stuff never changes...

morrownr, 2025-05-18:
@fs30000 

> Ah, linux for desktop still has a lot to go. Last time i tried it, and i always try it once in a while in these last 20 years, always some kind of annoying stuff. Like in gnome, to create a shortcut [...]

I agree with you about gnome and KDE. I try each every couple of years but I do not either. I have little tolerance for bugs and KDE just has too many bugs and gnome, well, it drives me crazy. The desktop that I prefer is Cinnamon. Next time you try a new Linux distro, try Linux Mint. Their flagship version uses Cinnamon.

deren, 2025-05-19:
Hi guys,

Can someone help to verity this patch? I can see a similar problem on my desktop and resolve it now. Hope someone can help to confirm this solution

```
diff --git a/drivers/net/wireless/mediatek/mt76/mt7925/main.c b/drivers/net/wireless/mediatek/mt76/mt7925/main.c
index 66f327781947..6d69132d3540 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7925/main.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7925/main.c
@@ -1593,34 +1593,39 @@ static void mt7925_sta_set_decap_offload(struct ieee80211_hw *hw,
                                         struct ieee80211_vif *vif,
                                         struct ieee80211_sta *sta,
                                         bool enabled)
 {
        struct mt792x_sta *msta = (struct mt792x_sta *)sta->drv_priv;
        struct mt792x_vif *mvif = (struct mt792x_vif *)vif->drv_priv;
        struct mt792x_dev *dev = mt792x_hw_dev(hw);
        unsigned long valid = mvif->valid_links;
        u8 i;

+       if (!msta->vif)
+               return;
+
        mt792x_mutex_acquire(dev);

        valid = ieee80211_vif_is_mld(vif) ? mvif->valid_links : BIT(0);

        for_each_set_bit(i, &valid, IEEE80211_MLD_MAX_NUM_LINKS) {
                struct mt792x_link_sta *mlink;

                mlink = mt792x_sta_to_link(msta, i);

                if (enabled)
                        set_bit(MT_WCID_FLAG_HDR_TRANS, &mlink->wcid.flags);
                else
                        clear_bit(MT_WCID_FLAG_HDR_TRANS, &mlink->wcid.flags);

+               if (!mlink->wcid.sta)
+                       continue;
                mt7925_mcu_wtbl_update_hdr_trans(dev, vif, sta, i);
        }

        mt792x_mutex_release(dev);
 }

```

morrownr, 2025-05-19:
@fs30000 @memfissoftware 

Do either or both of you have the ability to apply the above patch from @deren and report on the results? With positive results, this patch can go in rapidly.

memfissoftware, 2025-05-19:
@morrownr 
i can try it with ubuntu's 6.11.0-25 @deren  which kernel version do you use? also i have seen lot of update for mt7925 in ubuntu 6.11.0-26

deren, 2025-05-19:
@memfissoftware 

I check this issue on both v6.11 and mainline kernel. You may just play this patch in your environment directly.

fs30000, 2025-05-21:
> [@fs30000](https://github.com/fs30000) [@memfissoftware](https://github.com/memfissoftware)
> 
> Do either or both of you have the ability to apply the above patch from [@deren](https://github.com/deren) and report on the results? With positive results, this patch can go in rapidly.

Let me see if i can try this soon, will be busy these days...

sam8641, 2025-05-22:
I recently got MT7925 with M.2 Wifi to PCIE adapter about a week ago, with kernel oops problem.

The patch from @deren is a good fix on my side as well, with most of my Linux Desktop using Hostapd.

Without patch, hostapd with `ieee80211w=0` is a possible workaround for MT7925 kernel oops problem and MT7921 delayed reconnect problem.

I have manually patched and compiled Linux Kernel, 6.12.27, AMD64. All of my Linux is Debian 13 Trixie at the time of this reply. I have:
* Linux Desktop 1: MT7922, MT7921u, intel AX200, 6 ethernet ports.
* Linux Desktop 2: MT7925, Intel AC 9260.
* Linux Desktop 3: Old Wifi G Ath5k (168c:001c).
* Windows 10 Desktop: Intel AX200.
* Android: Pixel 6a, WiFi 6E HE160.

Without patch, I have seen my problem of MT7925 kernel oops and MT7921 delay connect, most frequently happens with hostapd `ieee80211w=2`, and a client lose connection by going too far away, then quickly go closer and reconnect.

MT7921, with the same part of a patch.
```
// drivers/net/wireless/mediatek/mt76/mt7921/main.c
// mt7921_sta_set_decap_offload
+	if (!msta->vif)
+		return;
```
When this part have been manually added to MT7921, and hostapd `ieee80211w=2` this solved my Android problem of WiFi roaming disconnects, and 5 second delay to reconnect.

I will likely keep my hostapd running to see if any problem happens.

memfissoftware, 2025-05-23:
i guess mt7921 problem may not be same @sam8641.
because i have tried to patch and it solved only one thing. now driver is not crashing but ap wifi disappears again even iw dev shows it is up and ap mode. 
but in reality it does not transmit any signal. 

could it be a deauth attack protection and not a driver issue anymore? from hostapd but only restart recovers that.

memfissoftware, 2025-05-23:
> * rfkill.
> 
> * Conflicts with network manager and hostapd.
> 
> * While hostapd is running, `iw reg set 00` may lose connection.
> 
> * Overheating may happen on continuous and maximum transmit speed. Some Wifi cards like MT7925 have temperature sensor. Try `sensors` or `watch sensors`.

@sam8641  I dont think problem about them. because i dont use network manager it is already headless system.
yes it has sensor but around 50 cel deg.
problem easy to reproduce just put mt7925 in managed mode and try to connect disconnect then during generally removing a sta with DEL_STATION on hostpad it crashes or signal lost even all look fine with iw dev etc.
sometimes it happens 5 time connect-disconnect sometimes 20 sometimes just it goes 3 days later etc when i try to use normally

btw also reverting "Update mt7925_mcu_uni_[tx,rx]_ba for MLO" did not improved the speed neither new fw.

fs30000, 2025-05-29:
> > Without patch, hostapd with `ieee80211w=0` is a possible workaround for MT7925 kernel oops problem and MT7921 delayed reconnect problem.
> 
> Tried ieee80211w=0 on my7921, didn't seem to have fixed the slowness.

Ok, upgraded  banana pi r4 to 6.12.30 and applied  this [patch](https://www.spinics.net/lists/linux-wireless/msg266240.html), so far the slowness and freezing seem to have disappeared. The present card is a mt7922.

I have yet to try the mt7925 and the qualcomm as well. Router is finally working good, so i'll leave it as it is. Anyway, if you do need more testing on the mt7925, just tell me.

morrownr, 2025-05-29:
> applied this patch, so far the slowness and freezing seem to have disappeared. The present card is a mt7922.

That is good to hear.

> I have yet to try the mt7925 and the qualcomm as well. Router is finally working good, so i'll leave it as it is. Anyway, if you do need more testing on the mt7925, just tell me.

That will be up to @deren . He has submitted both patches as can be seen at the following:

https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=*&q=mt76&archive=both&delegate=

I suspect additional testing would be a good thing. ndb usually does not go through patches on  daily basis but rather does the work on patches in batches one or two times per cycle unless it is something that needs to go directly to Linus. In fact, both of these patches would probably be accepted to be sent directly to Linus if requested by the patch submitter as both are needed fixes for kernel 6.16 as soon as the first rc is out.

morrownr, 2025-06-03:
@fs30000 

> I have been testing the mt7925 with latest firmware and kernel 6.12.30. Not crashing, but not good for daily use. A lot of freezes when opening sites.

My system with the mt7925 based M.2 card is currently running Ubuntu 25.04 with kernel 6.14. The original 6.14 that Ubuntu shipped with still had a problem but a new kernel flowed and I have seen a problem since. There have been some patches lately but do not know when or if they will all make it to kernel 6.12.

> I saw there were a lot of patches for the qualcomm QCNCM865, as i have one, i decided to test, but same error as months ago.

I don't have any ideas about this. I established this site for USB WiFi and since Qualcomm does not release WiFi capable chips...

> I'll put the mt7922 back on.

There was a recent patch to allow AP mode on the mt7922 to use 160 MHz channel width. I think it went into kernel 6.16 and I do not know if it was set to be backported.

fs30000, 2025-06-05:
> There was a recent patch to allow AP mode on the mt7922 to use 160 MHz channel width. I think it went into kernel 6.16 and I do not know if it was set to be backported.

Thanks for the info. I'll check it out later. The kernel repo i use for this banana r4,  is from Frank-W. Eventually 6.16 will arrive there, if it hasn't already. 

Thing is, the mt7922 is sucking again... I even compiled the hostapd's dev version, in hope it would fix something, it didn't. As i see it, AP mode is not a priority for chip companies, much less for arm64. Props to Openwrt people to fix and improve drivers. I have to use Debian on my main router tough.

memfissoftware, 2025-06-09:
@deren 
`Jun 09 08:25:47 n150 kernel: RIP: 0010:mt7925_sta_set_decap_offload+0xd3/0x180 [mt7925_common]
Jun 09 08:25:47 n150 kernel: Code: 00 00 00 b8 01 00 00 00 f3 48 0f bc c0 41 89 c6 3c 0e 77 b5 49 8d 87 30 02 00 00 48 89 45 b8 49 8b 87 18 06 00 00 41 0f b6 ce <66> 83 78 98 00 74 6d 48 63 c1 80 f9 0e 77 7b 49 8b 84 c7 a0 05 00
Jun 09 08:25:47 n150 kernel: RSP: 0018:ffffa08908da32c8 EFLAGS: 00010293
Jun 09 08:25:47 n150 kernel: RAX: 0000000000000000 RBX: ffff91e596432020 RCX: 0000000000000000
Jun 09 08:25:47 n150 kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
Jun 09 08:25:47 n150 kernel: RBP: ffffa08908da3320 R08: 0000000000000000 R09: 0000000000000000
Jun 09 08:25:47 n150 kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff91e593ca9d38
Jun 09 08:25:47 n150 kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff91e5835e8a98
Jun 09 08:25:47 n150 kernel: FS:  00007a8cf67d4740(0000) GS:ffff91e8efa00000(0000) knlGS:0000000000000000
Jun 09 08:25:47 n150 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 09 08:25:47 n150 kernel: CR2: ffffffffffffff98 CR3: 0000000113bd6000 CR4: 0000000000f50ef0
Jun 09 08:25:47 n150 kernel: PKRU: 55555554
Jun 09 08:25:47 n150 kernel: Call Trace:
Jun 09 08:25:47 n150 kernel:  <TASK>
Jun 09 08:25:47 n150 kernel:  ? show_regs+0x6c/0x80
Jun 09 08:25:47 n150 kernel:  ? __die+0x24/0x80
Jun 09 08:25:47 n150 kernel:  ? page_fault_oops+0x96/0x1b0
Jun 09 08:25:47 n150 kernel:  ? kernelmode_fixup_or_oops.isra.0+0x69/0x90
Jun 09 08:25:47 n150 kernel:  ? __bad_area_nosemaphore+0x1a1/0x2d0
Jun 09 08:25:47 n150 kernel:  ? radix_tree_lookup+0xd/0x20
Jun 09 08:25:47 n150 kernel:  ? start_flush_work+0x227/0x2e0
Jun 09 08:25:47 n150 kernel:  ? bad_area_nosemaphore+0x16/0x30
Jun 09 08:25:47 n150 kernel:  ? do_kern_addr_fault+0x78/0xa0
Jun 09 08:25:47 n150 kernel:  ? exc_page_fault+0x1b0/0x1c0
Jun 09 08:25:47 n150 kernel:  ? asm_exc_page_fault+0x27/0x30
Jun 09 08:25:47 n150 kernel:  ? mt7925_sta_set_decap_offload+0xd3/0x180 [mt7925_common]
Jun 09 08:25:47 n150 kernel:  ? mt7925_sta_set_decap_offload+0x50/0x180 [mt7925_common]
Jun 09 08:25:47 n150 kernel:  drv_sta_set_decap_offload+0x98/0x1e0 [mac80211]
Jun 09 08:25:47 n150 kernel:  ieee80211_check_fast_rx+0x315/0x420 [mac80211]
Jun 09 08:25:47 n150 kernel:  _sta_info_move_state+0x38e/0x3f0 [mac80211]
Jun 09 08:25:47 n150 kernel:  sta_info_move_state+0x13/0x20 [mac80211]
Jun 09 08:25:47 n150 kernel:  sta_apply_auth_flags.isra.0+0x5a/0x1e0 [mac80211]
Jun 09 08:25:47 n150 kernel:  sta_apply_parameters+0x26c/0x350 [mac80211]
Jun 09 08:25:47 n150 kernel:  ieee80211_add_station+0xde/0x1a0 [mac80211]
Jun 09 08:25:47 n150 kernel:  nl80211_new_station+0x4e3/0x780 [cfg80211]
Jun 09 08:25:47 n150 kernel:  genl_family_rcv_msg_doit+0xf7/0x160
Jun 09 08:25:47 n150 kernel:  genl_family_rcv_msg+0x182/0x250
Jun 09 08:25:47 n150 kernel:  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
Jun 09 08:25:47 n150 kernel:  ? __pfx_nl80211_new_station+0x10/0x10 [cfg80211]
Jun 09 08:25:47 n150 kernel:  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
Jun 09 08:25:47 n150 kernel:  genl_rcv_msg+0x4c/0xb0
Jun 09 08:25:47 n150 kernel:  ? __pfx_genl_rcv_msg+0x10/0x10
Jun 09 08:25:47 n150 kernel:  netlink_rcv_skb+0x5a/0x110
Jun 09 08:25:47 n150 kernel:  genl_rcv+0x28/0x50
Jun 09 08:25:47 n150 kernel:  netlink_unicast+0x245/0x390
Jun 09 08:25:47 n150 kernel:  netlink_sendmsg+0x213/0x470
Jun 09 08:25:47 n150 kernel:  ____sys_sendmsg+0x3a8/0x410
Jun 09 08:25:47 n150 kernel:  ___sys_sendmsg+0x9a/0xf0
Jun 09 08:25:47 n150 kernel:  __sys_sendmsg+0x89/0xf0
Jun 09 08:25:47 n150 kernel:  __x64_sys_sendmsg+0x1d/0x30
Jun 09 08:25:47 n150 kernel:  x64_sys_call+0x912/0x25f0
Jun 09 08:25:47 n150 kernel:  do_syscall_64+0x7e/0x170
Jun 09 08:25:47 n150 kernel:  ? __sys_setsockopt+0x76/0xe0
Jun 09 08:25:47 n150 kernel:  ? syscall_exit_to_user_mode+0x4e/0x250
Jun 09 08:25:47 n150 kernel:  ? do_syscall_64+0x8a/0x170
Jun 09 08:25:47 n150 kernel:  ? ksys_write+0xc9/0x100
Jun 09 08:25:47 n150 kernel:  ? syscall_exit_to_user_mode+0x4e/0x250
Jun 09 08:25:47 n150 kernel:  ? aa_sk_perm+0x46/0x240
Jun 09 08:25:47 n150 kernel:  ? copy_from_sockptr_offset.constprop.0+0x24/0x30
Jun 09 08:25:47 n150 kernel:  ? do_sock_setsockopt+0xbe/0x190
Jun 09 08:25:47 n150 kernel:  ? __sys_setsockopt+0x76/0xe0
Jun 09 08:25:47 n150 kernel:  ? syscall_exit_to_user_mode+0x4e/0x250
Jun 09 08:25:47 n150 kernel:  ? do_syscall_64+0x8a/0x170
Jun 09 08:25:47 n150 kernel:  ? ksys_write+0xc9/0x100
Jun 09 08:25:47 n150 kernel:  ? syscall_exit_to_user_mode+0x4e/0x250
Jun 09 08:25:47 n150 kernel:  ? do_syscall_64+0x8a/0x170
Jun 09 08:25:47 n150 kernel:  entry_SYSCALL_64_after_hwframe+0x76/0x7e
Jun 09 08:25:47 n150 kernel: RIP: 0033:0x7a8cf5f2c004
Jun 09 08:25:47 n150 kernel: Code: 15 19 6e 0d 00 f7 d8 64 89 02 b8 ff ff ff ff eb bf 0f 1f 44 00 00 f3 0f 1e fa 80 3d 45 f0 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 4c c3 0f 1f 00 55 48 89 e5 48 83 ec 20 89 55
Jun 09 08:25:47 n150 kernel: RSP: 002b:00007fff4ff6f918 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
Jun 09 08:25:47 n150 kernel: RAX: ffffffffffffffda RBX: 00005e3527caa380 RCX: 00007a8cf5f2c004
Jun 09 08:25:47 n150 kernel: RDX: 0000000000000000 RSI: 00007fff4ff6f950 RDI: 0000000000000005
Jun 09 08:25:47 n150 kernel: RBP: 00007fff4ff6f940 R08: 0000000000000004 R09: 00000000000000f0
Jun 09 08:25:47 n150 kernel: R10: 00007fff4ff6fa5c R11: 0000000000000202 R12: 00005e3527d051a0
[... cut ...]
Jun 09 08:25:47 n150 kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff91e593ca9d38
Jun 09 08:25:47 n150 kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff91e5835e8a98
Jun 09 08:25:47 n150 kernel: FS:  00007a8cf67d4740(0000) GS:ffff91e8efa00000(0000) knlGS:0000000000000000
Jun 09 08:25:47 n150 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 09 08:25:47 n150 kernel: CR2: ffffffffffffff98 CR3: 0000000113bd6000 CR4: 0000000000f50ef0
Jun 09 08:25:47 n150 kernel: PKRU: 55555554
Jun 09 08:25:47 n150 kernel: note: hostapd[90273] exited with irqs disabled`

morrownr, 2025-06-24:
@fs30000 @memfissoftware 

I am going to try to set aside some time to see if we can work through these AP problems you both are experiencing. I have a M.2 card that uses the mt7925 chip and a PCIe card that uses a mt7922 chip, however, the mt7922 uses the mt7921e driver so I can probably duplicate this problem with a USB WiFi adapter with a mt7921au chip.

I would like to be able to duplicate the problems and then, when the problems are duplicated, test the patches. I do request that we avoid statements such as: This makes me pissed, because don't devs test stuff, before commiting code to kernel?

It is not unusual for kernel devs to work problems at this site and I would prefer that the atmosphere here remain one where this continues. I understand frustration and I work hard to assist in fixing those problems but we can chase away people that are willing to help if we do not keep our cool.

You can see the two patches at the following site:

https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=*&q=mt76&archive=both&delegate=

[1/2] wifi: mt76: mt7925: prevent NULL pointer dereference in mt7925_sta_set_decap_offload()

2/2] wifi: mt76: mt7921: prevent decap offload config before STA initialization

We have already seen the one for the mt7925 earlier in this thread.

What would help me help you is if I know exactly what distros, kernels and setup you are using for AP mode so that I can try to duplicate your setups... or at least get close enough to duplicate the problem.

FYI: I've helped a lot of people with AP mode problems over the years. If the patches do not fix what you are seeing, maybe we can narrow to the problem to a specific patch or bit of code.

What I have noticed over the years is that often AP mode problems happen more often with specific AP mode setups and networking setups. I can tell you right now that the setup that I use with my mt7921au based adapters and the mt7921u driver are extremely rock solid. Post back with the specifics of your setup if you want to pursue this issue further.

Regards

morrownr, 2025-06-25:
@fs30000 

Thanks for the reply.  Some initial thoughts as I work my way through your post.

Banana Pi R4

I am doing some research on the bpiR4. I have deep knowledge of the RasPi4B, RasPi3B and a couple of other SBCs but not the bpiR4. It seems you can boot from multiple sources. Have you considered using an SD card to boot and try OpenWRT 24.10.2? OpenWRT is available for the bpiR4. This would give you something to compare with. The drivers for the mt7925 and mt7922 are available in OpenWRT. Link:

https://firmware-selector.openwrt.org/?version=24.10.2

BPI-Router-Images

That repo would take some time to understand and I do not have any bpi equipment. You say it is based on Debian 12. While I like Debian in general, Debian 12 is really out-of-date at this point in the cycle. What makes it problematic is that it was released with kernel 6.1 and starting with kernel 6.2, we saw a lot of wireless APIs get updates and additions. The worst period seemed to be from kernel 6.2 to 6.5. Someone had asked me to backport the mt7921u driver from kernel 6.12 to 5.15. I soon learned that doing so was impractical. I could backport the driver from 6.1 to 5.15 and it worked. It is hard to cross that 6.2 to 6.5 zone. Keep in mind that there is far more than a driver and firmware that makes a wireless device go. There are the wireless APIs and the networking and much more. Right now you are using kernel 6.15, the latest firmware and what else is left in that distro that is before the 6.2 to 6.5 zone of confusion?  Keep in mind that OpenWRT devs have full control over what levels of the various things that go into wireless support. Also note that OpenWRT skipped over kernel 6.1 because they knew what was going on so they jumped beyond kernel 6.5 to kernel 6.6 for their current release. They skipped 6.1.

In general, I recommend either fully staying with kernel 6.1 and before, to include firmware and drivers or fully after kernel 6.6 to include firmware and drivers.

My primary wifi router uses a ARM64 chip, the MediaTek MT7981B SOC. That is the same SOC used by the OpenWRT ONE router that Banana Pi builds and markets. With my mt7925 and mt7922 cards in managed mode, I get around 1.2 gbps thoughput as measured by iperf3. That is one stable wifi router. I would have bought the OpenWRT ONE but I bought my wifi router before I knew that OpenWRT had plans.

I also use my RasPi4B primarily as an AP. I use the RasPiOS. It uses the 6.12 kernel. I use hostapd and a mt7921au based usb wifi adapter. It is what I call indefinitely stable. I use systemd-resolved and systemd-networkd for networking.

Some things for you to think about.

morrownr, 2025-10-15:
Hi @lupine 

I just happened to be working another project where I think I can include working this problem. I am pretty good at getting patches into the kernel as long as we can work this to the point that we understand that the bug is in the driver. I have a mt7925 based USB WiFi adapter so hopefully I can reproduce this problem.

I am running Debian 13 on this box, my main dev box, but am using a RasPi4B with the latest RasPiOS for AP testing. It is using kernel 6.12 so if I can reproduce this bug with that setup, it will test us some things.

> 802.11ax, 5GHz, on channel 36, 80MHz width, GB regdomain

So, WiFi 6 on band 2 (5 GHz) on channel 36 and 80 MHz channel width. This is where I will start. This may go slow due to some health problems I have and I do hope that others can jump in to work various things as we proceed.

@morrownr

morrownr, 2025-10-16:
I've been testing with my main box and Android 14 phone for a few hours now and things are solid so far. I'll keep going.

I am running a very fresh install of the latest RasPiOS, kernel 6.12. I did not bother to upgrade the first because I wanted to see what would happen with the old firmware... and then later I could upgrade to the latest:

```
$ ethtool -i wlan0
driver: mt7925u
version: 6.12.47+rpt-rpi-v8
firmware-version: ____000000-20241104133053
expansion-rom-version: 
bus-info: 2-2:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

```

Do you know how to regress to the firmware files, 20241104133053? I'd like to rule out that a new bug in the latest firmware is the cause.

morrownr, 2025-10-16:
@lupine 

> Armed with a reliable reproducer, I tried downgrading the firmware. The bug was reproducible in 20241104133053 (yours), 20250305133013 (trixie), 20250721232943 (trixie-backports), and 20250825215925 (current linux-firmware version).

Making progress it would seem. This does suggest that the issue has nothing to do with the firmware. Let me suggest that we go with 20250825215925 so that our test systems are both on the latest firmware.

> I tried downgrading hostapd back down to the trixie-included 2.10, but couldn't get clients to see the AP at all with that version, so I went back to main.

Yes, hostapd 2.10, in my opinion, is a mistake on the part of the Debian devs. It is very old. We need at least 2.11. Would you mind reporting this to Debian? Now back to the topic: My AP test system is using:

```
$ hostapd -v
hostapd v2.12-devel-hostap_2_11-1128-g167aeab41
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2024, Jouni Malinen <j@w1.fi> and contributors

``` 

I just compiled this a few days ago as I was doing a clean install of RasPiOS. I have a guide to compiling it on the Main Menu. It might be good if you also test 2.12-devel to see if that makes a difference.

Could you post the hostapd.conf that you are using?

> I tried downgrading the kernel. It worked immediately with 6.12.48 and the 20250825215925 firmware; I was able to reconnect to my heart's content.

Okay, making more progress but we still do not know if the problem is in the mt7925u/e driver. Kernels contain far more than wifi drivers when it comes to the operation of wireless. There is the wireless stack. Let me suggest we get synced up on the following:

Firmware: Let's use the latest - 20250825215925
hostapd: Can I get you to compile the latest 2.12-devel?
hostapd.conf: Let's compare and decide on a common file to use.
kernel: 6.12 (we can move up to test as we need to)

> I wonder whether you can trigger it with the usb stick + 6.16 kernel?

I had a system die last week. I have a system with Ubuntu 25.10 (kernel 6.17) that I will test with. My main box uses Debian 13 (kernel 6.12 or 6.16) but it uses KDE and it appears KDE does not allow me to do what you are doing with gnome... that is, it only allow me to disconnect a system that is already connected. It actually seems strange to me that Ubuntu would allow what you are doing but I will test. One way or another, we should expect a disconnect followed by a reconnection.

lupine, 2025-10-17:
Hi again,

Yes, I'm on `hostapd` main, specifically commit `167aeab41b2cfc2bb73393e19ac19173a43fff8c`. Config file is in https://github.com/morrownr/USB-WiFi/issues/603#issuecomment-3405568651

I noticed Debian had kernel 6.17.2 in Experimental, so decided to give that a try and the bug isn't reproducible there, so whatever the bug is, it seems fixed, which is the important thing.

Going through the git log -  `git log v6.16..v6.17 -- drivers/net/wireless/mediatek/mt76/mt7925` - I noticed two potential candidates:

```
9f15701370ec15fbf1f6a1cbbf584b0018d036b5 wifi: mt76: mt7925: fix locking in mt7925_change_vif_links()
55424e7b9eeb141d9c8d8a8740ee131c28490425 wifi: mt76: mt7925: fix the wrong bss cleanup for SAP
```

I strongly suspect it's the second one; I guess I could compile debian's 6.16.3 kernel with just that patch applied to get a positive confirmation.

morrownr, 2025-10-18:
Follow on: I am going to start comparing your hostapd.conf to the one I am using. We have many different settings. Another possible source of issues could be the networking setup. I use systemd-networkd as outlined in my Bridged AP guide on the Main Menu. I have maintained this guide over several years and I do experiment at times and systemd-networks continue to be a modern, very solid solution in my experience.

I continue to be unable to duplicate the problem but am here for the long term to see if we can fully resolve this issue. This is my first testing with AP mode with the mt7925u driver and so far, so good. Most of my testing has been with Managed mode and just some basic monitor mode items. I was expecting more problems given the lack of wide testing but maybe we have a chip and driver that can serve us well for many years.

Are you interested in further testing with 6 GHz (I need to see if regdb shows support for GB) to see about publishing an example hostapd.conf for WiFi 7, 6 Ghz, 160 MHz channel width.

sam8641, 2025-10-18:
Used MT7925 PCI-E, and Android Pixel 6a. Newer MT7925 firmware made a huge difference for me, on the same 6.12.48 kernel.

Old firmware 20250425073109 I had problems with frequent command timeouts and disappearing AP.

hostapd config: `beacon_int=33 dtim_period=12` Used it for 6GHz, as this allows Pixel 6a to see and connect to 6GHz a lot faster with preferred channel scan. For 5GHz, ok to use the defaults.

Newer firmware, no command timeouts, but have 2 smaller problems.
- HW/SW Version: 0x8a108a10, Build Time: 20250825215832a
- WM Firmware Version: ____000000, Build Time: 20250825215925
- hostapd v2.12-devel-hostap_2_11-1008-gf747ae0d1+ (Sept 14, 2025)
- `Linux debian237 6.12.48-test1 #1 SMP PREEMPT_DYNAMIC Tue Sep 23 08:46:08 CDT 2025 x86_64 GNU/Linux` Manually compiled kernel.

**Problem 1:** If hostapd restarts, no reboot, no chip reset, then my Android pixel 6a, trying to speed test, will always repeatedly lose connection, reconnect, and show 0 Mbps. No command timeouts, AP remains visible. This problem appears limited to 80MHz or 160MHz.
- Found a workaround: killall hostapd, chip reset with `echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/chip_reset`, then start hostapd.

**Problem 2:** While Hostapd is running, and runtime-pm was 1, `echo 0 | tee /sys/kernel/debug/ieee80211/phy*/mt76/runtime-pm`, Clients can't find this AP anymore. Restart hostapd, and, go to problem 1.

When working: Download 1246 Mbps, Upload 1170 Mbps OpenSpeedTest on WiFi LAN, at 2 feet distance.

I had a Bluetooth MT7925 problem, when firmware files was changed without reboot, `usbreset` command fails with USB errors and no Bluetooth. Reboot won't work. Requires powercycle or shutdown the computer for MT7925 bluetooth to start working again.

Overall, an improvement on stability from older firmware. As long as I avoid triggering problems, it appears stable so far for the past 12 hours at 6GHz, 160MHz width, WPA3.

morrownr, 2025-10-18:
Hi @sam8641 

I was just going over @lupine 's hostapd.conf and I think that is what I need to see from you. I have matched settings for the most part with @lupine but not entirely. I have not been able to duplicate his problem so far... but then, he is getting to the point that he is almost running stable. In addition to hostapd.conf, there can be issues that come up because of how AP mode is set up in networking but let's work hostapd.conf first.

sam8641, 2025-10-19:
Here is my hostapd config, with wpa_passphrase edited. They may be mostly filled with some options from OpenWRT, and lots of lines commented out.

[hostapd-mt7925-6ghz.txt](https://github.com/user-attachments/files/22987664/hostapd-mt7925-6ghz.txt)

On 6 GHz, ieee80211n=1 and ieee80211ac=1 is ignored and not neccessary to have.

bjlockie, 2025-10-19:
Why did they even invent NOIR?
I doubt the signal from a router makes it very far outside.
It is like the Intel LAR abomination IMO.

morrownr, 2025-10-20:
@sam8641 

I had a little extra so I changed settings to 6 GHz to see what I could see.

Channel 65, WiFi 6, Width 80, Netgear A9000 - mt7925 adapter as AP:

```
$ iperf3 -c 192.168.1.24
Connecting to host 192.168.1.24, port 5201
[  5] local 192.168.1.169 port 37052 connected to 192.168.1.24 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  43.9 MBytes   368 Mbits/sec    0   1.28 MBytes       
[  5]   1.00-2.00   sec  45.9 MBytes   385 Mbits/sec    0   1.60 MBytes       
[  5]   2.00-3.00   sec  44.9 MBytes   376 Mbits/sec    0   1.70 MBytes       
[  5]   3.00-4.00   sec  44.5 MBytes   373 Mbits/sec    0   1.70 MBytes       
[  5]   4.00-5.00   sec  46.2 MBytes   388 Mbits/sec    0   1.70 MBytes       
[  5]   5.00-6.00   sec  46.6 MBytes   391 Mbits/sec    0   1.70 MBytes       
[  5]   6.00-7.00   sec  46.2 MBytes   388 Mbits/sec    0   1.70 MBytes       
[  5]   7.00-8.00   sec  46.6 MBytes   391 Mbits/sec    0   1.70 MBytes       
[  5]   8.00-9.00   sec  45.8 MBytes   384 Mbits/sec    0   1.70 MBytes       
[  5]   9.00-10.00  sec  46.1 MBytes   387 Mbits/sec    0   1.70 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   457 MBytes   383 Mbits/sec    0            sender
[  5]   0.00-10.01  sec   454 MBytes   380 Mbits/sec                  receiver

```

The speed is slow but this test is going through 2 walls and I am showing a signal level of -63 dBm. Whew. With a signal that bad, I am actually impressed that the connection and speed is very stable and usable. I have an Android 14 phone but it does not have 6 GHz. I have ran several speed checks with iperf3. I will leave my main box connected and see where this goes.

morrownr, 2025-10-22:
@sam8641 

Continuing on with band 4 (6 GHz) testing with my mt7925u based adapter:

Channel 65, WiFi 6, Width 160, Netgear A9000 - mt7925 adapter as AP:

Note that width has changed to 160.

```
$ iperf3 -c 192.168.1.24
Connecting to host 192.168.1.24, port 5201
[  5] local 192.168.1.128 port 38514 connected to 192.168.1.24 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  73.0 MBytes   612 Mbits/sec    0   1.24 MBytes       
[  5]   1.00-2.00   sec  70.1 MBytes   588 Mbits/sec    0   1.55 MBytes       
[  5]   2.00-3.00   sec  67.5 MBytes   566 Mbits/sec    0   1.55 MBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0   1.55 MBytes       
[  5]   4.00-5.00   sec  69.8 MBytes   585 Mbits/sec    0   1.74 MBytes       
[  5]   5.00-6.00   sec  70.2 MBytes   589 Mbits/sec    0   1.74 MBytes       
[  5]   6.00-7.00   sec  72.6 MBytes   609 Mbits/sec    0   1.74 MBytes       
[  5]   7.00-8.00   sec  72.4 MBytes   607 Mbits/sec    0   1.83 MBytes       
[  5]   8.00-9.00   sec  71.6 MBytes   601 Mbits/sec    0   1.83 MBytes       
[  5]   9.00-10.00  sec  72.6 MBytes   609 Mbits/sec    0   1.83 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   712 MBytes   598 Mbits/sec    0            sender
[  5]   0.00-10.01  sec   710 MBytes   595 Mbits/sec                  receiver

```

I am testing with a mt7922 based PCIe card now. It has a better quality antenna than what I was testing with and dBm is up to -56 now. Overall speed has increased to about 600 Mbps. That is still not a particularly a good speed due mostly to low signal level. The exactly same card in the same location will do over 1.2 Gbps when connected to my main router with band 2 (5 GHz). The testing here continues to confirm to me that 6 GHz has a more difficult time penatrating walls than does 5 GHz, but that should be expected.

Overall, my recent testing is showing the mt7925u driver to be very stable. I have not been able to duplicate any of the problems. That could be due to me not having an exact setup match, but I have been trying to get as close as I can, or it could be some differences in networking setup or in hostapd.conf setup. I can post my hostapd.conf if you guys want to look at it. I'll be the first one to admit that hostapd with 6 GHz is complicated. I'll continue testing this setup.

sam8641, 2025-10-25:
@morrownr 
For hostapd config tests with my `interface=wlp1s0` MT7925e:
- `ieee80211be` error, it should be `ieee80211be=1`
- `channel=65` Use 69. Some devices (my Android Pixel 6a) will not scan nor find anything on channel 65, as this channel is not Preferred Scanning Channel (PSC).
- Comments `PCS` without showing what this abbreviation is. With internet search, Preferred Scanning Channels (PSC) are more common.

Some problems on mt7925e (PCI-E) may sometimes not happen to mt7925u (USB) device.
- I have MT7925e PCI-E, MT7922 PCI-E, MT7921u USB.
- My android Pixel 6a with WiFiAnalyzer is able to see if Access Point (AP) is capable of WiFi 7, even though it can only do WiFi 6e speed.

I do have CLC disabled. Unsure if it helps MT7925e. Older MT7922 did improve signal strength on some channels.
File: /etc/modprobe.d/options.conf
```
options mt7925-common disable_clc=1
options mt7925e disable_aspm=1
options mt7921-common disable_clc=1
options mt7921e disable_aspm=1

```
As for my mt7925e (PCI-E) on Ryzen 3900x CPU, Linux 6.12.48, the newest firmware on MT7925e (20250825215832a + 20250825215925), it works fine for the past 3 days and remains stable while Hostapd stays up.
- Kernel 6.12.48 have most of the bug fix that it appears stable enough. This includes kernel panic or kernel oops bug fix.

**Stability:** A few days passed by, and MT7925e (PCI-E), on kernel 6.12.48 is stable and functional with newest firmware. Handles client disconnect and reconnect just fine.
- For my problem, hostapd restarts with no reboot and no chip_reset, makes my Pixel 6a client, lose connection trying to speed test.
- Workaround: stop Hostapd, `echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/chip_reset`, then start Hostapd.

**Speed:** Client: Android Pixel 6a
MT7925e with Edup AX antenna
- 2 feet: 1240 1140 Mbps, 21
- 10 feet: 828 383 Mbps, 21
- 20 feet: 467 55 Mbps, 21
- 10 feet: 941 333 Mbps, 133
- 20 feet: 351 122 Mbps, 133

With Asus antenna (Prime B550-E, TUF GAMING X570-PLUS WiFi)
- 10 feet: 1224 639 Mbps, 21
- 20 feet: 362 70 Mbps, 21
- 10 feet: 1198 570 Mbps, 37
- 20 feet: 272 30 Mbps, 37
- 10 feet: 925 711 Mbps, 69
- 20 feet: 246 60 Mbps, 69
- 2 feet: 1410 1050 Mbps, 133, MT7922
- 2 feet: 1188 1149 Mbps, 133
- 10 feet: 760 692 Mbps, 133, MT7922
- 10 feet: 962 569 Mbps, 133
- 20 feet: 335 304 Mbps, 133, MT7922
- 20 feet: 271 204 Mbps, 133

Looks like, Android Pixel 6a have faster upload on higher 6 GHz channels.

sam8641, 2025-10-25:
@morrownr
I was using your WiFi 7 example for testing with just a small changes.
```
op_class=134
channel=69
he_oper_centr_freq_seg0_idx=79
eht_oper_centr_freq_seg0_idx=79
# Lower beacon_int to speed up Android Pixel 6a 6 GHz finding this AP.
dtim_period=10
beacon_int=40
```

Discovered something that allows my Android to find and connect to non-preferred 6GHz channel 65, or any 6 GHz channels.
- 2 WiFi cards on 1 computer. Me: MT7925e + intel WiFi AC card.
- Config 1: 6 GHz, channel 65.
- Config 2: 2.4 or 5 GHz with `rnr=1` Reduced Neighbor Report (RNR).
- Run both hostapd config on a single Hostapd process.
- My Android Pixel 6a now found 6 GHz, channel 65 and can connect to it.

I often run a multiple commands on single line to restart hostapd to run 2 WiFi cards in AP mode. A single Hostapd process runs 2 WiFi devices.
`pkill -f /run/hostapd.pid; sleep 1; echo 1 > /sys/kernel/debug/ieee80211/phy1/mt76/chip_reset; sleep 1; /root/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd-mt7925.conf /etc/hostapd/hostapd-iwlwifi.conf &`
- I sometimes replace pkill with `killall hostapd`.

My Android Pixel 6a have a random missed 6 GHz AP problem when scanning, it also randomly fails to find neighbor's nearby 6 GHz AP with Android WiFiAnalyzer app.
- Workaround is `beacon_int=40` or lower, in hostapd config.
- This may be the US region "NO-IR" on the entire 6 GHz, limits to passive scan only.

My computer is Ryzen 3900x with MT7925e and an old intel WiFi AC.

My MT7925e problem list:
- 1: Solved: kernel panic/oops. Fixed with newer kernel.
- 2: Solved: Command timeout, disappearing AP. Outdated firmware.
- 3: Problem: Restart hostapd without chip_reset: Android disconnects during speed test. Workaround: manually chip_reset, then start hostapd.

> Is this the same problem that @DarkAlfar said a fix went into kernel 6.17?

I have seen problem number 2 happen to me before, solved with firmware update. My current, problem 3, is different.

For problem number 3, I can take a look at newer kernel version soon to check for possible fix, on Debian bookworm. Appears this specific disconnect problem is only happening to me.

lupine, 2025-10-25:
Sorry for dropping out of the conversation a bit - it's difficult to justify making changes now that it's solid for me!

I just wanted to add one note: The 6.17.2 kernel in Debian Experimental, which I mentioned above, is working great *but* it removed support for legacy iptables / xt modules, which (I found out several days later) disabled some firewall features when using, e.g. legacy iptables-persistent. It also had a knock-on effect which broke docker: https://github.com/moby/moby/issues/50615

Updating to nftables beforehand strongly recommended.

morrownr, 2025-10-26:
@sam8641 

I am pondering your last message above. Let me see if I have a handle on this:

You are seeing 2 problems:

- your AP that is using a mt7925e based card is showing a slow transfer speed based on what you are seeing at the client?

- About 50% of hostapd restarts is a connection failure. 

I have good luck working with some of the Mediatek kernels devs to get things fixed. I do not have a black belt in wireless coding but can help get things fixed. Do you mind if we concentrate on one problem first? If so, how about we start with the speed issue:

Remember that I am not there looking at what is going on so please be specific. 

 > Newer kernel appears to have slow download (MT7925e to client) speed problem as well.

What speeds are you seeing with which kernels?

How are the speed reading being taken?

What speeds are you expecting?

What is your distance from client to AP? I usually don't recommend closer than maybe 6 feet.

sam8641, 2025-10-26:
Speed tested using mostly OpenSpeedTest website on local network, Linux Apache server. Looking at Linux network monitor like `iptraf` and `iftop`, appears to speed test using multiple simultaneous download connections, then multiple simultaneous upload connections.

Also speed tested using `iperf3 -c 127.0.0.1 -t30 -P4` (with different IP address) for 4 simultaneous connections, sometimes with `-R` or `--bidir` to test speed in the other direction or both directions.
- Android: I use Termux app for command line, iperf3, works well with Linux iperf3.

Tested multiple times between 1 and 2 feet. Going more then 2 feet away, will make the speed starts to go slower. 6 GHz is weaker and shorter range. Need closer distance for fastest speed.

Speeds on each kernel.
MT7925e, Kernel 6.16.3: 1763 download, 1178 Mbps upload.
MT7925e, Kernel 6.17.5: 1266 download, 1173 Mbps upload.
MT7925e, Kernel 6.12.48: 1225 download, 1172 Mbps upload.
MT7922, Kernel 6.12.48: 1422 download, 1022 Mbps upload.

I was expecting the MT7925e to be equal or maybe faster speed compared to my MT7922, due to my client Wifi 6e limitation (Pixel 6a).

I have 2 computers with Hostapd WiFi:
- Ryzen 3900x, MT7925e, Intel WiFi 5, 2.5g ethernet.
- Ryzen 5700x, MT7922, Intel AX200, USB MT7921u, multiple 2.5g & 1g ethernet, connects to internet DOCSIS 3.1 modem.

sam8641, 2025-10-26:
Oops, my hostapd config slowed down upload speed.
Was: `he_6ghz_max_ampdu_len_exp=3`
Now: `he_6ghz_max_ampdu_len_exp=7`
5 Ghz: `vht_capab=[MAX-A-MPDU-LEN-EXP7]`

With this change, OpenSpeedTest on 6 GHz Channel 69 shows:
MT7925e 6.12.48, 1256 download, 1509 upload Mbps
MT7922 6.12.48,  1512 download, 1727 upload Mbps

`iperf3 -c 0.0.0.0 -t20 -P4` with correct IP address:
MT7925e iperf3, 6.12.48, 1.20 download, 1.50 upload Gbit/sec
MT7922 iperf3, 6.12.48, 1.42 download, 1.55 upload Gbit/sec

I did make sure the iw dev .. station dump is showing the maximum bitrate.
```
watch -n 0.1 /sbin/iw dev wlp1s0.4096 station dump
tx bitrate: 2401.9 MBit/s 160MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
rx bitrate: 2401.9 MBit/s 160MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
```

My upload speed problem have been solved.

The download speed problem on kernel 6.12.48 and kernel 6.17.5 remains a problem. Expected: Would like the speed of MT7925e to be the same or faster than MT7922.

sam8641, 2025-10-28:
Tested `he_6ghz_reg_pwr_type` option.

MT7925e, 6 GHz, Channel 69, 160 MHz, Kernel 6.12.48 Debian Trixie.
Distance to Android Pixel 6a: 9.5 feet.

`he_6ghz_reg_pwr_type=0` 1043 download, 483 upload Mbps, 9.5 feet.
`he_6ghz_reg_pwr_type=1` 1142 download, 479 upload Mbps, 9.5 feet.
`he_6ghz_reg_pwr_type=2` Connecting... connection failure.
`he_6ghz_reg_pwr_type=3` Connecting... connection failure.
`he_6ghz_reg_pwr_type=4` Connecting... connection failure.
- In all cases of connection failure, AP remains visible. to WiFi scans.

Hostapd default is `he_6ghz_reg_pwr_type=2`, which for me, is a connection failure to both AP, MT7925e and MT7922, with client Andorid Pixel 6a. This was tested on MT7925e on US wireless regulatory (NO-IR bypassed). This may possibly be a limitation of US wireless regulatory, or Android Pixel 6a limitation.
- Different countries with different wireless regulatory may behave differently.

Reminder: `he_6ghz_max_ampdu_len_exp=7` for WiFi6 and WiFi7 config, or leave it unset. Hostapd default appears to be `he_6ghz_max_ampdu_len_exp=7`.