morrownr/USB-WiFi #607: [Help]:

MarinaUps, 2025-05-12:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Uname-a

### lsusb

Sudo 1susb

### rfkill

Sudo refill list all

### dkms

```text
Sudo dkms status
```

### iw

```text
Sudo iw dev
```

### What happened?

Needs to provision

bjlockie, 2025-05-13:
It "uname -a" (space before dash) , "sudo lsusb" (ell not 1), "sudo rfkill". Did autocorrect mess up the post?
sudo is all lower case.

morrownr, 2025-05-14:
> Needs to provision

I have no idea what this is asking. I'll give this issue a couple of days and if not clarified, I will delete it.