morrownr/USB-WiFi #609: Instructions for mt7662u

kd496, 2025-05-15:
I had trouble installing mt7662u firmware (Alfa AWUS036ACM). What worked for me:

1. All four files are needed
- mt7662.bin
- mt7662_rom_patch.bin
- mt7662u.bin
- mt7662u_rom_patch.bin

2. Firmware files must be placed in /lib/firmware, not /lib/firmware/mediatek

morrownr, 2025-05-15:
Hi @kd496 

Thanks for the information. I think we may need to do some additional testing to ensure this information is correct. I may need to dig into the source to see what the driver says as each driver will have the location hard coded. 

Of the files you mention, here is where they are on my Debian 12 system:

```
/lib/firmware/mediatek$ ls -l *7662*
-rw-r--r-- 1 root root 91412 May  1  2023 mt7662u.bin
-rw-r--r-- 1 root root 20686 May  1  2023 mt7662u_rom_patch.bin
```

```
/lib/firmware$ ls -l *7662*
-rw-r--r-- 1 root root 81908 May  1  2023 mt7662.bin
-rw-r--r-- 1 root root 26350 May  1  2023 mt7662_rom_patch.bin
```

With this setup, my mt7612u adapters work fine. Are you using an adapter with a 7662 chip or 7612?

morrownr, 2025-05-18:
No problem with the title.

I have 7612u adapters but no 7662u adapters (include bluetooth). What I suspect has happened is that Mediatek changed where the *mt7662* firmware should be at some point. For me to make sure my guide is 100% will take some testing.

I am curious what distro and kernel you are using?