morrownr/USB-WiFi #613: [NEWS] New Project: add Realtek USB WiFi 6 and 7 driver to mainline

morrownr, 2025-05-22:
A new project has been started to bring new Linux Standards (mac80211) compliant drivers to the Linux kernel. The supported chips are planned to be:

rtl8851bu
rtl8852/32au
rtl8852/32bu
rtl8852/32cu
rtl8912au

The drivers for rtl8832bu and rtl8851bu are already available for early testing. Where can you find the drivers?

https://github.com/morrownr/rtw89

The drivers can be installed as out-of-kernel drivers. Follow the instructions in the rtw89 repo shown above.

Please help us with this project. What do we need?

1. @dubhater needs adapters based on the rtl8832au chip and the rtl8912au chip. 
2. @morrownr needs adapters based on the rtl8851bu chip and rtl8912au chip.
3. We need testers.

@dubhater is located in the EU so if anyone in that part of the world wants to send him one or both of the adapters that he needs, please contact him.

@morrownr is located in the central US so if anyone in this part of the world wants to send me one or both of the adapters that I need, please contact me.

Note: The WARNINGS for Linux users to avoid Realtek WiFi 6/7 based USB WiFi adapters are still posted and are still in effect. This could change at some point but there are issues besides having good drivers that affect the use of Realtek USB WiFi 6/7 adapters. If you have come to this site to obtain information about buying a new USB WiFi adapter, go to The Plug and Play List.

Kiwimannen, 2025-06-14:
This is great news , I live in Europe and can provide @dubhater with a rtl8832cu (not au) chip but most importantly a rtl8912au chip - you would have a choice of either a Fenvi or a Comfast model.

Kiwimannen, 2025-06-14:
I can - of course - also be a tester for rtl8832cu and rtl8912au (since I am keeping one of them) :)

morrownr, 2025-06-15:
@Kiwimannen 

Acknowledged. Keep an eye on the repo.

MrSorcus, 2025-09-15:
@morrownr  Thanks for the advice. Much appreciated.
Too bad the reviews of the BE92 Nano are not very good.
> Right now I have 4 of the 5 adapters needed to test the usb drivers in rtw89. I only one I am missing is the rtl8851/33bu but I have a plan for it if need be.

Ok. If you need any devices for future driver implementations, let me know. I will try to help and provide the necessary devices.

morrownr, 2025-10-04:
@MrSorcus 

Just to let you know, I now have the last adapter that I currently need on the way - rtl8851bu, so I am good for now. I think what could use most right now are more testers, especially testers that are somewhat knowledgeable of wireless coding but that is not so easy to find.