morrownr/USB-WiFi #614: Panda PAU0A is very slow

jducnuigeen, 2025-05-22:
I purchased a Panda PAU0A adaptor after seeing recommendations on this github repo, but I've had a terrible time getting it to work with any speed on two different access points in my office.
OS: Ubuntu 22.04.5
Uname -r:  6.8.0-59-generic
Link speed on access point A: 21Mb/s (2.4 GHz) using WPA2 personal
Link speed on access point B: 14 Mb/s (2.4 Ghz) using WPA2 Enterprise and TLS

On both of these access points, I'm getting at most about 200-400 **Kbps** speed.
If I connect to these same access points with my phone, I get 7.7-10 **Mbps** speed.

Things I've tried: I updated the firmware for the Panda adaptor using the instructions found on this github site for the MT7610u.bin firmware. No difference was made.

There seems to be glowing reviews of this adator, so I don't understand why it seems so slow on my machine.
The adaptor is plugged into an older HP desktop which doesn't have any wifi adaptors natively. I've tried plugging in the adaptor into both front and rear USB ports, but the slow speeds seem to be the same everywhere.

Thoughts on things to try?

morrownr, 2025-05-23:
Hi @jducnuigeen 

> I purchased a Panda PAU0A adaptor after seeing recommendations on this github repo...

I took a look in The Plug and Play List to check for sure and I do not see the Panda PAU0A listed. I do see the Panda PAU0B listed and I have one. I forget what chip the Panda PAU0A uses. You indicated it is the mt7610u, the same as the Panda PAU0B.

I've had really good luck with my Panda PAU0B. It should not be necessary to update the firmware for the mt7610u chip as it has not changed for years.

> Thoughts on things to try?

Maybe we should check signal strength first:

Install wavemon if not installed:

$ sudo apt install wavemon
$ wavemon

What is signal level?

jducnuigeen, 2025-05-25:
Yes, I saw the PAU0A uses the mt7610u chipset, so that was what I could purchase as many of these devices suggested on this repo are often out of stock. I assumed (maybe wrongly) that an adaptor using the same chipset as other adaptors on the list would behave the same. Maybe I wasted my money, I don't know.

Signal level is about -65 dbm. Link quality shows about 85%. I tried putting the usb adaptor on the end of an extension USB cord, but then I read this is not recommended either. So now it is on the back panel USB port.

morrownr, 2025-05-25:
> Signal level is about -65 dbm.

That is not good at all. -65 is about the level where I question if a connection can be maintained but if a connection is maintained, the throughput will be poor. 

I did a little research to refresh my memory on the Panda 0A and yes, poor range is the primary reason that this specific adapter did not make it into The Plug and Play List. The Panda 0B is very different in that I consider it to be a `Very Long Range` adapter. It is good.

> I assumed (maybe wrongly) that an adaptor using the same chipset as other adaptors on the list would behave the same.

My experience is that there can be a lot of difference between adapters with the same chips even with the same maker. This is one of those cases. This kind of thing happens with all adapter makers to one degree or another. I am very picky about what is added to The Plug and Play List as I want it to be a list of the best of the best. Other Panda adapters are on the list and Panda generally does a good job.

> I tried putting the usb adapter on the end of an extension USB cord, but then I read this is not recommended either.

I will argue that USB extension cables are a good thing. I use them extensively. They allow you to reposition the adapter to better locations and as long as the cable is not too long, there should be no problem. If you limit the cable to 4 feet or less, there should be no problem. Use the cable and get the adapter in a location at least a couple of feet away from a wall, in a high location and not close to objects, especially metal objects. You may need to test multiple locations to see what works best. Even small changes in location can make  a big difference. You can watch `wavemon` while changing locations to see what happens.

jducnuigeen, 2025-05-25:
I've put the adaptor back on the USB extension cord (about 1.5m long) and it is now up in a very high location, about 10m from the access point (which is on the ceiling).
Now the signal level is hovering between about -62 and -67.
I've decided to go ahead and purchase another adaptor, the ALFA AWUS036ACM Long Range AC1200 with the mt7612u chipset.
Thank very much for this insight, feedback, this help here and a very informative and detailed repo page !

jducnuigeen, 2025-05-25:
Unfortunately it looks like IT department is wanting to replace the access point so when I asked them about my problems a while back they said to just wait until the new one get installed. But good to know this mt7610u adaptor is probably only good for really close range anyway.
There are no walls between the adaptor and the access point.
When the new adaptor arrives I'll also put it on end of the extension and up high and I now better understand from what you're saying a good signal strength should be when I test it with wavemon.
Thank you very much for all of this information - it is really helpful !