morrownr/USB-WiFi #620: [Help]: MT7925B22M

skarpinis, 2025-06-02:
### Checklist

- [x] I acknowledge that support is provided on a best-effort basis.
- [x] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux archlinux-pc 6.14.9-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 29 May 2025 21:42:15 +0000 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 002: ID 046d:086b Logitech, Inc. BRIO 4K Stream Edition Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 258a:0150 SINOWEALTH Gaming KB  Bus 003 Device 007: ID 26ce:0a08 Generic USB Audio Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 009 Device 002: ID 046d:c547 Logitech, Inc. USB Receiver Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 011 Device 002: ID 26ce:01a2 ASRock LED Controller Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

0: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
nvidia/570.153.02, 6.14.9-arch1-1, x86_64: installed
```

### iw

```text
phy#0
	Interface wlp9s0
		ifindex 3
		wdev 0x1
		addr 3c:0a:f3:a0:48:dd
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

Bluetooth not working not showing: asrock x870 nova wifi
archlinux kernel: usb 3-7: device descriptor read/64, error -71
archlinux kernel: usb 3-7: device descriptor read/64, error -71
archlinux kernel: usb 3-7: device descriptor read/64, error -71
archlinux kernel: usb 3-7: device descriptor read/64, error -71
archlinux kernel: usb 3-7: device not accepting address 5, error -71
archlinux kernel: usb 3-7: device not accepting address 6, error -71

morrownr, 2025-06-02:
Hi @skarpinis 

Is this about a M.2 card based on a mt7925 chip?

I need you to run and post the results of:

$ lspci
$ modinfo mt7925e
$ sudo dmesg | grep mt7

My concern is that you are running old firmware for the mt7925.

skarpinis, 2025-06-03:
Hi @morrownr yes its about m2 integrated wifi & bluetooth, wifi works no problem but bluetooth is killed after next boot. I can recover only in bios "turn off bt" restart pc and again go in bios "turn on bt" restart or power off pc completelly from outlet for few minutes. I did tried update firmware as instructed here but did not helped. Tried suggested suspend/sleep service but it worked only once really weird issue here only happens in linux :( bluetooth when is working temporary using btusb driver.

morrownr, 2025-06-03:
@skarpinis 

>  really weird issue here only happens in linux...

That happens. Linux and Windows do not have a cooperative agreement to share bugs. ha ha

My mt7925 based M.2 card is also in an AMD processor system and I am not seeing this. There are some bios settings that could be involved. 

I have the following settings disabled:

Fast Boot
Quiet Boot

Since the BT is a USB supported thing, I have some usb related bios settings in this message:

Here is a message that I keep pinned in Issues. It has a lot about USB ports and USB related bios settings:

https://github.com/morrownr/USB-WiFi/issues/508

Also, here is a bug report message I posted recently to the RasPiOS repo. I am wondering if Arch has messed up and not turned on the driver to make the communications work. See Issue 1:

https://github.com/morrownr/USB-WiFi/issues/608

I would do cold boots after changing bios settings. If you try something and it does not work, return to the previous setting and continue...with the exception of Fast Boot and Quiet Boot as I think those should be off all of the time with Linux.

Let me know how it goes.

morrownr, 2025-06-03:
> suspend broken too

Can you elaborate on that? Suspend works fine here. What are you seeing?

The wifi driver supports a parameter:

parm: disable_aspm:disable PCI ASPM support (bool)

disable_aspm=y

Do you know how to set that up?

morrownr, 2025-06-03:
I was just pondering this issue and I thought of a couple of other things that could cause the range to be poor:

Go to:

https://github.com/morrownr/USB-WiFi/blob/main/home/FAQ.md

That is the site FAQ. Items 1 and 2 both could cause range problems. What do you get when you run:

$ iw reg get

closed 2025-06-04 by skarpinis